zigpy/zha-device-handlers #1921: [BUG] Why enable the invalid readings for Xiaomi?

glassbase, 2022-11-14:
**Describe the bug**
Why enable the invalid temp readings from the various Xiaomi sensors? See https://github.com/zigpy/zha-device-handlers/pull/1248

I personally have lumi.sensor_magnet.aq2 and lumi.sensor_wleak.aq1 and the temperatures reported are so far off in accuracy. I quickly compared my Xiaomi temp sensors to leak sensors temp readings in approx same area of house and the are off by 4-5C+.

To me, they should be disabled via quirks, not disabled in HA individually. 

Thoughts?

**To Reproduce**
n/a

**Expected behavior**
n/a

**Screenshots**
n/a

<details>
n/a

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**

dmulcahey, 2022-11-14:
They are device temperature not area temperature and i agree I think they’re pointless and we had removed them at one point but folks wanted them back so here we are 🤷🏻‍♂️. You will need to disable them in HA it is the only way to let folks choose.

glassbase, 2022-11-14:
I have leak sensors sitting on a cold basement floor report 25+C. And I live in northern Ontario.

glassbase, 2022-11-14:
If they are the internal device reading, then they aren't even valid if you compare then to a temp/humidity sensor type.

dmulcahey, 2022-11-14:
> If they are the internal device reading, then they aren't even valid if you compare them to a temp/humidity sensor type.

They are the temp of the SoC in the device. I’ll reopen it but I don’t have another solution for you. Some folks want them and some don’t 🤷🏻‍♂️

MattWestb, 2022-11-14:
If not like having it the GUI you can hidden or disable it under settings > Advanced settings and is no more problems.

If like have "real temperature" use one real sensor device for it.

closed 2023-05-20 by github-actions[bot]