zigpy/zha-device-handlers #1922: [BUG] Can't pair TS011F by _TZ3000_gjnozsaz

etienn01, 2022-11-14:
**Describe the bug**
When adding `TS011F by _TZ3000_gjnozsaz` in ZHA, it tells me the device is ready to use, but the light on the plug keeps flashing, meaning that it's still in paring mode. During that time, the device works for about 10 seconds (I can toggle the relay and it reports power consumption) and then it stops responding.

Firmware was upgraded to the latest version (`app_version` from 76 to 192) but it didn't solve the problem. The device seems to work properly in Z2M.

**To Reproduce**
Steps to reproduce the behavior:
1. Add device to ZHA
2. ZHA shows "The device is ready to use"
3. LED on the plug keeps flashing
4. After a few seconds, the device stops responding

**Expected behavior**
The pairing completes successfully.

**Screenshots**
[image]

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_gjnozsaz",
  "model": "TS011F",
  "class": "zhaquirks.tuya.ts011f_plug.Plug"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Paris",
[... 450 lines cut ...]
          "0xe001": {
            "endpoint_attribute": "tuya_external_switch_type",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

etienn01, 2022-11-17:
So I just put my ConBee II on an USB extension cord and now it's working.
It was plugged at the back of my NUC, I don't know why I never had any issue for over 3 years with my other devices.

closed 2022-11-17 by etienn01

dmulcahey, 2022-11-17:
> So I just put my ConBee II on an USB extension cord and now it's working.
> It was plugged at the back of my NUC, I don't know why I never had any issue for over 3 years with my other devices.

It’s literally in the documentation for the conbee… https://www.phoscon.de/en/conbee2/install

[image]

for some reason there is always confusion around this being necessary