zigpy/zha-device-handlers #1924: [Device Support Request] _TZE200_bvu2wnxz

thprim, 2022-11-15:
**Is your feature request related to a problem? Please describe.**
I bought multiple trvs "HAMA smart thermostat". Some of them are only detected as a device in ZHA without the necessary sensors for controlling the device.  It seems also, that HAMA has several devices in the produkt line which is not visible to which model number the device relates. So far, it's always a risk buying a device from HAMA and get it integrated into Homeassistent with the standard ZHA implementation.
I'm using the DeConz RaspBeeII with the ZHA integration. A lot of other devices are working well.

**Describe the solution you'd like**
Ideally, the devices can be added as a compatible device to ZHA, because I assume there are a lot of other installations with the same issue. An individuell working solution is not the best way. Other trv's from HAMA with the ID _TZE200_yw7cahqs are still working well.

<details>

<summary>Device signature NOT working</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_bvu2wnxz",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.61-v7l",
    "supervisor": "2022.10.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.28.3",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "watchman": {
[... 248 lines cut ...]
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
I have also an additional trv with the ID _TZE200_h4cgnbzg. An additional device support request can be submitted if necessary. The signature seems equal, mainly except the manufacturer ID / manufacturer code

javicalle, 2022-11-15:
Duplicate of #1818

thprim, 2022-11-15:
Thanks a lot, got my device _TZE200_bvu2wnxz up and running with the latest ts0601_trv_me167.py.zip file issued in the https://github.com/zigpy/zha-device-handlers/issues/1818
Really appreciated

javicalle, 2022-11-15:
Closing duplicate.

closed 2022-11-15 by javicalle