zigpy/zha-device-handlers #1925: [Device Support Request] Legrand Wireless Dimmer Switch (battery percentage = unknown)

stevenwfoley, 2022-11-15:
**Is your feature request related to a problem? Please describe.**
Device does not display remaining battery percentage.

**Describe the solution you'd like**
I would like to see the remaining battery percentage displayed for the device.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=1, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4129, maximum_buffer_size=89, maximum_incoming_transfer_size=63, server_mask=10752, maximum_outgoing_transfer_size=63, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000f",
        "0x0020",
        "0xfc01"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0006",
        "0x0008",
        "0x0019",
        "0xfc01"
      ]
    }
  },
  "manufacturer": " Legrand",
  "model": " Remote dimmer switch",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Detroit",
    "os_name": "Linux",
    "os_version": "5.15.74",
    "supervisor": "2022.10.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.28.3",
[... 349 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
It looks like the device only reports battery voltage in the power configuration cluster. Anecdotally, I believe that a value of 31 is a full/new battery.

javicalle, 2022-11-17:
There are a few implementations that convert voltage to percentage, the problem is calibrate the MIN/MAX values. According to Z2M, Legrand devices batteries are calibrated between 3.0 and 2.5 volts:
* https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/legrand.js

So I'm going to suggest the same approach for your device. When the batteries run out you can check if the values are adequate or not.

I suggest to take the current `legrand/dimmer.py` file and copy in your custom quirk folder. If you don't know how to set the custom quirk this is a good guide to follow:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Into the `dimmer.py` file you will need to perform the following steps:
* add the `PollControl` import:
```python
from zigpy.zcl.clusters.general import (
    Basic,
    BinaryInput,
    GreenPowerProxy,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    PollControl,
    Scenes,
)
```
* add the `PowerConfigurationCluster` import:
```python
from zhaquirks import PowerConfigurationCluster
```
* add at the bottom of file this 2 classes:
```python
class LegrandPowerConfigurationCluster(PowerConfigurationCluster):
    """Power configuration cluster for Legrand devices."""

    MIN_VOLTS = 2.5
    MAX_VOLTS = 3.0

class RemoteDimmer(CustomDevice):
    """Remote dimmer ."""

    signature = {
        MODELS_INFO: [(f" {LEGRAND}", " Remote dimmer switch")],
        ENDPOINTS: {
            1: {
                # "profile_id": 260,
                # "device_type": "0x0104",
                # "in_clusters": ["0x0000","0x0001","0x0003","0x000f","0x0020","0xfc01"],
                # "out_clusters": ["0x0000","0x0003","0x0006","0x0008","0x0019","0xfc01"]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMER_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfigurationCluster.cluster_id,
                    Identify.cluster_id,
                    BinaryInput.cluster_id,
                    PollControl.cluster_id,
                    LegrandCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Ota.cluster_id,
                    LegrandCluster.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    LegrandPowerConfigurationCluster,
                    Identify.cluster_id,
                    BinaryInput.cluster_id,
                    PollControl.cluster_id,
                    LegrandCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Ota.cluster_id,
                    LegrandCluster,
                ],
            }
        }
    }
```

Save the file, restart HA and check that the device signature have changed.

Until the device don't report a battery status, the percentage value will not be updated.
Battery device don't report the battery status very often. Maybe you can try to remove and put again the battery to check.

stevenwfoley, 2022-11-17:
Well that worked beautifully. Thank you so much!
I'll have to do some testing to perfect the battery voltage range, but it seems like a good start.

PR #1934 (2022-11-19): Add Legrand remote dimmer
New quirk for battery conversion of Legrand ` Remote dimmer switch`.
Also applies the `LegrandCluster` but not sure if usefull here.

Fixes: #1925
closed 2022-11-17 by stevenwfoley