zigpy/zha-device-handlers #1926: [Device Support Request] TS0601 by _TZE200_hhrtiq0x

BlackBadPinguin, 2022-11-15:
**Is your feature request related to a problem? Please describe.**
Hello, recently I bought a couple of Nedis TRV's in the hope of them being entirely supported by ZHA, and indeed mainly the integration works fine, the TRV correctly reports its temperature, as well as its setpoint I also can change the setpoint. However, I am missing the setting for window control, the actual mode (auto or manual) and the valve state in %.
**Describe the solution you'd like**
That Setting / reporting child lock, Setting / reporting window detection, Setting / reporting valve detection works.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_hhrtiq0x",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_trv.SiterwellGS361_Type2"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.74",
    "supervisor": "2022.10.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "homematicip_local": {
      "version": "1.21.2",
      "requirements": [
        "hahomematic==2022.11.1"
      ]
    },
[... 382 lines cut ...]
                "value": 2
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

MattWestb, 2022-11-15:
ZHA is only having basic functionality for most tuya TRVs then its nearly impossible squeezing then to Zigbee standard that ZHA is using.

I have some of this clones and one user  have doing one local quirk that is using all we can getting fom the TRV.

Configure local quirk and installing the quirk from this issue and you is getting nearly in ZHA and for getting the presets you need patching the climate.py in HA container.

https://github.com/zigpy/zha-device-handlers/discussions/1084#discussioncomment-1756246

BlackBadPinguin, 2022-11-15:
Thanks, this works great, however i needed to change the quirk a little bit to fix the deprecated `manufacturer_attributes`. I will attach the updated version. 
[ts0601_trv_siterwell.zip](https://github.com/zigpy/zha-device-handlers/files/10016658/ts0601_trv_siterwell.zip)

closed 2023-07-10 by github-actions[bot]