zigpy/zha-device-handlers #1927: [Device Support Request] Eglo PADROGIANO-Z (Awox) (2 in 1 White Leds + Color Leds)

tibuski, 2022-11-16:
Link : [Eglo PADROGIANO-Z ceiling light](https://www.eglo.com/en/ceiling-light-padrogiano-z-900481.html)

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0300",
        "0x1000",
        "0xfc57"
      ],
      "out_clusters": [
        "0x0006"
      ]
    },
    "3": {
      "profile_id": 4751,
      "device_type": "0x010d",
      "in_clusters": [
        "0x0004",
        "0xff50",
        "0xff51"
      ],
      "out_clusters": [
        "0xff50",
        "0xff51"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "AwoX",
  "model": "EBF_RGB_Zm",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.11.2",
    "dev": false,
    "hassio": false,
[... 497 lines cut ...]
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

**Additional context**
The Lamp consist of 2 parts : 
* 4 White Led panels 
*  Each Led Panel is surrounded by Color Leds

**Most missing feature :** being able to turn off the white leds and/or the color leds

I think, ideally, the Light should appear as 2 different lights for which we can change brightness and color/temp
As of now, the sliders only change the White Light brightness and turn on/off only works for the whole lamp.

MattWestb, 2022-11-25:
This light is little strange made and i have the 3 x CT spots and is making one CT light and one open/close sensor that is showing if the light is on or off.
The signature is :
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010c",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xfc57"
      ],
      "out_clusters": [
        "0x0006"
      ]
    },
    "3": {
      "profile_id": 4751,
      "device_type": "0x010c",
      "in_clusters": [
        "0xff50",
        "0xff51"
      ],
      "out_clusters": [
        "0xff50",
        "0xff51"
      ]
    }
  },
  "manufacturer": "AwoX",
  "model": "TLSR82xx",
  "class": "zigpy.device.Device"
}
```
My lights is working OK but i dont understand way ZHA is making the extra sensor that is no use if looking in the signature but i think its coming from the On/Off cluster `0x0006` on EP 1 as out that shall being one controller that is sending On/Off to lights.

I think your light need little patching of the color cluster for working OK with color and CT but i also think its need some attribute stetted on the `0xff50`, `0xff51` and `0xfc57` that we is not knowing how its working.
If some one have one AwoX Zigbee gateway and can sniffing how its initiating the devise it can helping us understanding how to configuring this device right.

Also we is expecting more strange things from this devices then its using the new chips from **Telink Semiconductor** that tuya also have starting to using and some device is working OK and other is heaving very strange bugs.
Also it looks they is using the same firmware for many devices and is only changing  some setting in the config and transforming it to one other device (IKEA is doing it with setting in the device user data but they is doing it in one very good was and its working OK).

tibuski, 2022-11-25:
Thank you very much for your reply !

I don't have the Awox Zigbee Gateway but if you need me to do any test and/or gather logs, please ask.

closed 2023-05-31 by github-actions[bot]

ZwoCa, 2025-02-21:
I know this issue has been closed for a long time now, but it's still something easily found when having problems controlling RGB values of the Eglo Connect-Z series devices.

I recently bought a VALDELAGRANO-Z pendant lamp (also using a Telink Semiconductor) and had the same problem, that while I could control white balance and luminance, there was no way to control the RGB channels.

The solution for this was updating the firmware. I installed the AwoX Smart CONTROL app (unfortunately, one must create an account to do so), paired the lamp and updated the firmware. It was a major version update (1.x  → 2.x) and it failed several times, I had to place the smartphone next to the controller to get it to work, but maybe it was just a question of luck.

After successfully updating the firmware and re-pairing the lamp in ZHA, it now gives me full control of the RGB channels:

[image]

I'm not sure if this applicable also for other lamps of this series, but I hope it is and helps someone out there.

tibuski, 2025-02-22:
Wow, thank you for providing feedback !
I'll give it a try and report here.