zigpy/zha-device-handlers #1933: [Device Support Request for HG0646]

erkr, 2022-11-17:
Please add a quirk to support the melinera HG06467 Christmas light strings sold by the LIDL.

The strings are supported by Z2M: https://www.zigbee2mqtt.io/devices/HG06467.html

And there exists an issue for this that became stale. That issue contains all information wrt these Christmas led strings: 
https://github.com/zigpy/zha-device-handlers/issues/604

Best 
Eric

MattWestb, 2022-11-19:
Hola @javicalle  !!

If you have time this device is very wanted for the moment and is one real "tuya hybrid" that is using normal Zigbee commands and some very tuya cooked ones.

In ground is one normal light that have pixel LEDs and the normal lights things is Zigbe commands and the pixel is having some special commands implanted.

I have not baying it then is was little over my budged but many users have doing it.

If you have time and can taking one look may user wold getting one nice Christmases if you can cocking one quirk for it.

Mvh Mattias

rkruisselbrink, 2022-12-13:
Is there anything we can do to help with this effort? I have this product as well and it would be awesome to control the effects via ZHA 🎄

erkr, 2022-12-14:
I can help to test with local quirks

erkr, 2023-01-01:
A lot of time left before Christmas 😉

MattWestb, 2023-01-13:
@TheJulianJES We was trying getting one of this to our US friends for2 years ago but we was not getting any that was willing baying and shipping it from Europe.
If you is getting one or some else in CE is having one and can getting it to you can you taking one look getting one working quirk for it ?
I think its one normal RGB light plus extra scenes for searing the pixel LEDs with tuya MCU commands.

MattWestb, 2023-01-13:
For users that having one of this "devices" and need it working next Christmas is the devs needing getting hand of one for testing and doing the coding.

In CE we have great quirk devs in Germany and in Spain and probably more if we asking little more around.

Pleas helping getting the beloved device working by sharing hardware for some time and getting it working with our systems.

Thanks in advance !!!

xanibale, 2023-01-18:
I am from Germany and have this Device, i would be willing to send it to any quirk dev, preferable in Germany, for some time.

MattWestb, 2023-01-18:
Great @xanibale I pinging @TheJulianJES hi is located in Ger and have the knowledge and if need hi is getting help from our US or Spanish devs.

I hoping for the next Christmas shall being white with much colors in the (dev)tree !!!

TheJulianJES, 2023-01-19:
At the moment, I don't have enough time to work on a (Tuya) device with a given time limit.

It'll also be hard to implement the effects properly and ZHA (HA Core) would probably also need to be modified for this.

xanibale, 2023-01-20:
If i would get it back until next Christmas, then there should be no problem, as it is now not usable for me anyway.
It would be very nice to use the colour/ white capabilities of the device. If that would make things easier.

erkr, 2023-01-21:
Yes, first fixing normal colour and brightness control would be a great step. Now the colours change when the brightness is adjusted.

erkr, 2023-07-08:
Christmas seem still far away, but we are 6 months further 🙏

rkruisselbrink, 2023-07-26:
I was also thinking about this. Is there some guide or docs on how we can test/debug this? I'm willing to give it a shot and spent some time on this, but i have no idea where to start 😅

d3nnis0, 2023-10-12:
I also own the those Christmas lights and I'am willing to invest some time to get them integrated until Christmas (Or even better some weeks before ;-) ) 

Is there already a kind of draft where I can try to contribute?

rkruisselbrink, 2023-10-24:
> Is there already a kind of draft where I can try to contribute?

Not that i know of. Any idea how to approach this?

erkr, 2023-10-30:
I noticed that the Christmas lighrs where added to the  Deconz integration: https://github.com/Kane610/deconz/issues/198

I hope some expert knows how to translate the information that seems to be available there into a handler for ZHA

Hedda, 2023-12-08:
> I noticed that the Christmas lights where added to the Deconz integration: [Kane610/deconz#198](https://github.com/Kane610/deconz/issues/198)

FYI, they also added support for HG06467 Lidl Silvercrest Christmas light effects to Home Assistant's deCONZ integration now too:

https://github.com/home-assistant/core/pull/102731

https://github.com/home-assistant/core/pull/105155

https://community.home-assistant.io/t/lidl-melinera-christmas-string-hg06467-lidl-livarno-lux/367

https://community.home-assistant.io/t/update-deconz-integration-to-support-flash-with-color/246904/

PS: Support for HG06467 Zigbee ID "TS0601" was as well added to zigbee-herdsman-converters for Zigbee2MQTT back in 2020:

https://github.com/Koenkk/zigbee-herdsman-converters/pull/1799

https://github.com/Koenkk/zigbee-herdsman-converters/issues/1792

enyone, 2023-12-12:
I tried to hack something together last year this same time (one week before Christmas). My neighbors are still not happy so also I am keen to debug any contributions to this draft.

https://gist.github.com/enyone/e1e0182f41ab9924d33fd24d77765aa6

rkruisselbrink, 2024-06-10:
I think we want to keep this open. I didn't have (make) time yet to checkout the implementation by @enyone

Melantrix, 2024-11-30:
> I tried to hack something together last year this same time (one week before Christmas). My neighbors are still not happy so also I am keen to debug any contributions to this draft.
> 
> https://gist.github.com/enyone/e1e0182f41ab9924d33fd24d77765aa6
> 
> See that `0xFC: foundation.ZCLCommandDef(...` I created to issue effect change command and a handler for that `if command_id == 0x00FC:` with a static data `tuya_payload.data = [0x00, 0x0A, 0x30, 0x31, 0x30, 0x30, 0x40, 0x40, 0x40, 0x40, 0x40, 0x40]` to play around.

I don't know if you've looked at this some more, but thanks to this i'm able to choose colors, however the colors i choose don't seem to correspond to the colors i choose (e.g. i choose red in HA, but the light turn yellow). it seems that color temperature is not set correctly. I'm able/willing to help troubleshoot/debug but i have no i idea what i'm looking at in the quirk.. :/ Can you maybe point me in the right direction?

enyone, 2024-12-10:
I still have these lights on my yard's bushes.

For now I've beard with that broken XY-mode, where actual string's LED color is not always quite the one one picks at RGB map palette.

This https://gist.github.com/enyone/e1e0182f41ab9924d33fd24d77765aa6 actually brings only (sightly broken) XY-mode available (where you can pick one led color for all string's LED using RGB map palette). Without that quirk only HS-mode is available (using Hue slider and/or buttons) but it does not work at all. For some reason this device reports HS-mode as only supported mode, whereas that quirk forces the modes XY-mode and CT-mode as supported ones.

Why XY-mode is broken, I don't know. I would guess this device uses some non-8-bit (but less) color depth.

Another thing that quirk tries to create support for is effects (flashing multi-color effects Lidl's Android App specifies). I have not found any documentation or similar implementation elsewhere to use as a reference, and fall back to reverse-engineering bytes using Lidl's Android App as a source of those bytes. One example set of bytes is present in that quirk code line 98:
`tuya_payload.data = [0x00, 0x0A, 0x30, 0x31, 0x30, 0x30, 0x40, 0x40, 0x40, 0x40, 0x40, 0x40]`

That array of bytes is send as a payload data to device using `TuyaManufCluster.Command` with:
`command_id=TUYA_SET_DATA (0x00)`
`function=TUYA_EFFECT_COMMAND (0x0603)`

See more from https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/__init__.py#L324

That array of bytes is just an example and more different samples (by reverse-engineering Lidl's Android App sent payloads) needs to be observed to create something working.

erkr, 2024-12-10:
z2m seems to support the light effects. So someone knowledgeable can reuse that for zha I guess

smoltron, 2024-12-23:
This seems to be broken in ZHA now. Last year it still kind of worked, I could control color. Now only on-off works as expected. The color setting/brightness changes colors randomly.

TjakBak, 2025-12-03:
Ho Ho Ho, Christmas is around the corner again, and Santa asked me to revive this 3-year-old thread to see if anyone following has managed to fix the quirk and get the effects selection working:)

Or is it officially time to throw these out and buy new ones? 

Anyway, Happy holidays everyone!!!

smoltron, 2025-12-04:
If one uses a device one week a year, it will take so many years until the device is done. Therefore, I would like to reuse this set as long as possible. It is also ecological.

enyone, 2026-01-06:
I still have these but my neighbor does not like blinking effects. Thus I must have been downgraded my expectations of ZHA support of this device and done nothing new.

Hedda, 2026-08-24:
> I got _TZE200_s8gkrkxk / TS0601 working with ZHA, including color, color temperature and all 16 effects.
> Custom quirk and details: https://github.com/ptc/Melinera_HG06467

Nice! Not sure they still sell this as a seasonal item at Lidl/Melinera but would be nice if you could try submit it as a pull request to this repo:

* https://github.com/zigpy/zha-device-handlers/pulls

See:

* https://github.blog/developer-skills/github/beginners-guide-to-github-creating-a-pull-request/

  * https://docs.github.com/en/pull-requests/how-tos/create-pull-requests/creating-a-pull-request

    * https://docs.github.com/en/pull-requests/reference/pull-requests

If you have a paid subscription to am AI/LLM then you could use it as a coding agent and have it guide you through the PR process:

* https://github.com/zigpy/zha-device-handlers/blob/dev/.github/copilot-instructions.md

Just make sure follow their newly published AI Policy:

* https://developers.home-assistant.io/blog/2026/07/20/ai-policy/

  * https://developers.home-assistant.io/docs/ai_policy/

erkr, 2026-08-24:
> I got _TZE200_s8gkrkxk / TS0601 working with ZHA, including color, color temperature and all 16 effects.
> Custom quirk and details: https://github.com/ptc/Melinera_HG06467

Thanks, it works great with my Christmas LED string 🎉