zigpy/zha-device-handlers #1941: [Device Support Request] Tuya TS110E (1 + 2 gang dimmer device)

maldegemev, 2022-11-19:
There is already a quirk for the TS110E (1 gang) dimmer device. I have some 1 and 2 gang devices and they are not working with the quirk which is available. When I create a custom quirk from the ts110e.py and add the signature of the 1 gang device I have, it is not apply due to invalid input cluster on the endpoint. The signature of the 1 gang device is:

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00",
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_zxbtub8r",
  "model": "TS110E",
  "class": "zigpy.device.Device"
}

Is it possible to add support for this device so I can add the min and max level dim values?

maldegemev, 2022-11-19:
This is the signature from the 2 gang device:

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_bvrlqyj7",
  "model": "TS0002",
  "class": "zhaquirks.tuya.ts000x.Switch_2G_GPP"
}

javicalle, 2022-11-19:
Any link to the manufacturer device or vendor site?

That is the proposal quirk for the 1 gang device.
It's a blind coded version because is not similar to the others `TS110E` devices. Is more similar to the `TS0601` and the quirk is adapted from them:

<details>
<summary>ts110e_2.py</summary>

```python
"""Tuya Dimmer TS110E."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaDimmerSwitch,
    TuyaZBExternalSwitchTypeCluster,
)
from zhaquirks.tuya.mcu import (
    TuyaInWallLevelControl,
    TuyaLevelControlManufCluster,
    TuyaOnOff,
    TuyaOnOffNM,
)

class DimmerSwitchWithNeutral1GangVar02(TuyaDimmerSwitch):
    """Tuya Dimmer Switch Module With Neutral 1 Gang."""

    signature = {
        MODELS_INFO: [("_TZ3210_zxbtub8r", "TS110E")],
        ENDPOINTS: {
            1: {
                # "profile_id": 260,
                # "device_type": "0x0101",
                # "in_clusters": ["0x0000","0x0003","0x0004","0x0005","0x0006","0x0008","0xef00",],
                # "out_clusters": ["0x000a","0x0019"]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMABLE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # "profile_id": 41440,
                # "device_type": "0x0061",
                # "in_clusters": [],
                # "out_clusters": ["0x0021"]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOff,
                    TuyaInWallLevelControl,
                    TuyaLevelControlManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</details>

Remove any previous quirk (for that device) from the folder and create a new file with the proposed one.
I'm not expecting to work at the first try but can generate the logs we need to see what is happening. Please, attach the relevant logs.

javicalle, 2022-11-19:
The 2 gang device is a `TS0002` device and already loads a quirk. Which is the problem here? What is working and what not?
Logs from the device are needed to understand wwhat the problem can be.

maldegemev, 2022-11-20:
The devices (1 and 2 gang) are from Aliexpress (https://nl.aliexpress.com/item/1005004580552856.html?spm=a2g0o.order_list.0.0.379879d2AAeWhS&gatewayAdapt=glo2nld). The 1 gang is working with zha, but is is not possible to set-up the min and max dim value through zha. This is possible when using this device with the tuya integration. The 2-gang have some problems with the status of the device. When the light is switched on, the status in home assistant is on for some time. The status changes to off after a while. The physical switch isn't switched off and the light is still on at the time of the status change.

maldegemev, 2022-11-20:
I tried the quirk you suggested. It isn't working yet. Here I have som logging from home assistant:

2022-11-20 10:40:12.533 INFO (MainThread) [zigpy.device] [0x8c56] Read model 'TS110E' and manufacturer '_TZ3210_zxbtub8r' from <Endpoint id=1 in=[groups:0x0004, scenes:0x0005, identify:0x0003, on_off:0x0006, level:0x0008, None:0xEF00, basic:0x0000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
2022-11-20 10:40:12.534 INFO (MainThread) [zigpy.device] [0x8c56] Discovered basic device information for <Device model='TS110E' manuf='_TZ3210_zxbtub8r' nwk=0x8C56 ieee=a4:c1:38:11:d4:4f:34:96 is_initialized=True>
2022-11-20 10:40:12.534 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model='TS110E' manuf='_TZ3210_zxbtub8r' nwk=0x8C56 ieee=a4:c1:38:11:d4:4f:34:96 is_initialized=True>
2022-11-20 10:40:12.535 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZ3210_zxbtub8r TS110E (a4:c1:38:11:d4:4f:34:96)
2022-11-20 10:40:12.536 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ts110_2.DimmerSwitchWithNeutral1GangVar02'>
2022-11-20 10:40:12.536 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint
2022-11-20 10:40:12.536 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ts110e_custom.DimmerSwitchWithNeutral1Gang'>
2022-11-20 10:40:12.537 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint
2022-11-20 10:40:12.537 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts110e.DimmerSwitchWithNeutral1Gang'>
2022-11-20 10:40:12.537 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint
2022-11-20 10:40:12.538 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace2'>
2022-11-20 10:40:12.539 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2, 242} {1, 242}
2022-11-20 10:40:12.539 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace1'>
2022-11-20 10:40:12.540 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2, 41, 42, 242, 51, 21, 31} {1, 242}
2022-11-20 10:40:12.540 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
2022-11-20 10:40:12.540 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1, 242}
2022-11-20 10:40:12.541 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
2022-11-20 10:40:12.541 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1, 242}
2022-11-20 10:40:12.541 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts0201.MoesTemperatureHumidtySensorWithScreen'>
2022-11-20 10:40:12.542 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 242}
2022-11-20 10:40:12.542 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
2022-11-20 10:40:12.542 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 242}
2022-11-20 10:40:12.542 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
2022-11-20 10:40:12.543 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 242}
2022-11-20 10:40:12.543 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
2022-11-20 10:40:12.543 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 242}
2022-11-20 10:40:12.544 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
2022-11-20 10:40:12.544 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {11, 13} {1, 242}
2022-11-20 10:40:12.553 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'button' component -> 'ZHAIdentifyButton' using ['identify']
2022-11-20 10:40:12.554 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'RSSISensor' using ['basic']
2022-11-20 10:40:12.554 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'LQISensor' using ['basic']
2022-11-20 10:40:12.559 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'OnOffTransitionTimeConfigurationEntity' using ['level']
2022-11-20 10:40:12.560 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'OnLevelConfigurationEntity' using ['level']
2022-11-20 10:40:12.560 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'OnTransitionTimeConfigurationEntity' using ['level']
2022-11-20 10:40:12.561 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'OffTransitionTimeConfigurationEntity' using ['level']
2022-11-20 10:40:12.561 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'DefaultMoveRateConfigurationEntity' using ['level']
[... cut ...]
2022-11-20 10:40:12.562 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'select' component -> 'ZHAStartupOnOffSelectEntity' using ['on_off']
2022-11-20 10:40:12.567 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0x8C56:a4:c1:38:11:d4:4f:34:96 entering async_device_initialized - is_new_join: True
2022-11-20 10:40:12.568 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0x8C56:a4:c1:38:11:d4:4f:34:96 has joined the ZHA zigbee network

maldegemev, 2022-11-20:
Sorry, I posted the wrong signature from the 2-gang device. I replaced it with the right one.

maldegemev, 2022-11-20:
[image]

This are the settings for the min and max dim value which you can set when using the tuya integration. These aren't available when using the zha integration.

javicalle, 2022-11-20:
Please, edit your comments and put the signature and logs inside ` ``` ` marks like this:
````
```
HERE all the code/logs
```
````

The min/max dim value is not configurable in HA UI (not yet).

Is the 1 gang device working with the proposed quirk or without it? The proposed quirk will not make the min/max available in HA.

The proposed quirk with the 2 gang would be:

<details open>
<summary>ts110e_2.py</summary>

```python
"""Tuya Dimmer TS110E."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaDimmerSwitch,
    TuyaZBExternalSwitchTypeCluster,
)
from zhaquirks.tuya.mcu import (
    TuyaInWallLevelControl,
    TuyaLevelControlManufCluster,
    TuyaOnOff,
    TuyaOnOffNM,
)

class DimmerSwitchWithNeutral1GangVar02(TuyaDimmerSwitch):
    """Tuya Dimmer Switch Module With Neutral 1 Gang."""

    signature = {
        MODELS_INFO: [("_TZ3210_zxbtub8r", "TS110E")],
        ENDPOINTS: {
            1: {
                # "profile_id": 260,
                # "device_type": "0x0101",
                # "in_clusters": ["0x0000","0x0003","0x0004","0x0005","0x0006","0x0008","0xef00",],
                # "out_clusters": ["0x000a","0x0019"]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMABLE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
[... 100 lines cut ...]
                    TuyaInWallLevelControl,
                    TuyaLevelControlManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOff,
                    TuyaInWallLevelControl,
                    TuyaLevelControlManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</summary>

MattWestb, 2022-11-20:
> The proposed quirk will not make the min/max available in HA.

You can manage the attributes on the `TuyaInWallLevelControl` cluster with normal Zigbee cluster commands / attributes.

javicalle, 2022-11-20:
Well, the `TuyaInWallLevelControl` only defines a `minimum_level` attribute, but I believe that the attribute contains both (min and max). If I can get some time I'll take a look at the Tuya integration.

maldegemev, 2022-11-20:
If it is possible through the TuyaInwallLevelControl cluster, it's fine for sure. I tried the new quirk but it gives still a input cluster mismatch. 

2022-11-20 12:56:28.960 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x3ACB), dst_ep=0, source_route=None, extended_timeout=False, tsn=44, profile_id=0, cluster_id=<ZDOCmd.Simple_Desc_req: 0x0004>, data=Serialized[b',\xcb:\xf2'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-11-20 12:56:28.961 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteChk.Req(Dst=0x3ACB, RtStatus=<RouteStatus.ACTIVE: 1>, Options=<RouteOptions.NO_ROUTE_CACHE|MTO_ROUTE: 3>)
2022-11-20 12:56:28.968 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteChk.Rsp(Status=<RoutingStatus.SUCCESS: 0>)
2022-11-20 12:56:28.970 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x3ACB), DstEndpoint=0, DstPanId=0x0000, SrcEndpoint=0, ClusterId=4, TSN=44, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=0, Data=b'\x2C\xCB\x3A\xF2')
2022-11-20 12:56:28.983 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-11-20 12:56:29.012 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SimpleDescRsp.Callback(Src=0x3ACB, Status=<Status.SUCCESS: 0>, NWK=0x3ACB, SimpleDescriptor=SizePrefixedSimpleDescriptor(endpoint=242, profile=41440, device_type=97, device_version=0, input_clusters=[], output_clusters=[33]))
2022-11-20 12:56:29.013 DEBUG (MainThread) [zigpy_znp.api] Command was not handled
2022-11-20 12:56:29.014 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MsgCbIncoming.Callback(Src=0x3ACB, IsBroadcast=<Bool.false: 0>, ClusterId=32772, SecurityUse=0, TSN=44, MacDst=0x0000, Data=b'\x00\xCB\x3A\x0A\xF2\xE0\xA1\x61\x00\x00\x00\x01\x21\x00')
2022-11-20 12:56:29.015 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x3ACB), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=44, profile_id=0, cluster_id=32772, data=Serialized[b',\x00\xcb:\n\xf2\xe0\xa1a\x00\x00\x00\x01!\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-11-20 12:56:29.016 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0x3ACB ieee=a4:c1:38:11:d4:4f:34:96 is_initialized=False> from ep 0 to ep 0, cluster 32772: b',\x00\xcb:\n\xf2\xe0\xa1a\x00\x00\x00\x01!\x00'
2022-11-20 12:56:29.021 INFO (MainThread) [zigpy.endpoint] [0x3acb:242] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=242, profile=41440, device_type=97, device_version=0, input_clusters=[], output_clusters=[33])
2022-11-20 12:56:29.023 DEBUG (MainThread) [zigpy.zcl] [0x3ACB:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), tsn=45, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-11-20 12:56:29.026 DEBUG (MainThread) [zigpy.zcl] [0x3ACB:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4, 5])
2022-11-20 12:56:29.030 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x3ACB), dst_ep=1, source_route=None, extended_timeout=False, tsn=45, profile_id=260, cluster_id=0, data=Serialized[b'\x00-\x00\x04\x00\x05\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-11-20 12:56:29.032 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x3ACB), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=0, TSN=45, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=0, Data=b'\x00\x2D\x00\x04\x00\x05\x00')
2022-11-20 12:56:29.049 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-11-20 12:56:29.054 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=45)
2022-11-20 12:56:29.078 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x7464](TLSR82xx): Device seen - marking the device available and resetting counter
2022-11-20 12:56:29.078 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x7464](TLSR82xx): Update device availability -  device available: True - new availability: True - changed: False
2022-11-20 12:56:29.102 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x3ACB, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=69, SecurityUse=<Bool.false: 0>, TimeStamp=5568277, TSN=0, Data=b'\x18\x2D\x01\x04\x00\x00\x42\x10\x5F\x54\x5A\x33\x32\x31\x30\x5F\x7A\x78\x62\ [... cut ...]
2022-11-20 12:56:29.706 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x3ACB), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=8, data=Serialized[b'\x188\x01\x00\x00\x86'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=69, rssi=None)
2022-11-20 12:56:29.706 DEBUG (MainThread) [zigpy.zcl] [0x3ACB:1:0x0008] Received ZCL frame: b'\x188\x01\x00\x00\x86'

javicalle, 2022-11-20:
Please, format your comments.

The device that is failing is the `_TZ3210_zxbtub8r` (the 1 gang) and I haven't changed this part.

I have just added the new quirk for the `_TZ3210_wdexaypg` (and its signature). Is the 2 gang device working?

maldegemev, 2022-11-21:
I tested the 2-gang device but when I remove and re-add the device the quirk isn't applied. Still got the input cluster mismatch error.

Here is a part of the logging:

2022-11-21 19:27:22.827 INFO (MainThread) [zigpy.device] [0x985e] Read model 'TS110E' and manufacturer '_TZ3210_wdexaypg' from <Endpoint id=1 in=[groups:0x0004, scenes:0x0005, identify:0x0003, on_off:0x0006, level:0x0008, None:0xEF00, basic:0x0000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
2022-11-21 19:27:22.828 INFO (MainThread) [zigpy.device] [0x985e] Discovered basic device information for <Device model='TS110E' manuf='_TZ3210_wdexaypg' nwk=0x985E ieee=a4:c1:38:e9:90:24:98:d6 is_initialized=True>
2022-11-21 19:27:22.828 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model='TS110E' manuf='_TZ3210_wdexaypg' nwk=0x985E ieee=a4:c1:38:e9:90:24:98:d6 is_initialized=True>
2022-11-21 19:27:22.829 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZ3210_wdexaypg TS110E (a4:c1:38:e9:90:24:98:d6)
2022-11-21 19:27:22.829 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ts110e_2.DimmerSwitchWithNeutral2GangVar02'>
2022-11-21 19:27:22.830 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint
2022-11-21 19:27:22.830 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace2'>
2022-11-21 19:27:22.830 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2022-11-21 19:27:22.830 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace1'>
2022-11-21 19:27:22.830 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2, 41, 42, 242, 51, 21, 31} {1, 2, 242}
2022-11-21 19:27:22.830 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
2022-11-21 19:27:22.831 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1, 2, 242}
2022-11-21 19:27:22.831 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
2022-11-21 19:27:22.831 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1, 2, 242}
2022-11-21 19:27:22.831 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts0201.MoesTemperatureHumidtySensorWithScreen'>
2022-11-21 19:27:22.831 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 2, 242}
2022-11-21 19:27:22.831 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
2022-11-21 19:27:22.831 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 2, 242}
2022-11-21 19:27:22.832 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
2022-11-21 19:27:22.832 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 2, 242}
2022-11-21 19:27:22.832 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
2022-11-21 19:27:22.832 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 2, 242}
2022-11-21 19:27:22.832 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
2022-11-21 19:27:22.832 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {11, 13} {1, 2, 242}
2022-11-21 19:27:22.836 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'button' component -> 'ZHAIdentifyButton' using ['identify']
2022-11-21 19:27:22.836 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'RSSISensor' using ['basic']
2022-11-21 19:27:22.836 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'LQISensor' using ['basic']
2022-11-21 19:27:22.837 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'OnOffTransitionTimeConfigurationEntity' using ['level']
2022-11-21 19:27:22.837 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'OnLevelConfigurationEntity' using ['level']
2022-11-21 19:27:22.837 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'OnTransitionTimeConfigurationEntity' using ['level']
2022-11-21 19:27:22.837 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'OffTransitionTimeConfigurationEntity' using ['level']
2022-11-21 19:27:22.838 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'DefaultMoveRateConfigurationEntity' using ['level']
2022-11-21 19:27:22.838 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'StartUpCurrentLevelConfigurationEntity' using ['level']
2022-11-21 19:27:22.838 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'select' component -> 'ZHAStartupOnOffSelectEntity' using ['on_off']
2022-11-21 19:27:22.840 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'button' component -> 'ZHAIdentifyButton' using ['identify']
[... cut ...]
2022-11-21 19:27:23.661 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x985E:2:0x0006]: Successfully configured reporting for '{'on_off': (0, 900, 1)}' on 'on_off' cluster: [ConfigureReportingResponseRecord(status=<Status.SUCCESS: 0>)]
2022-11-21 19:27:23.662 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x985E:2:0x0006]: finished channel configuration
2022-11-21 19:27:25.336 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.BindRsp.Callback(Src=0x985E, Status=<Status.SUCCESS: 0>)
2022-11-21 19:27:25.337 DEBUG (MainThread) [zigpy_znp.api] Command was not handled

closed 2023-07-10 by github-actions[bot]