zigpy/zha-device-handlers #1943: [BUG] Symfonisk sound remote - no events

Proche72, 2022-11-20:
**Describe the bug**
After connecting the Symfonisk remote to ZHA (over Conbee II) it is not responding - no zha_event created nor automation working.

**Expected behavior**
Some zha event to show up. Or functioning automation.

```yaml
SYMFONISK Sound Controller
by IKEA of Sweden
Connected via [dresden elektronik ConBee II](http://192.168.0.122:8123/config/devices/device/1ff8faaddfb643fa87108067d537aea0)
Firmware: 0x21024631
Zigbee info
IEEE: 94:34:69:ff:fe:65:98:bf
Nwk: 0xba57
Device Type: EndDevice
LQI: 255
RSSI: -56
Last Seen: 2022-11-20T17:12:12
Power Source: Battery or Unknown
Quirk: zhaquirks.ikea.symfonisk.IkeaSYMFONISK1
```

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Prague",
    "os_name": "Linux",
    "os_version": "5.15.61-v8",
    "supervisor": "2022.10.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "garbage_collection": {
      "version": "4.8.3",
      "requirements": [
        "python-dateutil>=2.8.2"
      ]
    },
    "xiaomi_miot": {
      "version": "0.7.1",
      "requirements": [
        "construct==2.10.56",
        "python-miio>=0.5.6",
        "micloud>=0.3"
      ]
    },
    "xiaomi_miio_airpurifier": {
      "version": "2022.3.3.2",
      "requirements": [
        "construct==2.10.56",
        "python-miio>=0.5.11"
      ]
    },
    "browser_mod": {
      "version": "2.1.2",
      "requirements": []
    },
    "skodaconnect": {
      "version": "v1.0.72",
[... 338 lines cut ...]
            "endpoint_attribute": "on_off",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0008": {
            "endpoint_attribute": "level",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

javicalle, 2022-11-21:
@Proche72 check the comment from https://github.com/zigpy/zha-device-handlers/issues/1949#issuecomment-1322331562

Proche72, 2022-11-22:
@javicalle thanks, I will try the reconfiguration and post results here.

Proche72, 2022-11-22:
@javicalle no luck, the reconfiguration failed:
[image]

javicalle, 2022-11-22:
Other user is reporting as working now:
* https://github.com/zigpy/zha-device-handlers/issues/1949#issuecomment-1324094890

An reddit user describe how keep awake the device during the proces:
* https://www.reddit.com/r/tradfri/comments/dr1cr2/tough_time_pairing_symfonisk_remote_found_a_fix/

Proche72, 2022-11-23:
No luck with mine. I tried also this:
-Take remote to be right beside hub
-Open up new window with developer tools in HA and subscribe to ‘zha_event’ (to see if you are able to sense remote inputs)
-Delete remote from ZHA
-Take battery out of remote
-Put battery back in
-Start pairing in ZHA
-Hold down paring button on remote
-Once remote pairs, rotate and click button to see if you see any activity from zha_event

It didn't even pair this way.

I have paired directly to the bulb I want to use it with so it works now but it is outside of HA and ZHA.

MattWestb, 2022-11-23:
> -Hold down paring button on remote

This is paring with touch link = direct to one light.

For paring you shall fast pressing the paring button fast 4 times and the LED start pulsing and start searching for open networks.

Proche72, 2022-11-23:
Yes, but it still doesn't work after pair this way and reconfiguration.

Proche72, 2022-11-24:
@MattWestb yep, unfortunately I lose also the light from zha if I link it directly so not great solution too :(

Proche72, 2022-12-04:
Latest update of zha fixed the issue. The symfonisk rottary switch now works flawlessly.

closed 2022-12-04 by Proche72