zigpy/zha-device-handlers #1944: [Device Support Request] Tuyau TZE200 yjjdcqsq

tho071088, 2022-11-20:
**Cheap Tuyau ZigBee Temperature Sensor.**
Device is detected but no sensors are fond.
[Link to device on Aliexpress ](https://a.aliexpress.com/_EuvXvTP) 

**Temperature, Humidity and Battery Status**
The three sensors are displayed on HA.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_yjjdcqsq",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
          "id": 81
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0004": {
            "endpoint_attribute": "groups",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0005": {
            "endpoint_attribute": "scenes",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xef00": {
            "endpoint_attribute": null,
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0001": {
                "attribute_name": "app_version",
                "value": 72
              },
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "_TZE200_yjjdcqsq"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "TS0601"
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
   }
 }
```

</details>

<details>
<summary>Additional logs</summary>

```
None
```

</details>

javicalle, 2022-11-20:
It can fit in the current `TuyaTempHumiditySensor` quirk.

Configure your local quirk folder. If you don't know how to set the custom quirk this is a good guide to follow:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Copy inside the current `ts0601_sensor.py` file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_sensor.py

Modify the `TuyaTempHumiditySensor` adding your device to the signature:
```python
    signature = {
        # <SimpleDescriptor endpoint=1, profile=260, device_type=81
        # device_version=1
        # input_clusters=[4, 5, 61184, 0]
        # output_clusters=[25, 10]>
        MODELS_INFO: [
            ("_TZE200_bjawzodf", "TS0601"),
            ("_TZE200_yjjdcqsq", "TS0601"),
        ],
```

Save changes, restart HA and repair the device.
There are 2 versions of the quirk, double check if the humidity values are fine.

tho071088, 2022-11-21:
Thanks @javicalle 

I'm still getting a problem :

I tried with TuyaTempHumiditySensor but humidity was false (10 times lower than expected), and battery is not working.
[image]

So I tried adding in TuyaTempHumiditySensor_Square, but it does not seems to work at all.
The quirk is not applied.
[image]

class TuyaTempHumiditySensor_Square(CustomDevice):

    # RelativeHumidity multiplier
    # RH_MULTIPLIER = 100

    signature = {
        MODELS_INFO: [
            ("_TZE200_yjjdcqsq", "TS0601"),
            ("_TZE200_qoy0ekbd", "TS0601"),
            ("_TZE200_znbl8dj5", "TS0601"),
        ],

Am I missing something ?

javicalle, 2022-11-21:
Do you have removed first your device from the `TuyaTempHumiditySensor`?
Also remove any `__pycache__` folder from your local quirk (double check that part).
Restart HA and repair the device.

tho071088, 2022-11-21:
Yes I tried twice, here is the py file.
Cache file deleted and HA restarted.
[ts0601_sensor.py.txt](https://github.com/zigpy/zha-device-handlers/files/10055726/ts0601_sensor.py.txt)

javicalle, 2022-11-21:
Do you have repaired the device?
To be sure, remove the device from HA, wait for a minute and pair the device again.
Look at the pairing logs if there's a cluster mismatch. If there is a problem, attach your logs.

tho071088, 2022-11-21:
There seems to be something in the logs :

Checking quirks for _TZE200_yjjdcqsq TS0601 (a4:c1:38:02:22:49:cb:32)
Considering <class 'ts0601_sensor.TuyaTempHumiditySensor_Square'>
Fail because device_type mismatch on at least one endpoint

Full log file : 
[log_zha.txt](https://github.com/zigpy/zha-device-handlers/files/10056096/log_zha.txt)

Should I create a new class like TuyaTempHumiditySensor ?
Maybe TuyaTempHumiditySensor_V2 with RH_MULTIPLIER = 100 ?
(There is still the battery issue)

tho071088, 2022-11-21:
I added this class, and it is working : 

class TuyaTempHumiditySensor_Rectangle(CustomDevice):
    """Custom device representing tuya temp and humidity sensor with e-ink screen."""

    # RelativeHumidity multiplier
    #RH_MULTIPLIER = 100

    signature = {
        # <SimpleDescriptor endpoint=1, profile=260, device_type=81
        # device_version=1
        # input_clusters=[4, 5, 61184, 0]
        # output_clusters=[25, 10]>
        MODELS_INFO: [("_TZE200_yjjdcqsq", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TemperatureHumidityManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    TemperatureHumidityManufCluster,  # Single bus for temp, humidity, and battery
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                    TuyaPowerConfigurationCluster2AAA,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

Should I do something more now ?

javicalle, 2022-11-21:
I have already taken care of it and I have taken the opportunity to make some adjustments.

Thanks for all the job @tho071088

closed 2022-11-21 by dmulcahey

javicalle, 2022-11-21:
@tho071088 be aware to remove your local quirk with the next monthly version of HA.

javicalle, 2023-03-28:
It depends on your device signature. Because of that always is better if you open a new issue with all the info from your device.
You can try to enable de local quirk folder and put the current `ts0601_sensor.py` file.
Then I would suggest to add your device to the `TuyaTempHumiditySensorVar03` quirk:
* https://github.com/zigpy/zha-device-handlers/blob/1fed017136e64274e8665a22ed7bd3556c1e0729/zhaquirks/tuya/ts0601_sensor.py#L175

Save changes, restart HA and repair yur device to see if the device signature changes.

pete1019, 2023-03-28:
> `TuyaTempHumiditySensorVar03`

Thanks. But Humidity, Tmperature and battery shown: UNKNOWN.
Time and Date was not updated either.
Suggestions?

Husk84, 2023-08-26:
Same issue here:
[image]
Here my signature:
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_yjjdcqsq",
  "model": "TS0601",
  "class": "ts0601_sensor.TuyaTempHumiditySensorVar03"
}

javicalle, 2023-08-27:
Why are you using a local quirk? It should not be necessary because the quirk is already in ZHA.
Remove the local quirk and repair the device.

Blackspell01, 2025-01-02:
> Why are you using a local quirk? It should not be necessary because the quirk is already in ZHA. Remove the local quirk and repair the device.

Without the quirk it does not show any entities

PR #1951 (2022-11-21): Add `_TZE200_yjjdcqsq` Tuya sensor
Add new device signature for the `_TZE200_yjjdcqsq` sensor

Fix: #1944
commit ad2eba0d (2022-11-21): Add `_TZE200_yjjdcqsq` Tuya sensor (#1951)

* Add `_TZE200_yjjdcqsq` Tuya sensor

Add new device signature for the `_TZE200_yjjdcqsq` sensor

Fix: #1944

* Restore removed clusters

No sure if are functional but I don't like to remove without knowing.