zigpy/zha-device-handlers #1945: [Device Support Request] tuya curtain motor _TZE200_xu4a5rhj

jeongsik-kim, 2022-11-20:
**Is your feature request related to a problem? Please describe.**
ts0601_cover.py needs a new quirk added to support this device properly.
Currently, it pairs with no entities attached.

**Describe the solution you'd like**
Be able to pair the device and control my curtains through HA.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_xu4a5rhj",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2022.11.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Asia/Seoul",
    "os_name": "Linux",
    "os_version": "4.4.180+",
    "supervisor": "2022.11.dev0802",
    "host_os": null,
    "docker_version": "20.10.3",
    "chassis": null,
    "run_as_root": true
  },
[... 325 lines cut ...]
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-11-20:
Another Tuya cover flavor...

My proposal quirk would be:

<details open>
<summary>ts0601_cover_2.py</summary>

```python
"""Tuya based cover and blinds."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)

class TuyaCover(TuyaWindowCover):
    """Tuya blind cover motor."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_xu4a5rhj", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</details>

Create a new `ts0601_cover_2.py` file in your custom quirk folder. If you don't know how to set the custom quirk this is a good guide to follow:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Save the file, restart HA and check that the device signature have changed.

It may be that the cover controls are inverted. In this case, if you have a cover remote, follow the manufacturer instructions and invert the cover direction.

jeongsik-kim, 2022-11-20:
> Another Tuya cover flavor...
> 
> My proposal quirk would be:
> 
> ts0601_cover_2.py
> ```python
> """Tuya based cover and blinds."""
> from zigpy.profiles import zha
> from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
> 
> from zhaquirks.const import (
>     DEVICE_TYPE,
>     ENDPOINTS,
>     INPUT_CLUSTERS,
>     MODELS_INFO,
>     OUTPUT_CLUSTERS,
>     PROFILE_ID,
> )
> from zhaquirks.tuya import (
>     TuyaManufacturerWindowCover,
>     TuyaManufCluster,
>     TuyaWindowCover,
>     TuyaWindowCoverControl,
> )
> 
> 
> class TuyaCover(TuyaWindowCover):
>     """Tuya blind cover motor."""
> 
>     signature = {
>         MODELS_INFO: [
>             ("_TZE200_xu4a5rhj", "TS0601"),
>         ],
>         ENDPOINTS: {
>             1: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     TuyaManufCluster.cluster_id,
>                 ],
>                 OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
>             },
>             242: {
>                 PROFILE_ID: 41440,
>                 DEVICE_TYPE: 97,
>                 INPUT_CLUSTERS: [],
>                 OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
>             },
>         },
>     }
>     replacement = {
>         ENDPOINTS: {
>             1: {
>                 DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     TuyaManufacturerWindowCover,
>                     TuyaWindowCoverControl,
>                 ],
>                 OUTPUT_CLUSTERS: [Ota.cluster_id],
>             },
>             242: {
>                 PROFILE_ID: 41440,
>                 DEVICE_TYPE: 97,
>                 INPUT_CLUSTERS: [],
>                 OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
>             },
>         },
>     }
> ```
> 
> Create a new `ts0601_cover_2.py` file in your custom quirk folder. If you don't know how to set the custom quirk this is a good guide to follow:
> 
> * [Editing (ZHA) quirks in a (HA) container or local. #693 (comment)](https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274)
> 
> Save the file, restart HA and check that the device signature have changed.
> 
> It may be that the cover controls are inverted. In this case, if you have a cover remote, follow the manufacturer instructions and invert the cover direction.

thanks 
The cover entities is created but it doesn't work.

jeongsik-kim, 2022-11-20:
Apply custom quirk device signature

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102",
        "0xef00"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_xu4a5rhj",
  "model": "TS0601",
  "class": "ts0601_cover_2.TuyaCover"
}
```

javicalle, 2022-11-20:
Try to repair the device. If it doesn't work, enable the debug logs and attach the relevant info:
https://www.home-assistant.io/integrations/zha/#debug-logging

jeongsik-kim, 2022-11-21:
> Try to repair the device. If it doesn't work, enable the debug logs and attach the relevant info: https://www.home-assistant.io/integrations/zha/#debug-logging

<details>

<summary>Debug-logging</summary>

```yaml
2022-11-21 14:00:04.601 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.still_target_energy, old_state=<state sensor.still_target_energy=6; unit_of_measurement=%, friendly_name=Still Target Energy @ 2022-11-21T14:00:03.529359+09:00>, new_state=<state sensor.still_target_energy=5; unit_of_measurement=%, friendly_name=Still Target Energy @ 2022-11-21T14:00:04.601319+09:00>>
2022-11-21 14:00:07.282 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x3C49:1:0x0b04]: async_update
2022-11-21 14:00:07.282 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x3C49:1:0x0b04]: Reading attributes in chunks: ['active_power', 'rms_voltage']
2022-11-21 14:00:07.288 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xFF40:1:0x0b04]: async_update
2022-11-21 14:00:07.288 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xFF40:1:0x0b04]: Reading attributes in chunks: ['active_power', 'rms_voltage']
2022-11-21 14:00:07.291 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x65EF:1:0x0b04]: async_update
2022-11-21 14:00:07.292 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x65EF:1:0x0b04]: Reading attributes in chunks: ['active_power', 'rms_voltage']
2022-11-21 14:00:07.297 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC1DE:1:0x0b04]: async_update
2022-11-21 14:00:07.298 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC1DE:1:0x0b04]: Reading attributes in chunks: ['active_power', 'rms_voltage']
2022-11-21 14:00:08.783 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.still_target_energy, old_state=<state sensor.still_target_energy=5; unit_of_measurement=%, friendly_name=Still Target Energy @ 2022-11-21T14:00:04.601319+09:00>, new_state=<state sensor.still_target_energy=4; unit_of_measurement=%, friendly_name=Still Target Energy @ 2022-11-21T14:00:08.783634+09:00>>
2022-11-21 14:00:09.487 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xC1DE](lumi.plug.mmeu01): Device seen - marking the device available and resetting counter
2022-11-21 14:00:09.488 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xC1DE](lumi.plug.mmeu01): Update device availability -  device available: True - new availability: True - changed: False
2022-11-21 14:00:10.601 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.espresense_door_uptime, old_state=<state sensor.espresense_door_uptime=305; unit_of_measurement=s, friendly_name=ESPresense Door Uptime @ 2022-11-21T13:59:55.651881+09:00>, new_state=<state sensor.espresense_door_uptime=320; unit_of_measurement=s, friendly_name=ESPresense Door Uptime @ 2022-11-21T14:00:10.601242+09:00>>
2022-11-21 14:00:10.602 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=binary_sensor.espresense_door, old_state=<state binary_sensor.espresense_door=on; ip=10.0.0.24, uptime=305, firm=esp32, rssi=-73, ver=v3.1.0, adverts=13, seen=11, reported=8, freeHeap=95260, maxAllocHeap=81908, memFrag=14.01637623, scanHighWater=1044, reportHighWater=1048, device_class=connectivity, friendly_name=ESPresense Door @ 2022-11-21T13:56:27.696473+09:00>, new_state=<state binary_sensor.espresense_door=on; ip=10.0.0.24, uptime=320, firm=esp32, rssi=-72, ver=v3.1.0, adverts=13, seen=11, reported=8, freeHeap=95260, maxAllocHeap=81908, memFrag=14.01637623, scanHighWater=1044, reportHighWater=1048, device_class=connectivity, friendly_name=ESPresense Door @ 2022-11-21T13:56:27.696473+09:00>>
2022-11-21 14:00:12.049 DEBUG (MainThread) [zigpy_deconz.api] Received command device_state_changed[<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-11-21 14:00:12.049 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-11-21 14:00:12.050 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, <DataIndicationFlags.Always_Use_NWK_Source_Addr: 1>)
2022-11-21 14:00:12.053 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_indication[30, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0x3c49>, 1, 260, 6, b'\x18\xbd\n\x00\x00\x10\x00', 0, 175, 255, 17, 94, 234, 2, -79]
2022-11-21 14:00:12.054 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [30, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0x3c49>, 1, 260, 6, b'\x18\xbd\n\x00\x00\x10\x00', 0, 175, 255, 17, 94, 234, 2, -79]
2022-11-21 14:00:12.054 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x3C49), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=6, data=Serialized[b'\x18\xbd\n\x00\ [... cut ...] 56, 5800, 4336, 5185, 4336]], is_empty=False, map_index=1, calibration_points=[{'vacuum': {'x': 0, 'y': 0}, 'map': {'x': 448, 'y': 832}}, {'vacuum': {'x': 1000, 'y': 0}, 'map': {'x': 528, 'y': 832}}, {'vacuum': {'x': 0, 'y': 1000}, 'map': {'x': 448, 'y': 752}}], entity_picture=/api/camera_proxy/camera.robo_map_1?token=e2332beec08d1a3a0fd3db770082db2a1a7c762d5d8b6eaa232b34378f5146f7, icon=mdi:map-search, friendly_name=robo Saved Map 1, supported_features=0 @ 2022-11-21T13:56:07.681918+09:00>>
```

javicalle, 2022-11-21:
Not familiar with the deconz logs, but I see a few HA commands with error (¿DEBUG?) reponses:
```
2022-11-21 14:00:14.262 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_confirm[12, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 119, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x690d endpoint=1>, 1, <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>, 0, 0, 0, 0]
2022-11-21 14:00:14.262 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 119: d0
2022-11-21 14:00:14.263 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x77 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x690d endpoint=1>, status: 0xd0
2022-11-21 14:00:14.263 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x690D:1:0x0102]: command failed: 'stop' args: '()' kwargs '{}' exception: 'Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'

2022-11-21 14:00:14.933 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_confirm[12, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 121, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x690d endpoint=1>, 1, <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>, 0, 0, 0, 0]
2022-11-21 14:00:14.934 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 121: d0
2022-11-21 14:00:14.934 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x79 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x690d endpoint=1>, status: 0xd0
2022-11-21 14:00:14.935 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x690D:1:0x0102]: command failed: 'down_close' args: '()' kwargs '{}' exception: 'Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'

2022-11-21 14:00:23.588 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_confirm[12, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 117, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x690d endpoint=1>, 1, <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>, 0, 0, 0, 0]
2022-11-21 14:00:23.588 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 117: d0
2022-11-21 14:00:23.589 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x75 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x690d endpoint=1>, status: 0xd0
2022-11-21 14:00:23.589 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x690D:1:0x0102]: command failed: 'up_open' args: '()' kwargs '{}' exception: 'Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'
```

If you have the cover remote, please enable debug logs and capture the remote events logs.

closed 2023-07-10 by github-actions[bot]

ricfelten, 2023-11-21:
Hello, I have the same curtain device as this issues subject. I am hoping someone can help get this proposed Zigbee quirk to work.

After configuring the quirk proposed above and i add the device and press an action on the new controls. It responds "Failed to call service cover/open_cover. Failed to open cover: <Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>"

I found some similar window curtains quirks here [zha-device-handlers project](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_cover.py). But this exact model number is not supported and i could not get the quirk to load editing it (don't have a full grasp on how this works yet). 

It seems the device is handled in this [kigbee2mqtt project issue](https://github.com/Koenkk/zigbee2mqtt/issues/17506).

See additional info below. Sorry for the formatting...

<details> 
<summary>Device Signature</summary>

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102",
        "0xef00"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_xu4a5rhj",
  "model": "TS0601",
  "class": "ts0601_cover_2.TuyaCover"
}

</details>
<details> 

<summary>Diagnostic Information</summary>2023-11-21 10:00:04.718 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration spotcast which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-21 10:00:04.719 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-21 10:00:04.719 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration blueiris which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-21 10:00:04.720 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration rinnai which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-21 10:00:04.720 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration browser_mod which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-21 10:00:04.721 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration nodered which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-21 10:00:13.439 WARNING (MainThread) [zhaquirks] Loaded custom quirks. Please contribute them to https://github.com/zigpy/zha-device-handlers
2023-11-21 10:00:16.367 ERROR (MainThread) [metno] Access to https://aa015h6buqvih86i1.api.met.no/weatherapi/locationforecast/2.0/complete returned error 'ClientConnectorError'
2023-11-21 10:00:17.031 ERROR (MainThread) [custom_components.hacs] Request exception for 'https://api.github.com/rate_limit' with - Cannot connect to host api.github.com:443 ssl:default [Try again]
Traceback (most recent call last):
  File "/config/custom_components/hacs/base.py", line 464, in async_can_update
    response = await self.async_github_api_method(self.githubapi.rate_limit)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hacs/base.py", line 522, in async_github_api_method
    raise HacsException(_exception)
custom_components.hacs.exceptions.HacsException: Request exception for 'https://api.github.com/rate_limit' with - Cannot connect to host api.github.com:443 ssl:default [Try again]
2023-11-21 10:00:19.963 ERROR (Thread-6) [root] Uncaught thread exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.11/site-packages/tuya_iot/openmq.py", line 158, in run
    self.__run_mqtt()
  File "/usr/local/lib/python3.11/site-packages/tuya_iot/openmq.py", line 172, in __run_mqtt
[... 454 lines cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
2023-11-21 10:03:27.311 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1cb3000c0005000207afffb0
2023-11-21 10:03:27.311 DEBUG (MainThread) [zigpy_deconz.api] Received command mac_poll[5, <DeconzAddress address_mode=AddressMode.NWK address=0xaf07>, 255, -80]
2023-11-21 10:03:27.318 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1cb4000c000500029addffbf
2023-11-21 10:03:27.318 DEBUG (MainThread) [zigpy_deconz.api] Received command mac_poll[5, <DeconzAddress address_mode=AddressMode.NWK address=0xdd9a>, 255, -65]

</details>