zigpy/zha-device-handlers #1947: [Device Support Request] Tuya 2 in 1 Zigbee Mini PIR Motion Detector (TS0601 by _TZE200_3towulqd)

fencing49, 2022-11-21:
**Is your feature request related to a problem? Please describe.**
Howdy friends, I would like support for this PIR motion sensor/  Light sensor. Currently when adding it via ZHA, it shows up as a smoke detector and triggering the motion detector yields no changes on HA dashboard. altering device type also produces the same results. device also displays battery at 0% despite that not being the case. basically it's a brick :)

**Describe the solution you'd like**
support for this particular motion sensor / light sensor

<details>

<summary>Device signature</summary>

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0500"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_3towulqd",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

```

</details>

<details>
<!-- Diagnostic information can be acquired by clicking on the "Download Diagnostics" button in the device settings view -->
<summary>Diagnostic information</summary>

```yaml

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "5.15.74",
    "supervisor": "2022.10.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "nodered": {
      "version": "0.5.2",
      "requirements": []
    },
    "blueiris": {
      "version": "1.0.12",
      "requirements": []
    },
[... 305 lines cut ...]
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

```

</details>

<details>
<summary>Additional logs</summary>

```
no logbook events found
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-11-21:
Duplicate of #1620

javicalle, 2022-11-21:
Let's keep the conversation inside https://github.com/zigpy/zha-device-handlers/issues/1620

closed 2022-11-21 by javicalle

fencing49, 2022-11-22:
> Let's keep the conversation inside #1620

this convo's quirk worked to fix it and it is now compatible.

thankyou!