zigpy/zha-device-handlers #1948: [BUG] Xiaomi Devices disconnects after some hours

tho071088, 2022-11-21:
Some Xiaomi ZigBee devices disconnects after some hours with no reasons.

Not working after some hours : 
lumi.sensor_switch
lumi.sensor_magnet
lumi.sensor_motion.aq2
lumi.sensor_motion

Working and not concerned by the issue : 
lumi.remote.b186acn02
lumi.sens
lumi.weather

**To Reproduce**
Steps to reproduce the behavior:
1. Add a device in the list
2. Wait some hours
3. Device appears disconnects and never connects again
4. I found nothing in the log file

**Expected behavior**
Device keeps working

**Screenshots**
[image]
[image]
[image]

<details>

<summary>Device signature (lumi.sensor_switch)</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003"
      ],
      "out_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.sensor_switch",
  "class": "zhaquirks.xiaomi.mija.sensor_switch.MijaButton"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
[... 320 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs (when adding new device)</summary>

```
[log_zha.txt](https://github.com/zigpy/zha-device-handlers/files/10056130/log_zha.txt)

```

</details>

**Additional context**
Other brands are working, batteries on devices are "new", putting them near the Zigbee reciever does not change anything.

dmulcahey, 2022-11-21:
https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623

Not an issue we can solve

closed 2022-11-21 by dmulcahey

MattWestb, 2022-11-21:
I have some Aqara Magnet and weather sensors that is working well **if**  they is having one good router to connecting to and shall being pared to the router you have near and **not** with the coordinator then if the coordinator is offline and they is trying communicating  and failing they is going offline.