zigpy/zha-device-handlers #1949: [BUG] No zha events with Symfonisk Dial

LCMilstein, 2022-11-21:
**Describe the bug**
I have a symfonisk dial remote set up with zha in Home Assistant and while I *do* see the proper automation actions to set up automations, I do NOT see any zha events associated with the paired device.

**To Reproduce**
Steps to reproduce the behavior:
1. Pair Symfonisk dial in zha
2. Confirm it is available
3. Set up 3 automations: (1) Press the knob trigger results in play/pause sonos action; (2) rotating counter-clockwise trigger results in volume down on sonos action; and (3) rotating clockwise trigger results in volume up sonos action.
4. Rotate and/or push the knob
5. Watch events in Developer Tools

**Expected behavior**
I expect the sonos to play/pause when the knob is pressed and to vol up/down depending on direction of rotatoin.

<details>

<summary>Device signature</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.11.1",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "4.4.180+",
    "run_as_root": true
  },

[COMPONENTS REDACTED]

    },
    "hacs": {
      "version": "1.28.3",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.2",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.84",
      "zigpy-deconz==0.19.0",
      "zigpy==0.51.5",
      "zigpy-xbee==0.16.2",
      "zigpy-zigate==0.10.3",
      "zigpy-znp==0.9.1"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
[... 285 lines cut ...]
</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
I notice that the Symfonisk becomes unavailable fairly often, but all of my testing has been done when I can confirm it *is* available... note that I have >65 zigbee devices, so I know the network is working.

MattWestb, 2022-11-21:
For all IKEA controllers you need doing one "reconfigure" from the device card after the device is being joined and waking the device up so it can receiving the commands.

If its being configure OK is shall doing chickens to the coordinator every hour like that from the device Logbook:
```
IKEA SYMFONISK Checkin event was fired
17:14:19 - 9 minutes ago
IKEA SYMFONISK Checkin event was fired
16:17:42 - 1 hour ago
```
If not doing chickens its having one no Zigbee 3 parent or have bad configuration.

If chickins is working OK is also sending events OK.

javicalle, 2022-11-21:
Duplicate of: #1943

dmulcahey, 2022-11-21:
@javicalle you can close these if you flag them as duplicates

closed 2022-11-21 by dmulcahey