zigpy/zha-device-handlers #1952: Aquara smart plug (lumi.plug.maeu01) fw.0029 don’t read sensor values (but fw.0020 does)

AndersWestman, 2022-11-21:
**Describe the bug**
I have an Aquara smart plug (lumi.plug.maeu01) with fw.0020 that is working great since the quirk
[zhaquirks.xiaomi.aqara.plug_eu.PlugMAEU01](https://community.home-assistant.io/t/aqara-plug-sp-euc01-lumi-plug-maeu01-not-updating/393110) earlier this year.

Now I have baught a second switch, which I can control as expected, but it will not read back any sensor values. They all show 0 all the time.

I notice two diffences between the “old” and the “new” plug.
Old plug Firmware: 0x00000020
New plug Firmware: 0x00000029

Old plug Sensors: Energy, Power, Electrical_measurement power_factor
New plug Sensors: Active power, Power factor, RMS voltage,Summation delivered

Can someone with the same plug please confirm if this is a problem with the newer firmware, or if I have a faulty unit.

**To Reproduce**
Steps to reproduce the behavior:
1. Pair an Aquara smart plug (lumi.plug.maeu01) with fw.0029
2. Try to read any sensor value
3. All values are read as 0

**Expected behavior**
Sensors should be read with correct values

**Screenshots**
If applicable, add screenshots to help explain your problem.

<details>

<summary>Device signature</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

javicalle, 2022-11-22:
Are the device signature for both devices equals? If they have diferent sensors, probably they have diferent signature.
`lumi.plug.maeu01` already have 3 diferents signatures, so another one would not be strange...

AndersWestman, 2022-11-22:
For some reason I'm not allowed to download the diagnostics for the older plug that works well.
The newer, non fully working plug, looks like this:

 ```
 "data": {
    "ieee": "**REDACTED**",
    "nwk": 38225,
    "manufacturer": "LUMI",
    "model": "lumi.plug.maeu01",
    "name": "LUMI lumi.plug.maeu01",
    "quirk_applied": true,
    "quirk_class": "zhaquirks.xiaomi.aqara.plug_eu.PlugMAEU01",
    "manufacturer_code": 4447,
    "power_source": "Mains",
    "lqi": 175,
    "rssi": -72,
    "last_seen": "2022-11-22T21:42:30",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0051",
          "in_clusters": [
            "0x0000",
            "0x0002",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x0009",
            "0x0702",
            "0x0b04",
            "0xfcc0"
          ],
          "out_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "21": {
          "profile_id": 260,
          "device_type": "0x0051",
          "in_clusters": [
            "0x000c"
          ],
          "out_clusters": []
        },
        "242": {
          "profile_id": 41440,
          "device_type": "0x0061",
          "in_clusters": [],
          "out_clusters": [
            "0x0021"
          ]
        }
      }
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "button.lumi_lumi_plug_maeu01_identify_2",
        "name": "LUMI lumi.plug.maeu01"
      },
      {
        "entity_id": "sensor.lumi_lumi_plug_maeu01_active_power",
        "name": "LUMI lumi.plug.maeu01"
      },
[... 403 lines cut ...]
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {}
      },
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

```

maverich, 2022-11-23:
I also have a similar issue except that I have a working device with firmware `0x00000029` and not working but not reporting any data kind of device with the same version. I think issue is related to some kind of mapping because the entity id given to the working device is `sensor.lumi_lumi_plug_maeu01_electricalmeasurement` as its but my new not-reporting one has `sensor.lumi_lumi_plug_maeu01_active_power`. In between my two installations, something has changed but I don't know what yet. If you found any solutions please share here as well :/

DlargeX, 2022-11-24:
I have the same problem with firmware 0000029. On/Off is working but Sensors for Power not.

javicalle, 2022-11-24:
Can any of you test with another quirk?

Enable your local quirk configuration an copy in your folder the current `plug_eu.py`:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/plug_eu.py

At the bottom of file, add the this quirk:
<details open>
<summary>PlugEU_Var07</summary>

```python
class PlugEU_Var07(XiaomiCustomDevice):
    """Aqara plug EU (variant 07)."""

    def __init__(self, *args, **kwargs):
        """Init."""
        self.voltage_bus = Bus()
        self.consumption_bus = Bus()
        self.power_bus = Bus()
        super().__init__(*args, **kwargs)

    signature = {
        MODELS_INFO: [
            (LUMI, "lumi.plug.maeu01"),
        ],
        ENDPOINTS: {
            # "profile_id": 260,
            # "device_type": "0x0051",
            # "in_clusters": ["0x0000","0x0002","0x0003","0x0004","0x0005","0x0006","0x0009","0x0702","0x0b04","0xfcc0"],
            # "out_clusters": ["0x000a","0x0019"]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    DeviceTemperature.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Alarms.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    OppleCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # "profile_id": 260,
            # "device_type": "0x0051",
            # "in_clusters": ["0x000c"],
            # "out_clusters": []
            21: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [AnalogInput.cluster_id],
            },
            # "profile_id": 41440,
            # "device_type": "0x0061",
            # "in_clusters": [],
            # "out_clusters": ["0x0021"]
            242: {
                PROFILE_ID: XIAOMI_PROFILE_ID,
                DEVICE_TYPE: XIAOMI_DEVICE_TYPE,
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    BasicCluster,
                    DeviceTemperature.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Alarms.cluster_id,
                    MeteringCluster,
                    ElectricalMeasurementCluster,
                    OppleCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            21: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [AnalogInputCluster],
            },
            242: {
                PROFILE_ID: XIAOMI_PROFILE_ID,
                DEVICE_TYPE: XIAOMI_DEVICE_TYPE,
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</details>

Save changes, delete any `__pycache__` folder, restart HA and repair the device.

AndersWestman, 2022-11-24:
> Can any of you test with another quirk?

I have tried to test your quirk, but it looks like it is not used.
After restart of HA and re-pairing it still show `Quirk: plug_eu.PlugMAEU01` and behave like before.

I have never used a local quirk before, so probebly I have made somthing wrong...

This is what I have done:

- Created a folder "zha_quirks" under config
- Created a file named `plug_eu.py` in the new folder
- Copied your new quirk into that new, emty file
- Added the following to my configuration.yalm
```
zha:
  enable_quirks: true
  custom_quirks_path: /config/zha_quirks
```
- Checked the configuration and restarted HA
- Removed the non working plug and re-pair it
- The Zigbee info section tells that it still use `Quirk: plug_eu.PlugMAEU01`

What have I messed up...?

javicalle, 2022-11-25:
Ouuuch, I didn't realized that the signature was already with the quirk loaded 🤦🏻‍♀️
Forget about the custom quirk.

I have read something about the coordinator update. Which Zigbee coordinator do you have?

AndersWestman, 2022-11-25:
> I have read something about the coordinator update. Which Zigbee coordinator do you have?

Yes, I recent saw that to. Issue #1897 looks like it covers the same problem.
I use the ConBeeII so it can absolutely be relevant for that update. I'm not sure which firmware I have on the stick, it was bought in October 2021.
Referring to [this changelog](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog)
 it should have some off the latest stabile versions - assuming it was updated then sold...

- Anyone who knows which firmware version that has this update?
- Is there any easy way to check which fw-version that is on the stick right now?

However, I will give the update a try, cross my fingers that it not destroy my ZigBee network, and report back the resolut here.

MattWestb, 2022-11-25:
From: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog
```
0x266b0700 Beta 2021-04-29
Support custom Node Descriptor manufacturer code when requesting device has Xiaomi T1 MAC address prefix
```
Its also implanted in later first gen adapters in the latest release.
but with deCONZ adapters the latest firmware is recommended.
Look in DE wiki for instructions how doing upgrade in Windows or Linux.

AndersWestman, 2022-11-25:
```
0x266b0700 Beta 2021-04-29
Support custom Node Descriptor manufacturer code when requesting device has Xiaomi T1 MAC address prefix
```
> Its also implanted in later first gen adapters in the latest release. but with deCONZ adapters the latest firmware is recommended. Look in DE wiki for instructions how doing upgrade in Windows or Linux.

Thanks for pointing that out! I have read through that but totally missed it...

DlargeX, 2022-11-25:
Is there a solution now, I have the Sonoff Stick and reading out the values doesn't work?

AndersWestman, 2022-11-25:
After updating the ConBeeII firmware to the today latest version, 26780700, and re-pairing the device, the sensors started to work!
Power (W) worked right away, while voltage and "Summation delivered" (kWh) took a couple of minutes to show up.
Power factor is still unavailable "as usual". 
Thanks all for your effort!

closed 2022-11-25 by AndersWestman