zigpy/zha-device-handlers #1953: [BUG]  Curtain motor _TZE200_r0jdjrvi has no entities

caradas, 2022-11-22:
**Describe the bug**
Curtain motor has no entities.

**Expected behavior**
Entities for opening and closing curtain and if possible slider to set partially closed

<details>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_r0jdjrvi",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
````

</details>

<details>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.74",
    "supervisor": "2022.10.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
[... 252 lines cut ...]
          }
        }
      },
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
````
</details>

According to https://github.com/zigpy/zha-device-handlers/issues/1602#issuecomment-1322763054 its already fixed.

javicalle, 2022-11-22:
Sorry for closing, but not, not fixed yet.

If you don't mind, go for a fresh start and focused just in your device.
I have recall all the post and it seems to be a tricky device. If I have read all the info, there must be a posible quirk but is possible that the `stop` button don't work.

Lets begin...

My proposed quirk:
<details open>
<summary>ts0601_cover_2.py</summary>

```python
"""Tuya MCU based cover and blinds."""
from typing import Dict, Optional, Union

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TUYA_MCU_COMMAND,
    TuyaLocalCluster,
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaClusterData,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaOnOffNM,
)

# Maps OPEN/CLOSE/STOP cover commands from Tuya to Zigbee
# https://github.com/zigpy/zigpy/blob/master/zigpy/zcl/clusters/closures.py#L558
# https://developer.tuya.com/en/docs/iot-device-dev/zigbee-curtain-switch-access-standard?id=K9ik6zvra3twv#title-7-DP1%20and%20DP4%20Curtain%20switch%201%20and%202
TUYA2ZB_COMMANDS = {
    0x0000: 0x0000,
    0x0001: 0x0002,
    0x0002: 0x0001,
}

class TuyaWindowCovering(NoManufacturerCluster, WindowCovering, TuyaLocalCluster):
    """Tuya MCU WindowCovering cluster."""

    """Add additional attributes for direction"""
    attributes = WindowCovering.attributes.copy()
    attributes.update(
[... 231 lines cut ...]
        }
    }
```

</details>

There is a good guide to create your local quirk:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Just create inside your local quirk folder a new file (ie: `ts0601_cover_2.py`) and put the code inside.

After any future changes to the quirk you will need to remove any `__pycache__` folder, restart HA and repair the device.

I will try to help as much as I can, but it will take a lot of tests and a lot of patience (my time is limited).
It will also be necessary to activate the logs for the integration and collect the relevant information. Instructions here:
* https://www.home-assistant.io/integrations/zha/#debug-logging

If you have a remote control for the blind, it will also be useful to obtain information on the commands.

One last request, please, put the logs and code examples between ` ``` ` marks like this:
````
```
Any code o logs requested
```
````

caradas, 2022-11-23:
Thanks for your time and help :)
I have changed the if in line 88 to: 
```
        # (upopen, downclose, stop)
        if command_id == (0x0002):  # ¿0x0003: continue?
            cluster_data = TuyaClusterData(
```
After that i noticed that the dasboard buttons aren't working.

Also shortened the log to the only thing thats happening while the motor is active:

Remote: Open + Stop:
```
2022-11-23 22:15:16.598 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0xB562, Relays=[0x7929])
2022-11-23 22:15:16.604 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0xB562, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=49, SecurityUse=<Bool.false: 0>, TimeStamp=15684090, TSN=0, Data=b'\x19\xC0\x01\x00\x01\x03\x02\x00\x04\x00\x00\x00\x51', MacSrcAddr=0x7929, MsgResultRadius=28)
2022-11-23 22:15:16.605 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB562), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\x19\xc0\x01\x00\x01\x03\x02\x00\x04\x00\x00\x00Q'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=49, rssi=None)
2022-11-23 22:15:16.605 DEBUG (MainThread) [zigpy.zcl] [0xB562:1:0xef00] Received ZCL frame: b'\x19\xc0\x01\x00\x01\x03\x02\x00\x04\x00\x00\x00Q'
2022-11-23 22:15:16.606 DEBUG (MainThread) [zigpy.zcl] [0xB562:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=192, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-11-23 22:15:16.607 DEBUG (MainThread) [zigpy.zcl] [0xB562:1:0xef00] Decoded ZCL frame: TuyaWindowCoverManufCluster:get_data(data=TuyaCommand(status=0, tsn=1, datapoints=[TuyaDatapointData(dp=3, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'Q\x00\x00\x00', *payload=81))]))
2022-11-23 22:15:16.608 DEBUG (MainThread) [zigpy.zcl] [0xB562:1:0xef00] Received command 0x01 (TSN 192): get_data(data=TuyaCommand(status=0, tsn=1, datapoints=[TuyaDatapointData(dp=3, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'Q\x00\x00\x00', *payload=81))]))
2022-11-23 22:15:16.609 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xB562:1:0x0102]: Attribute report 'Window Covering'[accurate_calibration] = 81
```

Remote Open without stop till fully open:
```
2022-11-23 22:18:30.401 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0xB562, Relays=[0x7929])
2022-11-23 22:18:30.407 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0xB562, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=47, SecurityUse=<Bool.false: 0>, TimeStamp=16289752, TSN=0, Data=b'\x19\xC7\x01\x00\x01\x03\x02\x00\x04\x00\x00\x00\x00', MacSrcAddr=0x7929, MsgResultRadius=28)
2022-11-23 22:18:30.408 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB562), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\x19\xc7\x01\x00\x01\x03\x02\x00\x04\x00\x00\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=47, rssi=None)
2022-11-23 22:18:30.408 DEBUG (MainThread) [zigpy.zcl] [0xB562:1:0xef00] Received ZCL frame: b'\x19\xc7\x01\x00\x01\x03\x02\x00\x04\x00\x00\x00\x00'
2022-11-23 22:18:30.409 DEBUG (MainThread) [zigpy.zcl] [0xB562:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=199, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-11-23 22:18:30.409 DEBUG (MainThread) [zigpy.zcl] [0xB562:1:0xef00] Decoded ZCL frame: TuyaWindowCoverManufCluster:get_data(data=TuyaCommand(status=0, tsn=1, datapoints=[TuyaDatapointData(dp=3, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))]))
2022-11-23 22:18:30.409 DEBUG (MainThread) [zigpy.zcl] [0xB562:1:0xef00] Received command 0x01 (TSN 199): get_data(data=TuyaCommand(status=0, tsn=1, datapoints=[TuyaDatapointData(dp=3, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))]))
2022-11-23 22:18:30.410 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xB562:1:0x0102]: Attribute report 'Window Covering'[accurate_calibration] = 0
```

Remote Close + Stop:
```
Open till completly open:
2022-11-23 22:18:30.401 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0xB562, Relays=[0x7929])
2022-11-23 22:18:30.407 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0xB562, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=47, [... cut ...]
2022-11-23 22:35:24.762 DEBUG (MainThread) [zigpy.zcl] [0xB562:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=223, command_id=11, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-11-23 22:35:24.763 DEBUG (MainThread) [zigpy.zcl] [0xB562:1:0xef00] Decoded ZCL frame: TuyaWindowCoverManufCluster:Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
```

javicalle, 2022-11-24:
Sorry but this seems to be a completely diferent cover from the quirk. The previous quirk was for a 2 cover device.
I would need the full device description (manufacturer/vendor link).

From your logs I would say that cover reports the status position in some way. I would expect to have one report from every remote command but I only can see one from your logs.

My assumption is that the device will report the 'target status' when pushing the `open`/`close` buttons and the stop button will report the current position if stopped halfway. That could lead to a strange behavior in HA, but we'll see about that later.

Let's try with a 'classical' approach. Replace completely the local quirk with this one:
<details open>
<summary>ts0601_cover_2.py</summary>

```python
"""Tuya based cover and blinds."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)

class TuyaCover0601_GP(TuyaWindowCover):
    """Tuya cover controller device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_r0jdjrvi", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

</details>

Save changes, delete any `__pycache__` folder, restart HA and repair the device.
Double check if the device report the button press o just when it stop moving. Maybe is a RF remote?
Look for any other reports from the device to see if there are other device attributes there.

caradas, 2022-11-24:
It is this device: https://www.aliexpress.com/item/1005003796260309.html As EU Plug, Motor only variant. 
The datasheet says something like "Frequency 2.4G/433Mhz" so yes the remote should be rf, this also explains the Wire antenna hanging out of the device. It also says Product model "CMD900E"
Any log output with the remote happens after the curtains stops regardless if stopped by the remote or track ends.

Pressing the stop button on the remote without the curtain moving no log output is generated.

Log for Stop in HA:
```` 

2022-11-24 21:50:48.744 DEBUG (MainThread) [zhaquirks.tuya] a4:c1:38:44:24:da:54:d3 Sending Tuya Cluster Command.. Manufacturer is _TZE200_r0jdjrvi Cluster Command is 0x0002, Arguments are ()
2022-11-24 21:50:48.745 DEBUG (MainThread) [zhaquirks.tuya] a4:c1:38:44:24:da:54:d3 Sending Tuya Command. Paylod values [endpoint_id : 1, Status : 0, TSN: 0, Command: 0x0401, Function: 0, Data: [1, 1]]
2022-11-24 21:50:48.745 DEBUG (MainThread) [zigpy.zcl] [0x6A13:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4417, tsn=118, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-11-24 21:50:48.745 DEBUG (MainThread) [zigpy.zcl] [0x6A13:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 1]))
2022-11-24 21:50:48.746 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x6A13), dst_ep=1, source_route=None, extended_timeout=False, tsn=118, profile_id=260, cluster_id=61184, data=Serialized[b'\x05A\x11v\x00\x00\x00\x01\x04\x00\x01\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-11-24 21:50:48.747 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x6A13), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=118, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=0, Data=b'\x05\x41\x11\x76\x00\x00\x00\x01\x04\x00\x01\x01')
2022-11-24 21:50:48.760 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-11-24 21:50:48.765 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=118)
2022-11-24 21:50:48.786 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x6A13, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=39, SecurityUse=<Bool.false: 0>, TimeStamp=3669657, TSN=0, Data=b'\x18\x76\x0B\x00\x83', MacSrcAddr=0x7929, MsgResultRadius=28)
2022-11-24 21:50:48.786 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x6A13), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\x18v\x0b\x00\x83'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=39, rssi=None)
2022-11-24 21:50:48.786 DEBUG (MainThread) [zigpy.zcl] [0x6A13:1:0xef00] Received ZCL frame: b'\x18v\x0b\x00\x83'
2022-11-24 21:50:48.786 DEBUG (MainThread) [zigpy.zcl] [0x6A13:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=118, command_id=11, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-11-24 21:50:48.787 DEBUG (MainThread) [zigpy.zcl] [0x6A13:1:0xef00] Decoded ZCL frame: TuyaManufacturerWindowCover:Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
```` 

Log for UP in HA: 

```` 
2022-11-24 21:53:42.314 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=1794, SrcAddr=0x2AD3, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=97, SecurityUse=<Bool.false: 0>, TimeStamp=4211952, TSN=0, Data=b'\x08\x22\x0A\x00\x00\x25\x34\x0A\x00\x00\x00\x00', MacSrcAddr=0x2AD3, MsgResultRadius=29)
2022-11-24 21:53:42.314 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x2AD3), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=1794, data=Serialized[b'\x08"\n\x00\x00%4\n\x00\x00\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=97, rssi=None)
2022-11-24 21:53:42.315 DEBUG (MainThread) [zigpy.zcl] [0x2AD3:1:0x0702] Received ZCL frame: b'\x08"\n\x00\x00%4\n\x00\x00\x00\x00'
2022-11-24 21:53:42.315 DEBUG (MainThread) [zigpy.zcl] [0x2AD3:1:0x0702] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, dis [... cut ...]
2022-11-24 21:53:42.406 DEBUG (MainThread) [zhaquirks.tuya] a4:c1:38:44:24:da:54:d3 Sending Tuya Command. Paylod values [endpoint_id : 1, Status : 0, TSN: 0, Command: 0x0401, Function: 0, Data: [1, 0]]
2022-11-24 21:53:42.407 DEBUG (MainThread) [zigpy.zcl] [0x6A13:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4417, tsn=131, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
````

RowCZ, 2022-11-27:
Hi @javicalle,
I also have problem with my cover motor with same designation. :-/
I would love to assist to help you find the solution but because I am not exactly good at this I would appreciate any instruction what you need and how can I help.
I was able to connect it to ZHA, I am catching LQI a RSSI Diagnostics but that is all.
I have latest HA 2022.11.4 and with active built in quirks.
Here is at least what I was able to dig from diagnostics about this device

```
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 204,
    "manufacturer": "_TZE200_r0jdjrvi",
    "model": "TS0601",
    "name": "_TZE200_r0jdjrvi TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4417,
    "power_source": "Mains",
    "lqi": 180,
    "rssi": -55,
    "last_seen": "2022-11-27T00:31:18",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0051",
          "in_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "out_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "242": {
          "profile_id": 41440,
          "device_type": "0x0061",
          "in_clusters": [],
          "out_clusters": [
            "0x0021"
          ]
        }
      }
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "sensor.tze200_r0jdjrvi_ts0601_rssi",
        "name": "_TZE200_r0jdjrvi TS0601"
      },
      {
        "entity_id": "sensor.tze200_r0jdjrvi_ts0601_lqi",
        "name": "_TZE200_r0jdjrvi TS0601"
      }
    ],
    "neighbors": [
      {
        "device_type": "Coordinator",
        "rx_on_when_idle": "On",
        "relationship": "Sibling",
        "extended_pan_id": "**REDACTED**",
        "ieee": "**REDACTED**",
        "nwk": "0x0000",
        "permit_joining": "Unknown",
[... 79 lines cut ...]
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
```

With this custom quirk active https://github.com/zigpy/zha-device-handlers/issues/1953#issuecomment-1326805887
I at least have open close and stop buttons active but they do nothing. I also have an inactive Switch.

And with this one https://github.com/zigpy/zha-device-handlers/issues/1953#issuecomment-1324221269
I have same controls active as above but on a press of any of them I get "Failed to call service cover/open_cover. argument of type 'int' is not iterable" notification in HA.
I also have active switch in OFF position and when I turn it ON, nothing happens and after few seconds it returns to OFF position. But when I turn it ON and OFF again (before it returns itself), motor starts spin in clockwice direction and it runs until some kind of timeout. But when it runs and I switch it to ON position again it stops. It also stops when I start to run motor manually (making few turns in one or another direction, it react to external movement of curtain and continues that direction) and switch it to ON position.
But even if I am able to do something, there are still no records in device Logbook, only "No logbook events found." message.
So in this case, "switch ON" is probably command to stop and "switch OFF" is probably command to open or close curtains and this quirk is close on its way to be functional. (I am almost able to make workaround in node-red :D but still missing second direction :( )

I also setup ZHA logging you mentioned before and tried to check my log but it is pretty overcrowded since I have dozens of devices connected to HA so if you specify which part is interresting for you I will try to find it or be able at least to clean it a little bit.

Thanks a lot for all your work.

javicalle, 2022-11-27:
Probably we need to map the motion commands.
You can look in the logs for your device NMK. It will show as `[0x00cc]`

javicalle, 2022-11-27:
After checking it, I will return to the new implementation, so please use this quirk to test:
<details open>
<summary>ts0601_cover_2.py</summary>

```python
"""Tuya MCU based cover and blinds."""
from typing import Dict, Optional, Union

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TUYA_MCU_COMMAND,
    TuyaLocalCluster,
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaClusterData,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaOnOffNM,
)

# Maps OPEN/CLOSE/STOP cover commands from Tuya to Zigbee
# https://github.com/zigpy/zigpy/blob/master/zigpy/zcl/clusters/closures.py#L558
# https://developer.tuya.com/en/docs/iot-device-dev/zigbee-curtain-switch-access-standard?id=K9ik6zvra3twv#title-7-DP1%20and%20DP4%20Curtain%20switch%201%20and%202
TUYA2ZB_COMMANDS = {
    0x0000: 0x0000,
    0x0001: 0x0002,
    0x0002: 0x0001,
}

class TuyaWindowCovering(NoManufacturerCluster, WindowCovering, TuyaLocalCluster):
    """Tuya MCU WindowCovering cluster."""

    """Add additional attributes for direction"""
    attributes = WindowCovering.attributes.copy()
    attributes.update(
        {
            0xF000: ("curtain_switch", t.enum8, True),  # 0: open, 1: stop, 2: close
            0xF001: (
                "accurate_calibration",
                t.enum8,
                True,
            ),  # 0: calibration started, 1: calibration finished
[... 223 lines cut ...]
                    TuyaWindowCovering,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

</details>

Most of the functionallity has been commented, I will enable it later.

Is expected that lift_percentage would be updated.

I don't know what switch is this that you refer, any info will be welcomed. 
Please, enable the debug logs and attach the related ones:
* https://www.home-assistant.io/integrations/zha/#debug-logging

@RowCZ if you can attach the logs related to the switch operation will be helpfull.

RowCZ, 2022-11-27:
@javicalle thanks for quick response
I used your new quirk and tried to catch relevant logs while tried:

switch ON - switch OFF (motor starts runnik CW)
switch OFF - (motor stops)
Up - error occurs
Stop - error occurs
Down - error occurs

[rowcz__TZE200_r0jdjrvi_home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/10099343/rowcz__TZE200_r0jdjrvi_home-assistant.log)

With new quirk I also noticed that I have finally some messages in Logbook on device page but there are only messages from yesterday (about 15 hours ago and older) so something in past worked at least that much that made some records there and new quirk was at least able to load that records but wasn't able to made new one.

I hope this info will help.
Thanks again

caradas, 2022-11-27:
> Is expected that lift_percentage would be updated.

I can confirm that lift_percantage is now showing up and updating correctly after the curtain moved.

> switch ON - switch OFF (motor starts runnik CW) switch OFF - (motor stops) Up - error occurs Stop - error occurs Down - error occurs
I can't confirm that. The Switch entity does nothing for me.

caradas, 2022-12-07:
Thanks for the update.

Percentage update is still working.
None of the Buttons do anything.

UP:
````
2022-12-07 22:44:45.632 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='current_position_lift_percentage', attr_value=100, expect_reply=True)
2022-12-07 22:44:45.632 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] get_dp_mapping --> found DP: 3
2022-12-07 22:44:45.632 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] from_cluster_data: 3, DPToAttributeMapping(ep_attribute='window_covering', attribute_name='current_position_lift_percentage', dp_type=<TuyaDPType.VALUE: 2>, converter=None, dp_converter=None, endpoint_id=None)
2022-12-07 22:44:45.632 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] ztype: 100
2022-12-07 22:44:45.632 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] from_value: [4, 0, 0, 0, 100]
2022-12-07 22:44:45.632 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] raw: b'\x00\x00\x00d'
2022-12-07 22:44:45.632 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] tuya_command: TuyaCommand(status=0, tsn=79, datapoints=[TuyaDatapointData(dp=3, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00d', *payload=1677721600))])
2022-12-07 22:44:45.633 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4417, tsn=80, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-12-07 22:44:45.633 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=79, datapoints=[TuyaDatapointData(dp=3, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00d', *payload=1677721600))]))
2022-12-07 22:44:45.693 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] Received ZCL frame: b'\x18P\x0b\x00\x83'
2022-12-07 22:44:45.693 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=80, command_id=11, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-07 22:44:45.694 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] Decoded ZCL frame: TuyaWindowCoverManufCluster:Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
````
Down:
```` 
2022-12-07 22:49:00.991 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='current_position_lift_percentage', attr_value=0, expect_reply=True)
2022-12-07 22:49:00.991 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] get_dp_mapping --> found DP: 3
2022-12-07 22:49:00.991 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] from_cluster_data: 3, DPToAttributeMapping(ep_attribute='window_covering', attribute_name='current_position_lift_percentage', dp_type=<TuyaDPType.VALUE: 2>, converter=None, dp_converter=None, endpoint_id=None)
2022-12-07 22:49:00.992 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] ztype: 0
2022-12-07 22:49:00.992 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] from_value: [4, 0, 0, 0, 0]
2022-12-07 22:49:00.992 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] raw: b'\x00\x00\x00\x00'
2022-12-07 22:49:00.992 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] tuya_command: TuyaCommand(status=0, tsn=101, datapoints=[TuyaDatapointData(dp=3, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))])
2022-12-07 22:49:00.993 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4417, tsn=102, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-12-07 22:49:00.993 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=101, datapoints=[TuyaDatapointData(dp=3, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))]))
2022-12-07 22:49:01.058 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] Received ZCL frame: b'\x18f\x0b\x00\x83'
2022-12-07 22:49:01.058 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=102, command_id=11, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-07 22:49:01.058 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] Decoded ZCL frame: TuyaWindowCoverManufCluster:Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
````
Stop:

```` 
[... cut ...]
2022-12-07 22:49:52.003 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] from_cluster_data: None, None
2022-12-07 22:49:52.003 WARNING (MainThread) [zigpy.zcl] [0xB579:1:0xef00] No cluster_dp found for 1, curtain_switch
2022-12-07 22:49:52.005 DEBUG (MainThread) [zigpy.zcl] [0xB579:1:0xef00] tuya_command: None
2022-12-07 22:49:52.005 WARNING (MainThread) [zigpy.zcl] [0xB579:1:0xef00] no MCU command for data TuyaClusterData(endpoint_id=1, cluster_attr='curtain_switch', attr_value=1, expect_reply=True)
````

RowCZ, 2022-12-07:
@javicalle hi and thanks for the update
I modified the quirk and here is the results

switch is without any change
switch ON - switch OFF (motor starts runnig CW)
switch OFF - (motor stops)

Up - no error this time, in log book I can see message that Cover is opening but no movement on motor (button change to inactive state)
Stop - failed to call service error occurs as before
Up - no error this time, in log book I can see message that Cover is closing but no movement on motor (button change to inactive state)

log: 
[rowcz__TZE200_r0jdjrvi_home-assistant_2022-12-07.log](https://github.com/zigpy/zha-device-handlers/files/10180444/rowcz__TZE200_r0jdjrvi_home-assistant_2022-12-07.log)

Here is a also a part of the screen with controls and logbook 
[image]

caradas, 2022-12-07:
Good News i got the arrows working with the following:

```python
"""Tuya MCU based cover and blinds."""
from typing import Dict, Optional, Union

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TUYA_MCU_COMMAND,
    TuyaLocalCluster,
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaClusterData,
    TuyaDPType,
    TuyaMCUCluster,
)

# Maps OPEN/CLOSE/STOP cover commands from Tuya to Zigbee
# https://github.com/zigpy/zigpy/blob/master/zigpy/zcl/clusters/closures.py#L558
# https://developer.tuya.com/en/docs/iot-device-dev/zigbee-curtain-switch-access-standard?id=K9ik6zvra3twv#title-7-DP1%20and%20DP4%20Curtain%20switch%201%20and%202
TUYA2ZB_COMMANDS = {
    0x0000: 0x0000,
    0x0001: 0x0002,
    0x0002: 0x0001,
}

class TuyaWindowCovering(NoManufacturerCluster, WindowCovering, TuyaLocalCluster):
    """Tuya MCU WindowCovering cluster."""

    """Add additional attributes for direction"""
    attributes = WindowCovering.attributes.copy()
    attributes.update(
        {
            0xF000: ("curtain_switch", t.enum8, True),  # 0: open, 1: stop, 2: close
            0xF001: ("accurate_calibration", t.enum8, True),  # 0: calibration started, 1: calibration finished
            0xF002: ("motor_steering", t.enum8, True),  # 0: default, 1: reverse
            0xF003: ("travel", t.uint16_t, True),  # 30 to 9000 (units of 0.1 seconds)
        }
    )

    async def command(
        self,
        command_id: Union[foundation.GeneralCommand, int, t.uint8_t],
[... 214 lines cut ...]
        }
    }
```
Key was to set following in commands:
```python
manufacturer=-1,
```

Unfortunally the percentage slider is only a display but doesn't turn the motor:
Log:

````
2022-12-08 00:48:27.526 DEBUG (MainThread) [zigpy.zcl] [0x50CE:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='current_position_lift_percentage', attr_value=75, expect_reply=True, manufacturer=-1)
2022-12-08 00:48:27.526 DEBUG (MainThread) [zigpy.zcl] [0x50CE:1:0xef00] get_dp_mapping --> found DP: 3
2022-12-08 00:48:27.526 DEBUG (MainThread) [zigpy.zcl] [0x50CE:1:0xef00] from_cluster_data: {3: DPToAttributeMapping(ep_attribute='window_covering', attribute_name='current_position_lift_percentage', dp_type=<TuyaDPType.VALUE: 2>, converter=None, dp_converter=None, endpoint_id=None)}
2022-12-08 00:48:27.527 DEBUG (MainThread) [zigpy.zcl] [0x50CE:1:0xef00] ztype: 75
2022-12-08 00:48:27.527 DEBUG (MainThread) [zigpy.zcl] [0x50CE:1:0xef00] from_value: [4, 0, 0, 0, 75]
2022-12-08 00:48:27.527 DEBUG (MainThread) [zigpy.zcl] [0x50CE:1:0xef00] raw: b'K\x00\x00\x00'
2022-12-08 00:48:27.527 DEBUG (MainThread) [zigpy.zcl] [0x50CE:1:0xef00] tuya_commands: [TuyaCommand(status=0, tsn=72, datapoints=[TuyaDatapointData(dp=3, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'K\x00\x00\x00', *payload=75))])]
2022-12-08 00:48:27.529 DEBUG (MainThread) [zigpy.zcl] [0x50CE:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=73, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-12-08 00:48:27.529 DEBUG (MainThread) [zigpy.zcl] [0x50CE:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=72, datapoints=[TuyaDatapointData(dp=3, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'K\x00\x00\x00', *payload=75))]))
2022-12-08 00:48:27.596 DEBUG (MainThread) [zigpy.zcl] [0x50CE:1:0xef00] Received ZCL frame: b'\x18I\x0b\x00\x00'
2022-12-08 00:48:27.596 DEBUG (MainThread) [zigpy.zcl] [0x50CE:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=73, command_id=11, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-08 00:48:27.597 DEBUG (MainThread) [zigpy.zcl] [0x50CE:1:0xef00] Decoded ZCL frame: TuyaWindowCoverManufCluster:Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
````

RowCZ, 2022-12-08:
@caradas Nice!
It works for me as well even Stop button works now (but still causes error).
Switch become inactive (but is not needed anymore)
And I have no percentage slider at all 😮 is there way how to add it?

[image]

Here is log from version that cardas provided:
[rowcz__TZE200_r0jdjrvi_home-assistant_2022-12-07-2.log](https://github.com/zigpy/zha-device-handlers/files/10181287/rowcz__TZE200_r0jdjrvi_home-assistant_2022-12-07-2.log)

RowCZ, 2022-12-08:
weird, I removed device, delete pycache, restarted HA and paired again but I still don't have a progress bar (that mysterious switch is gone now)

[image]
[image]

javicalle, 2022-12-08:
A lot of updates here.

The switch wasn't really necessary, just an attempt to understand things. Once resolved it is not necessary and should not return.

That `manufacturer=-1` is the same that would try with the `NoManufacturerCluster` not sure why isn't working. You can get the same with `manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID`.

The lift percentage is mapped against the DP_3. The device don't complain about it but it can be that is not the good one. According to Z2M it could be the DP_2, so I would try to replace the `dp_to_attribute` mapping from `3` to `2`.

Let's see if we can fix the cover functions and then I try to see what happens to the missing slider.

RowCZ, 2022-12-08:
@javicalle 
I am bit confused, I am not sure if I uderstand it right, I changed it here from 3 to 2 but with no effect

```
dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaWindowCovering.ep_attribute,
            "curtain_switch",
            dp_type=TuyaDPType.ENUM,
        ),
        2: DPToAttributeMapping(
            TuyaWindowCovering.ep_attribute,
            "current_position_lift_percentage",
            dp_type=TuyaDPType.VALUE,
        ),
```

bezmi, 2022-12-11:
I've only just started with the zigbee stuff, but upon testing zigbee2mqtt, it appears that all of the buttons as well as position control work pretty well. For position control to work, I had to first set endstop positions by letting the motor move in one direction and physically stopping rotation to signal the end of the track. Repeat for the other direction and the position slider starts working.

Is there any way to bring the config from zigbee2mqtt into the  custom quirk here so we can get all of the info required to get this working with zha?

RowCZ, 2022-12-11:
> ... I had to first set endstop positions by letting the motor move in one direction and physically stopping rotation to signal the end of the track. Repeat for the other direction and the position slider starts working...
That can be the reason why I have no slider, I didn't bother with instalation since I was unable to make it work at all. :D

bezmi, 2022-12-12:
clarification: this was position control with zigbee2mqtt. Haven't been able to reliably get the slider to show up in zha.

sid12345678910, 2022-12-12:
> Good News i got the arrows working with the following:

so i have this model _TZE200_65moe6rh

This code works strangely with mine. Mine came with a remote. It seems the code works only if the motor doesn't hit its "endstops" Once is it hits the endstop HA stops responding. But if its in the middle I can control the motor in HA. All the commands work Up, Down and stop but only if its in-between both endstops. Once it hits the endstop nothing responds...

Also the slider reports the correct position but I cant move it around.

javicalle, 2023-01-30:
That quirk is a little old now.
You can try adding the `cluster_name` in the `TuyaClusterData`, more or less like this:
```python
        # (upopen, downclose, stop)
        if command_id in (0x0002, 0x0000, 0x0001):  # ¿0x0003: continue?
            cluster_data = TuyaClusterData(
                endpoint_id=self.endpoint.endpoint_id,
                cluster_name=TuyaMCUCluster.ep_attribute,
                cluster_attr="curtain_switch",
                attr_value=TUYA2ZB_COMMANDS[command_id],  # convert tuya2zigbee command
                expect_reply=expect_reply,
                manufacturer=-1,
            )
```
and 
```python
        # (go_to_lift_percentage)
        elif command_id == 0x0005:
            lift_value = args[0]

            cluster_data = TuyaClusterData(
                endpoint_id=self.endpoint.endpoint_id,
                cluster_name=TuyaMCUCluster.ep_attribute,
                cluster_attr="current_position_lift_percentage",
                attr_value=lift_value,
                expect_reply=expect_reply,
                manufacturer=-1,
            )
```

IIRC that quirk is very incomplete

kolmakova, 2023-01-31:
@javicalle Thanks! I've fixed a slider control by changing `current_position_lift_percentage` DP from 3 to 2 [as in Tuya documentation](https://developer.tuya.com/en/docs/iot/zigbee-curtain-switch-access-standard?id=K9ik6zvra3twv&bd_vid=10971872461674271794#title-8-DP2%20and%20DP5%20Percentage%20control%201%20and%202) and added your solution to fix `TypeError: getattr(): attribute name must be string`, so slider is working now. Here's the quirk (for my **_TZE200_nogaemzt**). Don't forget to add another model name to the `MODELS_INFO`.

<details>
    <summary>ts0601_cover.py</summary>

```py
"""Tuya MCU based cover and blinds."""
from typing import Dict, Optional, Union

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TUYA_MCU_COMMAND,
    TuyaLocalCluster,
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaClusterData,
    TuyaDPType,
    TuyaMCUCluster,
)

# Maps OPEN/CLOSE/STOP cover commands from Tuya to Zigbee
# https://github.com/zigpy/zigpy/blob/master/zigpy/zcl/clusters/closures.py#L558
# https://developer.tuya.com/en/docs/iot-device-dev/zigbee-curtain-switch-access-standard?id=K9ik6zvra3twv#title-7-DP1%20and%20DP4%20Curtain%20switch%201%20and%202
TUYA2ZB_COMMANDS = {
    0x0000: 0x0002,
    0x0001: 0x0000,
    0x0002: 0x0001,
}

class TuyaWindowCovering(NoManufacturerCluster, WindowCovering, TuyaLocalCluster):
    """Tuya MCU WindowCovering cluster."""

    """Add additional attributes for direction"""
    attributes = WindowCovering.attributes.copy()
    attributes.update(
        {
            0xF000: ("curtain_switch", t.enum8, True),  # 0: open, 1: stop, 2: close
            0xF001: ("accurate_calibration", t.enum8, True),  # 0: calibration started, 1: calibration finished
            0xF002: ("motor_steering", t.enum8, True),  # 0: default, 1: reverse
            0xF003: ("travel", t.uint16_t, True),  # 30 to 9000 (units of 0.1 seconds)
        }
    )

[... 204 lines cut ...]
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaWindowCoverManufCluster,
                    TuyaWindowCovering,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```
</details>

Seems like the only problem now is that controls have the right direction, but the curtains state and slider control are inverted. To open curtains via slider, I need to move a slider to the 0 so state results in Closed (and vice versa). And if the curtain state is Closed (but it's actually Open), Close button is inactive. Also, commands (buttons) don't result in updating a slider value and curtain state (maybe this doesn't relate to the invertion).

Tried to add `tuya_cover_inverted_by_default = True` to the `TuyaCover0601_GP` but no result. Any ideas?

javicalle, 2023-01-31:
If you have a physical remote for that cover try to reverse the motor spin according to the manufacturer.
If don't, try updating the `motor_steering` attribute from the cluster `TuyaWindowCovering`. Switch between 0 and 1 values to test.

Drmatterpl, 2023-02-20:
Hi

> That can be the reason why I have no slider, I didn't bother with instalation since I was unable to make it work at all. :D

can you share your latest quirk code? I have spend a looooot of time and it seems  that this is the only quirk that works with my motor - the one by @caradas

Dextrik, 2023-02-20:
Hi, @Drmatterpl 
I can share mine, everything is working, but i don't think so this quirk it's optimal.

<details>
<summary><strong>ts0601_cover.py</strong></summary>
<br>

```
"""Tuya MCU based cover and blinds."""
from typing import Dict, Optional, Union

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TUYA_MCU_COMMAND,
    TuyaLocalCluster,
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaClusterData,
    TuyaDPType,
    TuyaMCUCluster,
)

# Maps OPEN/CLOSE/STOP cover commands from Tuya to Zigbee
# https://github.com/zigpy/zigpy/blob/master/zigpy/zcl/clusters/closures.py#L558
# https://developer.tuya.com/en/docs/iot-device-dev/zigbee-curtain-switch-access-standard?id=K9ik6zvra3twv#title-7-DP1%20and%20DP4%20Curtain%20switch%201%20and%202
TUYA2ZB_COMMANDS = {
    0x0000: 0x0000,
    0x0001: 0x0002,
    0x0002: 0x0001,
}

class TuyaWindowCovering(NoManufacturerCluster, WindowCovering, TuyaLocalCluster):
    """Tuya MCU WindowCovering cluster."""

    """Add additional attributes for direction"""
    attributes = WindowCovering.attributes.copy()
    attributes.update(
        {
            0xF000: ("curtain_switch", t.enum8, True),  # 0: open, 1: stop, 2: close
            0xF001: ("accurate_calibration", t.enum8, True),  # 0: calibration started, 1: calibration finished
            0xF002: ("motor_steering", t.enum8, True),  # 0: default, 1: reverse
            0xF003: ("travel", t.uint16_t, True),  # 30 to 9000 (units of 0.1 seconds)
        }
[... 200 lines cut ...]
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaWindowCoverManufCluster,
                    TuyaWindowCovering,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```
</details>

Drmatterpl, 2023-02-20:
Thanks! it works for me.
I have all I need - operating slider, no errors when hitting the buttons.
Before I had some errors with stop button - Failed to call service cover/stop_cover....

javicalle, 2023-02-20:
The quirk seems to be pretty incomplete:
* DP `current_position_lift_percentage` is duplicated
* no attributes implemented

There must be some `No cluster_dp found for ...` errors in the logs. They would give to us some info about capabilities and how implement it.

MattWestb, 2023-02-20:
Take the power of the device and wait one minute and putting it back and look if some interesting DPs is s coning in the log.

RowCZ, 2023-02-20:
> Hi
> 
> > That can be the reason why I have no slider, I didn't bother with instalation since I was unable to make it work at all. :D
> 
> @RowCZ can you share your latest quirk code? I have spend a looooot of time and it seems that this is the only quirk that works with my motor - the one by @caradas

This is last that worked for me
[TS0601_Covertest.txt](https://github.com/zigpy/zha-device-handlers/files/10785032/TS0601_Covertest.txt)

Dextrik, 2023-02-20:
> Take the power of the device and wait one minute and putting it back and look if some interesting DPs is s coning in the log.

@MattWestb, @javicalle  this is log when i plug the device back to power and than press close, but i don't see any errors.
<details>
<summary>log</summary>
<br>

```

2023-02-20 18:58:15.515 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x826D](TS0601): Device seen - marking the device available and resetting counter
2023-02-20 18:58:15.516 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x826D](TS0601): Update device availability -  device available: True - new availability: True - changed: False
2023-02-20 18:58:16.165 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x826D, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=116, SecurityUse=<Bool.false: 0>, TimeStamp=6971197, TSN=0, Data=b'\x08\xBA\x0A\x01\x00\x20\x46\xE2\xFF\x20\x36\xE4\xFF\x20\x01', MacSrcAddr=0x826D, MsgResultRadius=29)
2023-02-20 18:58:16.168 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x826D), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=0, data=Serialized[b'\x08\xba\n\x01\x00 F\xe2\xff 6\xe4\xff \x01'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=116, rssi=None)
2023-02-20 18:58:16.168 DEBUG (MainThread) [zigpy.zcl] [0x826D:1:0x0000] Received ZCL frame: b'\x08\xba\n\x01\x00 F\xe2\xff 6\xe4\xff \x01'
2023-02-20 18:58:16.169 DEBUG (MainThread) [zigpy.zcl] [0x826D:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=186, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-02-20 18:58:16.170 DEBUG (MainThread) [zigpy.zcl] [0x826D:1:0x0000] Decoded ZCL frame: Basic:Report_Attributes(attribute_reports=[Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=70)), Attribute(attrid=0xFFE2, value=TypeValue(type=uint8_t, value=54)), Attribute(attrid=0xFFE4, value=TypeValue(type=uint8_t, value=1))])
2023-02-20 18:58:16.170 DEBUG (MainThread) [zigpy.zcl] [0x826D:1:0x0000] Received command 0x0A (TSN 186): Report_Attributes(attribute_reports=[Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=70)), Attribute(attrid=0xFFE2, value=TypeValue(type=uint8_t, value=54)), Attribute(attrid=0xFFE4, value=TypeValue(type=uint8_t, value=1))])
2023-02-20 18:58:16.170 DEBUG (MainThread) [zigpy.zcl] [0x826D:1:0x0000] Attribute report received: app_version=70, 0xFFE2=54, 0xFFE4=1
2023-02-20 18:58:16.176 DEBUG (MainThread) [zigpy.zcl] [0x826D:1:0x0000] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=186, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-02-20 18:58:16.176 DEBUG (MainThread) [zigpy.zcl] [0x826D:1:0x0000] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
2023-02-20 18:58:16.177 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x826D), dst_ep=1, source_route=None, extended_timeout=False, tsn=186, profile_id=260, cluster_id=0, data=Serialized[b'\x18\xba\x0b\n\x00'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-02-20 18:58:16.182 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x826D), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=0, TSN=186, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x18\xBA\x0B\x0A\x00')
2023-02-20 18:58:16.203 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2023-02-20 18:58:16.390 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=186)
2023-02-20 18:58:17.681 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.wemostemp_humidity, old_state=<state sensor.wemostemp_humidity=41.6; state_class=measurement, unit_of_measurement=%, device_class=humidity, friendly_name=wemostemp_Humidity @ 2023-02-20T18:57:47.704762+01:00>, new_state=<state sensor.wemostemp_humidity=41.7; state_class=measurement, unit_of_measurement=%, device_class=humidity, friendly_name=wemostemp_Humidity @ 2023-02-20T18:58:17.681224+01:00>>
2023-02-20 18:58:17.766 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x826D, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=127, SecurityUse=<Bool.false: 0>, TimeStamp=7071309, TSN=0, Data=b'\x09\xBB\x11\x00\x03\x40', MacSrcAddr=0x826D, MsgResultRadius=29)
[... cut ...] ment=kWh, device_class=energy, friendly_name=rgbw2bedroom - Total Daily Energy @ 2023-02-20T18:58:56.697141+01:00>>
2023-02-20 18:58:56.947 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xDA58](TS0601): Device seen - marking the device available and resetting counter
2023-02-20 18:58:56.948 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xDA58](TS0601): Update device availability -  device available: True - new availability: True - changed: False

```

</details>

Drmatterpl, 2023-03-03:
after the HA update to 2023.3 it is not working anymore, need to remove the quirk - zha did not start

2023-03-03 17:20:48.301 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry socket://192.168.1.98:8888 for zha
File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 100, in async_setup_entry
File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
File "/config/custom_zha_quirks/ts601_from_Dextrik.py", line 35, in <module>
from zhaquirks.tuya.mcu import (
ImportError: cannot import name 'TuyaDPType' from 'zhaquirks.tuya.mcu' (/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/mcu/__init__.py)

javicalle, 2023-03-03:
Check here:
https://github.com/zigpy/zha-device-handlers/issues/2250#issuecomment-1453978417

Drmatterpl, 2023-03-04:
> @Drmatterpl, check here: [#2250 (comment)](https://github.com/zigpy/zha-device-handlers/issues/2250#issuecomment-1453978417)

Thank YOU!!!
works like a harm, added # to few lines and all is fine.

	Line  38: #    TuyaDPType,
	Line 152: #            dp_type=TuyaDPType.ENUM,
	Line 157: #            dp_type=TuyaDPType.VALUE,
	Line 162: #            dp_type=TuyaDPType.ENUM,

javicalle, 2023-03-09:
> but the buttons are always open the curtain.

Not sure which the problem is. Can you elaborate?

> Is it maybe because:
> data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x02', *payload=2)
> Shouldn't it be TuyaDPType.ENUM?

No, it shouldn't. But...

Maybe you can try with that change in your code:
```python
            cluster_data = TuyaClusterData(
                endpoint_id=self.endpoint.endpoint_id,
                cluster_name=self.ep_attribute,
                cluster_attr="curtain_switch",
                attr_value=int(TUYA2ZB_COMMANDS[command_id]),  # convert tuya2zigbee command
                expect_reply=expect_reply,
                manufacturer=-1,
            )
```

(just wrap `TUYA2ZB_COMMANDS[command_id]` with the `int()`).

Save changes, delete any `__pycache__`  folder in your local quirk folder and restart HA.

caradas, 2023-03-17:
Got it working again. 
By comparing the old Logs with new ones i found that with the removal of specified "TuyaDPType.ENUM" commands are now 4 byte(int32)  instead of 1 byte and  the motor is ignoring the trailing bytes to the command.
As i did not find a way to shorten the command to 1 byte i simply moved the command in the first byte of the int32.

I would call this a hackfix:
````
TUYA2ZB_COMMANDS = {
    0x0000: 0x00000000, #Open
    0x0001: 0x02000000, #Close
    0x0002: 0x01000000, #Stop
}
````

Is there a simple way to reverse the displayed lift percentage?

caradas, 2023-03-17:
Yes i have a remote, up arrow on the remote opens, same goes for up arrow in ha.
Only the percentage is not intuitive as 100% is fully open

javicalle, 2023-03-17:
First of all, you were right:
> Shouldn't it be TuyaDPType.ENUM?

Yes it should, and maybe is the reason of the behavior.

Let's see if this can fix something

My proposal is add and replace this definitions:
```python
class TuyaCC(t.enum8):
    """Tuya cover commands."""

    OPEN = 0x00
    STOP = 0x01
    CLOSE = 0x02

class ZclCC(t.enum8):
    """ZCL cover commands."""

    OPEN = 0x00
    CLOSE = 0x01
    STOP = 0x02

TUYA2ZB_COMMANDS = {
    ZclCC.OPEN: TuyaCC.OPEN,
    ZclCC.CLOSE: TuyaCC.CLOSE,
    ZclCC.STOP: TuyaCC.STOP,
}
```

Also, remove the `int()` wrap from https://github.com/zigpy/zha-device-handlers/issues/1953#issuecomment-1462707698
```python
            cluster_data = TuyaClusterData(
                endpoint_id=self.endpoint.endpoint_id,
                cluster_name=self.ep_attribute,
                cluster_attr="curtain_switch",
                attr_value=TUYA2ZB_COMMANDS[command_id],  # convert tuya2zigbee command
                expect_reply=expect_reply,
                manufacturer=-1,
            )
```

Save changes, remove any `__pycache__` folder and restart.

javicalle, 2023-03-17:
BTW, do you have any `motor_steering (0xF002)` attribute in the `TuyaWindowCovering` cluster? Can you read the value and try to check setting 0/1 values?

caradas, 2023-03-20:
Your Code proposal is working as expected. Thank you.
```
2023-03-20 13:50:02.029 DEBUG (MainThread) [zigpy.zcl] [0x3F47:1:0xef00] raw: b'\x01'
``` 

> BTW, do you have any `motor_steering (0xF002)` attribute in the `TuyaWindowCovering` cluster? Can you read the value and try to check setting 0/1 values?

[image]
Log:
```` 
2023-03-20 13:55:32.972 DEBUG (MainThread) [zigpy.zcl] [0x3F47:1:0x0102] Couldn't normalize 61442 attribute with enum8.undefined_0x00 value
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 481, in read_attributes
    value = self.attributes[record.attrid].type(record.value.value)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/basic.py", line 368, in __call__
    value = int(value)
ValueError: invalid literal for int() with base 10: 'enum8.undefined_0x00'
````

danidask, 2023-08-06:
It works. Let me sum up all the steps and the updated quirk file:
You'll need the HA File editor, can be installed via settings->add-ons
in the file editor, open config/configuration.yaml an add:
```
zha:
  custom_quirks_path: /config/custom_zha_quirks/
```
This will look for quirks in that folder, so create the folder custom_zha_quirks inside config (can be also done with File editor) and inside create a file named TZE200_r0jdjrvi.py and copy the content of this file:
[TZE200_r0jdjrvi.py.txt](https://github.com/zigpy/zha-device-handlers/files/12270123/TZE200_r0jdjrvi.py.txt)
Restart Home Assistant

P.D for pairing mode, press the button in the base of the motor 3 times and then keep it pressed for the forth time for like two seconds. green led will start blinking

utech-git, 2023-09-23:
Hello, guys.

I was having problem pairing and using a similar model curtain motor. But the the quirk file that @danidask posted on his last comment worked like a charm. I just had to change TZE200 to TZE204 on line 261, because my motor model is _TZE204_r0jdjrvi.

The one thing that still isn't working is the curtain state whenever i use the set_position service (either by calling it manually or by selecting the appropriate position on the scroller in the UI).

If click on the half of the bar (50%) and the curtain goes to that position, the state of the curtain keeps on "Opening" forever. I have to click the Stop button for the state to change to Open. The same thing happens with the Closing state.

Is there a way to fix this?

tbhova, 2023-09-24:
You might need to reverse the motor direction reporting. IIRC, this is an
option in the quirk.

javicalle, 2023-09-24:
> Is there a way to fix this?

https://github.com/home-assistant/core/pull/99646

Onepamopa, 2023-10-06:
> It works. Let me sum up all the steps and the updated quirk file: You'll need the HA File editor, can be installed via settings->add-ons in the file editor, open config/configuration.yaml an add: [...]

Added as described, but there is no slider?
Latest ha version...

bschmidt006, 2023-10-11:
In case is helps someone in the future, I was able to use the above quirk to integrate a different model number: _TZE200_xu4a5rhj

I simply changed the model number in the quirk from _TZE200_r0jdjrvi to _TZE200_xu4a5rhj.

Open/Stop/Close seems to work fine.

I did have to set the motor reverse flag to get those going the correct direction.

KirkKirk, 2023-10-26:
> Hello, guys.
>
> I was having problem pairing and using a similar model curtain motor. But the the quirk file that @danidask posted on his last comment worked like a charm. I just had to change TZE200 [...]

Hi @utech-git 
We have the same model (_TZE204_r0jdjrvi) and have encountered the same issue. I don't use rails and the only solution I have found is to set limits by forcing the motor to stop. However, I have to do this every time after a power outage because the motor forgets the limits. Have you found any other solution to set the limits?

utech-git, 2023-10-26:
> Hi @utech-git We have the same model (_TZE204_r0jdjrvi) and have encountered the same issue. I don't use rails and the only solution I have found is to set limits by forcing the motor to stop. [...]

Well, every time after a power outage it loses the limits. But I don't do anything to get the limits back. Once the power is back, the first time I open the curtain it goes slowly to the beggining of the rail and when it reachs it automatically knows where the start limit is. The same way when I first close the curtain, it goes to the end of the rail and set the finish limit. And there it goes.

It knows where the end/beggining is when it feels resistance and can't go any further (just like you said when you force the motor to stop).

utech-git, 2023-10-26:
> In case is helps someone in the future, I was able to use the above quirk to integrate a different model number: _TZE200_xu4a5rhj
>
> I simply changed the model number in the quirk from [...]

Could you please tell me how to set this "reverse flag"? I couldnt find it on the ZHA.
I had to first pair the motor on a tuya zigbee gateway, set the reverse flag on the tuya app, then delete it from tuya and pair on ZHA.

bschmidt006, 2023-10-26:
> Could you please tell me how to set this "reverse flag"? I couldnt find it on the ZHA. I had to first pair the motor on a tuya zigbee gateway, set the reverse flag on the tuya app, then delete it [...]

In ZHA - select the device. Under device info - hit the 3 dots -> Manage zigbee device.

1. Ensure you're on the CLUSTERS tab.
2. Select the TuyaWindowCovering cluster.
3. Select the motor_steering attribute
4. Read attribute to see what it's set to.
5. Flip the value if needed and hit Write Attribute.

This is reflect in the quirk here:
```
"""Add additional attributes for direction"""
    attributes = WindowCovering.attributes.copy()
    attributes.update(
        {
            0xF000: ("curtain_switch", t.enum8, True),  # 0: open, 1: stop, 2: close
            0xF001: ("accurate_calibration", t.enum8, True),  # 0: calibration started, 1: calibration finished
            **0xF002: ("motor_steering", t.enum8, True),  # 0: default, 1: reverse**
            0xF003: ("travel", t.uint16_t, True),  # 30 to 9000 (units of 0.1 seconds)
        }
    )
```

utech-git, 2023-10-30:
> > > In case is helps someone in the future, I was able to use the above quirk to integrate a different model number: _TZE200_xu4a5rhj
> > > I simply changed the model number in the quirk from _TZE200_r0jdjrvi to _TZE200_xu4a5rhj.
> > > Open/Stop/Close seems to work fine.
> > > I did have to set the motor reverse flag to get those going the correct direction.
> > 
> > 
> > Could you please tell me how to set this "reverse flag"? I couldnt find it on the ZHA. I had to first pair the motor on a tuya zigbee gateway, set the reverse flag on the tuya app, then delete it from tuya and pair on ZHA.
> 
> In ZHA - select the device. Under device info - hit the 3 dots -> Manage zigbee device.
> 
> 1. Ensure you're on the CLUSTERS tab.
> 2. Select the TuyaWindowCovering cluster.
> 3. Select the motor_steering attribute
> 4. Read attribute to see what it's set to.
> 5. Flip the value if needed and hit Write Attribute.
> 
> This is reflect in the quirk here:
> 
> ```
>             0xF002: ("motor_steering", t.enum8, True),  # 0: default, 1: reverse
> ```

I did try to change this "True" to "False" on the quirk, at first. But nothing happened.

Now I tried on the Manage Zigbee Device menu... when I click "Read Attribute" it shows "None". I try to write attribute to 0 or 1, but when I read it again it continues "None" and the motor is still not reversed.

Maybe it doesn't work with this model, but thanks any way! :)

Onepamopa, 2023-10-30:
@utech-git works fine with zigbee2mqtt ... I'm using it until zha gets enough supported devices working properly ....

ATaylor5596, 2024-01-01:
I have a similar motor, model TZE200_idkvxabg. Thanks to everyone here I was able to get it to work with the most recent quirk, however, my open/close percentage stays out of sync with the remote control. Switching the motor direction will always have the remote and HA be opposite controls. Does anyone have any idea how to invert the display percentages?

IntegersOfK, 2024-03-02:
> I did try to change this "True" to "False" on the quirk, at first. But nothing happened.
>
> Now I tried on the Manage Zigbee Device menu... when I click "Read Attribute" it shows "None". I try to [...]

I have the same issue, _TZE204_r0jdjrvi model doesn't appear to accept any 0, 1, True, or False for reversing the motor. I wonder if there's a way to tell Home Assistant to treat it as the opposite. Did you figure out any other way to deal with that? I do have a remote but there's nothing described on the docs as to how one might reverse it on the device itself.

yuejun92, 2024-11-21:
Mine is also on reverse and the reverse flag is not working. I read the attribute is "enum8.undefined_0x00" and i changed to "enum8.undefined_0x01", my motor is still on reverse. Please assist.

Tribalwarrior, 2024-12-14:
> It works. Let me sum up all the steps and the updated quirk file: You'll need the HA File editor, can be installed via settings->add-ons in the file editor, open config/configuration.yaml an add: [...]

Worked, Thanks!

CreaturePH, 2025-01-06:
> It works. Let me sum up all the steps and the updated quirk file: You'll need the HA File editor, can be installed via settings->add-ons in the file editor, open config/configuration.yaml an add: [...]

I was looking for a way to add my curtain drive, "Roximo" (TS0601 _TZE204_57hjqelq), I just substituted the data of my model and everything works fine! Thanks, guys! Full code listing for the model, maybe it will be useful to someone else: 
[_TZE204_57hjqelq.py.txt](https://github.com/user-attachments/files/18320588/_TZE204_57hjqelq.py.txt)

SirDboy, 2025-04-21:
To change the direction of the motor click the "set"-button twice, holding the second click until the motor reacts–voila, motor direction changed!