zigpy/zha-device-handlers #1954: [Device Support Request] TS0601 PC321-TY 3-phase Clamp Power Meter

Dogmeat88, 2022-11-22:
Hi I would like to request device support for the TS0601 PC321-TY 3-phase Clamp Power Meter - TZE200_nslr42tt
[https://www.aliexpress.com/item/1005004719888092.html](https://s.click.aliexpress.com/e/_DksL5Lp)

zigbee2mqtt device support thread : [https://github.com/Koenkk/zigbee2mqtt/issues/14790](https://github.com/Koenkk/zigbee2mqtt/issues/14790)

Thanks.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_nslr42tt",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Pacific/Auckland",
    "os_name": "Linux",
    "os_version": "5.15.74",
    "supervisor": "2022.10.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
[... 571 lines cut ...]
            "unsupported_attributes": {}
          }
        }
      },
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

Dogmeat88, 2023-03-31:
Is there anything I can do to help move this thing along?

Any-berg, 2023-08-27:
Please try [matseeplus_ts0601.py.txt](https://github.com/zigpy/zha-device-handlers/files/12448568/matseeplus_ts0601.py.txt) - just remember to remove the .txt suffix from the filename.

This is still far from perfect. For example, HA seems to only create sensors/entities for phase A attributes, although corresponding phase B and C attributes are available. I had to `SKIP_CONFIGURATION`, because configuration was getting bogged by unhandled '_unsupported_attribute_handled' events and FOREIGN_KEY constraint failures, and in the end only Power factor entity was added. Current situation looks a bit better:

[image]

Note that the device does not actually report Apparent power. I could have hidden that sensor but instead I calculated it from Active power and Power factor. Instantaneous demand should not be reported, either, but I could not hide it without hiding also Summation delivered.

Suheaa, 2023-11-22:
matseeplus_ts0601.py.txt - his file works great. Can I get one that works for 3 phases?

mattbyte, 2024-04-17:
As a quick and dirty solution, I have created duplicate clusters on different endpoints to get all the data to come through:
[ts0601_3_phase_energy_meter_edited.zip](https://github.com/zigpy/zha-device-handlers/files/15009421/ts0601_3_phase_energy_meter_edited.zip)

It will creates some redundant entities that can be hidden (the meter only provides one frequency for all phases for example). The order of duplicate entries are Total, Phase A, Phase B, Phase C. Only device status is not presented. Might try to use Quirks V2 to improve this at some point.

Suheaa, 2024-04-17:
Great. That's enough information. And the excess can be hidden.

closed 2024-10-21 by github-actions[bot]

MagicalTux, 2025-01-22:
This was closed for inactivity I guess, but getting support for this would be nice!