zigpy/zha-device-handlers #1959: [BUG] Hive Home MOT003 battery readout wrong

titchy, 2022-11-23:
**Describe the bug**
The battery value reported for the MOT003 device is very wrong, usually a new battery will report 8%, when I looked at the values of the new battery it was reporting 3.2 volts. The batteries are Lithium 3V. In the data from ZHA debug I see it shows battery voltage as 10 x the value of the battery voltage, i.e. 32 for a 3.2 volt reading. but the value in HA is showing as 8%

**To Reproduce**
Steps to reproduce the behavior:
1. pair MOT003 device with new battery
2. Click on device in HA
3. Look at battery readout and see 8%

**Expected behavior**
The battery should show 100%

**Screenshots**
[image]
This screenshot results from this debug data:
"0x0001": {
            "endpoint_attribute": "power",
            "attributes": {
              "0x0020": {
                "attribute_name": "battery_voltage",
                "value": 30
              },
              "0x0021": {
                "attribute_name": "battery_percentage_remaining",
                "value": 15
              },
              "0x0031": {
                "attribute_name": "battery_size",
                "value": 2
              },
              "0x0033": {
                "attribute_name": "battery_quantity",
                "value": 1
              }
            }

and this one:
[image]
from this data:
"0x0001": {
            "endpoint_attribute": "power",
            "attributes": {
              "0x0020": {
                "attribute_name": "battery_voltage",
                "value": 29
              },
              "0x0021": {
                "attribute_name": "battery_percentage_remaining",
                "value": 14
              },
              "0x0031": {
                "attribute_name": "battery_size",
                "value": 2
              },
              "0x0033": {
                "attribute_name": "battery_quantity",
                "value": 1
              }
            }

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4153, maximum_buffer_size=80, maximum_incoming_transfer_size=80, server_mask=0, maximum_outgoing_transfer_size=80, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "6": {
      "profile_id": 260,
[... 370 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

closed 2023-05-29 by github-actions[bot]

leopold005, 2023-07-07:
I'm having this exact same issue, with one door sensor and one motion sensor.  All the others are fine, it just seems to be the new ones I've added recently.  Does anyone have a fix for it?

leopold005, 2023-07-07:
I've done a couple of checks on devices, and it seems the ones with the battery issue have a much earlier firmware, so maybe that's the answer