zigpy/zha-device-handlers #1960: [BUG]

DerekVonk, 2022-11-23:
**Describe the bug**
I'm noticing issues with the Aqara smart plug sensors. I've seen multiple (open) other issues which are about power consumption sensors and such. In order to supply more information to these issues (e.g. https://github.com/zigpy/zha-device-handlers/issues/1952) which I also have on my instance.
I want to retrieve my devices' diagnostic data. 
HomeAssistant doesn't download the requested diagnostics.

**To Reproduce**
Steps to reproduce the behavior:
1. Have an Aqara smart plug (lumi.plug.maeu01 with fw.0029) paired
2. Go to 'Devices' config and info overview page'
3. Click on 'three vertical dots menu icon, next to 'Reconfigure' '
4. Select 'Download diagnostics' from the dropdown
5. See that nothing happens and no diagnostic file is downloaded
6. check the logs for errors

**Expected behavior**
A download diagnostics file with .json or .txt extension containing diagnostic information on the selected device

**Screenshots**
screenshots of device info:
[image]

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.plug.maeu01",
  "class": "zhaquirks.xiaomi.aqara.plug_eu.PlugMAEU01"
}```

</details>

<details>
<!-- Diagnostic information can be acquired by clicking on the "Download Diagnostics" button in the device settings view -->
<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Logger: aiohttp.server
Source: components/zha/diagnostics.py:153
First occurred: 09:58:34 (2 occurrences)
Last logged: 10:31:31

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 82, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 136, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/diagnostics/__init__.py", line 234, in get
    data = await info[sub_type.value](hass, config_entry, device)
  File "/usr/src/homeassistant/homeassistant/components/zha/diagnostics.py", line 101, in async_get_device_diagnostics
    device_info[CLUSTER_DETAILS] = get_endpoint_cluster_attr_data(zha_device)
  File "/usr/src/homeassistant/homeassistant/components/zha/diagnostics.py", line 123, in get_endpoint_cluster_attr_data
    ATTR_IN_CLUSTERS: {
  File "/usr/src/homeassistant/homeassistant/components/zha/diagnostics.py", line 126, in <dictcomp>
    **get_cluster_attr_data(cluster),
  File "/usr/src/homeassistant/homeassistant/components/zha/diagnostics.py", line 152, in get_cluster_attr_data
    UNSUPPORTED_ATTRIBUTES: {
  File "/usr/src/homeassistant/homeassistant/components/zha/diagnostics.py", line 153, in <dictcomp>
    f"0x{cluster.find_attribute(u_attr).id:04x}": {
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 181, in find_attribute
    return self.attributes[name_or_id]
KeyError: 900```

</details>

**Additional context**
Add any other context about the problem here.

TheJulianJES, 2023-01-06:
Can you still reproduce this on Home Assistant Core 2023.1.0?

DerekVonk, 2023-01-11:
Nope, updated to core 2023.1.2 and currently the issue's no longer happening 🙌

closed 2023-01-11 by DerekVonk