zigpy/zha-device-handlers #1963: Support for Back Light on/off and Child Lock on/off for BlitzWolf-SHP15

zleba, 2022-11-24:
These parameters are available in cluster 6 as attributes 32769 (Back Light) and 32768 (Child Lock).
Always 0 means off, 1 means on.

<details>

```yaml
   "nwk": 43382,
    "manufacturer": "_TZ3000_u5u4cakc",
    "model": "TS011F",
    "name": "_TZ3000_u5u4cakc TS011F",
    "quirk_applied": true,
    "quirk_class": "device_plug.Plug",
    "manufacturer_code": 4098,
    "power_source": "Mains",
```

</details>

javicalle, 2022-11-27:
`backlight_mode` already implemented in mainstream but not included in any release
* https://github.com/home-assistant/core/pull/81218

I can't see the `child_lock` implemented yet.

TheJulianJES, 2023-01-04:
Implemented in current HA versions.

closed 2023-01-04 by TheJulianJES