zigpy/zha-device-handlers #1964: [BUG] TS110E eventually reports being turned off, but it is still on

janick, 2022-11-25:
**Describe the bug**
My "At Sunset" automation sucessfully turns one my dimmers at 20% intensity, but after some time, they report as being off. However, the lights are still on. 

#1878 https://community.home-assistant.io/t/2022-7-7-zigbee-dimmers-immediately-turn-off-after-turning-on-via-automation-call-service/445784/26

**To Reproduce**
Logbook
November 24, 2022
Breezeway Sconce turned off
6:28:40 PM - 8 minutes ago
Breezeway Sconce turned on triggered by automation At Sunset triggered
5:46:17 PM - 1 hour ago
Breezeway Sconce turned off
5:29:40 PM - 1 hour ago
Breezeway Sconce turned on triggered by automation At Sunset sunset
4:35:05 PM - 2 hours ago

**Expected behavior**
The HA device should be one, reflecting the state of the dimmer

**Screenshots**
[It shows on just after the automation turns it on](https://imgur.com/a/HTGEvDh)
[45 mins later, it shows off -- but the light is physically on](https://imgur.com/tFbMjuw)

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_wdexaypg",
  "model": "TS110E",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Failed - Network Error
```
That may be a hint? I was able to download it before. Will download it when I'm able.

</details>

<details>
<summary>Additional logs</summary>

Logs that seem relevant below. Hmmmm looks like my enabling 'debug' logs didn't take and will have to restart my instance...

```
Logger: zigpy.zcl
Source: runner.py:119
First occurred: November 21, 2022 at 4:44:20 PM (2648 occurrences)
Last logged: 6:45:53 PM

[0x8BF0:1:0xef00] Unknown cluster command 6 b'\x00\x00k\x02\x00\x04\x00\x00\x00G'
[0x8BF0:1:0xef00] Unknown cluster command 6 b'\x00\x00l\x02\x00\x04\x00\x00\x00T'
[0x8BF0:1:0xef00] Unknown cluster command 6 b'\x00\x00n\x02\x00\x04\x00\x00\x00d'
[0x8BF0:1:0xef00] Unknown cluster command 6 b'\x00\x00l\x02\x00\x04\x00\x00\x00S'
[0x8BF0:1:0xef00] Unknown cluster command 6 b'\x00\x00k\x02\x00\x04\x00\x00\x00H'
```

</details>

janick, 2022-11-25:
I was looking at the ts110e.py quirks and the following comment caught my attention:

```Python
class DimmerSwitchWithNeutral1Gang(TuyaDimmerSwitch):
    """Tuya Dimmer Switch Module With Neutral 1 Gang."""
```

In my case, it is a 2-gang dimmer.

janick, 2022-11-26:
After the automation turned on the light and while it was still showing up as "ON" on the dashboard, I sent a "scan_device" service call. Here are the results. However, the light immediately showed up as "OFF" on the dashboard afterward.

```JSON
{
  "ieee": "a4:c1:38:6f:00:30:63:6a",
  "nwk": "0x6c21",
  "model": "TS110E",
  "manufacturer": "_TZ3210_wdexaypg",
  "manufacturer_id": "0x4417",
  "endpoints": [
    {
      "id": 1,
      "device_type": "0x0101",
      "profile": "0x0104",
      "in_clusters": {
        "0x0000": {
          "cluster_id": "0x0000",
          "title": "Basic",
          "name": "basic",
          "attributes": {
            "0x0000": {
              "attribute_id": "0x0000",
              "attribute_name": "zcl_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "REPORT|READ",
              "access_acl": 5,
              "attribute_value": 3
            },
            "0x0001": {
              "attribute_id": "0x0001",
              "attribute_name": "app_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "REPORT|READ",
              "access_acl": 5,
              "attribute_value": 65
            },
            "0x0002": {
              "attribute_id": "0x0002",
              "attribute_name": "stack_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "REPORT|READ",
              "access_acl": 5,
              "attribute_value": 0
            },
            "0x0003": {
              "attribute_id": "0x0003",
              "attribute_name": "hw_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "REPORT|READ",
              "access_acl": 5,
              "attribute_value": 1
            },
            "0x0004": {
              "attribute_id": "0x0004",
[... 757 lines cut ...]
              "attribute_id": "0x0000",
              "attribute_name": "0",
              "value_type": [
                "0x48",
                "Array",
                "Discrete"
              ],
              "access": "REPORT|READ",
              "access_acl": 5
            }
          },
          "commands_received": {},
          "commands_generated": {}
        }
      },
      "out_clusters": {}
    },
    {
      "id": 242,
      "device_type": "0x0061",
      "profile": "0xa1e0"
    }
  ]
}
```

janick, 2022-11-26:
It seems that turning the dimmer on without an explicit brightness then turning it on with a specified brightness works around the issue:

```YAML
  - service: light.turn_on
    data: {}
    target:
      entity_id:
        - light.breezeway_sconce
 - service: light.turn_on
    data:
      brightness_pct: 20
    target:
      entity_id:
        - light.breezeway_sconce
```

javicalle, 2022-11-26:
Hi there, a few comments from my side.
Your device isn't loading any quirk (`"class": "zigpy.device.Device"`). If the dimming level is working fine from HA I would say that isn't needed either.

Maybe the device needs a fix for that `on_off` cluster (despite it is working fine).
Here I can see 2 approachs:
1. create a quirk that calls the `turn_on` command before dimming the value. That already has been done in other devices that really need it. Not sure about the device behavior, maybe brighness changes while turnning on are apreciated.
2. the are a [`ForceOnLight`](https://github.com/home-assistant/core/blob/6cfb40f080279a45c8e630419742cc2f3faedecb/homeassistant/components/zha/light.py#L895) device that would do the same, but not sure how can link your device to ZHA device definition. In any case, probably will be the same behavior that with the quirk (a `turn_on` call with a `move_to_level`)

Not sure which can be the proper one.

janick, 2022-11-27:
I'm trying to create a quirk for it and I'm unable to get it to match. I know the py file is getting loaded because I had to work through syntax errors when restarting ZHA, but the device signature keeps returning "class": "zigpy.device.Device"

```Python
class DimmerSwitchWithNeutral2Gang(TuyaDimmerSwitch):
    """Tuya Dimmer Switch Module With Neutral 2 Gang."""

    signature = {
        MODELS_INFO: [
            ("_TZ3210_wdexaypgu", "TS110E"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMABLE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },

            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMABLE_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },

            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaLevelControl,
                    TuyaManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time, Ota],
            },
            2: {
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaLevelControl,
                    TuyaManufCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

javicalle, 2022-11-27:
Your code seems good to me.
Have you imported the `GreenPowerProxy `?
Which are the syntax errors?

janick, 2022-11-27:
No more errors, those were obvious to fix. When I [manually reload ZHA](https://imgur.com/a/G0Wljcx), it does so successfully and without errors.

Unless I need to do something more drastic to HA to have it apply the quirk??

janick, 2022-11-27:
Just did a full restart. No dice.

javicalle, 2022-11-27:
Double check that there is no a `__pycache__` folder in your local quirk folder. If exists, delete the folder and restart HA.

In the documentation is explained how to enable the debug logs:
* https://www.home-assistant.io/integrations/zha/#debug-logging

I have reviewed it and can't find any reason why the quirk isn't loading. Check at the debug logs.
I wouldn't use the `TuyaLevelControl` in the replacement part. If the device is dimming the correct values you don't need it.

janick, 2022-11-27:
There was a __pycache__ folder. Removed and restarted with ZHA debug enabled.

My quirks file is loaded:

```
2022-11-27 09:36:15.076 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.zhongxing.motion
2022-11-27 09:36:15.078 DEBUG (MainThread) [zhaquirks] Loading custom quirks from /config/custom_zha_quirks
2022-11-27 09:36:15.082 DEBUG (MainThread) [zhaquirks] Loading custom quirks module ts004f
2022-11-27 09:36:15.085 DEBUG (MainThread) [zhaquirks] Loading custom quirks module ts110e
2022-11-27 09:36:15.090 DEBUG (MainThread) [homeassistant.components.zha] ZHA storage file does not exist or was already removed
2022-11-27 09:36:31.010 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace2'>
2022-11-27 09:36:31.011 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace1'>
...
```
but I never see a debug message stating that the registry is considering by quirky class.

The next instance of "ts110e" in the log file is:
```
2022-11-27 09:36:36.377 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'select' component -> 'ZHAStartupOnOffSelectEntity' using ['on_off']
2022-11-27 09:36:36.381 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] [0xF662](_TZ3210_wdexaypg TS110E) restored as 'available', last seen: 0:22:23 ago, consider_unavailable_time: 7200 seconds
2022-11-27 09:36:36.386 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'button' component -> 'ZHAIdentifyButton' using ['identify']
```
Should I delete the device entirely and re-pair it??

javicalle, 2022-11-27:
Yes, after restarting, repair your device and check if the device signature has changed.

janick, 2022-11-28:
I have another brand new dimmer of the exact same model I just paired.

[Nope. Still a plain zigpy device.](https://imgur.com/a/VCFyza3)

janick, 2022-11-29:
Does that mean I need to impair/delete all instances of the same device for ZHA to recognize a quirk?

javicalle, 2022-11-29:
No, just the device you want to test.
If enable the debug logs, the pairing logs could tell to us what is happening with the quirk.

javicalle, 2022-12-01:
The quirk must not be loading. Review the HA boot logs and look for any errors or exceptions during boot that indicate an error in the quirk or ZHA.
I can't find a reason why the device isn't loading the quirk.

janick, 2022-12-04:
Updated the logs settings to the following and restarted.

```YAML
logger:
    default: info
    logs:
#        homeassistant.components.mqtt: debug
#        homeassistant.core: debug
        homeassistant.components.zha: debug
        bellows.zigbee.application: debug
        bellows.ezsp: debug
        zigpy: debug
        zigpy_deconz.zigbee.application: debug
        zigpy_deconz.api: debug
        zigpy_xbee.zigbee.application: debug
        zigpy_xbee.api: debug
        zigpy_zigate: debug
        zigpy_znp: debug
```

then added another brand new never-paired dimmer. Here are the full log messages. Doesn't even appear to consider the quirk...

```
2022-12-04 13:01:33.274 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'button' component -> 'ZHAIdentifyButton' using ['identify']
2022-12-04 13:01:33.275 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'RSSISensor' using ['basic']
2022-12-04 13:01:33.275 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'sensor' component -> 'LQISensor' using ['basic']
2022-12-04 13:01:33.278 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'OnOffTransitionTimeConfigurationEntity' using ['level']
2022-12-04 13:01:33.278 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'OnLevelConfigurationEntity' using ['level']
2022-12-04 13:01:33.278 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'OnTransitionTimeConfigurationEntity' using ['level']
2022-12-04 13:01:33.278 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'OffTransitionTimeConfigurationEntity' using ['level']
2022-12-04 13:01:33.279 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'DefaultMoveRateConfigurationEntity' using ['level']
2022-12-04 13:01:33.279 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'StartUpCurrentLevelConfigurationEntity' using ['level']
2022-12-04 13:01:33.279 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'select' component -> 'ZHAStartupOnOffSelectEntity' using ['on_off']
2022-12-04 13:01:33.283 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'button' component -> 'ZHAIdentifyButton' using ['identify']
2022-12-04 13:01:33.285 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'OnOffTransitionTimeConfigurationEntity' using ['level']
2022-12-04 13:01:33.286 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'OnLevelConfigurationEntity' using ['level']
2022-12-04 13:01:33.286 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'OnTransitionTimeConfigurationEntity' using ['level']
2022-12-04 13:01:33.287 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'OffTransitionTimeConfigurationEntity' using ['level']
2022-12-04 13:01:33.287 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'DefaultMoveRateConfigurationEntity' using ['level']
2022-12-04 13:01:33.288 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'number' component -> 'StartUpCurrentLevelConfigurationEntity' using ['level']
2022-12-04 13:01:33.288 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] 'select' component -> 'ZHAStartupOnOffSelectEntity' using ['on_off']
2022-12-04 13:01:33.293 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0x72CC:a4:c1:38:2f:75:e7:bb:9e entering async_device_initialized - is_new_join: True
2022-12-04 13:01:33.294 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0x72CC:a4:c1:38:2f:75:e7:bb:9e has joined the ZHA zigbee network
2022-12-04 13:01:33.295 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x72CC](TS110E): started configuration
2022-12-04 13:01:33.296 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x72CC:ZDO](TS110E): 'async_configure' stage succeeded
2022-12-04 13:01:33.313 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x72CC:1:0x0006]: Performing cluster binding
2022-12-04 13:01:33.320 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x72CC:1:0x0008]: Performing cluster binding
2022-12-04 13:01:33.323 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x72CC:1:0x0003]: Configuring cluster attribute reporting
2022-12-04 13:01:33.324 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x72CC:1:0x0003]: finished channel configuration
2022-12-04 13:01:33.324 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x72CC:1:0x0000]: Configuring cluster attribute reporting
2022-12-04 13:01:33.324 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x72CC:1:0x0000]: finished channel configuration
2022-12-04 13:01:33.325 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x72CC:1:0xef00]: Performing cluster binding
2022-12-04 13:01:33.328 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x72CC:1:0x0019]: finished channel configuration
2022-12-04 13:01:33.328 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x72CC:2:0x0006]: Performing cluster binding
[... cut ...]
2022-12-04 13:01:43.583 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x72CC:2:0x0008]: received attribute: 61440 update with value: 1000
2022-12-04 13:01:43.583 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x72CC:2:0x0008]: received attribute: 64515 update with value: 11
2022-12-04 13:01:43.584 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x72CC:2:0x0008]: received attribute: 64516 update with value: 1000
2022-12-04 13:01:43.688 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x72CC:2:0x0008]: received attribute: 64514 update with value: 0
```

janick, 2022-12-04:
I have confirmed it is loading my quirk file without errors:

```
2022-12-04 13:17:36.335 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.zhongxing
2022-12-04 13:17:36.339 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.zhongxing.motion
2022-12-04 13:17:36.341 DEBUG (MainThread) [zhaquirks] Loading custom quirks from /config/custom_zha_quirks
2022-12-04 13:17:36.344 DEBUG (MainThread) [zhaquirks] Loading custom quirks module ts004f
2022-12-04 13:17:36.347 DEBUG (MainThread) [zhaquirks] Loading custom quirks module ts110e
```

janick, 2022-12-04:
Checking the source code, I see that one of the considered quirk is defined as follows:

```Python
class MoesTemperatureHumidtySensorWithScreen(CustomDevice):
    """Moes temperature and humidity sensor with screen."""

    signature = {
        #  <SimpleDescriptor endpoint=1, profile=260, device_type="0x0302"
        #  input_clusters=["0x0000", "0x0001", "0x0003", "0x0402", "0x0405", "0xe002"]
        #  output_clusters=["0x0003", "0x000a", "0x0019"]>
        ENDPOINTS: {
            1: {
```

I note the absence of "MODEL INFO" which makes it available via the "[None][None]" look up:

```Python
        _LOGGER.debug(
            "Checking quirks for %s %s (%s)",
            device.manufacturer,
            device.model,
            device.ieee,
        )
        for candidate in itertools.chain(
            self.registry[device.manufacturer][device.model],
            self.registry[device.manufacturer][None],
            self.registry[None][device.model],
            self.registry[None][None],
        ):
            _LOGGER.debug("Considering %s", candidate)
```

So the problem appears to be a mismatch in device.manufacturer and/or device.model. However, I do not see the debug message that should have been issued by the first statement which would have given valuable info...

janick, 2022-12-04:
Annnnnnnd it was an extra "u" in the manufacturer id...

```
  "manufacturer": "_TZ3210_wdexaypg",
  "model": "TS110E",
  "class": "ts110e.DimmerSwitchWithNeutral2Gang"
```

janick, 2022-12-04:
The bug is with the move_to_level_with_on_off command to the LevelControl cluster. If I do a 'light.toggle' service call to the dimmer switch, it does not physically turn on. I think I need to send an explicit "turn on" command to the OnOff cluster if a move_to_level_with_on_off command with a level > 0 is sent.

```
2022-12-04 14:38:11.596 INFO (MainThread) [homeassistant.components.automation.test_ts110e] Test TS110E: Running automation actions
2022-12-04 14:38:11.596 INFO (MainThread) [homeassistant.components.automation.test_ts110e] Test TS110E: Executing step call service
2022-12-04 14:38:11.602 DEBUG (MainThread) [homeassistant.components.zha.entity] light.tz3210_wdexaypg_ts110e_light: setting transitioning flag to True
2022-12-04 14:38:11.603 DEBUG (MainThread) [zigpy.zcl] [0xE3AD:1:0x0008] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=180, command_id=4, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-12-04 14:38:11.604 DEBUG (MainThread) [zigpy.zcl] [0xE3AD:1:0x0008] Sending request: move_to_level_with_on_off(level=51, transition_time=0)
2022-12-04 14:38:11.647 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xE3AD), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=35, profile_id=260, cluster_id=8, data=Serialized[b'\x18\xb4\x0b\x04\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=244, rssi=-39)
2022-12-04 14:38:11.648 DEBUG (MainThread) [zigpy.zcl] [0xE3AD:1:0x0008] Received ZCL frame: b'\x18\xb4\x0b\x04\x00'
2022-12-04 14:38:11.649 DEBUG (MainThread) [zigpy.zcl] [0xE3AD:1:0x0008] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=180, command_id=11, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-04 14:38:11.652 DEBUG (MainThread) [zigpy.zcl] [0xE3AD:1:0x0008] Decoded ZCL frame: LevelControl:Default_Response(command_id=4, status=<Status.SUCCESS: 0>)
2022-12-04 14:38:11.653 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xE3AD:1:0x0008]: executed 'move_to_level_with_on_off' command with args: '()' kwargs: '{'level': 51, 'transition_time': 0}' result: Default_Response(command_id=4, status=<Status.SUCCESS: 0>)
2022-12-04 14:38:11.654 DEBUG (MainThread) [homeassistant.components.zha.entity] light.tz3210_wdexaypg_ts110e_light: starting transitioning timer for 0.25
2022-12-04 14:38:11.655 DEBUG (MainThread) [homeassistant.components.zha.entity] light.tz3210_wdexaypg_ts110e_light: turned on: {'move_to_level_with_on_off': Default_Response(command_id=4, status=<Status.SUCCESS: 0>)}
2022-12-04 14:38:11.905 DEBUG (MainThread) [homeassistant.components.zha.entity] light.tz3210_wdexaypg_ts110e_light: transition complete - future attribute reports will write HA state
2022-12-04 14:38:12.455 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xE3AD), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=36, profile_id=260, cluster_id=8, data=Serialized[b'\x08\xd6\n\x00\x00 3'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=244, rssi=-39)
2022-12-04 14:38:12.456 DEBUG (MainThread) [zigpy.zcl] [0xE3AD:1:0x0008] Received ZCL frame: b'\x08\xd6\n\x00\x00 3'
2022-12-04 14:38:12.457 DEBUG (MainThread) [zigpy.zcl] [0xE3AD:1:0x0008] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=214, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-04 14:38:12.461 DEBUG (MainThread) [zigpy.zcl] [0xE3AD:1:0x0008] Decoded ZCL frame: LevelControl:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint8_t, value=51))])
2022-12-04 14:38:12.462 DEBUG (MainThread) [zigpy.zcl] [0xE3AD:1:0x0008] Received command 0x0A (TSN 214): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint8_t, value=51))])
2022-12-04 14:38:12.463 DEBUG (MainThread) [zigpy.zcl] [0xE3AD:1:0x0008] Attribute report received: current_level=51
2022-12-04 14:38:12.463 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xE3AD:1:0x0008]: received attribute: 0 update with value: 51
2022-12-04 14:38:12.465 DEBUG (MainThread) [zigpy.zcl] [0xE3AD:1:0x0008] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=214, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[... cut ...]
            # getting the level value
            if kwargs and "level" in kwargs:
                level = kwargs["level"]
            elif args:
                level = args[0]
            else:
                level = 0
            
            if level > 0:
                return await self.endpoint.OnOff.command(
                    0x0001,
                    None,
                    manufacturer=manufacturer,
                    expect_reply=expect_reply,
                    tsn=tsn,
                )
```

javicalle, 2022-12-05:
> The bug is with the move_to_level_with_on_off command to the LevelControl cluster.

Yes, the device should trigger the `on_off` command in the `OnOff` cluster and update its status but it is not doing it, so the bug is in the device.

I believe that ZHA update is the way to go because already have an implementation for this kind of devices.

> it smells like I'm going to have to use a Bus and listener.

Don't go this way, there is no need. You can fix your code like:
```python
self.endpoint.on_off.command(...
```

janick, 2022-12-05:
I'm getting a "Multiple values for argument 'level'" error...

```Python
    async def command(
        self,
        command_id: Union[foundation.GeneralCommand, int, t.uint8_t],
        *args,
        manufacturer: Optional[Union[int, t.uint16_t]] = None,
        expect_reply: bool = True,
        tsn: Optional[Union[int, t.uint8_t]] = None,
        **kwargs: Any,
    ):
        """Override the default Cluster command."""
        self.debug(
            "Sending Cluster Command. Cluster Command is %x, Arguments are %s %s",
            command_id,
            args,
            kwargs
        )
        # If command is move_to_level_with_on_off with level > 0, explicitly turn on
        if command_id == 0x0004:
            # getting the level value
            if kwargs and "level" in kwargs:
                level = kwargs["level"]
            elif args:
                level = args[0]
            else:
                level = 0
            
            if level > 0:
                await self.endpoint.on_off.command(0x0001)
                
        return await super().command(
            command_id, *args, manufacturer, expect_reply, 1, tsn, **kwargs
        )
```

javicalle, 2022-12-07:
I have to said one more time that this is not the way to go but...

```python
        return await super().command(
            command_id, level, manufacturer, expect_reply, tsn, level=level
        )
```

janick, 2022-12-07:
I agree a ZHA fix would be better but this I can fix now :-)

It seems like Tuya devices use a plethora of manufacturer names. It will be a pain to keep up. This begs the question: why not make all devices with a "LevelControl" cluster a "ForceOnLights" devices?

janick, 2023-02-12:
Since upgrading to HA 2023.2, these dimmers have been working just fine, without any errors from the zigpy layer.

```
"manufacturer": "_TZ3210_wdexaypg",
  "model": "TS110E",
  "class": "zigpy.device.Device"
```

I'm going to close this issue as "fixed".

closed 2023-02-12 by janick