zigpy/zha-device-handlers #1966: TS011F by _TZ3000_okaz9tjs

Fierydwarf, 2022-11-27:
Previously i had issues with my TS011F by _TZ3000_okaz9tjs Tuya plug not showing the correct values - the issues were resolved here (https://github.com/zigpy/zha-device-handlers/issues/1764) by adding quirks.
Now i have another issue - the energy monitoring plug is now not available to the home assistant energy dashboard via Energy Dashboard > Add Device. it was before using quirks to modify the values. 

For the device to be avaliable after modifying it using quirks. I hav a few of these plugs and would like to resolve this before adding them

<details>

<summary>Device signature</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=0, server_mask=11264, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe001"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_okaz9tjs",
  "model": "TS011F",
  "class": "ts011f_plug.Plug_v2"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "5.15.61-v8",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "pyscript": {
      "version": "1.3.3",
      "requirements": [
        "croniter==1.3.4",
        "watchdog==2.1.6"
      ]
    },
    "uptime_kuma": {
[... 797 lines cut ...]
              }
            }
          }
        },
        "out_clusters": {}
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-11-27:
You are using a local quirk. After reading the related issue not sure which is the quirk content.

Have you tried to remove and pair the device again?
Also you can try the reconfigure option from the device view.

Fierydwarf, 2022-11-27:
This section on the related issue resolved it 

> 
You would need enable the local quirk configuration. There are some info for example here:

https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274
Once configured, copy the current ts011f_plug.py file inside the custom_zha_quirks folder:

https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f_plug.py
Add the proposed quirk at the end of the file. That would be my proposal:

class Plug_v2(CustomDevice):
    """Another TS011F Tuya plug."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # "profile_id": 260,
            # "device_type": "0x0051",
            # "in_clusters": ["0x0000", "0x0003", "0x0004", "0x0005", "0x0006", "0x0702", "0x0b04", "0xe001"],
            # "out_clusters": []
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
Once added you need to restart HA, remove the device and pair it again.

As part of the process, i removed and repaired the device. -which resulted in the correct values being displayed. Ive also tried the reconfiguration option after repairing.
Do you think i should remove and re-add again?

javicalle, 2022-11-30:
@dmulcahey Sorry to ping you directly, but I'm not familiar with the energy dashboard.

The related quirk just modify:
* `DEVICE_TYPE` from SMART_PLUG to ON_OFF_PLUG_IN_UNIT
* clusters definitions, but just the `multiplier` and `divisor`:
```python
class TuyaZBMeteringCluster(CustomCluster, Metering):
    """Divides the kWh for tuya."""

    MULTIPLIER = 0x0301
    DIVISOR = 0x0302
    _CONSTANT_ATTRIBUTES = {MULTIPLIER: 1, DIVISOR: 100}

class TuyaZBElectricalMeasurement(CustomCluster, ElectricalMeasurement):
    """Divides the Current for tuya."""

    AC_CURRENT_MULTIPLIER = 0x0602
    AC_CURRENT_DIVISOR = 0x0603
    _CONSTANT_ATTRIBUTES = {AC_CURRENT_MULTIPLIER: 1, AC_CURRENT_DIVISOR: 1000}

```

No `cluster_id` or `ep_attribute` modification here, just the cluster class.
Can you point me to the HA code responsible for the `Add device` for the Energy Dashboard.

Thanks in advanced.

dmulcahey, 2022-11-30:
Not sure what you mean. If it exposes a summation sensor you just need to go to the dashboard and add the device

MattWestb, 2022-11-30:
@javicalle The quirked device is not setting the multiplier and divisor so the quirk is adding them so the dashboard is shewing the right measurements.
Power measurements and more sensors if sending one value and then have one fixed` / `and` * ` for presenting the right formatted to the host system.

The rest is OK Zigbee and is working OK (not all tuyas is very bad ;-)) ).

javicalle, 2022-11-30:
User claims that it was possible to add the device to the Energy dashboard before apply the quirk.
Once the quirk is applied to the device, the device was not avaliable to add to the Energy Dashboard.

I'm not familiar with the energy dashboard because I don't have it enabled in my instalation, but I can't see any reason why the behavior would change after applying the quirk. And I am not able to locate where is this functionallity in HA to take a look.

MattWestb, 2022-11-30:
Restart HA so its loading all things OK and can hocking up the device with  the new configuration.

javicalle, 2022-11-30:
@Fierydwarf can you do a HA restart and try to add the device again?

Fierydwarf, 2022-12-01:
@javicalle  Sure. To confirm what you'd want me to do. Remove the device from ZHA, reboot, re-add the device , reboot?

javicalle, 2022-12-01:
I'm going to suggest:
* remove the device from ZHA
* reboot HA
* re-add the device

At this point, try again to add the device to the Energy dashboard.

javicalle, 2022-12-01:
Could be that the device is not avaliable to add to dashboard if already included?

Fierydwarf, 2022-12-08:
> Could be that the device is not avaliable to add to dashboard if already included?

Unfortunately, removing and re-adding didn't work. I'll just not add it to the energy monitor. At least the numbers are correct.

javicalle, 2022-12-08:
Maybe the problem is with the use of a local quirk.
I just checked that the previous issue was closed but no PR was made.
Can you create the PR with the changes?

rofo69, 2022-12-09:
I have this device and using the information above i created this quirk:-

`"""TS011F plug."""

from zigpy.profiles import zha
from zigpy.quirks import CustomCluster,CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaNewManufCluster,
    TuyaZBE000Cluster,
    TuyaZBElectricalMeasurement,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBMeteringCluster,
    TuyaZBOnOffAttributeCluster,
)
from zhaquirks.tuya.mcu import EnchantedDevice

class TuyaZBMeteringCluster(CustomCluster, Metering):
    """Divides the kWh for tuya."""

    MULTIPLIER = 0x0301
    DIVISOR = 0x0302
    _CONSTANT_ATTRIBUTES = {MULTIPLIER: 1, DIVISOR: 100}

class TuyaZBElectricalMeasurement(CustomCluster, ElectricalMeasurement): 
    #Divides the Current for tuya.
    AC_CURRENT_MULTIPLIER = 0x0602 
    AC_CURRENT_DIVISOR = 0x0603 
    _CONSTANT_ATTRIBUTES = {AC_CURRENT_MULTIPLIER: 1, AC_CURRENT_DIVISOR: 1000}

class Plug(CustomDevice):
    """Another TS011F Tuya plug."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # "profile_id": 260,
            # "device_type": "0x0051",
            # "in_clusters": ["0x0000", "0x0003", "0x0004", "0x0005", "0x0006", "0x0702", "0x0b04", "0xe001"],
            # "out_clusters": []
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }

`
I added these two classes into the quirk:-

class TuyaZBMeteringCluster(CustomCluster, Metering):
class TuyaZBElectricalMeasurement(CustomCluster, ElectricalMeasurement): 

which I understand are for adjusting the incorrect power metering figures, but mine are still way out.

It looks like the device has taken the quirk, and added three missing configuration items (backlight mode, child lock and power on state) but I'm not sure what to do to fix the numbers:-

[image]

rofo69, 2022-12-09:
FYI, for some reason the RMS current values are starting to report correctly, so all that is left is the total summation values.
I also removed those 2 classes I referred to and checked the cluster attributes so see those values were already correct.
I can confirm that before the quirk was installed, the device was selectable as an energy source for the energy dashboard, but it no longer appears now its using the quirk.

mattsaxon, 2022-12-14:
> Previously i had issues with my TS011F by _TZ3000_okaz9tjs Tuya plug not showing the correct values - the issues were resolved here (#1764) by adding quirks. Now i have another issue - the energy [...]

I've not added the quirk yet, but I can't add these plugs to the energy dashboard. I'm using Home Assistant 2022.12.1 at the moment

Fierydwarf, 2022-12-14:
> > Previously i had issues with my TS011F by _TZ3000_okaz9tjs Tuya plug not showing the correct values - the issues were resolved here (#1764) by adding quirks. Now i have another issue - the energy monitoring plug is now not available to the home assistant energy dashboard via Energy Dashboard > Add Device. it was before using quirks to modify the values.
> 
> I've not added the quirk yet, but I can't add these plugs to the energy dashboard either. Perhaps this is an change that was caused by a newer HA, I'm using Home Assistant 2022.12.1 at the moment

I have that issue too! I added another plug to test if it I could add it to the energy dashboard with out the quirk and nothing.

Fierydwarf, 2022-12-14:
> Maybe the problem is with the use of a local quirk. I just checked that the previous issue was closed but no PR was made. Can you create the PR with the changes?

PR?

mattsaxon, 2022-12-14:
PR, a GitHub Pull Request, to take the local quirk and include it in the zigpy standard delivery

mattsaxon, 2023-01-02:
OK, I've added the local quirk as @Fierydwarf has and had a look around. Seems to me the issue is that the plug doesn't support the following atribute

`              "0x0300": {
                "attribute_name": "unit_of_measure"
              },`

Can anyone instruct me how to mark a quirk return this when the device doesn't support it?

MattWestb, 2023-01-02:
look how the other attributes that is missing from the device is fixed here 
https://github.com/zigpy/zha-device-handlers/blob/071190e7ebaff6e3601c328726be585820f8adb0/zhaquirks/tuya/__init__.py#L989-L995
And adding the 0x0300 and the definition is here in the system  that is imprted 
https://github.com/zigpy/zigpy/blob/a5ae12efc6510df9c487e775bef7af8937ad85f5/zigpy/zcl/clusters/smartenergy.py#L208..
The value you shall using i think is `0x00` `kWh (Kilowatt Hours) & kW (Kilowatts) in pure binary format`. 
If you need other values look in ZCL R8 `Table 10-72. UnitofMeasure Attribute Enumerations`.

javicalle, 2023-01-03:
Here you have a couple of examples:
* https://github.com/zigpy/zha-device-handlers/blob/7788036230986cebe8e65281d7f58c7566172c8a/zhaquirks/tuya/ts0601_din_power.py#L120-L130
* https://github.com/zigpy/zha-device-handlers/blob/7788036230986cebe8e65281d7f58c7566172c8a/zhaquirks/tuya/ts0601_valve.py#L32-L43

The later one (`TuyaValveWaterConsumed`) also defines the `metering_device_type` (0x0306), mapped here:
* https://github.com/home-assistant/core/blob/8a0fb21988fe4c92a849e5af39423b7c8120ec90/homeassistant/components/zha/core/channels/smartenergy.py#L82-L96

mattsaxon, 2023-01-05:
Thanks @javicalle and @MattWestb , I've successfully tested locally and submitted a PR for both this issue and #1764.

@Fierydwarf do you want to see if it resolves your HA dashboard issues for me it has.

closed 2023-01-10 by dmulcahey

arsenyspb, 2023-03-05:
@mattsaxon could you leave code review & your suggestions in [pull/2008](https://github.com/zigpy/zha-device-handlers/pull/2008) - based on [issues/2005](https://github.com/zigpy/zha-device-handlers/issues/2005) there is a need to add `MULTIPLIER` and `DIVISOR` too? @javicalle I see you're already in the PR, maybe can you propose similar quirk for it?...

mattsaxon, 2023-03-05:
Id be pleased to take a look, but am away from computer for 3 weeks

javicalle, 2023-03-05:
> see you're already in the PR, maybe can you propose similar quirk for it?...

Not sure which PR are you refering.
In PR #2008 I have asked for some changes but no response from author.

jesus51998, 2024-02-07:
Hi, I have a device kind “_TZ3000_okaz9tjs”, I followed the thread, but my device, don’t work well,
the energy and current are multiplier for 100 or 1000.
Somebody can help my.
1.- Delete device.
2.- delete folder  __cache__
3.- copy file https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f_plug.py in my folder /config/custom_zha_quirks/
4.- In configuration.yaml insert 
zha:
   enable_quirks: true
   custom_quirks_path: /config/custom_zha_quirks/
5.- Reboot HA.
6.- Add device

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=0, server_mask=11264, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x0702",
        "0x0b04",
        "0x1000",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZ3000_okaz9tjs",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}

GuidoC1, 2024-03-08:
@jesus51998 I have the same device and applied the quirk provided in [this post](https://github.com/zigpy/zha-device-handlers/issues/1764#issuecomment-1946842958). It works fine for me and multiple other users. Maybe you can give the custom quirk provided in that post a try?

GuidoC1, 2024-03-08:
**For reference** (as others might be looking for the same quick)**:**
For the custom quirk for "device_type": "0x0100", as Jesus51998 was looking for, see [this post](https://github.com/zigpy/zha-device-handlers/issues/1764#issuecomment-1946842958). It has been confirmed to work by mutliple people.

PR #2069 (2023-01-10): Add support for TS011F by _TZ3000_okaz9tjs
Fixes #1764, fixes #1966 

Note, I doubt this is fully ready, but it is functional.

Potential discussions are if the new Plug class named correctly, is adding _v2 appropriate?

And should class TuyaZBMeteringCluster_v2() be in this file or say, in __init__.py