zigpy/zha-device-handlers #1968: [Device Support Request] _TZ3000_46t1rvdu no power metering

redQs, 2022-11-27:
**Is your feature request related to a problem? Please describe.**
Hey, i just addat an Tuya TS0001 _TZ3000_46t1rvdu i can klick on the switch but i it dont give me any power metering its all the time 0

**Describe the solution you'd like**
how can i fix that problem? 

<details>

<summary>Device signature</summary>

```{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_46t1rvdu",
  "model": "TS0001",
  "class": "zhaquirks.tuya.ts000x.Switch_1G_Metering"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
[image]

javicalle, 2022-11-29:
Despite the device signature it seems that device don't support any electrical metering.
Z2M configs declares the device just as a `OnOff` switch:
* https://github.com/Koenkk/zigbee-herdsman-converters/blob/33d7003c2cabfe963d0a34eb2b1447e26abed004/devices/tuya.js#L814

Do you have any evidence or vendor declaration that device supports metering?

redQs, 2022-11-30:
i ask them, they say it in the advertisement.

Why does it show up from itself in home assistant when it doesnt support it?

HENN0, 2022-12-04:
It hsould have power metering 
There is a 1 Ohm resistor between Nin and Nout

[image]

javicalle, 2022-12-04:
Have you tried to remove the device from HA and pair it again?
If not working, we need the logs to see what can be happening. Check the documentation to enable it and attach the relevant info:
* https://www.home-assistant.io/integrations/zha/#debug-logging

HENN0, 2022-12-04:
I don't use Home-Assistent
Zigbee2mqtt and as home server FHEM
As the power is not shown in zigbee2mqtt, it cannot been shown in FHEM and i guess also in HA

But i did repair it already after switching to dev git

redQs, 2022-12-05:
Do you need this?
`{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.74",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.28.4",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "govee": {
      "version": "0.2.2",
      "requirements": [
        "govee-api-laggat==0.2.2",
        "dacite==1.6.0"
      ]
    },
    "smartthinq_sensors": {
      "version": "0.24.2",
      "requirements": [
        "pycountry>=20.7.3",
        "xmltodict>=0.12.0",
        "cchardet>=2.1.7"
      ]
    },
    "localtuya": {
      "version": "4.1.1",
      "requirements": []
    },
    "scheduler": {
      "version": "v0.0.0",
      "requirements": []
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.4",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.87",
      "zigpy-deconz==0.19.1",
      "zigpy==0.51.6",
      "zigpy-xbee==0.16.2",
      "zigpy-zigate==0.10.3",
      "zigpy-znp==0.9.1"
    ],
    "usb": [
      {
        "vid": "10C4",
[... 433 lines cut ...]
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      },
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}`

redQs, 2022-12-05:
```{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.74",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.28.4",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "govee": {
      "version": "0.2.2",
      "requirements": [
        "govee-api-laggat==0.2.2",
        "dacite==1.6.0"
      ]
    },
    "smartthinq_sensors": {
      "version": "0.24.2",
      "requirements": [
        "pycountry>=20.7.3",
        "xmltodict>=0.12.0",
        "cchardet>=2.1.7"
      ]
    },
    "localtuya": {
      "version": "4.1.1",
      "requirements": []
    },
    "scheduler": {
      "version": "v0.0.0",
      "requirements": []
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.4",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.87",
      "zigpy-deconz==0.19.1",
      "zigpy==0.51.6",
      "zigpy-xbee==0.16.2",
      "zigpy-zigate==0.10.3",
      "zigpy-znp==0.9.1"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
[... 433 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      },
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

redQs, 2022-12-05:
<summary>Additional logs</summary>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.74",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.28.4",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "govee": {
      "version": "0.2.2",
      "requirements": [
        "govee-api-laggat==0.2.2",
        "dacite==1.6.0"
      ]
    },
    "smartthinq_sensors": {
      "version": "0.24.2",
      "requirements": [
        "pycountry>=20.7.3",
        "xmltodict>=0.12.0",
        "cchardet>=2.1.7"
      ]
    },
    "localtuya": {
      "version": "4.1.1",
      "requirements": []
    },
    "scheduler": {
      "version": "v0.0.0",
      "requirements": []
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.4",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.87",
      "zigpy-deconz==0.19.1",
      "zigpy==0.51.6",
      "zigpy-xbee==0.16.2",
      "zigpy-zigate==0.10.3",
      "zigpy-znp==0.9.1"
    ],
    "usb": [
[... 436 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      },
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

redQs, 2022-12-05:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

<details>

<summary>Device signature</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.74",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.28.4",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "govee": {
      "version": "0.2.2",
      "requirements": [
        "govee-api-laggat==0.2.2",
        "dacite==1.6.0"
      ]
    },
    "smartthinq_sensors": {
      "version": "0.24.2",
      "requirements": [
        "pycountry>=20.7.3",
        "xmltodict>=0.12.0",
[... 470 lines cut ...]
      },
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

redQs, 2022-12-05:
or you need this right? :D
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/10149889/home-assistant.log)

pervolianinen, 2022-12-05:
same on TS0004
_TZ3000_ltt60asa. in device no functional of voltage and etc. but in HA sensors with 0 data. 

<details>

<summary>Device signature</summary>

```{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_ltt60asa",
  "model": "TS0004",
  "class": "tuya.ts000x.Switch_4G_Metering"
}
```

</details>

javicalle, 2022-12-05:
Can you try to remove from HA the device, wait a minute and pair it again?

HENN0, 2022-12-05:
Yes, dosnt help

"dateCode":"","friendlyName":"Stecker1","hardwareVersion":1,"ieeeAddr":"0xa4c138fb4cba77fd","manufacturerID":4417,"manufacturerName":"_TZ3000_gjnozsaz","model":"TS011F_plug_1","networkAddress":35969,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":0.81,"indicator_mode":"

javicalle, 2022-12-06:
Ok, that's what I can see in the logs (device: `0xCE12`):

1. the device is initializing:
```
2022-12-05 01:10:45.598 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCE12:1:0x0702]: initializing channel: from_cache: True
2022-12-05 01:10:45.598 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCE12:1:0x0702]: initializing cached channel attributes: ['demand_formatting', 'divisor', 'metering_device_type', 'multiplier', 'summation_formatting', 'unit_of_measure']
2022-12-05 01:10:45.599 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCE12:1:0x0702]: Reading attributes in chunks: ['demand_formatting', 'divisor', 'metering_device_type', 'multiplier', 'summation_formatting']
2022-12-05 01:10:45.599 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCE12:1:0x0702]: Reading attributes in chunks: ['unit_of_measure']
2022-12-05 01:10:45.599 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCE12:1:0x0702]: initializing uncached channel attributes: ['instantaneous_demand', 'current_summ_delivered', 'status'] - from cache[True]
2022-12-05 01:10:45.599 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCE12:1:0x0702]: Reading attributes in chunks: ['instantaneous_demand', 'current_summ_delivered', 'status']
2022-12-05 01:10:45.599 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCE12:1:0x0702]: Performing channel specific initialization: ['instantaneous_demand', 'current_summ_delivered', 'status']
2022-12-05 01:10:45.599 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCE12:1:0x0702]: finished channel initialization
```

2. the device read the attribute's values
```
2022-12-05 01:10:45.735 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCE12:1:0x0702]: initializing channel: from_cache: False
2022-12-05 01:10:45.735 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCE12:1:0x0702]: initializing cached channel attributes: ['demand_formatting', 'divisor', 'metering_device_type', 'multiplier', 'summation_formatting', 'unit_of_measure']
2022-12-05 01:10:45.735 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCE12:1:0x0702]: Reading attributes in chunks: ['demand_formatting', 'divisor', 'metering_device_type', 'multiplier', 'summation_formatting']
2022-12-05 01:10:45.735 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCE12:1:0x0702]: Reading attributes in chunks: ['unit_of_measure']
2022-12-05 01:10:45.735 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCE12:1:0x0702]: initializing uncached channel attributes: ['instantaneous_demand', 'current_summ_delivered', 'status'] - from cache[False]
2022-12-05 01:10:45.735 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCE12:1:0x0702]: Reading attributes in chunks: ['instantaneous_demand', 'current_summ_delivered', 'status']
2022-12-05 01:10:45.735 DEBUG (MainThread) [zigpy.zcl] [0xCE12:1:0x0702] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), tsn=50, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-12-05 01:10:45.736 DEBUG (MainThread) [zigpy.zcl] [0xCE12:1:0x0702] Sending request: Read_Attributes(attribute_ids=[1024, 0, 512])
2022-12-05 01:10:45.736 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xCE12), dst_ep=1, source_route=None, extended_timeout=False, tsn=50, profile_id=260, cluster_id=1794, data=Serialized[b'\x002\x00\x00\x04\x00\x00\x00\x02'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
```

3. the device report the values for the uncached values:
```
2022-12-05 01:10:50.874 DEBUG (MainThread) [zigpy.zcl] [0xCE12:1:0x0702] Received ZCL frame: b'\x182\x01\x00\x04\x86\x00\x00\x00%\x00\x00\x00\x00\x00\x00\x00\x02\x00\x18\x00'
2022-12-05 01:10:50.875 DEBUG (MainThread) [zigpy.zcl] [0xCE12:1:0x0702] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=50, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-05 01:10:50.875 DEBUG (MainThread) [zigpy.zcl] [0xCE12:1:0x0702] Decoded ZCL frame: Metering:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0400, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>), ReadAttributeRecord(attrid=0x0000, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint48_t, value=0)), ReadAttributeRecord(attrid=0x0200, status=<Status.SUCCESS: 0>, value=TypeValue(type=bitmap8, value=<bitmap8.0: 0>))])
2022-12-05 01:10:50.876 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x9055), DstEndpoint=11, DstPanId=0x0000, SrcEndpoint=1, ClusterId=768, TSN=52, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: [... cut ...]
2022-12-05 01:10:50.877 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCE12:1:0x0702]: Performing channel specific initialization: ['instantaneous_demand', 'current_summ_delivered', 'status']
2022-12-05 01:10:50.877 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xCE12:1:0x0702]: finished channel initialization
```

Here, the `instantaneous_demand` report as `UNSUPPORTED_ATTRIBUTE`. The `current_summ_delivered` and `status` report 0 as value.

Nothing more. No errors, no extrange reports. It seems that device don't report its values, even with the Tuya spell. If the problem is not the spell cast I don't know what else can be.

CharlyNaseweis, 2022-12-06:
Hi there, 

 I have the same issue: no power reading. I have the power cable to "L-in" and "N-in" and a normal light bulb (old school, no LED ;-)) connected. 
 HA shows history of the power consumption, but no graph...
[image]

Also I importet the quirk file, restarted HA, new configuration, remove, restart, new pairing... nothing did help.

Any idea is welcome, 
thx, 
c

alepee, 2022-12-24:
Hi same here, I got no information regarding Power Meter. Plus, some device became unresponsive when switching them on/off (`MAC_NO_ACK:233`), even if they are online and reporting their status to HA.

I can help on investigate the topic.

**Logfile** from ZHA debug scoped to device IEEE
[TS0001-0x154B.log](https://github.com/zigpy/zha-device-handlers/files/10298376/TS0001-0x154B.log)

<details><summary>Device diagnostics</summary>
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.12.8",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Paris",
    "os_name": "Linux",
    "os_version": "5.15.76-v8",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.4",
    "docker_version": "20.10.19",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.5",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.89",
      "zigpy-deconz==0.19.2",
      "zigpy==0.52.3",
      "zigpy-xbee==0.16.2",
      "zigpy-zigate==0.10.3",
      "zigpy-znp==0.9.2"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "1A86",
        "pid": "55D4",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus v2"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
[... 377 lines cut ...]
            "unsupported_attributes": {}
          },
          "0xe001": {
            "endpoint_attribute": "tuya_external_switch_type",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
</details>

khorchanov, 2023-02-09:
Hi folks,
Same issue here, device paired and works good except for the metering part: 
[image]

<details>
  <summary>Signature</summary>

```json
{
    "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
    "endpoints": {
        "1": {
            "profile_id": 260,
            "device_type": "0x0100",
            "in_clusters": [
                "0x0000",
                "0x0003",
                "0x0004",
                "0x0005",
                "0x0006",
                "0x0702",
                "0x0b04",
                "0xe000",
                "0xe001"
            ],
            "out_clusters": [
                "0x000a",
                "0x0019"
            ]
        }
    },
    "manufacturer": "_TZ3000_46t1rvdu",
    "model": "TS0001",
    "class": "zhaquirks.tuya.ts000x.Switch_1G_Metering"
}
```
</details>

<details>
  <summary>Diag</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.2.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Paris",
    "os_name": "Linux",
    "os_version": "5.15.84-v8",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "keyatome": {
      "version": "7.4.1",
      "requirements": [
        "pykeyatome==2.1.1"
      ]
    },
    "alarmo": {
      "version": "v1.9.7",
[... 649 lines cut ...]
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```
</details>

@alepee I think your reconfigure issue is because of bad signal

MattWestb, 2023-02-13:
> It hsould have power metering There is a 1 Ohm resistor between Nin and Nout
> 
> [image]

The 1 Ohm is needed for getting power working the you taking it away the neutral circuit is broken.

But if you is looking is the power meting circuits not mounted so the hardware support is not on you device PCB = can and have never working.

jasimancas, 2023-03-07:
Hi, I just installed this same device and I see that you have the same problems as me with the sensors, I will try to sniff the traffic of the device to see what appears.

My sensors all give me 0.

screem20056, 2023-06-21:
Same issue here the device is connected it worked for a while as on_off for the other entities i just disabled them but after an hour or so the device stopped working from HA switch it still receives updates meaning if any one turned it on or off manually HA can see that but controlling it from HA is not possible 
thanks

erkr, 2023-08-01:
Same issue, I think this device should be excluded from the current quirk that suggest it supports energie metering

javicalle, 2023-08-01:
> setting the backlight mode doesn't work either, but don't know how to remove it

The easy way would be to disable the entity.

This comes from the `TuyaZBOnOffAttributeCluster` cluster:
https://github.com/zigpy/zha-device-handlers/blob/e87f1172496a8482ed0d68570cf41c341b0d820f/zhaquirks/tuya/__init__.py#L989-L996

Its hard to match all the properties from every device, so one approach is implement always most of them despite that for some/most of the devices would not work. 
Is more user friendly to have to disable these not working entities that to have to ask here if someone can implement their specific device case to add some functionality (like `power_on_state` or `child_lock`). Trust me that maintainers can't implement device one by one to make available only the working properties.

erkr, 2023-08-01:
@javicalle Thanks for explaining! No problem to disable the backlight option. I hope my quirk can be used as a basis to exclude the metering part for this device. Best Eric

rossco555, 2023-09-19:
I have the same issue with the same device.

closed 2024-03-24 by github-actions[bot]