zigpy/zha-device-handlers #1969: [Device Support Request] Legrand Adorne WNAL63W1 Wireless Dimmer

brantmacga, 2022-11-28:
**Is your feature request related to a problem? Please describe.**
I am unable to use this wireless dimmer in ZHA. I can bind the dimming function only, but it is reversed in that when I press the lower button, the brightness increases, and vice-versa. I cannot get the on/off function to work. 

**Describe the solution you'd like**
Would like to be able to bind this wireless dimmer to a zigbee lamp. It has a single on/off push button, and raise/lower dimmer buttons. It is battery powered.  Here is a link to the product. https://www.legrand.us/wiring-devices/designer-switches-and-outlets/adorne-wireless-smart-dimmer-with-netatmo-white/p/wnal63w1

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=1, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4129, maximum_buffer_size=89, maximum_incoming_transfer_size=63, server_mask=10752, maximum_outgoing_transfer_size=63, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000f",
        "0x0020",
        "0xfc01"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0006",
        "0x0008",
        "0x0019",
        "0xfc01"
      ]
    }
  },
  "manufacturer": " Legrand",
  "model": " Remote dimmer switch",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "5.15.61-v8",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
[... 314 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
I have a ZHA quirk that i installed on my system, but it did not work for this product. I believe you can see it in the diagnostics.

stevenwfoley, 2022-11-30:
My understanding is that once you bind the devices directly together they can communicate independently of HA/ZHA, so its up to the firmware on the devices to communicate properly. In this case where things are not working as you expect, you might actually need to use HA/ZHA to interpret and facilitate proper communication between the devices. What lamp device are you using?

FYI, according to the diagnostics info, it does not look like you are using a quirk.

#1934 was recently merged to provide specific support for the Radiant version of this Legrand switch, but the signatures are identical, so I would expect this to apply to your Adorne switch as well. I'm not sure if this code has made it into a release yet though. I have some additional code I added to my quirk to support all the button triggers natively in HA Automations. I've attached the file that you'll need to rename and remove the .txt extension.
[legrand_wireless_dimmer.py.txt](https://github.com/zigpy/zha-device-handlers/files/10127047/legrand_wireless_dimmer.py.txt)

brantmacga, 2022-11-30:
Steven,

I appreciate the help. One thing I realized afterwards is that I didn't remove & re-pair my dimmer after adding the quirk. 

I did try to use the legrand_wireless_dimmer.py quirk, and it still doesn't work properly. The adorne wireless dimmers are different from the Radiant dimmers in that they have a single on/off push-button and not an up & down push button. I'm not sure what else internally is different; I've got some Radiant switches & wireless switches but I can't remember if Legrand sent me any wireless Radiant dimmers. 

I'm trying to bind it to a hue lamp. All I'm really trying to accomplish is being able to use the Bticino toggle switch relays along with some wireless dimmers. If the netatmo integration is ever fixed, I could install the hue lamps and dimmers directly to the Legrand gateway, and setup an automation in HA for the Bticino relays using Zigbee2MQTT. I've already tested the toggle switch automation through Z2M and it works. I just need the dimmers there now. It would be nice though to also have the Bticino relay in ZHA so I could bind it directly to the hue lamps.

brantmacga, 2022-11-30:
Here is a pic of what I see when trying to reconfigure the device. 

[image]

brantmacga, 2022-11-30:
Here is my diagnostic info with the quirk

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "5.15.61-v8",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.4",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.87",
      "zigpy-deconz==0.19.1",
      "zigpy==0.51.6",
      "zigpy-xbee==0.16.2",
      "zigpy-zigate==0.10.3",
      "zigpy-znp==0.9.1"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "1A86",
        "pid": "55D4",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus v2"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
[... 251 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x0008": {
            "endpoint_attribute": "level",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xfc01": {
            "endpoint_attribute": "legrand_cluster",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

stevenwfoley, 2022-11-30:
That looks like the zigbee interview/joining issue I experienced. I had very sporadic results joining the device to ZHA with the firmware shipped on the wireless dimmer. I had to purchase the Legrand Smart Gateway to get the firmware updated to the latest, which according to the Legrand tech support fixed a lot of zigbee connectivity issues with 3rd party coordinators. The firmware I'm running is 31 as shown in the Legrand Control app; ZHA displays it as 0x001f4203.

I could sometimes get better results by continuously pressing the config button during the joining process.

brantmacga, 2022-12-01:
> That looks like the zigbee interview/joining issue I experienced. I had very sporadic results joining the device to ZHA with the firmware shipped on the wireless dimmer. I had to purchase the [...]

I just paired it to the Legrand gateway, its version 28. I'll try to force it to 31 and start over. thanks again

brantmacga, 2022-12-01:
also, not sure if this is significant info, but when I'm pressing buttons on the adorne dimmer, I see the following in the logbook:

If I press dimmer paddle "lower", I see...

> Legrand Remote dimmer switch Stop event was fired with parameters: {'options_mask': None, 'options_override': None}

Followed by...

> Legrand Remote dimmer switch Move event was fired with parameters: {'move_mode': <MoveMode.Up: 0>, 'rate': 100, 'options_mask': None, 'options_override': None}

When I press dimmer paddle "raise", I see...

> Legrand Remote dimmer switch Stop event was fired with parameters: {'options_mask': None, 'options_override': None}

followed by...

> Legrand Remote dimmer switch Move event was fired with parameters: {'move_mode': <MoveMode.Down: 1>, 'rate': 100, 'options_mask': None, 'options_override': None}

brantmacga, 2022-12-01:
When I press the single on/off button, there is no event data. I pressed it multiple times, got nothing. 

Press "dim lower" and I see.....

> event_type: zha_event
data:
  device_ieee: 64:62:66:ff:fe:61:a0:4b
  unique_id: 64:62:66:ff:fe:61:a0:4b:1:0x0008
  device_id: 59b0239b4e83b7f2c65e002adf970589
  endpoint_id: 1
  cluster_id: 8
  command: move
  args:
    - 0
    - 100
  params:
    move_mode: 0
    rate: 100
    options_mask: null
    options_override: null
origin: LOCAL
time_fired: "2022-12-01T03:07:02.962111+00:00"
context:
  id: 01GK5TDRXJHEAAT24ZW1ZQKNZQ
  parent_id: null
  user_id: null

and then...

> event_type: zha_event
data:
  device_ieee: 64:62:66:ff:fe:61:a0:4b
  unique_id: 64:62:66:ff:fe:61:a0:4b:1:0x0008
  device_id: 59b0239b4e83b7f2c65e002adf970589
  endpoint_id: 1
  cluster_id: 8
  command: stop
  args: []
  params:
    options_mask: null
    options_override: null
origin: LOCAL
time_fired: "2022-12-01T03:07:03.038186+00:00"
context:
  id: 01GK5TDRZYDYFAZCKNE4H601C9
  parent_id: null
  user_id: null

When I press "dim raise" I see.....

> event_type: zha_event
data:
  device_ieee: 64:62:66:ff:fe:61:a0:4b
  unique_id: 64:62:66:ff:fe:61:a0:4b:1:0x0008
  device_id: 59b0239b4e83b7f2c65e002adf970589
  endpoint_id: 1
  cluster_id: 8
  command: move
  args:
    - 1
    - 100
  params:
    move_mode: 1
    rate: 100
    options_mask: null
    options_override: null
origin: LOCAL
time_fired: "2022-12-01T03:07:47.052562+00:00"
context:
  id: 01GK5TF3ZCWA868DCV8FTKG3WX
  parent_id: null
  user_id: null

followed by.....

> event_type: zha_event
data:
  device_ieee: 64:62:66:ff:fe:61:a0:4b
  unique_id: 64:62:66:ff:fe:61:a0:4b:1:0x0008
  device_id: 59b0239b4e83b7f2c65e002adf970589
  endpoint_id: 1
  cluster_id: 8
  command: stop
  args: []
  params:
    options_mask: null
    options_override: null
origin: LOCAL
time_fired: "2022-12-01T03:07:47.064904+00:00"
context:
  id: 01GK5TF3ZR34JT8QRXWCDTBKDH
  parent_id: null
  user_id: null

stevenwfoley, 2022-12-01:
This is probably because your OnOff cluster is not bound. Your image shows a yellow X on both the OnOff and LevelControl clusters, so I would have not expected events on either cluster, but perhaps the LevelControl bound during the initial pairing. I'm by no means an expert, but I think you'll need to get both the clusters bound for this to work. If re-pairing is not successful you can also try manual binding.

-Navigate to the device, click the 3 dots menu next to the Reconfigure button, and click Manage zigbee device.
-Select the Bindings tab.
-Under the Bindable Devices drop-down, select your zigbee coordinator device.
-Click the Bind button.
Since the dimmer is a battery powered SED (sleepy end device) you'll have to press the config button and/or the other buttons to wake it up to receive the command. In my experience, the bind button just keeps spinning even though it succeeds. And I retry it a few times, you can close the dialog to clear the spinning bind button for a retry.

brantmacga, 2022-12-01:
I was unsuccessful at binding. I really appreciate all the help. I'm not sure what else I can do at this point.

brantmacga, 2022-12-23:
I'm attaching my debug log here. Hoping to get this one sorted out also. On/Off doesn't work, and the DIM_UP / DIM_DOWN functions are reversed on the switch. When I press Dim Down, it reports a DIM_UP and vice versa. Thanks. 
[home-assistant_zha_2022-12-23T02-06-00.909Z.log](https://github.com/zigpy/zha-device-handlers/files/10291521/home-assistant_zha_2022-12-23T02-06-00.909Z.log)

brantmacga, 2023-01-02:
I was able to create an external converter in Zigbee2MQTT to support this dimmer. 

Is there any information I could use from that to create a working quirk in ZHA? 

When I press the on/off button on the dimmer, I see this in the Z2M log:
MQTT publish: topic 'zigbee2mqtt/Patio Light 1', payload '{"brightness":120,"color_mode":"xy","linkquality":148,"power_on_behavior":null,"state":"OFF","update":{"installed_version":"1.93.11","state":"available"},"update_available":null}'

closed 2023-07-08 by github-actions[bot]

sharwell, 2024-05-02:
@brantmacga I have three of these dimmers in the house currently. I've found the reliability of the switches using the Legrand hub to be _terrible_, and am trying to test out using the Home Assistant SkyConnect device to directly communicate with them instead. I'm new to Home Assistant and to ZHA, but not new to programming. Please feel free to reach out if you are interested in working on this with me.