zigpy/zha-device-handlers #1970: [Device Support Request] Bticino 3577C Toggle Switch Relay

brantmacga, 2022-11-28:
**Is your feature request related to a problem? Please describe.**
I can pair this toggle switch device, but am unable to bind to a lamp or use in automations. It works correctly in Zigbee2MQTT.

**Describe the solution you'd like**
I would like to be able to bind this to a zigbee lamp to toggle on/off, or use in automations. Link to product - https://catalogue.bticino.com/BTI-3577C-EN

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=1, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4129, maximum_buffer_size=89, maximum_incoming_transfer_size=63, server_mask=10752, maximum_outgoing_transfer_size=63, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000f",
        "0x0020",
        "0xfc01"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0006",
        "0x0019",
        "0xfc01"
      ]
    }
  },
  "manufacturer": " Legrand",
  "model": " Remote toggle switch",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "5.15.61-v8",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
[... 303 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-12-01:
Not sure if that quirk could fix the related issue because other Legrand devices are quite standar zigbee.

If you want to try, there is my proposal:
<details open>
</summary>RemoteSwitch</summary>

```python
class RemoteSwitch(CustomDevice):
    """Remote switch."""

    signature = {
        MODELS_INFO: [(f" {LEGRAND}", " Remote toggle switch")],
        ENDPOINTS: {
            1: {
                # "profile_id": 260,
                # "device_type": "0x0103",
                # "in_clusters": ["0x0000","0x0001","0x0003","0x000f","0x0020","0xfc01"],
                # "out_clusters": ["0x0000","0x0003","0x0006","0x0019","0xfc01"]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfigurationCluster.cluster_id,
                    Identify.cluster_id,
                    BinaryInput.cluster_id,
                    PollControl.cluster_id,
                    LegrandCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    OnOff.cluster_id,
                    Ota.cluster_id,
                    LegrandCluster.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    LegrandPowerConfigurationCluster,
                    Identify.cluster_id,
                    BinaryInput.cluster_id,
                    PollControl.cluster_id,
                    LegrandCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    OnOff.cluster_id,
                    Ota.cluster_id,
                    LegrandCluster,
                ],
            }
        }
    }
```

</details>

You can enable the local quirks and copy the current `legrand/dimmer.py` file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/legrand/dimmer.py

Add the proposed quirk at the end of the file.
Save changes, restart HA and repair the device.

If not working, enable the debud logs and attach the relavant info.

brantmacga, 2022-12-02:
thanks.... Still not working. The 'On/off' and 'PollControl' are not binding.

javicalle, 2022-12-02:
I can't say anything about the binding part, but is the switch status updating in HA?
Maybe an `EventableCluster` can be implemented...

brantmacga, 2022-12-03:
it doesn't appear to be using the quirk. I say this because when I look at my remote dimmer, if I check "Zigbee info" I see on the last line that its using a quirk. I do not see the same information on the remote toggle switch, and nothing about its behavior has changed.

javicalle, 2022-12-04:
I have added an `OnOffEventsCluster` to the proposed quirk. It is used in the replacement part.

As a remote switch this device has not a switch entity in HA, isn't it?

The new quirk may trigger `zha_event` with the button pressed.

brantmacga, 2022-12-04:
> I have added an `OnOffEventsCluster` to the proposed quirk. It is used in the replacement part.
> 
> As a remote switch this device has not a switch entity in HA, isn't it?
> 
> The new quirk may trigger `zha_event` with the button pressed.

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "5.15.61-v8",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.4",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.87",
      "zigpy-deconz==0.19.1",
      "zigpy==0.51.6",
      "zigpy-xbee==0.16.2",
      "zigpy-zigate==0.10.3",
      "zigpy-znp==0.9.1"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "1A86",
        "pid": "55D4",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus v2"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
[... 235 lines cut ...]
          },
          "0xfc01": {
            "endpoint_attribute": "manufacturer_specific",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

It didn't work that time either. I don't think the switch is seeing the quirk. 

My process was to delete the toggle switch, add new lines to the .py file, restart HA, then pair the toggle switch. When I click on "Zigbee Info" on the toggle switch page, I'm not seeing the quirk like I do on the wireless dimmer; the wireless dimmer doesn't work either though. 

Thanks.

javicalle, 2022-12-04:
The device is not loading the quirk.
Double check that there is no a `__pycache__` folder in your local quirk folder.
Also look for any exception or errors in the HA startup logs. Maybe there is a errata in the local quirk code.

brantmacga, 2022-12-04:
> The device is not loading the quirk.
> Double check that there is no a `__pycache__` folder in your local quirk folder.
> Also look for any exception or errors in the HA startup logs. Maybe there is a errata in the local quirk code.

not seeing the __pycache__ folder, and I just read through all of the logs and did not see any errors related to that switch. Certainly possible I'm overlooking something. 

I noticed the quirk calls it an ON_OFF_LIGHT_SWITCH; the device model in the diagnostics refers to it only as a Toggle switch. Would that have anything to do with it?

brantmacga, 2022-12-04:
ok I deleted the device, paired again, and I saw this in the logs while adding to ZHA....

> [0x0A92:1:0x0003] Decoded ZCL frame: Identify:identify_query()
[0x0A92:1:0x0003] Received command 0x01 (TSN 1): identify_query()
[0x0A92:1:0x0003] No explicit handler for cluster command 0x01: identify_query()
[0x3208:1:0x0006]: failed to get attributes '['on_off']' on 'on_off' cluster: Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>
[0x3208:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>')]

brantmacga, 2022-12-10:
Any idea what these errors could mean ?

javicalle, 2022-12-22:
The errors seem to be some communication error. Not sure if related.
Can you enable the debug logs and attach the logs from pairing the device.

brantmacga, 2022-12-22:
[home-assistant_zha_2022-12-22T21-50-58.603Z.log](https://github.com/zigpy/zha-device-handlers/files/10290627/home-assistant_zha_2022-12-22T21-50-58.603Z.log)

Log attached. Thanks!

javicalle, 2022-12-23:
That part is very significative:
```
2022-12-22 16:47:07.573 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry ConBee II, s/n: DE2483634 - dresden elektronik ingenieurtechnik GmbH for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 372, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 100, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/zha_quirks/legrand_wireless_dimmer.py", line 170, in <module>
    class OnOffEventsCluster(EventableCluster, OnOff):
NameError: name 'EventableCluster' is not defined
```

Edit the quirk and add the `EventableCluster` import statement replacing this line:
```python
from zhaquirks import EventableCluster, PowerConfigurationCluster
```

Save changes, delete any `__pycache__` folder from your local quirk folder, restart HA and pair the device again.

brantmacga, 2022-12-23:
I'm sorry, I don't quite fully understand where I should be placing that text within the quirk .py file

brantmacga, 2022-12-23:
I added the text line above the quirk for the toggle switch. Log is attached. thank you. 
[home-assistant_zha_2022-12-23T00-26-40.166Z.log](https://github.com/zigpy/zha-device-handlers/files/10291123/home-assistant_zha_2022-12-23T00-26-40.166Z.log)

javicalle, 2022-12-23:
> I'm sorry, I don't quite fully understand where I should be placing that text within the quirk .py file

At the top of file, where are all the import statements you need to replace this line with the proposed one:
* https://github.com/zigpy/zha-device-handlers/blob/c2b2893d027fdcd3a225996fdf382d2ac72ff799/zhaquirks/legrand/dimmer.py#L20

javicalle, 2022-12-23:
I don't see any error in the logs. I don't see any command or `zha_event`

Have you validated in the Device view if the device is loading the quirk?
Have you tested if the device is generating events from the device actions?

brantmacga, 2022-12-23:
Ok it is now loading the quirk...
Quirk: legrand_wireless_dimmer.RemoteSwitch

And it is firing events in the log. 

What I am missing now is the "toggle" action when using for automations. I have "Binary Input Turned On" or "Binary Input Turned Off". 

When I pair it through Z2M, I have a "toggle" option that allows me to use it like a normal switch with anything. Thank you again!

brantmacga, 2022-12-23:
I will also add that in the event log, I see "Legrand Remote toggle switch Toggle event was fired". 

I tried binding it to a Hue lamp to toggle on/off and was unsuccessful.

javicalle, 2022-12-23:
The quirk has nothing to do with the direct binding between devices.

In ZHA you can use triggers in automations but will not create a switch or input.
I have updated the code from my comment to add the triggers:
```python
    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: DOUBLE_PRESS},
    }
```

But these are just an example. We need to know the events and attributes for that device. So we need that you test the posible events (click, double-click, long-press, release, whatever...) and attach the events to see what can be done and how.

MattWestb, 2022-12-23:
@brantmacga Do you getting command 0x02 (toggle) in the log then pressing the button ?
If yes make one device automation with it as @javicalle was explaining in the example above.
If the device is not sending toggle you cant do device automatons for it then the device is not sending it.

Z2M is making may things that is emulated function like brightness for one dimmer remote that is only emulated and is not one real thing and is being broken if using more dimmers or other device is changing the real one.

brantmacga, 2022-12-23:
@MattWestb in the Logbook, I see 'Legrand Remote toggle switch Toggle even was fired'. If I use the same device paired with Z2M, I do have a "Toggle" option in automations.

brantmacga, 2022-12-23:
@javicalle I will try the updated code when get home later today. thank you!

MattWestb, 2022-12-23:
one example of DA with toggle:
https://github.com/zigpy/zha-device-handlers/blob/c2b2893d027fdcd3a225996fdf382d2ac72ff799/zhaquirks/tuya/ts004f.py#L134

brantmacga, 2022-12-23:
I got this error with the new code:

Logger: homeassistant.config_entries
Source: zha_quirks/legrand_wireless_dimmer.py:228 
First occurred: 11:55:33 AM (1 occurrences) 
Last logged: 11:55:33 AM

Error setting up entry ConBee II, s/n: DE2483634 - dresden elektronik ingenieurtechnik GmbH for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 372, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 100, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/zha_quirks/legrand_wireless_dimmer.py", line 173, in <module>
    class RemoteSwitch(CustomDevice):
  File "/config/zha_quirks/legrand_wireless_dimmer.py", line 228, in RemoteSwitch
    (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
NameError: name 'BUTTON_1' is not defined

brantmacga, 2022-12-23:
@javicalle I haven't been able to get it to work correctly yet with the code. It did give me an error about attributes not being define (BUTTON_1) and (DOUBLE_PRESS); I added those to the list below {from zhaquirks.const import} and got ZHA to load correctly, but now it won't load the switch attributes when I try to pair it; and it actually won't show that its paired within ZHA. 

I also tried @MattWestb 's example in its place, added the definition for (COMMAND_TOGGLE) and (BUTTON_1), and now the switch will pair, and I can try to create an automation with it, but it doesn't work. When I check the log, no ZHA Events are being received. 

Still thinking it through.

brantmacga, 2022-12-23:
@javicalle I think I'm going to leave it alone for now.... It is binding with zigbee lamps and switching them correctly. If I can get my wireless dimmer to work reported here https://github.com/zigpy/zha-device-handlers/issues/1969 then I really don't have a need for the automation toggle trigger. I could just bind everything together in ZHA and be done with it. Thanks for all the help.

javicalle, 2022-12-23:
Just one question, is still needed the quirk?
I mean, is you just need the binding, can it be done without the quirk?

brantmacga, 2022-12-23:
I was unable to bind without using the quirk, so yes, definitely needed

Thanks

Buci007, 2023-01-15:
Hello! Thank you for your information. I tried the suggested quirk. Added the code at the and of the original legrand/dimmer.py and updated the line 20 "from zhaquirks import EventableCluster, PowerConfigurationCluster". The file put in the zha_custom_quirks and restart HA.  The coordinator crashed. If I tried with the original dimmer.py file the system restart without any failure. My switch device_type: "0x104" and model: Remote switch".  I changed these 2 part in the code, but the result was same. Crash again. Ps.: I use already custom quirk for termostat valve: ts0601_trv_moes.py.
[home-assistant_2023-01-15T19-35-31.842Z.log](https://github.com/zigpy/zha-device-handlers/files/10421039/home-assistant_2023-01-15T19-35-31.842Z.log)

javicalle, 2023-01-15:
You need to import all the `device_automation_triggers` constants:
```python
from zhaquirks.const import (
    BUTTON_1,
    COMMAND,
    DEVICE_TYPE,
    DOUBLE_PRESS,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LONG_PRESS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SHORT_PRESS,
)
```

Buci007, 2023-01-16:
Good morning! Thank you. I updated the quirk. Deleted all the not needed devices to be able better overview it and updated the remote switch part with my device_type: "0x104" and model: " Remote switch"
The quirk is loading. The sytem started.
[home-assistant_2023-01-16T06-43-03.867Z.log](https://github.com/zigpy/zha-device-handlers/files/10423197/home-assistant_2023-01-16T06-43-03.867Z.log)
I added the switch new, but based on his signature does not use the quirk.
[Signature.txt](https://github.com/zigpy/zha-device-handlers/files/10423203/Signature.txt)
[dimmer.txt](https://github.com/zigpy/zha-device-handlers/files/10423245/dimmer.txt)
Thank you in advance your response.

javicalle, 2023-01-16:
Your device signature don't match the quirk.
Let's continue in #2098

closed 2023-07-22 by github-actions[bot]