zigpy/zha-device-handlers #1973: [Device Support Request] TUYA DIN ENERGY METERING

seblang, 2022-11-29:
**Is your feature request related to a problem? Please describe.**
I can pair this tuya consumption energy metering  device but i have no sensor

**Describe the solution you'd like**
i want to be able to see the consumption metering

<details>

<summary>Device signature</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` {
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_cjbofhxw",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` {
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Paris",
    "os_name": "Linux",
    "os_version": "5.15.74",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
[... 303 lines cut ...]
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

seblang, 2022-11-30:
product link

https://fr.aliexpress.com/item/1005004845740618.html?spm=a2g0o.productlist.0.0.7f0c1c051vU9BM&ad_pvid=202211300039372484257705916200029604571_1&s=p

javicalle, 2022-11-30:
There are already 2 implementations for Tuya DIN metering devices.
I would suggest you to try with the `HikingPowerMeter` one.

There is a good guide to create your local quirk:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Just copy inside your local quirk folder the current `ts0601_din_power.py` file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_din_power.py

Edit the content adding your device signature at the `MODELS_INFO` like this:
```python
class HikingPowerMeter(TuyaSwitch):
    """Hiking Power Meter Device - DDS238-2."""

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4098
        #                       maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264
        #                       maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
        # device_version=1
        # input_clusters=[0x0000, 0x0004, 0x0005, 0xef00]
        # output_clusters=[0x000a, 0x0019]
        MODELS_INFO: [
            ("_TZE200_bkkmqmyo", "TS0601"),
            ("_TZE204_cjbofhxw", "TS0601"),
        ],
        ENDPOINTS: {
```

Save changes, restart HA and repair the device.

seblang, 2022-11-30:
Hi Thank you 

Done but result is not perfect :-)

[image]

If I ask a reconfiguration I have this :

[image]

My product have no switch :

[image]

available info according vendor

[image]

Can you help me to update mu quirks please

javicalle, 2022-11-30:
We can remove the switch, this part is the easy one.
For all the other entities we need to collect the info that report the device and map it against the measured values.

To do that, will be necessary to activate the logs for the integration and collect the relevant information. Instructions here:
* https://www.home-assistant.io/integrations/zha/#debug-logging

You will get multiple reports from your device, NMK: `0x7667`

Every report will be a meassurement and you have to analize it to map against the correct one. In example, the current voltage will be a value around 220/240

seblang, 2022-11-30:
Hi debug activated, but I am not sur how to understand the log file:

`2022-11-30 19:57:34.371 WARNING (MainThread) [zigpy.zcl] [0x7667:1:0xef00] Unknown cluster command 37 b'\t\x00'
2022-11-30 19:57:34.374 DEBUG (MainThread) [zigpy.zcl] [0x7667:1:0xef00] Received command 0x25 (TSN 64): b'\t\x00'
2022-11-30 19:57:34.374 DEBUG (MainThread) [zigpy.zcl] [0x7667:1:0xef00] No explicit handler for cluster command 0x25: b'\t\x00'
2022-11-30 19:57:34.519 DEBUG (MainThread) [bellows.zigbee.application] Received incomingRouteRecordHandler frame with [0x7667, e0:79:8d:ff:fe:de:3b:6a, 99, -69, []]
2022-11-30 19:57:34.519 DEBUG (MainThread) [bellows.zigbee.application] Processing route record request: (0x7667, e0:79:8d:ff:fe:de:3b:6a, 99, -69, [])
2022-11-30 19:57:34.566 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=181), 99, -69, 0x7667, 255, 255, b'\t@%\t\x00']
2022-11-30 19:57:34.567 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x7667), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=181, profile_id=260, cluster_id=61184, data=Serialized[b'\t@%\t\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=99, rssi=-69)
2022-11-30 19:57:34.567 DEBUG (MainThread) [zigpy.zcl] [0x7667:1:0xef00] Received ZCL frame: b'\t@%\t\x00'
2022-11-30 19:57:34.568 DEBUG (MainThread) [zigpy.zcl] [0x7667:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=64, command_id=37, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-11-30 19:57:34.569 WARNING (MainThread) [zigpy.zcl] [0x7667:1:0xef00] Unknown cluster command 37 b'\t\x00'
2022-11-30 19:57:34.572 DEBUG (MainThread) [zigpy.zcl] [0x7667:1:0xef00] Received command 0x25 (TSN 64): b'\t\x00'
2022-11-30 19:57:34.573 DEBUG (MainThread) [zigpy.zcl] [0x7667:1:0xef00] No explicit handler for cluster command 0x25: b'\t\x00'
2022-11-30 19:57:38.100 DEBUG (MainThread) [bellows.zigbee.application] Received incomingRouteRecordHandler frame with [0x7667, e0:79:8d:ff:fe:de:3b:6a, 94, -70, []]
2022-11-30 19:57:38.100 DEBUG (MainThread) [bellows.zigbee.application] Processing route record request: (0x7667, e0:79:8d:ff:fe:de:3b:6a, 94, -70, [])
2022-11-30 19:57:38.149 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=182), 94, -70, 0x7667, 255, 255, b'\tA%\x0b\x00']
2022-11-30 19:57:38.149 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x7667), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=182, profile_id=260, cluster_id=61184, data=Serialized[b'\tA%\x0b\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=94, rssi=-70)
2022-11-30 19:57:38.150 DEBUG (MainThread) [zigpy.zcl] [0x7667:1:0xef00] Received ZCL frame: b'\tA%\x0b\x00'
2022-11-30 19:57:38.150 DEBUG (MainThread) [zigpy.zcl] [0x7667:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=65, command_id=37, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-11-30 19:57:38.151 WARNING (MainThread) [zigpy.zcl] [0x7667:1:0xef00] Unknown cluster command 37 b'\x0b\x00'
2022-11-30 19:57:38.154 DEBUG (MainThread) [zigpy.zcl] [0x7667:1:0xef00] Received command 0x25 (TSN 65): b'\x0b\x00'
2022-11-30 19:57:38.155 DEBUG (MainThread) [zigpy.zcl] [0x7667:1:0xef00] No explicit handler for cluster command 0x25: b'\x0b\x00'
2022-11-30 19:57:38.296 DEBUG (MainThread) [bellows.zigbee.application] Received incomingRouteRecordHandler frame with [0x7667, e0:79:8d:ff:fe:de:3b:6a, 99, -69, []]
2022-11-30 19:57:38.297 DEBUG (MainThread) [bellows.zigbee.application] Processing route record request: (0x7667, e0:79:8d:ff:fe:de:3b:6a, 99, -69, [])
2022-11-30 19:57:38.346 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=183), 99, -69, 0x7667, 255, 255, b'\tA%\x0b\x00']
2022-11-30 19:57:38.347 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x7667), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, ext [... cut ...] ved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=66, command_id=37, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-11-30 19:57:42.725 WARNING (MainThread) [zigpy.zcl] [0x7667:1:0xef00] Unknown cluster command 37 b'\r\x00'
2022-11-30 19:57:42.729 DEBUG (MainThread) [zigpy.zcl] [0x7667:1:0xef00] Received command 0x25 (TSN 66): b'\r\x00'
2022-11-30 19:57:42.730 DEBUG (MainThread) [zigpy.zcl] [0x7667:1:0xef00] No explicit handler for cluster command 0x25: b'\r\x00'`

javicalle, 2022-11-30:
The device is calling some custom command `0x25` every 4 minutes with one repetition.
I have no idea what is this command. There is nothing in ZHA and I can't find any relation in Z2M.

Is any entity updating values?
Any other logs than these ones?

seblang, 2022-11-30:
no

seblang, 2022-11-30:
I have cancel the quirk to see if I have some other info

`Journaux complets
2022-11-30 23:04:53.593 DEBUG (MainThread) [zigpy.appdb] [0x7667:1:0x0000] Attribute id: 4 value: _TZE204_cjbofhxw
2022-11-30 23:04:53.593 DEBUG (MainThread) [zigpy.appdb] [0x7667:1:0x0000] Attribute id: 5 value: TS0601
2022-11-30 23:04:58.525 DEBUG (MainThread) [zigpy.appdb] [0x7667:1:0x0000] Attribute id: 4 value: _TZE204_cjbofhxw
2022-11-30 23:04:58.525 DEBUG (MainThread) [zigpy.appdb] [0x7667:1:0x0000] Attribute id: 5 value: TS0601
2022-11-30 23:04:58.525 DEBUG (MainThread) [zigpy.appdb] [0x7667:1:0x0000] Attribute id: 1 value: 70
2022-11-30 23:04:58.525 DEBUG (MainThread) [zigpy.appdb] [0x7667:1:0x0000] Attribute id: 65506 value: 31
2022-11-30 23:04:58.525 DEBUG (MainThread) [zigpy.appdb] [0x7667:1:0x0000] Attribute id: 65508 value: 0
2022-11-30 23:04:58.525 DEBUG (MainThread) [zigpy.appdb] [0x7667:1:0x0000] Attribute id: 65503 value:
2022-11-30 23:04:58.525 DEBUG (MainThread) [zigpy.appdb] [0x7667:1:0x0000] Attribute id: 65534 value: 0
2022-11-30 23:05:06.984 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] [0x7667](_TZE204_cjbofhxw TS0601) restored as 'available', last seen: 0:01:36 ago, consider_unavailable_time: 7200 seconds
2022-11-30 23:05:07.010 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x7667](TS0601): started initialization
2022-11-30 23:05:07.010 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:ZDO](TS0601): 'async_initialize' stage succeeded
2022-11-30 23:05:07.149 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0x0000]: initializing channel: from_cache: True
2022-11-30 23:05:07.149 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0x0000]: finished channel initialization
2022-11-30 23:05:07.149 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0xef00]: initializing channel: from_cache: True
2022-11-30 23:05:07.150 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0xef00]: finished channel initialization
2022-11-30 23:05:07.150 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0x0019]: initializing channel: from_cache: True
2022-11-30 23:05:07.150 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0x0019]: finished channel initialization
2022-11-30 23:05:07.186 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0x0000]: 'async_initialize' stage succeeded
2022-11-30 23:05:07.186 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0xef00]: 'async_initialize' stage succeeded
2022-11-30 23:05:07.186 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0x0019]: 'async_initialize' stage succeeded
2022-11-30 23:05:07.226 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x7667](TS0601): power source: Mains
2022-11-30 23:05:07.226 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x7667](TS0601): completed initialization
2022-11-30 23:05:07.253 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x7667](TS0601): started initialization
2022-11-30 23:05:07.253 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:ZDO](TS0601): 'async_initialize' stage succeeded
2022-11-30 23:05:07.388 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0x0000]: initializing channel: from_cache: False
2022-11-30 23:05:07.388 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0x0000]: finished channel initialization
2022-11-30 23:05:07.388 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0xef00]: initializing channel: from_cache: False
2022-11-30 23:05:07.389 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0xef00]: finished channel initialization
2022-11-30 23:05:07.389 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0x0019]: initializing channel: from_cache: False
2022-11-30 23:05:07.389 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0x0019]: finished channel initialization
2022-11-30 23:05:07.393 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0x0000]: 'async_initialize' stage succeeded
2022-11-30 23:05:07.393 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0xef00]: 'async_initialize' stage succeeded
2022-11-30 23:05:07.393 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7667:1:0x0019]: 'async_initialize' stage succeeded
2022-11-30 23:05:07.416 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x7667](TS0601): power source: Mains
2022-11-30 23:05:07.416 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x7667](TS0601): completed initialization
2022-11-30 23:05:07.653 DEBUG (MainThread) [bellows.zigbee.application] Received incomingRouteRecordHandler frame with [0x7667, e0:79:8d:ff:fe:de:3b:6a, 99, -69, []]
2022-11-30 23:05:07.653 DEBUG (MainThread) [bellows.zigbee.application] Processing route record request: (0x7667, e0:79:8d:ff:fe:de:3b:6a, 99, -69, [])
2022-11-30 23:05:07.656 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint [... cut ...]
2022-11-30 23:05:22.674 DEBUG (MainThread) [zigpy.zcl] [0x7667:1:0xef00] Received ZCL frame: b'\t\x17%5\x00'
2022-11-30 23:05:22.675 DEBUG (MainThread) [zigpy.zcl] [0x7667:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=23, command_id=37, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)`

How can I use this information to customize my quirk
please

seblang, 2022-12-01:
How can I interpret the log?

seblang, 2022-12-01:
`2022-12-01 21:03:52.052 DEBUG (MainThread) [zigpy.appdb] [0x1c87:1:0x0000] Attribute id: 4 value: _TZE204_cjbofhxw
2022-12-01 21:03:52.052 DEBUG (MainThread) [zigpy.appdb] [0x1c87:1:0x0000] Attribute id: 5 value: TS0601
2022-12-01 21:03:57.241 DEBUG (MainThread) [zigpy.appdb] [0x1c87:1:0x0000] Attribute id: 4 value: _TZE204_cjbofhxw
2022-12-01 21:03:57.241 DEBUG (MainThread) [zigpy.appdb] [0x1c87:1:0x0000] Attribute id: 5 value: TS0601
2022-12-01 21:03:57.241 DEBUG (MainThread) [zigpy.appdb] [0x1c87:1:0x0b04] Attribute id: 1539 value: 1000
2022-12-01 21:03:57.241 DEBUG (MainThread) [zigpy.appdb] [0x1c87:1:0x0b04] Attribute id: 1538 value: 1
2022-12-01 21:03:57.241 DEBUG (MainThread) [zigpy.appdb] [0x1c87:1:0x0b04] Attribute id: 1536 value: 2
2022-12-01 21:03:57.241 DEBUG (MainThread) [zigpy.appdb] [0x1c87:1:0x0702] Attribute id: 768 value: 0
2022-12-01 21:03:57.242 DEBUG (MainThread) [zigpy.appdb] [0x1c87:1:0x0000] Attribute id: 1 value: 70
2022-12-01 21:03:57.242 DEBUG (MainThread) [zigpy.appdb] [0x1c87:1:0x0000] Attribute id: 65506 value: 31
2022-12-01 21:03:57.242 DEBUG (MainThread) [zigpy.appdb] [0x1c87:1:0x0000] Attribute id: 65508 value: 0
2022-12-01 21:03:57.242 DEBUG (MainThread) [zigpy.appdb] [0x1c87:1:0x0000] Attribute id: 65503 value: t�+��+e��+��+
2022-12-01 21:04:01.505 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] [0x1C87](_TZE204_cjbofhxw TS0601) restored as 'available', last seen: 0:01:34 ago, consider_unavailable_time: 7200 seconds
2022-12-01 21:04:01.530 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x1C87](TS0601): started initialization
2022-12-01 21:04:01.530 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:ZDO](TS0601): 'async_initialize' stage succeeded
2022-12-01 21:04:01.664 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0006]: initializing channel: from_cache: True
2022-12-01 21:04:01.664 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0006]: initializing cached channel attributes: ['start_up_on_off']
2022-12-01 21:04:01.664 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0006]: Reading attributes in chunks: ['start_up_on_off']
2022-12-01 21:04:01.664 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0006]: initializing uncached channel attributes: ['on_off'] - from cache[True]
2022-12-01 21:04:01.664 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0006]: Reading attributes in chunks: ['on_off']
2022-12-01 21:04:01.664 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0006]: finished channel initialization
2022-12-01 21:04:01.665 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0b04]: initializing channel: from_cache: True
2022-12-01 21:04:01.665 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0b04]: initializing cached channel attributes: ['ac_current_divisor', 'ac_current_multiplier', 'ac_power_divisor', 'ac_power_multiplier', 'ac_voltage_divisor', 'ac_voltage_multiplier', 'ac_frequency_divisor', 'ac_frequency_multiplier', 'measurement_type', 'power_divisor', 'power_multiplier']
2022-12-01 21:04:01.665 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0b04]: Reading attributes in chunks: ['ac_current_divisor', 'ac_current_multiplier', 'ac_power_divisor', 'ac_power_multiplier', 'ac_voltage_divisor']
2022-12-01 21:04:01.665 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0b04]: Reading attributes in chunks: ['ac_voltage_multiplier', 'ac_frequency_divisor', 'ac_frequency_multiplier', 'measurement_type', 'power_divisor']
2022-12-01 21:04:01.665 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0b04]: Reading attributes in chunks: ['power_multiplier']
2022-12-01 21:04:01.665 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0b04]: initializing uncached channel attributes: ['active_power', 'active_power_max', 'apparent_power', 'rms_current', 'rms_current_max', 'rms_voltage', 'rms_voltage_max', 'ac_frequency', 'ac_frequency_max'] - from cache[True]
2022-12-01 21:04:01.665 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0b04]: Reading attributes in chunks: ['active_power', 'active_power_max', 'apparent_power', 'rms_current', 'rms_current_max']
2022-12-01 21:04:01.666 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0b04]: Reading attributes in chunks: ['rms_voltage', 'rms_voltage_max', 'ac_frequency', 'ac_frequency_max']
2022-12-01 21:04:01.666 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0b04]: finished channel initialization
2022-12-01 21:04:01.666 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0000]: initializing channel: from_cache: True
2022-12-01 21:04:01.666 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0000]: finished channel initialization
2022-12-01 21:04:01.666 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1C87:1:0x0702]: initializing channel: from_cache: True
[... cut ...]
2022-12-01 21:04:07.927 DEBUG (MainThread) [zigpy.zcl] [0x1C87:1:0xef00] Received ZCL frame: b'\t\x1c%\xf3\x00'
2022-12-01 21:04:07.928 DEBUG (MainThread) [zigpy.zcl] [0x1C87:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=28, command_id=37, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)`

Warre118, 2022-12-15:
Hi 
Are there any updates on this request?
I have the same device installed yesterday but can't get it to work.
From what I understand ZHA needs to send a a reply on command 0x25 to confirm that there is a cloud connection, before it starts sending any data, as described here https://github.com/Koenkk/zigbee2mqtt/issues/15359#issuecomment-1342957715

I have some programming skills, but this is too difficult for me to figure out, can someone help with this?
Big chance that the rest of the ts0601_din_power.py quirk works as soon as that this command is send. (if not, big chance that I can figure out the attributes and map it correctly as soon as it is being send by the device)

MattWestb, 2022-12-15:
@javicalle Look in the link in the post before its very interesting and the "found" is implanted and tested by the same user as was doing the same with our "tuya magic" casting for Hubitat and Z2M and ZHA (deCONZ not) have implanting it from his work.
I think "tuya cloud service status" need being implanted in your MCU commands and i think some other devices can needing it 2 (the LCD temperature sensors and some TRVs that  is not reporting with very long intervals or not at all).

Thanks for digging and implanting your findings in Hubitat so other open system can using it @kkossev (our tuya magician) !!!

MattWestb, 2022-12-15:
From https://developer.tuya.com/en/docs/iot/tuya-zigbee-module-uart-communication-protocol?id=K9ear5khsqoty#title-32-MCU%20requests%20gateway%20connection%20status%20(0x25).
Command 0x25 = Tuya gateway connection status
And command 0x26 = Configure Zigbee network policy is interesting.

javicalle, 2022-12-15:
Ok, let's try it...

That's my (quick & dirty) proposal:

1. copy the current `ts0601_din_power.py` file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_din_power.py

2. edit the `HikingManufClusterDinPower` and add the following code (ie: between `attributes` declaration and the `_update_attribute` method):
```python
        .../...
        HIKING_POWER_FACTOR_ATTR: ("power_factor", t.uint16_t, True),
    }

    client_commands = TuyaManufClusterAttributes.client_commands.copy()
    client_commands.update(
        {
            0x25: foundation.ZCLCommandDef(
                "mcu_connection_status",
                {},
                True,
                is_manufacturer_specific=True,
            ),
        }
    )

    def handle_mcu_connection_status(self) -> foundation.Status:
        """Handle gateway connection status requests (0x25)."""

        payload_rsp = 0x01  # 0x00 not connected to internet | 0x01 connected to internet | 0x02 time out

        self.create_catching_task(
            super().command(0x25, payload_rsp, expect_reply=False)
        )

        return foundation.Status.SUCCESS

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        .../...
```

3. add the import foundation statement at the top of file:
```python
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
```

Save changes, restart HA and repair the device.

There is some instructions:
* https://github.com/zigpy/zha-device-handlers/issues/1973#issuecomment-1331903614

If something isn't working, enable the debug logs and attach the relevant info.
Put the logs between ` ``` ` marks or as attached file to easy reading.

Warre118, 2022-12-15:
Hi, 
Thank you for the quick response, your help is much apperciated.
I made the necessary changes to the custom quirk, however, there still seems to be an issue with the 0x25 command.
since the warning changed, I'm quite sure the changes had an effect.

Now I get `WARNING (MainThread) [zigpy.zcl] [0x93D7:1:0xef00] Data remains after deserializing ZCL frame: b'\xb9\x00'` 

I copied the most relevant lines from the debug log and pasted them below.
If you need the complete file, happy to include this.

` 
2022-12-15 23:46:50.850 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x93D7, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=40, SecurityUse=<Bool.false: 0>, TimeStamp=1369556, TSN=0, Data=b'\x09\x40\x25\xB9\x00', MacSrcAddr=0x93D7, MsgResultRadius=29)
2022-12-15 23:46:50.855 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x93D7), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\t@%\xb9\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=40, rssi=None)
2022-12-15 23:46:50.857 DEBUG (MainThread) [zigpy.zcl] [0x93D7:1:0xef00] Received ZCL frame: b'\t@%\xb9\x00'
2022-12-15 23:46:50.859 DEBUG (MainThread) [zigpy.zcl] [0x93D7:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=64, command_id=37, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-15 23:46:50.862 DEBUG (MainThread) [zigpy.zcl] [0x93D7:1:0xef00] Decoded ZCL frame: HikingManufClusterDinPower:mcu_connection_status()
2022-12-15 23:46:50.863 WARNING (MainThread) [zigpy.zcl] [0x93D7:1:0xef00] Data remains after deserializing ZCL frame: b'\xb9\x00'
2022-12-15 23:46:50.873 DEBUG (MainThread) [zigpy.zcl] [0x93D7:1:0xef00] Received command 0x25 (TSN 64): mcu_connection_status()
2022-12-15 23:46:50.874 DEBUG (MainThread) [zigpy.zcl] [0x93D7:1:0xef00] No explicit handler for cluster command 0x25: mcu_connection_status()
 `

Warre118, 2022-12-16:
The warnings are gone now, but the device is still only sending the x25 messages.
I also see the "no explicit handler" message at the bottom of the log.
not sure if it helps, but here is the debug log where i got below snippet from.
[home-assistant_zha_2022-12-16T00-39-37.512Z.log.txt](https://github.com/zigpy/zha-device-handlers/files/10241816/home-assistant_zha_2022-12-16T00-39-37.512Z.log.txt)

```
2022-12-16 01:39:24.728 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x3DFB, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=40, SecurityUse=<Bool.false: 0>, TimeStamp=501062, TSN=0, Data=b'\x09\x3B\x25\x5D\x00', MacSrcAddr=0x3DFB, MsgResultRadius=29)
2022-12-16 01:39:24.732 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x3DFB), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\t;%]\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=40, rssi=None)
2022-12-16 01:39:24.734 DEBUG (MainThread) [zigpy.zcl] [0x3DFB:1:0xef00] Received ZCL frame: b'\t;%]\x00'
2022-12-16 01:39:24.737 DEBUG (MainThread) [zigpy.zcl] [0x3DFB:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=59, command_id=37, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-16 01:39:24.740 DEBUG (MainThread) [zigpy.zcl] [0x3DFB:1:0xef00] Decoded ZCL frame: HikingManufClusterDinPower:mcu_connection_status(data=TuyaTsn(status=93, tsn=0))
2022-12-16 01:39:24.741 DEBUG (MainThread) [zigpy.zcl] [0x3DFB:1:0xef00] Received command 0x25 (TSN 59): mcu_connection_status(data=TuyaTsn(status=93, tsn=0))
2022-12-16 01:39:24.743 DEBUG (MainThread) [zigpy.zcl] [0x3DFB:1:0xef00] No explicit handler for cluster command 0x25: mcu_connection_status(data=TuyaTsn(status=93, tsn=0))
```

javicalle, 2022-12-16:
Have you updated also this part?
```python
def handle_mcu_connection_status(self, payload: TuyaTsn) -> foundation.Status:
```

Warre118, 2022-12-16:
Just double checked, but that looks ok.
I copy/pasted the whole block of code over the old block, did a restart of home assistant, and re-paired the device.
Just to be sure, below a printscreen of the adapted class.

[image]

I also see a timeout error during initialisation, but this is for the 0x0006 cluster, so I don't think that is relevant for now. 

```
2022-12-16 08:38:30.300 WARNING (MainThread) [homeassistant.components.zha.core.channels.base] [0x3DFB:16:0x0006]: async_initialize: all attempts have failed: [TimeoutError(), TimeoutError(), TimeoutError(), TimeoutError()]
```

javicalle, 2022-12-16:
Not sure if related but it lacks an break like before the `_update_attribute` method.
I'll check later.

Warre118, 2022-12-19:
Hi, @javicalle have you been able to take a look?
I had another look over the weekend, but I can't figure out why the handle is not recognized.
If I base myself on the  `handle_mcu_version_response` in [this _ini_.py](https://github.com/zigpy/zha-device-handlers/blob/5ae95df8c1b1052ce89ab50b916e95ce77711487/zhaquirks/tuya/mcu/__init__.py), as far as I can tell everything looks ok.

The only thing I see, is that in this comand the `client_commands` are copied from the ` TuyaNewManufCluster` class instead of the `TuyaManufClusterAttributes` class. so I copied the command from the `TuyaManufCluster` class, (that seemed the most logic position to me) but the results are the same.

Warre118, 2022-12-19:
ok, I think I got the handle working, based on the logs below. 
However the device is still repeating the `x25` command, so the exepected response does not seem correct.
[here](https://github.com/zigpy/zha-device-handlers/files/10257654/ts0601_din_power.TXT) you find the complete custom quirk file. Below the lines I changed to get it working.

```
from zhaquirks.tuya import TuyaManufClusterAttributes, TuyaOnOff, TuyaSwitch, TuyaNewManufCluster

class HikingManufClusterDinPower(TuyaManufClusterAttributes, TuyaNewManufCluster):

client_commands = TuyaNewManufCluster.client_commands.copy()
```

```
2022-12-19 09:34:14.694 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x3DFB, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=40, SecurityUse=<Bool.false: 0>, TimeStamp=9681462, TSN=0, Data=b'\x09\x27\x25\x11\x00', MacSrcAddr=0x3DFB, MsgResultRadius=29)
2022-12-19 09:34:14.699 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x3DFB), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b"\t'%\x11\x00"], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=40, rssi=None)
2022-12-19 09:34:14.702 DEBUG (MainThread) [zigpy.zcl] [0x3DFB:1:0xef00] Received ZCL frame: b"\t'%\x11\x00"
2022-12-19 09:34:14.704 DEBUG (MainThread) [zigpy.zcl] [0x3DFB:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=39, command_id=37, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-19 09:34:14.706 DEBUG (MainThread) [zigpy.zcl] [0x3DFB:1:0xef00] Decoded ZCL frame: HikingManufClusterDinPower:mcu_thisisatest(data=TuyaTsn(status=17, tsn=0))
2022-12-19 09:34:14.707 DEBUG (MainThread) [zigpy.zcl] [0x3DFB:1:0xef00] Received command 0x25 (TSN 39): mcu_thisisatest(data=TuyaTsn(status=17, tsn=0))
2022-12-19 09:34:14.723 ERROR (MainThread) [zigpy.zcl] [0x3DFB:1:0xef00] Traceback (most recent call last):
2022-12-19 09:34:14.734 DEBUG (MainThread) [zigpy.zcl] [0x3DFB:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=39, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-19 09:34:14.736 DEBUG (MainThread) [zigpy.zcl] [0x3DFB:1:0xef00] Sending reply: Default_Response(command_id=37, status=<Status.SUCCESS: 0>)
2022-12-19 09:34:14.738 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x3DFB), dst_ep=1, source_route=None, extended_timeout=False, tsn=39, profile_id=260, cluster_id=61184, data=Serialized[b"\x18'\x0b%\x00"], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-12-19 09:34:14.742 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x3DFB), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=39, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x18\x27\x0B\x25\x00')

```

kkossev, 2022-12-20:
> However the device is still repeating the 0x25 command, so the exepected response does not seem correct
The payload for the 0x25 command answer should be 3 bytes - the previously received serial number (2 bytes) plus 0x01. See the [sniffed communication](https://github.com/Koenkk/zigbee2mqtt/issues/15359#issuecomment-1344528114) to Tuya gateway. So either the Tuya documentation or the MatSee Plus implementation for this device may be wrong,

However, even with the correct answer to 0x25 command, the device will still continue to query the ZIgbee coordinator every 5 seconds.
The LED will blink shortly every 3 seconds if the 0x25 command answer was OK. Otherwise will stay constantly lit.

Warre118, 2022-12-21:
Thank you for the info.
The LED does not blink, it is permanent on, and In the logs I only see the x25 command coming trough in the form of `Data=b'\x09\x27\x25\x11\x00'` 

I'm way over my head here, so below logic could be totatly wrong.
To get a specific example, I'll use the data in the logs of [above comment](https://github.com/zigpy/zha-device-handlers/issues/1973#issuecomment-1357301237).
If the received string is `Data=b'\x09\x27\x25\x11\x00'`  then, from what I understand, I have to reply with `x27\x11\x01`.
Now my question is how do I build this reply? 

The data received from the device is stored via  the `class TuyaTsn(t.Struct):`
If I check the `status: t.uint8_t` I get `x11`
If I check the `tsn: t.uint8_t`  I always get `0`
Based on that, I think only the second part of the data string (`x25\x11\x00`) is stored.

For now this is my code, but I'm quite sure the `payload_rsp` needs to be defined differently.
p.s. in [this](https://github.com/zigpy/zha-device-handlers/issues/1973#issuecomment-1353750980) comment the code block is colorcoded, How do I do that? I can't find this in the Github documentation. 😅

````
    class TuyaTsn(t.Struct):
        """Tuya request data."""

        status: t.uint8_t
        tsn: t.uint8_t

    client_commands = TuyaNewManufCluster.client_commands.copy()
    client_commands.update(
        {
            0x25: foundation.ZCLCommandDef(
                "mcu_thisisatest",
                {"data": TuyaTsn},
                True,
                is_manufacturer_specific=True,
            ),
        }
    )

    def handle_mcu_thisisatest(self, payload: TuyaTsn) -> foundation.Status:
        """Handle gateway connection status requests (0x25)."""
        payload_rsp: t.uint32_t
        payload_rsp = payload.status, payload.tsn, 0x01  # 0x00 not connected to internet | 0x01 connected to internet | 0x02 time out
        
        self.debug("handle_mcu_thisisatest response: %s", payload_rsp)
        
        self.create_catching_task(
            super().command(0x25, payload_rsp, expect_reply=False)
        )

        return foundation.Status.SUCCESS
````

````
2022-12-21 09:42:14.966 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0xC817, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=76, SecurityUse=<Bool.false: 0>, TimeStamp=3890105, TSN=0, Data=b'\x09\x53\x25\xEC\x00', MacSrcAddr=0xDF0C, MsgResultRadius=28)
2022-12-21 09:42:14.967 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xC817), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\tS%\xec\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=76, rssi=None)
2022-12-21 09:42:14.968 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Received ZCL frame: b'\tS%\xec\x00'
2022-12-21 09:42:14.968 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=83, command_id=37, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-21 09:42:14.970 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Decoded ZCL frame: HikingManufClusterDinPower:mcu_thisisatest(data=TuyaTsn(status=236, tsn=0))
2022-12-21 09:42:14.970 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Received command 0x25 (TSN 83): mcu_thisisatest(data=TuyaTsn(status=236, tsn=0))
2022-12-21 09:42:14.971 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] handle_mcu_thisisatest response: (0, 236, 1)
2022-12-21 09:42:14.974 ERROR (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy/quirks/__init__.py", line 173, in command
    command = self.server_commands[command_id]
KeyError: 37

2022-12-21 09:42:14.975 DEBUG (Thread-13) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f32d0ec5540>, 'UPDATE devices_v11\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_last_seen_delta', {'ts': 1671612134.967824, 'ieee': e0:79:8d:ff:fe:de:3b:71, 'min_last_seen_delta': 30.0})
2022-12-21 09:42:14.979 DEBUG (Thread-13) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f32d0ec5540>, 'UPDATE devices_v11\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_last_seen_delta', {'ts': 1671612134.967824, 'ieee': e0:79:8d:ff:fe:de:3b:71, 'min_last_seen_delta': 30.0}) completed
2022-12-21 09:42:14.982 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=83, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[... cut ...]
2022-12-21 09:42:14.988 DEBUG (Thread-13) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f32d0ec5540>)
2022-12-21 09:42:14.990 DEBUG (Thread-13) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f32d0ec5540>) completed
2022-12-21 09:42:15.007 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-12-21 09:42:15.053 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=83)
````

javicalle, 2022-12-21:
Sorry for the late response.

> The only thing I see, is that in this comand the `client_commands` are copied from the `TuyaNewManufCluster` class instead of the `TuyaManufClusterAttributes` class. so I copied the command from the `TuyaManufCluster` class, (that seemed the most logic position to me) but the results are the same.

No, that is not correct. The `TuyaNewManufCluster` is another diferent approach (for the same target) but don't apply here.

> However the device is still repeating the `x25` command

Repeating from time to time is not a problem. The problem is if there are retries (with the same `tsn` number). Can you check it?

In any case, the response message (`Data=b'\x18\x27\x0B\x25\x00'`) is not what I would expect.

> ```
> 2022-12-21 09:42:14.974 ERROR (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Traceback (most recent call last):
>   File "/usr/local/lib/python3.10/site-packages/zigpy/quirks/__init__.py", line 173, in command
>     command = self.server_commands[command_id]
> KeyError: 37
> ```

That's interesting...
We need to define also the `server_commands` to be able to send the response.

Putting all together that would be:

```python
"""Tuya Din Power Meter."""
from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaManufClusterAttributes, TuyaOnOff, TuyaSwitch

TUYA_TOTAL_ENERGY_ATTR = 0x0211
TUYA_CURRENT_ATTR = 0x0212
TUYA_POWER_ATTR = 0x0213
TUYA_VOLTAGE_ATTR = 0x0214
TUYA_DIN_SWITCH_ATTR = 0x0101

SWITCH_EVENT = "switch_event"

"""Hiking Power Meter Attributes"""
HIKING_DIN_SWITCH_ATTR = 0x0110
HIKING_TOTAL_ENERGY_DELIVERED_ATTR = 0x0201
HIKING_TOTAL_ENERGY_RECEIVED_ATTR = 0x0266
HIKING_VOLTAGE_CURRENT_ATTR = 0x0006
HIKING_POWER_ATTR = 0x0267
HIKING_FREQUENCY_ATTR = 0x0269
HIKING_POWER_FACTOR_ATTR = 0x026F
HIKING_TOTAL_REACTIVE_ATTR = 0x026D
HIKING_REACTIVE_POWER_ATTR = 0x026E

class TuyaManufClusterDinPower(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Tuya Power Meter device."""

    attributes = {
        TUYA_TOTAL_ENERGY_ATTR: ("energy", t.uint16_t, True),
        TUYA_CURRENT_ATTR: ("current", t.int16s, True),
        TUYA_POWER_ATTR: ("power", t.uint16_t, True),
[... 279 lines cut ...]
                    Scenes.cluster_id,
                    HikingManufClusterDinPower,
                    TuyaElectricalMeasurement,
                    TuyaPowerMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            16: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
```

PS: you can color your code with:
````
```python
YOUR CODE HERE
```
````

javicalle, 2022-12-21:
> If I check the `status: t.uint8_t` I get `x11`

This value change over time? Or is exactly `x11` always?

Warre118, 2022-12-21:
The message is repeated five times,  then there is a timeout of 5 seconds, and then the  TSN number goes up with one decimal, and the status number goes up with two decimals.
Below an example.

With the code above the `No explicit handler for cluster command 0x25: mcu_connection_status`  pops up.

```
2022-12-21 22:56:28.421 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xC817), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\tu%\xd4\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=127, rssi=None)
2022-12-21 22:56:28.423 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Received ZCL frame: b'\tu%\xd4\x00'
2022-12-21 22:56:28.427 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=117, command_id=37, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-21 22:56:28.431 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Decoded ZCL frame: HikingManufClusterDinPower:mcu_connection_status(payload=TuyaConnectionStatus(data_1=212, data_2=0))
2022-12-21 22:56:28.434 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Received command 0x25 (TSN 117): mcu_connection_status(payload=TuyaConnectionStatus(data_1=212, data_2=0))
2022-12-21 22:56:28.441 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] No explicit handler for cluster command 0x25: mcu_connection_status(payload=TuyaConnectionStatus(data_1=212, data_2=0))
------------------------------------------------------------------------------------------------------------------------------------
2022-12-21 22:56:32.082 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xC817), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\tv%\xd6\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=127, rssi=None)
2022-12-21 22:56:32.083 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Received ZCL frame: b'\tv%\xd6\x00'
2022-12-21 22:56:32.084 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=118, command_id=37, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-21 22:56:32.085 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Decoded ZCL frame: HikingManufClusterDinPower:mcu_connection_status(payload=TuyaConnectionStatus(data_1=214, data_2=0))
2022-12-21 22:56:32.086 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Received command 0x25 (TSN 118): mcu_connection_status(payload=TuyaConnectionStatus(data_1=214, data_2=0))
2022-12-21 22:56:32.087 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] No explicit handler for cluster command 0x25: mcu_connection_status(payload=TuyaConnectionStatus(data_1=214, data_2=0))

```

MattWestb, 2022-12-21:
If one unicats is not being acked the sender is sending it with the same TSN after some hundred millS.
Id the system (the Zibee stack if ZCL or the quirk id not ZCL) sending one default resonance to the device ?
If not is the device doing right and the the system is doing wrong.

Default response from quirk is implanted in tuya INIT must only using the right device class for using it.

javicalle, 2022-12-21:
Thanks for testing.
I'm a bit confused here.

Can you try replacing the copy() parr like this?
```python
client_commands = TuyaManufCluster.client_commands.copy()
```
```python
server_commands = TuyaManufCluster.server_commands.copy()
```

Warre118, 2022-12-22:
Hi javicalle, That gives me the same error message.

If I use the `TuyaNewManufCluster` , the handle is recognized, but now I get a traceback.
I had to import the class and define it like described [here](https://github.com/zigpy/zha-device-handlers/issues/1973#issuecomment-1357301237)
Below you find the log, and the adapted code that generated this log.

My reasoning is that the 'TuyaNewManufCluster' contains a similar error handling message as I see in the logs except it specifically says  _tuya_ handler.

In the code I also changed
````python
def handle_mcu_connection_status(self, payload: TuyaTsn)  -> foundation.Status:
````
to
````python
def handle_mcu_connection_status(self, payload: TuyaConnectionStatus) -> foundation.Status:
````

```
2022-12-22 08:29:11.019 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0xC817, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=58, SecurityUse=<Bool.false: 0>, TimeStamp=1497354, TSN=0, Data=b'\x09\x6C\x25\x1E\x00', MacSrcAddr=0xA4E2, MsgResultRadius=28)
2022-12-22 08:29:11.023 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xC817), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\tl%\x1e\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=58, rssi=None)
2022-12-22 08:29:11.026 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Received ZCL frame: b'\tl%\x1e\x00'
2022-12-22 08:29:11.030 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=108, command_id=37, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-22 08:29:11.036 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Decoded ZCL frame: HikingManufClusterDinPower:mcu_connection_status(payload=TuyaConnectionStatus(data_1=30, data_2=0))
2022-12-22 08:29:11.038 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Received command 0x25 (TSN 108): mcu_connection_status(payload=TuyaConnectionStatus(data_1=30, data_2=0))
2022-12-22 08:29:11.040 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/zigbee/application.py", line 588, in on_af_message
    self.packet_received(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 855, in packet_received
    self.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 361, in handle_message
    return sender.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 370, in handle_message
    return endpoint.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 224, in handle_message
    handler(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 373, in handle_message
    self.handle_cluster_request(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py", line 429, in handle_cluster_request
    return super().handle_cluster_request(
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py", line 377, in handle_cluster_request
    return super().handle_cluster_request(
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py", line 1431, in handle_cluster_request
    status = getattr(self, handler_name)(*args)
  File "/config/custom_zha_quirks/ts0601_din_power.py", line 198, in handle_mcu_connection_status
    payload_rsp = TuyaConnectionStatusRsp()
NameError: name 'TuyaConnectionStatusRsp' is not defined
```

```python
"""Tuya Din Power Meter."""
from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaManufClusterAttributes, TuyaOnOff, TuyaSwitch, TuyaNewManufCluster

TUYA_TOTAL_ENERGY_ATTR = 0x0211
[... 300 lines cut ...]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    HikingManufClusterDinPower,
                    TuyaElectricalMeasurement,
                    TuyaPowerMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            16: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }

```

javicalle, 2022-12-22:
To fix the exception replace at `handle_mcu_connection_status` this part:
```python
        payload_rsp = HikingManufClusterDinPower.TuyaConnectionStatusRsp()
```

javicalle, 2022-12-22:
Also, update to false the new command definition at the `server_commands` like this:
```python
    server_commands.update(
        {
            0x25: foundation.ZCLCommandDef(
                "mcu_connection_status_rsp",
                {"payload": TuyaConnectionStatusRsp},
                False,
                is_manufacturer_specific=True,
            ),
        }
    )
```

javicalle, 2022-12-22:
Z2M support for command here:
https://github.com/Koenkk/zigbee-herdsman/commit/22c41a81f4f6ef49538f2ea2cae27bdbe69a6b0b

Not sure about attributes meanings, but match with our proposal.

Warre118, 2022-12-22:
Ok I think we are getting somewhere.
I see different packets in the logs coming from the device now, and the LED blinks every few seconds. 
As @MattWestb described this should be the expected behaviour.

I see a traceback on command 0x02 though, and I still don't see the correct entities on the device page.

I'll start with re-pairing the device, and then I'll try to figure out what the different messages mean, or at least filter the log a bit post the relevant lines here.

```
2022-12-22 16:40:26.803 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xC817), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\tO\x02\x00\xbc\x13\x02\x00\x04\x00\x00\t\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=80, rssi=None)
2022-12-22 16:40:26.808 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Received ZCL frame: b'\tO\x02\x00\xbc\x13\x02\x00\x04\x00\x00\t\x01'
2022-12-22 16:40:26.814 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=79, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-22 16:40:26.818 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Decoded ZCL frame: HikingManufClusterDinPower:set_data_response(data=TuyaCommand(status=0, tsn=188, datapoints=[TuyaDatapointData(dp=19, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x01\t\x00\x00', *payload=2305))]))
2022-12-22 16:40:26.823 DEBUG (MainThread) [zigpy.zcl] [0xC817:1:0xef00] Received command 0x02 (TSN 79): set_data_response(data=TuyaCommand(status=0, tsn=188, datapoints=[TuyaDatapointData(dp=19, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x01\t\x00\x00', *payload=2305))]))
2022-12-22 16:40:26.829 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/zigbee/application.py", line 588, in on_af_message
    self.packet_received(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 855, in packet_received
    self.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 361, in handle_message
    return sender.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 370, in handle_message
    return endpoint.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 224, in handle_message
    handler(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 373, in handle_message
    self.handle_cluster_request(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py", line 437, in handle_cluster_request
    tuya_cmd = args[0].command_id
AttributeError: 'TuyaCommand' object has no attribute 'command_id'

```

MattWestb, 2022-12-22:
## Feliz navidad @javicalle !!!

javicalle, 2022-12-22:
That error comes from putting the `TuyaManufClusterAttributes` and `TuyaNewManufCluster` together.

Warre118, 2022-12-22:
Ok I found three datapoints in the logs for this device.
Following datapoints should be mapped:

````
dp=18  ==>  ampere/1000   ==>(payload=8487 with a stovetop on with the stovetop off it is  1395)
dp=19  ==>  wattage/10    ==>(payload=19306 with a stovetop on, with the stovetop off it is 1668)
dp=20  ==>  volts/10      ==>(payload=2315 with a stovetop on, with the stopvetop off it is 2337)
````
They all reported an unsupported attribute of `134` with `command id 2

I think that is all the data needed to add it to the quirk, but I need some help to do so.
I made an attempt, but that didn't work out so well. 😅

Below four snippets of the received messages. 
Here are the full set of logs if required [home-assistant_zha_2022-12-22T20-33-43.502Z.log](https://github.com/zigpy/zha-device-handlers/files/10290440/home-assistant_zha_2022-12-22T20-33-43.502Z.log)

````
2022-12-22 20:51:27.573 DEBUG (MainThread) [zigpy.zcl] [0xDF79:1:0xef00] Received ZCL frame: b'\t\x1a\x02\x00\xf9\x12\x02\x00\x04\x00\x00\x06\n'
2022-12-22 20:51:27.574 DEBUG (MainThread) [zigpy.zcl] [0xDF79:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=26, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-22 20:51:27.575 DEBUG (MainThread) [zigpy.zcl] [0xDF79:1:0xef00] Decoded ZCL frame: HikingManufClusterDinPower:set_data_response(data=TuyaCommand(status=0, tsn=249, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\n\x06\x00\x00', *payload=1546))]))
2022-12-22 20:51:27.576 DEBUG (MainThread) [zigpy.zcl] [0xDF79:1:0xef00] Received command 0x02 (TSN 26): set_data_response(data=TuyaCommand(status=0, tsn=249, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\n\x06\x00\x00', *payload=1546))]))
2022-12-22 20:51:27.577 DEBUG (MainThread) [zigpy.zcl] [0xDF79:1:0xef00] No datapoint handler for TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\n\x06\x00\x00', *payload=1546))
2022-12-22 20:51:27.579 DEBUG (MainThread) [zigpy.application] Previously delayed request is now running, delayed by 4.99s
2022-12-22 20:51:27.582 DEBUG (MainThread) [zigpy.zcl] [0xDF79:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=26, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-22 20:51:27.582 DEBUG (MainThread) [zigpy.zcl] [0xDF79:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>)
2022-12-22 20:51:27.583 DEBUG (Thread-16) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7fdd69441a40>)
2022-12-22 20:51:27.584 DEBUG (Thread-16) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7fdd69441a40>) completed
2022-12-22 20:51:27.584 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xDF79), dst_ep=1, source_route=None, extended_timeout=False, tsn=26, profile_id=260, cluster_id=61184, data=Serialized[b'\x18\x1a\x0b\x02\x86'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-12-22 20:51:27.585 DEBUG (MainThread) [zigpy.application] Max concurrency (8) reached, delaying request (43 enqueued)
2022-12-22 20:51:27.586 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0xDF79, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=69, SecurityUse=<Bool.false: 0>, TimeStamp=737917, TSN=0, Data=b'\x09\x1A\x02\x00\xF9\x12\x02\x00\x04\x00\x00\x06\x0A', MacSrcAddr=0xA4E2, MsgResultRadius=28)
2022-12-22 20:51:27.587 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=2820, SrcAddr=0x342E, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=109, SecurityUse=<Bool.false: 0>, TimeStamp=750741, TSN=0, Data=b'\x18\x6A\x0A\x08\x05\x21\x39\x02', MacSrcAddr=0x342E, MsgResultRadius=29)
2022-12-22 20:51:27.588 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0xDF79, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=69, SecurityUse=<Bool.false: 0>, TimeStamp=750957, TSN=0, Data=b'\x09\x1A\x02\x00\xF9\x12\x02\x00\x04\x00\x00\x06\x0A', MacSrcAddr=0xA4E2, MsgResultRadius=28)
2022-12-22 20:51:27.589 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0xDF79, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=69, SecurityUse=<Bool.false: 0>, TimeStamp=762478, TSN=0, Data=b'\x09\x1A\x02\x00\xF9\x12\x02\x00\x04\x00\x00\x06\x0A', MacSrcAddr=0xA4E2, MsgResultRadius=28)
[... cut ...] =61184, data=Serialized[b'\x18\x1e\x0b\x02\x86'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-12-22 20:51:34.683 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xDF79), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=30, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x18\x1E\x0B\x02\x86')
````

javicalle, 2022-12-22:
How have you fixewd the previous error?

I'm working in another approach for the quirk.
Totally experimental and barelly tested.

<details>
<summary>ts0601_din_power_mcu.py</summary>

```python
"""Tuya Din Power Meter."""
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaOnOff,
)

class TuyaPowerMeasurement(LocalDataCluster, ElectricalMeasurement):
    """Custom class for power, voltage and current measurement."""

    AC_CURRENT_MULTIPLIER = 0x0602
    AC_CURRENT_DIVISOR = 0x0603

    _CONSTANT_ATTRIBUTES = {AC_CURRENT_MULTIPLIER: 1, AC_CURRENT_DIVISOR: 1000}

class TuyaElectricalMeasurement(LocalDataCluster, Metering):
    """Custom class for total energy measurement."""

    POWER_WATT = 0x0000

    _CONSTANT_ATTRIBUTES = {
        0x0300: POWER_WATT,  # unit_of_measure
        0x0302: 100,  # divisor
    }

class DinPowerManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with din power datapoints."""

    class TuyaConnectionStatus(t.Struct):
        """Tuya request data."""

        tsn: t.uint8_t
        status: t.LVBytes

    client_commands = TuyaMCUCluster.client_commands.copy()
    client_commands.update(
        {
            0x25: foundation.ZCLCommandDef(
                "mcu_connection_status",
                {"payload": TuyaConnectionStatus},
                True,
[... 130 lines cut ...]
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    DinPowerManufCluster,
                    TuyaElectricalMeasurement,
                    TuyaPowerMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            16: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
```

</details>

This quirk would replace the previous one. If you want to try you can rename the file extension from the previous one and create a new one, this way you can return to the previous quirk at any moment.

Warre118, 2022-12-22:
yeah I just removed `TuyaManufClusterAttributes` from the class, that seemed to do the job.

> How have you fixed the previous error?
> 
> I'm working in another approach for the quirk. Totally experimental and barelly tested.
> 
> ts0601_din_power_mcu.py
> This quirk would replace the previous one. If you want to try you can rename the file extension from the previous one and create a new one, this way you can return to the previous quirk at any moment.

The new quirk works looks good, the output is exactly the same as in the previous logs.

javicalle, 2022-12-22:
Have the quirk created any entity in HA?
Reading at Z2M don't seems match the DPs from quirk

Warre118, 2022-12-22:
Nope no new entities.

I think that is what I see in the logs, shouldn't they be mapped like this? https://github.com/zigpy/zha-device-handlers/issues/1973#issuecomment-1363379122

I'm just not sure where/how to define this in the quirk though.

javicalle, 2022-12-22:
And the old ones which are?

Warre118, 2022-12-22:
Below the entities, but the device does not have a switch.
The power factor reports permanent `unkown` state if I enable it.
I can try and re-pair the device, and see what happens with the new quirk, one sec.

[image]

cryptocelt, 2022-12-22:
I am facing the same issue, with the same adapter, how can I help?
This one is mine, I get the same results as @Warre118:
("_TZE204_cjbofhxw", "TS0601")

javicalle, 2022-12-22:
Every time you made changes in a quirk, you need to delete any `__pycache__` folder from your local quirk installation, restart HA and re pair the device.

cryptocelt, 2022-12-22:
[image]

cryptocelt, 2022-12-22:
> How have you fixed the previous error?
>
> I'm working in another approach for the quirk. Totally experimental and barelly tested.
>
> ts0601_din_power_mcu.py
> This quirk would replace the previous one. If [...]

Here are the log files trying the latest version (cache folder deleted and HA restarted).

[home-assistant_2022-12-22T22-37-42.241Z.log](https://github.com/zigpy/zha-device-handlers/files/10290778/home-assistant_2022-12-22T22-37-42.241Z.log)

javicalle, 2022-12-22:
That would be the quirk for `_TZE204_cjbofhxw` according to Z2M:

<details>
<summary>ts0601_din_power_mcu.py</summary>

```python
"""Tuya Din Power Meter."""
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaOnOff,
)

class TuyaPowerMeasurement(LocalDataCluster, ElectricalMeasurement):
    """Custom class for power, voltage and current measurement."""

    AC_CURRENT_MULTIPLIER = 0x0602
    AC_CURRENT_DIVISOR = 0x0603

    _CONSTANT_ATTRIBUTES = {AC_CURRENT_MULTIPLIER: 1, AC_CURRENT_DIVISOR: 1000}

class TuyaElectricalMeasurement(LocalDataCluster, Metering):
    """Custom class for total energy measurement."""

    POWER_WATT = 0x0000

    _CONSTANT_ATTRIBUTES = {
        0x0300: POWER_WATT,  # unit_of_measure
        0x0302: 1000,  # divisor
    }

class DinPowerManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with din power datapoints."""

    class TuyaConnectionStatus(t.Struct):
        """Tuya request data."""

        tsn: t.uint8_t
        status: t.LVBytes

    client_commands = TuyaMCUCluster.client_commands.copy()
    client_commands.update(
        {
            0x25: foundation.ZCLCommandDef(
                "mcu_connection_status",
                {"payload": TuyaConnectionStatus},
                True,
                is_manufacturer_specific=True,
            ),
        }
[... 214 lines cut ...]
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    DinPowerManufCluster,
                    TuyaElectricalMeasurement,
                    TuyaPowerMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            16: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
```

</details>

Reference:
* https://gist.github.com/Koenkk/8fc146f998e2bd9986df783164f4f3df

Warre118, 2022-12-22:
okay I removed the device from HA,  removed the `__pycache__`, restarted HA, and re-paired the device.

cryptocelt, 2022-12-22:
> That would be the quirk for `_TZE204_cjbofhxw` according to Z2M:
> 
> ts0601_din_power_mcu.py
> Reference:
> 
> * https://gist.github.com/Koenkk/8fc146f998e2bd9986df783164f4f3df

:/

Still not working.
I also tried removing and re-pairing it.
[home-assistant_2022-12-22T22-44-16.860Z.log](https://github.com/zigpy/zha-device-handlers/files/10290797/home-assistant_2022-12-22T22-44-16.860Z.log)
[zha-a9bb694b462bbe34a98b8c714d6263f1-_TZE204_cjbofhxw TS0601-e22ad2f4ef65be0dff13126b521f13a8.json.txt](https://github.com/zigpy/zha-device-handlers/files/10290798/zha-a9bb694b462bbe34a98b8c714d6263f1-_TZE204_cjbofhxw.TS0601-e22ad2f4ef65be0dff13126b521f13a8.json.txt)

[image]
[image]

Warre118, 2022-12-22:
okay, I removed the device, then removed the `__pycache__`, restarted HA, and re-paired the device.
The switch is gone now, but the rest remains the same.

I don't get the `No datapoint handler for TuyaDatapointData` messages anymore though.
Below the logs of an incoming message for datapoint 19

````
2022-12-22 23:49:18.203 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0xA3CF, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=7, SecurityUse=<Bool.false: 0>, TimeStamp=14670277, TSN=0, Data=b'\x09\x74\x02\x00\x8A\x13\x02\x00\x04\x00\x00\x07\x85', MacSrcAddr=0xA3CF, MsgResultRadius=29)
2022-12-22 23:49:18.207 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xA3CF), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\tt\x02\x00\x8a\x13\x02\x00\x04\x00\x00\x07\x85'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=7, rssi=None)
2022-12-22 23:49:18.210 DEBUG (MainThread) [zigpy.zcl] [0xA3CF:1:0xef00] Received ZCL frame: b'\tt\x02\x00\x8a\x13\x02\x00\x04\x00\x00\x07\x85'
2022-12-22 23:49:18.214 DEBUG (MainThread) [zigpy.zcl] [0xA3CF:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=116, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-22 23:49:18.220 DEBUG (MainThread) [zigpy.zcl] [0xA3CF:1:0xef00] Decoded ZCL frame: TuyaManufClusterDinPower:set_data_response(data=TuyaCommand(status=0, tsn=138, datapoints=[TuyaDatapointData(dp=19, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x85\x07\x00\x00', *payload=1925))]))
2022-12-22 23:49:18.224 DEBUG (MainThread) [zigpy.zcl] [0xA3CF:1:0xef00] Received command 0x02 (TSN 116): set_data_response(data=TuyaCommand(status=0, tsn=138, datapoints=[TuyaDatapointData(dp=19, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x85\x07\x00\x00', *payload=1925))]))
2022-12-22 23:49:18.235 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f0d1cccf040>, 'UPDATE devices_v11\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_last_seen_delta', {'ts': 1671749358.209509, 'ieee': e0:79:8d:ff:fe:de:3b:71, 'min_last_seen_delta': 30.0})
2022-12-22 23:49:18.233 DEBUG (MainThread) [zigpy.zcl] [0xA3CF:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=116, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-22 23:49:18.238 DEBUG (MainThread) [zigpy.zcl] [0xA3CF:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2022-12-22 23:49:18.241 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xA3CF), dst_ep=1, source_route=None, extended_timeout=False, tsn=116, profile_id=260, cluster_id=61184, data=Serialized[b'\x18t\x0b\x02\x00'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-12-22 23:49:18.248 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xA3CF), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=116, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x18\x74\x0B\x02\x00')
````

javicalle, 2022-12-23:
Could you check if the DEBUG logs for `zhaquirks` are enabled?

Refernce:
* https://www.home-assistant.io/integrations/zha#debug-logging

javicalle, 2022-12-23:
I had forgotten about the `EnchantedDevice` 🤦🏻‍♂️ 

I have updated the quirk from my previous comment. Just the new import and the hierarchy in `class TuyaPowerMeter(EnchantedDevice):`. 

Update, save, delete, restart & repair.

cryptocelt, 2022-12-23:
[zha-a9bb694b462bbe34a98b8c714d6263f1-_TZE204_cjbofhxw TS0601-e22ad2f4ef65be0dff13126b521f13a8.json (2).txt](https://github.com/zigpy/zha-device-handlers/files/10291240/zha-a9bb694b462bbe34a98b8c714d6263f1-_TZE204_cjbofhxw.TS0601-e22ad2f4ef65be0dff13126b521f13a8.json.2.txt)
[home-assistant_2022-12-23T00-50-38.283Z.log](https://github.com/zigpy/zha-device-handlers/files/10291241/home-assistant_2022-12-23T00-50-38.283Z.log)

Debug logs enabled.
Still not working, but I seriously appreciate your help.

Warre118, 2022-12-23:
I'm at work right now, so I couldn't re-pair the device.
In the device Diagnostics however I see the datapoints supported and getting updates now.

I even managed to add the entities by writing a way out of spec number(just to make sure it gets updated correctly) to the atribute, and then reload ZHA. 
After doing so, the device reports the values just fine.
[image]

[image]

it's just a guess, but Could it be that the device is not "ready"  when it has to generate the entities? 
Can we find that back in the logs?
The device only starts sending data, after it gets a positive 0x25 response. I can imagine it messes with things during initialization. 

Nonetheless, during noon, I'll clean the cache, and re-pair the device. 
Then we'll see from there.

````
         "0x0b04": {
            "endpoint_attribute": "electrical_measurement",
            "attributes": {
              "0x0505": {
                "attribute_name": "rms_voltage",
                "value": 235
              },
              "0x0508": {
                "attribute_name": "rms_current",
                "value": 946
              },
              "0x050b": {
                "attribute_name": "active_power",
                "value": 75
              },
              "0x0602": {
                "attribute_name": "ac_current_multiplier",
                "value": 1
              },
              "0x0603": {
                "attribute_name": "ac_current_divisor",
                "value": 1000
              }
            },
````

cryptocelt, 2022-12-23:
So, the seller suggests using z2m instead of zha, as it apparently works there.

He pointed me at:
https://github.com/Koenkk/zigbee2mqtt/issues/15359
But I would prefer sticking with zha :/

cryptocelt, 2022-12-24:
And it now works. 
Not sure what did the trick, a couple of HA restarts and device re-configurations were involved, but I now get data from the device.
I am know trying to figure out a way to reset the energy counter, but the quirk seems to work just fine. 
Any idea if I will will I have to disable the quirk once the driver hits the main release?

MattWestb, 2022-12-24:
For resetting the energy counter i think one reset of the device is needed.
I suggesting delete the device and resenting it and restart HA if doing it very well re power the host  system. Then HA is stable deleting the ` __pycache__ ` folder in your local quirk directors and then joining the device and see if its working OK.

Then the quirk is merged in the main ZHA you dont need the local quirk so deleting it in the local quirk folder and / or in the HA config for not getting problem in the future then ZHA is getting updates that can braking your local quirk.

cryptocelt, 2022-12-24:
I am not sure this is the proper place to post this, but i am getting a statistics_not_defined warning when i try to setup power in the energy dashboard. Is that something declared in the quirk?

I did use the power value in a Riemann sum integral, then that into a utility meter helper, and then that in the energy dashboard.

01fr, 2023-01-17:
Greetings.

Any update on this issue? I have tried all the quirks but never managed to get it working. The last quirk causes my Tuya smart plugs (TS011F by_TZ3000_gvn91tmx) to give out wrong power measurements.

javicalle, 2023-01-18:
@01fr I really can't recall all the info from this issue.
I would suggest to try with the quirk from here:
* https://github.com/zigpy/zha-device-handlers/issues/1973#issuecomment-1363412395

You would need to add your device in the `MODELS_INFO` like this:
```python
        MODELS_INFO: [
            ("_TZE200_bkkmqmyo", "TS0601"),
            ("_TZ3000_gvn91tmx", "TS0601"),
        ],
```
but just in one device at time (`TuyaPowerMeter` or `HikingPowerMeter`).

If not working, I would suggest to create a new issue with all the device info: signature (without any quirk loaded), logs, vendor page...

[82 more replies]

closed 2026-01-01 by github-actions[bot]