zigpy/zha-device-handlers #1975: [Device Support Request] Aqara Cube T1 Pro (lumi.remote.cagl02)

malhalla, 2022-12-01:
Device not working properly, zha_events only fired for "rotate left", "rotate right" and "shake".
Not "flip to side", "slide" etc

Expected that device to be contained in 'zhaquirks.xiaomi.aqara.cube_aqgl01.CubeCAGL02'
Tried with raspberry ii and Sonor ZBDongle-E, same behaviour.

Device signature and diagnostics attached.

[zha-09d2981b9eb9c743a9d295267e126c51-LUMI lumi.remote.cagl02-077d41276e29fae304e20428d234b81e.json.txt](https://github.com/zigpy/zha-device-handlers/files/10130197/zha-09d2981b9eb9c743a9d295267e126c51-LUMI.lumi.remote.cagl02-077d41276e29fae304e20428d234b81e.json.txt)

[Device Signature.odt](https://github.com/zigpy/zha-device-handlers/files/10130201/Device.Signature.odt)

robertdijk, 2023-01-02:
Looking at the signature (also the one I got from my device) there are only a few clusters, and not even a `MultistateInput` or `AnalogInput` like the previous version. Just this from my understanding:
```python
signature = {
    MODELS_INFO: [(LUMI, "lumi.remote.cagl02")],
    ENDPOINTS: {
        1: {
            PROFILE_ID: zha.PROFILE_ID,
            DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT_SWITCH,
            INPUT_CLUSTERS: [
                BasicCluster.cluster_id,
                Identify.cluster_id,
                OnOff.cluster_id,
            ],
            OUTPUT_CLUSTERS: [
                BasicCluster.cluster_id,
                Identify.cluster_id,
            ],
        },
    },
}
```
How would we get all the data necessary for all the actions?

malhalla, 2023-01-05:
This is solved, no action needed. 
The device is working in two different modes, which can be switched by pressing the pairing button 5 times in short time. 
After doing so, the existing quirk is working as expected.

closed 2023-01-05 by malhalla

mclex79, 2023-01-05:
I can't get this fix to switch things...can you please give more context on the presses and any other steps needed?

mclex79, 2023-01-05:
Nevermind, got it!!  Thanks a bunch!

mclex79, 2023-01-05:
> How did you resolve?

Using the method above.  The clicks are full clicks where the led lights up every time...5 in a row

mclex79, 2023-01-06:
Unfortunately it's reverted back to only recognizing the original rotations and shaking only after only 2.5 hours.  Anybody else having issues with this?

Michael-Busack, 2023-01-06:
Are you able to pair the t1 pro cube using z2m instead of Zha ?

I keep getting unsupported device 

Regards

Michael

robertdijk, 2023-01-06:
> > How did you resolve?
> 
> Using the method above. The clicks are full clicks where the led lights up every time...5 in a row

Do you first reset it? or first remove it? And then change the mode? Or first change the mode and then remove it? Because every time I re-add it, the signature is still the same.

antonyat, 2023-01-06:
> Do you first reset it? or first remove it? And then change the mode? Or first change the mode and then remove it? Because every time I re-add it, the signature is still the same.
> 
> And do you get a confirmation of it changing mode? Because after 5 clicks (quite slow to let the LED blink every click), nothing seems to change.

I connected it, then switched the mode. It doesn't give you any confirmation when it changes, just the blue light. 

I came across it because I was on the device page of the cube and noticed from the logbook that it had started to recognize face change events after the 5 button and blue light.

robertdijk, 2023-01-06:
> I connected it, then switched the mode. It doesn't give you any confirmation when it changes, just the blue light.
> 
> I came across it because I was on the device page of the cube and noticed from the logbook that it had started to recognize face change events after the 5 button press and blue light.

Did the signature change?
And is then also the zha quirk applied?

For me before and after pressing the button 5 times, it have a switch entity that toggles when shaken or double tapped, and turns on when flipped 90 or 180 degrees. But I can't see the command, or the side it is on...

mclex79, 2023-01-07:
You don't see anything in the device notes except and indication that it
flipped.  But when you go to do automation it will have the options for
each side and will register individual sides now.

codyde, 2023-01-11:
So weird - ive tried this a dozen times today and cant get it to register anything other than shakes or rotates when connected via zha. No slide motions, no flip actions, no knock on number actions. I defintely see the options in the automation - but no event fires when I use them. I thought maybe it was a logging problem so I built a few automations to test and nothing still.

mclex79, 2023-01-11:
I think that means that you didn't get the device switched over from the other mode.  It was all about fully pressing the button five times before it would switch.

codyde, 2023-01-11:
Is there any sort of device log I can pull to confirm which mode im currently set onto? Each time I tried I did the 5-press process - but like you said, theres no confirmation it "took" so its hard to gauge if it worked or not.

mclex79, 2023-01-11:
Sorry.. no clue.  I just kept repeating it .. Probably 5x before I did it right

codyde, 2023-01-11:
Whoa - finally got it. I sat in the developer tools screen in HA and clicked the button on the cube as fast as I could while still having it recognize the button press and eventually after a few tries it worked. Flips and knocks recognizing now.

W1ngR1d3r, 2023-01-13:
> Whoa - finally got it. I sat in the developer tools screen in HA and clicked the button on the cube as fast as I could while still having it recognize the button press and eventually after a few tries it worked. Flips and knocks recognizing now.

I have tried like 30 times and still not success. Could you please describe step by step how you did it? (what screen you were at in developers tools? what did you do in the cube and in HA?
Thanks!!!

Giuliointhetube, 2023-01-14:
> Whoa - finally got it. I sat in the developer tools screen in HA and clicked the button on the cube as fast as I could while still having it recognize the button press and eventually after a few tries it worked. Flips and knocks recognizing now.

So no long press??? Just fast as possible 5 times??

codyde, 2023-01-14:
So it wasn't just a rapid fire 5 press. I kept the event monitor open in HA development tools, filtering on zha_event. If I waited each time for the blue to flash it was still to slow. Instead, id watch for the "attribute changed" event to hit in HA and then press the button again, and wait again - for a total of 5 times. Id say it ended up being probably a half second between pushes. It was definitely fast though. 

No indicator that it worked - I just double tapped the cube, after doing the sequence, on my desk and saw the "knock" event fire.

Giuliointhetube, 2023-01-14:
> So it wasn't just a rapid fire 5 press. I kept the event monitor open in HA development tools, filtering on zha_event. If I waited each time for the blue to flash it was still to slow. Instead, id [...]

It works also for me and in a simple way. Just 5 fast press! Thank you!

gioele-antoci, 2023-01-14:
Did anyone have their cube mess with the ZHA network? My cube off the box toggles random zigbee devices on/off ?!

TheJulianJES, 2023-01-14:
Try unbinding it from your Zigbee Coordinator or try to remove it from group 0.

gioele-antoci, 2023-01-15:
> remove it from group 0.

I am not sure how to do this, can you point me to a resource or a keyword I can google to do this?

W1ngR1d3r, 2023-01-15:
This cube is driving me crazy. I did everything you said a lot of times. 
Today I managed to connect the cube and see in automations all the events BUT no event is triggered. 
I really don’t know what else to do.

dedeckerni, 2023-01-18:
This new cube t1 pro is my first zigbee device and first zigbee experience. Also had the issues with changing the mode and tried several times, but it was always stuck in the shake/rotate left/rotate right modes.  Tried several times pushing the button 5 times, tried with different intervals but it didn't work.

In some attempt I then pressed a couple of more times (like 7 or 8) with I guess 300 ms interval (guesstimate) and now the mode is ok: flip, slide, knock, rotate left/right, ... 
getting in the zha_event with the correct properties in HA.

Using a Sonoff ZBdongle-E and currently started with ZHA. Also using a 5 meter usb 2.0 extension cable to get to a central spot in my living area and seems to work great so far.

So for those that fail switching the mode: keep trying and do different sequences + maybe some more then 5 pushes.
It automagically started to work.

SirGoodenough, 2023-01-28:
The T1 pro cube is not functioning properly.  New functions such as the 'Throw' and the 'Hold' are not working. 'Side-up' is also not working.
See https://github.com/JJPro/CTP-R01-converter
For the ZHA Version for hints on how to make this work properly.
I would but do not have the knowledge to do so.

maxemoose, 2023-01-30:
To anyone still having issues with this: update your Zigbee coordinator firmware. Immediately after that the "press five times" solution will work for you.

I had a Sonoff Zigbee Dongle P that was running the firmware it arrived with. ZHA worked fine, no obvious issues, though the Aqara Cube T1 Pro would only pair as a switch and only shakes turned the switch on or off. 

When looking in Developer Tools -> Events to debug the issue, listening to zha_event showed nothing. Ever. That's when I updated the coordinator firmware. Now listening to zha_event shows everything it should. Applied the "press five times" solution and now my Cube is working.

petrusb83, 2023-02-04:
Still having issues. I have a conbee II stick. Seems a firmware upgrade for it is not a straight forward process...

henk65, 2023-02-19:
> To anyone still having issues with this: update your Zigbee coordinator firmware. Immediately after that the "press five times" solution will work for you.

Since I also have the G2 camera with Aqara hub built-in I don't need the "press button 5x" trick, I can just connect it to the hub and change the mode (action mode / scene mode) in the app.
But I didn't get the Cube T1 Pro to work in HA, it only showed a "switch" entity and only the "shake" would work to toggle that switch. Nothing more, and nothing to see when listening to zha_event.
Now I updated the Conbee II firmware from version 26580700 to 26780700 and reconnected the cube everything is working!

petrusb83, 2023-02-19:
Can you post instructions for updating the firmware?

henk65, 2023-02-20:
> Can you post instructions for updating the firmware?

In most cases you do NOT want to update the Conbee II firmware directly on the machine (Raspberry Pi, NUC, whatever you have) where HA is running. Don't even try if you don't know if it's possible or not, it's not going to be easy.
What I did? Pulled the Conbee II out of my HA NUC (without shutting it down...), stick it into my Windows 10 PC, waited for the USB driver to get installed (automatic on Windows 10) and then followed the following instructions, that's all.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-windows
Download GCFFlasher from the provided link and follow the steps "Update on a system without deCONZ software".

After that I put it back in my NUC and without restarting anything the Conbee II worked right away. After that I removed the Aqara Cube from ZHA, paired it again and everything was different! No switch entity anymore, you don't see any entities, only the standard entities like battery (not working), identify button, lqi and rssi. But in the logbook next to it you can see lots of things happening if you move the cube, and in the event listener in developer tools (zha_event) even more! That is, IF your cube is connected to ZHA and NOT to the Aqara Hub (I tried both) and if the cube is in "action mode", NOT the new "scene mode".

Meaglin84, 2023-03-12:
> In most cases you do NOT want to update the Conbee II firmware directly on the machine (Raspberry Pi, NUC, whatever you have) where HA is running. Don't even try if you don't know if it's possible [...]

I've tried this but for me nothing has changed unfortunately

monstergubl, 2023-08-04:
> > How did you resolve?
> 
> Using the method above. The clicks are full clicks where the led lights up every time...5 in a row

So i first long press for device search after that 5 times in a row ? right

rlue, 2024-10-15:
Struggled with this all day today. Finally got it working after I upgraded my Zigbee coordinator's firmware (SONOFF Zigbee USB 3.0 Dongle Plus-P), and then 2) switched the cube into Action Mode, though both steps were fraught with setbacks. If I could go back in time, I would 200% tell past-me not to buy this thing, and just stick with voice commands.

## Setting the Cube to Action Mode

I tried the press-LINK-five-times trick over and over and just couldn't get it to work. But apparently [you can toggle Scene/Action Mode on the Aqara Cube in the web frontend of Zigbee2MQTT](https://www.zigbee2mqtt.io/devices/CTP-R01.html#mode-switch)—the only snag is that setting up a fresh instance of Zigbee2MQTT and linking the Cube to your coordinator will bork your Zigbee network settings in Home Assistant. Setting it all up again is a real pain, but after burning a whole day on this, I was just glad to get it working.

In brief, I:

* shut down my running instance of Home Assistant
* launched a fresh instance of MQTT and Zigbee2MQTT
* added the Cube to my Zigbee network in Zigbee2MQTT
* set the Cube to Action Mode in Zigbee2MQTT
* shut down my instances of MQTT and Zigbee2MQTT
* re-launched Home Assistant
* removed all my ZHA devices
* re-added the ZHA integration from scratch

## Updating firmware on a SONOFF coordinator

Obviously, you can ignore this section if you have a different coordinator. It's not entirely clear to me if this is necessary at all—I tried this first (based on [this discussion](https://community.home-assistant.io/t/aqara-cube-t1-pro-no-zha-event-fired/582051/4)) and it didn't work, so I didn't get a chance to try Action Mode + my coordinator's original (2021) firmware.

### What didn't work: Flashing on Windows

The [official SONOFF docs](https://sonoff.tech/wp-content/uploads/2023/02/SONOFF-Zigbee-3.0-USB-dongle-plus-firmware-flashing.pdf) and [this YouTube tutorial from Smart Home Junkie](https://www.youtube.com/watch?v=KBAGWBWBATg) suggest using TI FLASH-PROGRAMMER-2 to flash the firmware, but it would **crash every time** for me.

Based on [this GitHub issue](https://github.com/Koenkk/Z-Stack-firmware/issues/397), it looks like the TI utility is just broken. (If you Google "ti flash-programmer crashes", you'll find lots of results on TI's own forums to that effect.) Some people on that thread indicated that flashing with [ZigStar MultiTool](https://zig-star.com/radio-docs/zigstar-multi-tool/) worked for them, but I didn't have any luck with it.

### What did work: Flashing on Linux

The [cc2538-bsl](https://github.com/JelmerT/cc2538-bsl/blob/master/cc2538-bsl.py) python script finally worked for me. I had to run with sudo (because of some device permissions issues) and `pip install intelhex` first, but it worked. Found it via [this discussion on the HA Community boards](https://community.home-assistant.io/t/sonoffs-zigbee-3-0-usb-dongle-plus-firmware/420558/5).

I think the script will handle restarting the device into bootloader mode for you.

Tijgerd, 2024-12-12:
to get it working in ZHA I asked good'ol chatgpt to make a custom quirk :)
Works like a charm:
[aqara_cube_t1_pro.zip](https://github.com/user-attachments/files/18111574/aqara_cube_t1_pro.zip)

jonfurniss, 2025-01-04:
I was having the same issue as others, Home Assistant was only recognizing Shake and Rotate, and was finally able to solve it.

Cube version: Aqara Cube T1 Pro
Zigbee dongle: Sonoff ZBDongle-E

On the default Sonoff firmware, I tried about 20 different times pressing the reset button various number of times over 5, nothing ever changed.

I finally got around to updating the Sonoff firmware, which was less painful than expected.

I used [version 7.4.3](https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/blob/master/Dongle-E/NCP_7.4.3/README.md)

I flashed the firmware from a Windows 10 PC with [SecureCRT](https://www.vandyke.com/cgi-bin/releases.php?product=securecrt), following [these instructions from Sonoff](https://sonoff.tech/wp-content/uploads/2023/02/SONOFF-Zigbee-3.0-USB-dongle-plus-firmware-flashing.pdf).

The only tricky part with it was needing to hit the boot button on the device, which isn't normally accessible, so I had to remove the 4 screws on the case and take the board out. You then have access to the boot and reset buttons. Flashing the firmware worked on the first try.

When I got back to Home Assistant, 5 quick button presses did not work, despite all of them registering in the ZHA event monitor.

I then super quickly pushed the button 10 times (the monitor seemed to only pick up 2 of those presses, and the light on the cube did not flash each time), and the monitor suddenly started picking up face changes and sliding.

petrusb83, 2025-08-13:
> Hi, does anyone as battery state setup for cube. All other function works
> but unknow on Battery and Firmware.
> I use Sonoff ZBDongle-E version 7.4.4.0

I was just playing with mine last night again. Still no possibility to see
the battery life. My guess is you will need a Aqara Hub in between.