zigpy/zha-device-handlers #1976: [BUG] Negative temperatures wrap around to 6000°C with Tuya Smart Air Detector 6 in 1

gudvinr, 2022-12-01:
**Describe the bug**

[Tuya Smart Air Detector 6 in 1](https://zigbee.blakadder.com/Tuya_DCR-KQG.html) has working temperature range 0-60°C by specs.  
But handling of temps out of specs leads to wacky statistics.

Issue here is that apparently either raw reported values or their source representation uses unsigned 16 bit numbers and negative values written as [2's complement](https://en.wikipedia.org/wiki/Two%27s_complement) signed integer, which makes this numbers in range of ~65500-ish.

This leads to reported temperatures and temperatures displayed in HA to be above 6000°C which by any means doesn't make any sense and makes charts unusable since all other sensors will have temperatures around 0°C at this scale.

However, despite temperatures being in unsupported range, I think it will be more convenient to convert them back from 2's complement to signed form. For example, it could be done like so:
```(((x&0xFFFF)^0x8000) - 0x8000)```

`x` here is 16 bit _integer_ value. Hence if it is, let's say, 65528, then we'll get -8 as the result.  
Which converts to "real" temperature value of -0.8°C instead of current 6552.8°C.

It brings results back to normal range and, despite them not being inside supported value range, still makes them usable.

Actually, values themselves, after being converted, are quite accurate.  
For example, when sensor reports temperature 6549.9°C, it gives -3.7°C after conversion using formula above. And sensor rated for negative temps gives -3.5°C (I added this as 3rd chart for the reference).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to place with negative ambient temps
2. Plug sensor there
3. Connect it to HA
4. See high temperatures

**Expected behavior**
Don't see temperatures outside of normal ambient temperatures range

**Screenshots**
[image]

[image]

[image]

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0x040d",
        "0x042a",
        "0x042b",
        "0x042e",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_dwcarsat",
  "model": "TS0601",
  "class": "ts0601_air_quality.TuyaCO2Sensor2GPP"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
```

</details>

**Additional context**

javicalle, 2022-12-01:
So, if we considerer all the temp measures as 2's complement the device range would be [409.5 ºC, -409.6 ºC], isn't it? 
I think that wouldn't be a problem if all temperatures are 'converted' for that device and would be in inside the manufacturer work range.

I see that you have a local quirk. Do you have already implemented the conversion? 
You can create the PR with the modification, this way we can comment the changes.

gudvinr, 2022-12-01:
> I see that you have a local quirk. Do you have already implemented the conversion?

No, that's quirk for swapping DP for PM2.5 and VOC sensor as described [here](https://github.com/zigpy/zha-device-handlers/pull/1740#issuecomment-1243663998) and also to filter out bad PM2.5 values.  
I maybe add PR for that eventually but I am not sure if what I did there is correct.

I am not quite sure how data mapping works inside ZHA and HA so it'd be quite hard for me to do. From numbers alone it is very clear that at some point these numbers surely were 16 bit integers.  
But inside `ts0601_air_quality.py` temp multiplied by 10. Does this mean that raw value from device arrives as float divided by 10? Or it is something needed for HA and those are actually in some different form?

Shulyaka, 2023-02-04:
@gudvinr Please test the fix and reopen the issue if needed.

gudvinr, 2023-02-04:
In #2113 issue happens to be about uint32 vs int32, and in my case values seem to be uint16 vs int16.

See how their clamps to 400M ( which is 4'294'967'295 divided by 10) vs my 6K (which is 65536 divided by 10)

javicalle, 2023-02-05:
Enable the debug logs from the zha integration and try to get a report from the device.

gudvinr, 2023-02-05:
Adding `zigpy: debug` to logs configuration did the trick:

```[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00E', *payload=69))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 94): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00E', *payload=69))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00E', *payload=69))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00E', *payload=69))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 104): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00E', *payload=69))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00E', *payload=69))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 110): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00E', *payload=69))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00C', *payload=67))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 112): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00C', *payload=67))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00D', *payload=68))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 118): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00D', *payload=68))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00D', *payload=68))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 120): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00D', *payload=68))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00B', *payload=66))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 126): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00B', *payload=66))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00D', *payload=68))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 132): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00D', *payload=68))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00B', *payload=66))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 134): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00B', *payload=66))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00C', *payload=67))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 144): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00C', *payload=67))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00B', *payload=66))]))
[... cut ...]
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00D', *payload=68))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 189): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00D', *payload=68))]))
```

`dp=18` is temp value.

javicalle, 2023-02-05:
Thanks for the logs. I suppose that these are values around 6.8ºC (positive)
Are you able to force negative values? Maybe putting them in the freezer?
If i get some negative raw data (and the real values) I could check against the new code.

gudvinr, 2023-02-05:
Yes, 'tis temperature multiplied by ten which I can double check with sensor located in same area.

Not sure if I can get a good reading from the freezer but I put sensor outside, it'll take some time to cool down but I'll send some logs when it's done.

Shulyaka, 2023-02-05:
May be cover it with some ice?

gudvinr, 2023-02-05:
Good ol' waiting when sun goes down when it's below zero outside does wonders:

<details><summary>logs</summary>

```
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00M', *payload=77))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 169): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00M', *payload=77))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00N', *payload=78))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 171): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00N', *payload=78))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xfa', *payload=65530))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 218): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xfa', *payload=65530))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xfd', *payload=65533))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 228): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xfd', *payload=65533))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xfc', *payload=65532))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 230): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xfc', *payload=65532))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xfc', *payload=65532))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xff', *payload=65535))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 236): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xff', *payload=65535))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xff', *payload=65535))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 238): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xff', *payload=65535))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xfa', *payload=65530))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 244): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xfa', *payload=65530))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xfb', *payload=65531))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 250): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xfb', *payload=65531))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xf8', *payload=65528))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 252): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xf8', *payload=65528))]))
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xf6', *payload=65526))]) [... cut ...]
[0xA5F5:1:0xef00] Decoded ZCL frame: TuyaCO2ManufCluster2:set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xef', *payload=65519))]))
[0xA5F5:1:0xef00] Received command 0x02 (TSN 80): set_data_response(data=TuyaCommand(status=0, tsn=0, datapoints=[TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\xff\xef', *payload=65519))]))
```
</details>

Outdoor temperature should be around -3°C to -4°C

Shulyaka, 2023-02-05:
Does it look like the device sends a 32-bit value, but only 16 bit are set?

gudvinr, 2023-02-05:
Looks like it. But operation I posted right at the beginning takes care of that:  
```
>>> x = int.from_bytes(b'\x00\x00\xff\xef', byteorder='big')
>>> (((x&0xFFFF)^0x8000) - 0x8000)
-17
```

PR #2175 (2023-02-04): tuya: fix negative values for some devices
When using tuya datapoint of type `TuyaDPType.VALUE` we are currently assuming it to be of unsigned type. This seems to be an incorrect assumption.
This PR interpret them as a signed type.
Probably fixes #1976 and the negative temperature issue from #2113
Fixes https://github.com/zigpy/zha-device-handlers/issues/2113
PR #2188 (2023-03-28): Fix Tuya signed 16 bit temperature values
Another approach to fix the 16 bits reports from some devices.
Supersed #2180
Fixes: #1976
(And will not break any sensor while temp < 4095 ºC)
closed 2023-03-28 by TheJulianJES