zigpy/zha-device-handlers #1979: [BUG] Aqara C1 Pet Feeder Not Showing Up Correctly After Added

wraith1385, 2022-12-01:
**Describe the bug**
Added the Aqara C1 Pet feeder after the latest HA Core Release.  It only has the features in the screenshot.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to ZHA Devices in HA
2. Click on 'Add Device'
3. Hold down reset button on Aqara C1 for 5 seconds.  
4. Wait for device to be recognized and added.

**Expected behavior**
Expect all features shown in https://github.com/home-assistant/core/pull/82340

**Screenshots**
[image]

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0002",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x000a",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Aqara",
  "model": "aqara.feeder.acn001",
  "class": "zhaquirks.xiaomi.aqara.feeder_acn001.AqaraFeederAcn001"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 5471,
    "manufacturer": "Aqara",
    "model": "aqara.feeder.acn001",
    "name": "Aqara aqara.feeder.acn001",
    "quirk_applied": true,
    "quirk_class": "zhaquirks.xiaomi.aqara.feeder_acn001.AqaraFeederAcn001",
    "manufacturer_code": 4447,
    "power_source": "Mains",
    "lqi": 72,
    "rssi": null,
    "last_seen": "2022-12-01T15:54:27",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0002",
          "in_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x000a",
            "0xfcc0"
          ],
          "out_clusters": [
            "0x0003",
            "0x0019"
          ]
        },
        "242": {
          "profile_id": 41440,
          "device_type": "0x0061",
          "in_clusters": [],
          "out_clusters": [
            "0x0021"
          ]
        }
      }
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "button.aqara_aqara_feeder_acn001_identify",
        "name": "Aqara aqara.feeder.acn001"
      }
    ],
    "neighbors": [],
    "endpoint_names": [
      {
        "name": "ON_OFF_OUTPUT"
      },
      {
        "name": "unknown 97 device_type of 0xa1e0 profile id"
      }
    ],
    "user_given_name": null,
    "device_reg_id": "dbbd0b0f31c86d817dc34c2fa4884357",
    "area_id": "143cd80a4f4f488ab5f2bc9591e1c9ba",
    "cluster_details": {
      "1": {
        "device_type": {
          "name": "ON_OFF_OUTPUT",
          "id": 2
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0001": {
                "attribute_name": "app_version",
                "value": 33
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "aqara.feeder.acn001"
              }
            },
            "unsupported_attributes": {}
          },
          "0x0003": {
            "endpoint_attribute": "identify",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0004": {
            "endpoint_attribute": "groups",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0005": {
            "endpoint_attribute": "scenes",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xfcc0": {
            "endpoint_attribute": "opple_cluster",
            "attributes": {
              "0x138b": {
                "attribute_name": "portions_dispensed",
                "value": 0
              },
              "0x138c": {
                "attribute_name": "weight_dispensed",
                "value": 0
              },
              "0x138d": {
                "attribute_name": "error_detected",
                "value": false
              },
              "0x138e": {
                "attribute_name": "disable_led_indicator",
                "value": 0
              },
              "0x138f": {
                "attribute_name": "child_lock",
                "value": 0
              },
              "0x1390": {
                "attribute_name": "feeding_mode",
                "value": 0
              },
              "0x1391": {
                "attribute_name": "serving_size",
                "value": 1
              },
              "0x1392": {
                "attribute_name": "portion_weight",
                "value": 8
              },
              "0xfff1": {
                "attribute_name": "feeder_attr",
                "value": {
                  "__type": "<class 'zigpy.types.basic.LVBytes'>",
                  "repr": "b'\\x00\\x05\\x12\\x08\\x00\\x08\\xc8 7F09000100,7F0D000100,7F13000100'"
                }
              }
            },
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0003": {
            "endpoint_attribute": "identify",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      },
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Coordinator Information:
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 0,
    "manufacturer": "Texas Instruments",
    "model": "CC1352/CC2652, Z-Stack 3.30+ (build 20210708)",
    "name": "Texas Instruments CC1352/CC2652, Z-Stack 3.30+ (build 20210708)",
    "quirk_applied": false,
    "quirk_class": "zigpy_znp.zigbee.device.ZNPCoordinator",
    "manufacturer_code": 0,
    "power_source": "Mains",
    "lqi": null,
    "rssi": null,
    "last_seen": "2022-12-01T14:48:33",
    "available": true,
    "device_type": "Coordinator",

wraith1385, 2022-12-01:
Tagging @TheJulianJES

tube0013, 2022-12-01:
Are you on 2022.12 beta? that's where it should be supported

wraith1385, 2022-12-01:
No, I'm on the latest 2022.11 Core release.  I looked at the release details and it's mentioned when you dig down into the ZHA details.

TheJulianJES, 2022-12-01:
The quirk is applied correctly but your Z-Stack firmware seems outdated and probably doesn't send Xiaomi manufacturing codes during pairing.
You'll need to update the firmware and then re-pair the feeder.

wraith1385, 2022-12-01:
Ah, ok.  So I need to update the coordinator firmware and either get on the 2022.12 beta or wait for the release.

Apologies for misunderstanding the release notes.

tube0013, 2022-12-01:
you may want to update the esp32/esphome fw on the coord too before trying a coordinator fw update. there are new beta images in my repo which do much better with the coordinator FW updates over network serial.  <https://github.com/tube0013/tube_gateways>

closed 2022-12-01 by wraith1385