zigpy/zha-device-handlers #1980: [Device Support Request] Livarno / Lidl ceiling lamp with module TS0504B

J3ru, 2022-12-01:
I can’t set any color temperature, the RGB works, but quite dim, when I set the “normal” white color it can go very bright. But when using the normal white light and trying to set color temperature, it does nothing.

I'd like to be able to control the white light temperature

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_sroezl0s",
  "model": "TS0504B",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.11.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "5.15.74",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "vm",
[... 402 lines cut ...]
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

RafaelSchridi, 2022-12-02:
Same problems here also using a ZBDongle-E

<details>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_sroezl0s",
  "model": "TS0504B",
  "class": "zigpy.device.Device"
}
```

</details>

<details>
<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Unsupported Third Party Container",
    "version": "2022.11.4",
    "dev": false,
    "hassio": false,
    "virtualenv": true,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "5.15.74-1-lts",
    "run_as_root": false
  },
  "custom_components": {
    "basestation": {
      "version": "0.0.1",
      "requirements": []
    },
    "localtuya": {
      "version": "4.1.1",
[... 388 lines cut ...]
            "unsupported_attributes": {}
          }
        }
      },
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

javicalle, 2022-12-09:
That is my proposal for yours devices:

<details open>
<summary>rgbcct.py</summary>

```python
class DimmableLamp(CustomDevice):
    """Lidl dimmable lamp."""

    signature = {
        MODEL: "TS0504B",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=257
            # input_clusters=[0, 3, 4, 5, 6, 8, 768, 4096]
            # output_clusters=[10, 25]>
            1: {
                # "profile_id": 260,
                # "device_type": "0x010d",
                # "in_clusters": ["0x0000","0x0003","0x0004","0x0005","0x0006","0x0008","0x0300","0x1000","0xef00"],
                # "out_clusters": ["0x000a","0x0019"]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
                    TuyaNewManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # "profile_id": 41440,
                # "device_type": "0x0061",
                # "in_clusters": [],
                # "out_clusters": ["0x0021"]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    LidlRGBCCTColorCluster,
                    LightLink.cluster_id,
                    TuyaNewManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</details>

You will need to import the `TuyaNewManufCluster`:
```python
from zhaquirks.tuya import TuyaNewManufCluster
```

Configure your local quirk folder.
Copy the current `rgbcct.py` file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/lidl/rgbcct.py
Add the proposed quirk and the import to the file.
Save changes, remove the device from HA, restart HA and pair the device again.

J3ru, 2022-12-09:
> That is my proposal for yours devices:
> 
> rgbcct.py
> You will need to import the `TuyaNewManufCluster`:
> 
> ```python
> from zhaquirks.tuya import TuyaNewManufCluster
> ```
> 
> Configure your local quirk folder. Copy the current `rgbcct.py` file:
> 
> * https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/lidl/rgbcct.py
>   Add the proposed quirk and the import to the file.
>   Save changes, remove the device from HA, restart HA and pair the device again.

Thanks so much, will try this ASAP, can you elaborate on the importing TuyaNewManufCluster not too sure how to do that.
Local quirck I think I found some documentation on. Sorry still new to all this.

javicalle, 2022-12-09:
There is a guide about enabling custom quirks:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

The proposed quirk will go at the bottom of the file, but the import statement must be at the top (with all the others statements).
The upper part of the file must see like this (with the new import statement last):
```python
"""Quirk for LIDL RGB+CCT bulb."""
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.lighting import Color
from zigpy.zcl.clusters.lightlink import LightLink

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaNewManufCluster
```

J3ru, 2022-12-09:
> There is a guide about enabling custom quirks:
> 
> * [Editing (ZHA) quirks in a (HA) container or local. #693 (comment)](https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274)
> 
> The proposed quirk will go at the bottom of the file, but the import statement must be at the top (with all the others statements). The upper part of the file must see like this (with the new import statement last):
> 
> ```python
> """Quirk for LIDL RGB+CCT bulb."""
> from zigpy.profiles import zha
> from zigpy.quirks import CustomCluster, CustomDevice
> from zigpy.zcl.clusters.general import (
>     Basic,
>     GreenPowerProxy,
>     Groups,
>     Identify,
>     LevelControl,
>     OnOff,
>     Ota,
>     Scenes,
>     Time,
> )
> from zigpy.zcl.clusters.lighting import Color
> from zigpy.zcl.clusters.lightlink import LightLink
> 
> from zhaquirks.const import (
>     DEVICE_TYPE,
>     ENDPOINTS,
>     INPUT_CLUSTERS,
>     MODELS_INFO,
>     OUTPUT_CLUSTERS,
>     PROFILE_ID,
> )
> from zhaquirks.tuya import TuyaNewManufCluster
> ```

I might be doing something wrong, but getting this after restart:
Error setting up entry SONOFF Zigbee 3.0 USB Dongle Plus V2 for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 372, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 100, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/zha_quirks/rgbcct.py", line 101, in <module>
    class DimmableLamp(CustomDevice):
  File "/config/zha_quirks/rgbcct.py", line 105, in DimmableLamp
    MODEL: "TS0504B",
NameError: name 'MODEL' is not defined

javicalle, 2022-12-09:
I have updated my previous comment. There is a new `MODEL` import in the middle of the `from zhaquirks.const import`.
Update the quirk and repeat.

J3ru, 2022-12-10:
> That is my proposal for yours devices:
> 
> rgbcct.py
> ```python
> class DimmableLamp(CustomDevice):
>     """Lidl dimmable lamp."""
> 
>     signature = {
>         MODEL: "TS0504B",
>         ENDPOINTS: {
>             # <SimpleDescriptor endpoint=1 profile=260 device_type=257
>             # input_clusters=[0, 3, 4, 5, 6, 8, 768, 4096]
>             # output_clusters=[10, 25]>
>             1: {
>                 # "profile_id": 260,
>                 # "device_type": "0x010d",
>                 # "in_clusters": ["0x0000","0x0003","0x0004","0x0005","0x0006","0x0008","0x0300","0x1000","0xef00"],
>                 # "out_clusters": ["0x000a","0x0019"]
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,
>                     Identify.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     OnOff.cluster_id,
>                     LevelControl.cluster_id,
>                     Color.cluster_id,
>                     LightLink.cluster_id,
>                     TuyaNewManufCluster.cluster_id,
>                 ],
>                 OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
>             },
>             242: {
>                 # "profile_id": 41440,
>                 # "device_type": "0x0061",
>                 # "in_clusters": [],
>                 # "out_clusters": ["0x0021"]
>                 PROFILE_ID: 41440,
>                 DEVICE_TYPE: 97,
>                 INPUT_CLUSTERS: [],
>                 OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
>             },
>         },
>     }
>     replacement = {
>         ENDPOINTS: {
>             1: {
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,
>                     Identify.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     OnOff.cluster_id,
>                     LevelControl.cluster_id,
>                     LidlRGBCCTColorCluster,
>                     LightLink.cluster_id,
>                     TuyaNewManufCluster,
>                 ],
>                 OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
>             },
>             242: {
>                 PROFILE_ID: 41440,
>                 DEVICE_TYPE: 97,
>                 INPUT_CLUSTERS: [],
>                 OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
>             },
>         },
>     }
> ```
> 
> You will need to import the `TuyaNewManufCluster` and the `MODEL`:
> 
> ```python
> from zhaquirks.const import (
>     DEVICE_TYPE,
>     ENDPOINTS,
>     INPUT_CLUSTERS,
>     MODEL,
>     MODELS_INFO,
>     OUTPUT_CLUSTERS,
>     PROFILE_ID,
> )
> from zhaquirks.tuya import TuyaNewManufCluster
> ```
> 
> Configure your local quirk folder. Copy the current `rgbcct.py` file:
> 
> * https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/lidl/rgbcct.py
> 
> Add the proposed quirk and the import to the file. Save changes, remove the device from HA, restart HA and pair the device again.

ZHA is now working again, however it hasn't fixed the issue.

javicalle, 2022-12-11:
Can you attach the device diagnostic information?

J3ru, 2022-12-11:
> Can you attach the device diagnostic information?

'''yaml

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.12.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "5.15.74",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "localtuya": {
      "version": "4.1.1",
      "requirements": []
    },
    "hacs": {
      "version": "1.28.3",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.5",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.88",
      "zigpy-deconz==0.19.2",
      "zigpy==0.52.3",
      "zigpy-xbee==0.16.2",
      "zigpy-zigate==0.10.3",
      "zigpy-znp==0.9.2"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "1A86",
        "pid": "55D4",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus v2"
        ]
      },
      {
        "vid": "10C4",
[... 331 lines cut ...]
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      },
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

'''

J3ru, 2022-12-23:
> Can you attach the device diagnostic information?

Do you need any additional information?

Yakie996, 2022-12-27:
I got the same problem with the lidl light, I cant even turn it off in HA i'm using the Skyconnect stick.
[image]

javicalle, 2022-12-27:
@J3ru your device is loading the quirk but seems not being working as expected.
@Yakie996 have you enabled and configured the local quirk for your device?

To go any forward we need the device logs.
Enable the debug logs and look for the logs and attach the relevant info when you operate the device:
* https://www.home-assistant.io/integrations/zha/#debug-logging

Yakie996, 2022-12-30:
> @J3ru your device is loading the quirk but seems not being working as expected. @Yakie996 have you enabled and configured the local quirk for your device?
>
> To go any forward we need the device logs. [...]

Hi,

I did this 
```
zha:
  custom_quirks_path: /config/custom_zha_quirks/
Making one folder with the name custom_zha_quirks in you HA config directory.
```
But i don't understand which part of your code I need to put in the quirk? 

Sorry this is all new to me.

javicalle, 2022-12-30:
Is all explained here:
https://github.com/zigpy/zha-device-handlers/issues/1980#issuecomment-1344659534

Before you do anything, can you post your device diagnostic info to be sure that is the same device?

J3ru, 2022-12-30:
> @J3ru your device is loading the quirk but seems not being working as expected. @Yakie996 have you enabled and configured the local quirk for your device?
>
> To go any forward we need the device logs. [...]
I hope this gives some relevanty information.

`2022-12-30 12:34:29.271 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=light, service=turn_on, service_data=entity_id=light.lamp_overloop_zolder_light, color_temp_kelvin=2000>
2022-12-30 12:34:29.273 DEBUG (MainThread) [homeassistant.components.zha.entity] light.lamp_overloop_zolder_light: setting transitioning flag to True
2022-12-30 12:34:29.274 DEBUG (MainThread) [zigpy.zcl] [0x15F2:1:0x0006] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=13, command_id=1, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-12-30 12:34:29.275 DEBUG (MainThread) [zigpy.zcl] [0x15F2:1:0x0006] Sending request: on()
2022-12-30 12:34:29.275 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x15F2), dst_ep=1, source_route=None, extended_timeout=False, tsn=13, profile_id=260, cluster_id=6, data=Serialized[b'\x01\r\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-12-30 12:34:29.276 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x15f2, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=13), 14, b'\x01\r\x01')
2022-12-30 12:34:29.297 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'0062'
2022-12-30 12:34:29.309 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'00f21504010600010100010000620e0000'
2022-12-30 12:34:29.310 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 5618, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=98), 14, <EmberStatus.SUCCESS: 0>, b'']
2022-12-30 12:34:29.330 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'0004010600010140010000b0dcd3f215ffff05080d0b010002'
2022-12-30 12:34:29.330 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=176), 220, -45, 0x15f2, 255, 255, b'\x08\r\x0b\x01\x00']
2022-12-30 12:34:29.330 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x15F2), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=176, profile_id=260, cluster_id=6, data=Serialized[b'\x08\r\x0b\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=220, rssi=-45)
2022-12-30 12:34:29.331 DEBUG (MainThread) [zigpy.zcl] [0x15F2:1:0x0006] Received ZCL frame: b'\x08\r\x0b\x01\x00'
2022-12-30 12:34:29.331 DEBUG (MainThread) [zigpy.zcl] [0x15F2:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=13, command_id=11, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-30 12:34:29.331 DEBUG (MainThread) [zigpy.zcl] [0x15F2:1:0x0006] Decoded ZCL frame: OnOff:Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
2022-12-30 12:34:29.335 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x15F2:1:0x0006]: executed 'on' command with args: '()' kwargs: '{}' result: Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
2022-12-30 12:34:29.335 DEBUG (MainThread) [zigpy.zcl] [0x15F2:1:0x0300] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=15, command_id=10, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-12-30 12:34:29.335 DEBUG (MainThread) [zigpy.zcl] [0x15F2:1:0x0300] Sending request: move_to_color_temp(color_temp_mireds=500, transition_time=0, options_mask=None, options_override=None)
2022-12-30 12:34:29.336 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x15F2), dst_ep=1, source_route=None, extended_timeout=False, tsn=15, profile_id=26 [... cut ...] evation=14.7, azimuth=179.13, rising=True, friendly_name=Sun @ 2022-12-30T12:31:15.724361+01:00>, new_state=<state sun.sun=above_horizon; next_dawn=2022-12-31T07:00:21.372603+00:00, next_dusk=2022-12-30T16:09:39.664388+00:00, next_midnight=2022-12-30T23:35:23+00:00, next_noon=2022-12-31T11:35:09+00:00, next_rising=2022-12-31T07:42:03.981151+00:00, next_setting=2022-12-30T15:27:52.534516+00:00, elevation=14.71, azimuth=179.94, rising=False, friendly_name=Sun @ 2022-12-30T12:31:15.724361+01:00>>
`

RafaelSchridi, 2022-12-30:
I was finally able to test the quirk but unfortunately it doesn't work.

Device info does show the quirk in use:
``` 
Quirk: rgbcct.DimmableLamp
```

The only thing related to ZHA in the logs is 

```
Logger: zigpy.device
Source: runner.py:120
First occurred: 18:32:44 (4 occurrences)
Last logged: 18:32:44

Failed to parse message (b'8184018b') on cluster 32799, because Data is too short to contain 1 bytes
``` 
and 
```
Logger: homeassistant.components.zha.core.channels.base
Source: components/zha/core/channels/base.py:486
Integration: Zigbee Home Automation (documentation, issues)
First occurred: 18:32:39 (26 occurrences)
Last logged: 18:32:45

[0xB1F0:1:0x0300]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0xCCB0:2:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0xB1F0:1:0x0008]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0xCCB0:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0x3476:3:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
```

RafaelSchridi, 2022-12-30:
I've been looking around, and I don't think this device can change colour temperature. Here is an issues discussing the device with Z2M: https://github.com/Koenkk/zigbee2mqtt/issues/14439

Which would make it a really odd device, I've never seen a light that does RGB but not colour temp. 
Is it even possible to disable colour temp control in HA and still have RGB control?

javicalle, 2022-12-30:
It seems that device 'just' allow to turn on and off the white leds when color temp called.
In Z2M seems that are keeping the control to allow enable the white leds. So if the temp color control is disabled maybe you could not turn on the white leds.

RafaelSchridi, 2022-12-30:
Ah that is an annoying requirement, is it possible to just limit the colour range to 4000k-4000k?

javicalle, 2022-12-30:
> Ah that is an annoying requirement, is it possible to just limit the colour range to 4000k-4000k?

Ummm, maybe with:
```python

class Lidl4000KColorCluster(CustomCluster, Color):
    """Lidl RGB+4000K Lighting custom cluster."""

    # Set correct capabilities to ct, xy, hs
    # LIDL bulbs do not correctly report this attribute (comes back as None in Home Assistant)
    _CONSTANT_ATTRIBUTES = {
        0x400A: 0b11001,
        0x400B: 306,  # color_temp_physical_min
        0x400C: 306,  # color_temp_physical_max
    }
```

Add that class in file (upper than the device definition). Then use the new class in the replacement part for your device:
```python
    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Lidl4000KColorCluster,
                    LightLink.cluster_id,
                    TuyaNewManufCluster,
                ],
```

Save changes, restart HA. Not sure if re-pair the device will be needed. Check the attribute values in the device diagnostic or manage cluster form.

MattWestb, 2022-12-30:
Off topic but curious:
The IKEA first gen CWS (or its pre first gen then its having one not IKEA standard Zigbee module) was RGB + W and was emulating TC and is working OK but not good then its so bright in all modes.

J3ru, 2022-12-30:
> I've been looking around, and I don't think this device can change colour temperature. Here is an issues discussing the device with Z2M: [Koenkk/zigbee2mqtt#14439](https://github.com/Koenkk/zigbee2mqtt/issues/14439)
> 
> Which would make it a really odd device, I've never seen a light that does RGB but not colour temp. Is it even possible to disable colour temp control in HA and still have RGB control?
> 
> EDIT: if you look on the box it has ranges for RGB and colour temp but on the colour temp range there is an arrow that indicated it only does 4000k. Man that is some dirty marketing and has a lot of people confused it seems.

So chances are that its locked at 4000k, but what I then don't get is why are other colors way more dimm than the "normal" white light?
And yes they totally got me with this, I mean it can do RGB but not variants of white which is I dare say the standard?

RafaelSchridi, 2022-12-31:
> So chances are that its locked at 4000k, but what I then don't get is why are other colors way more dimm than the "normal" white light? And yes they totally got me with this, I mean it can do RGB [...]

They're probably separate led strips, get some highpower single color white strips so they can put 3000 lumens on the box and then seperate RGB so people think they can do both at the same time.

closed 2023-07-10 by github-actions[bot]