zigpy/zha-device-handlers #1981: [BUG] TS0003 lights effect each other

admons, 2022-12-02:
**Describe the bug**
I have a Tuya 3 gang with neutral light switch (TS0003 _TZ3000_odzoiovu) connect to zha. When I turn on/off 1 light, it effect all other lights too.

**Expected behavior**
Every light should be independant

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_odzoiovu",
  "model": "TS0003",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

javicalle, 2022-12-04:
The quirk from https://github.com/zigpy/zha-device-handlers/pull/1983/files would fix the issue.
It will be helpful if you can validate it against your device.

Thanks in advance.

admons, 2022-12-04:
Thanks for the help @javicalle ,
I tried it and it didn't solved the problem with my module, although the description is equal to the one I have.
I tried 2 different (identical) modules, so the problem is not with the module itself.

javicalle, 2022-12-04:
Have you tried to remove the device from HA and paired it again?
Is the device loading the quirk? You can check it from the device's view.

admons, 2022-12-04:
The device is loading the quirk. Verified it

admons, 2022-12-06:
@dmulcahey This bug still exists, I only said that HA is loading the quirk.

javicalle, 2022-12-06:
The issue has been closed because I marked as related to the #1983 PR.
I reopened it now.

javicalle, 2022-12-07:
No metering issue here #1968

javicalle, 2022-12-07:
@admons Have you tried to remove the device from HA and pair it again?

admons, 2022-12-08:
Yes I have but it didn't work

javicalle, 2022-12-08:
Can you enable the debug logs and attach the relevant info?
https://www.home-assistant.io/integrations/zha/#debug-logging

It will be helpful the logs from the HA interaction and the ones from the physical device.

admons, 2022-12-09:
As I had 1MB of logs after few seconds from when I restarted HA, I'm attaching only the diff between before I turned on the switch and after

```
2022-12-09 21:10:54.100 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=light, service=toggle, service_data=entity_id=light.tz3000_odzoiovu_ts0003_light_4>
2022-12-09 21:10:54.107 DEBUG (MainThread) [zigpy.zcl] [0x8A1B:1:0x0006] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=150, command_id=1, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-12-09 21:10:54.109 DEBUG (MainThread) [zigpy.zcl] [0x8A1B:1:0x0006] Sending request: on()
2022-12-09 21:10:54.109 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x8A1B), dst_ep=1, source_route=None, extended_timeout=False, tsn=150, profile_id=260, cluster_id=6, data=Serialized[b'\x01\x96\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-12-09 21:10:54.111 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x8a1b, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=150), 151, b'\x01\x96\x01')
2022-12-09 21:10:54.123 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'008b'
2022-12-09 21:10:54.136 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'001b8a040106000101000100008b970000'
2022-12-09 21:10:54.137 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 35355, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=139), 151, <EmberStatus.SUCCESS: 0>, b'']
2022-12-09 21:10:54.178 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'000401060001010001000018ffdc1b8affff0518960b0100'
2022-12-09 21:10:54.181 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=24), 255, -36, 0x8a1b, 255, 255, b'\x18\x96\x0b\x01\x00']
2022-12-09 21:10:54.182 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x8A1B), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=24, profile_id=260, cluster_id=6, data=Serialized[b'\x18\x96\x0b\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-36)
2022-12-09 21:10:54.183 DEBUG (MainThread) [zigpy.zcl] [0x8A1B:1:0x0006] Received ZCL frame: b'\x18\x96\x0b\x01\x00'
2022-12-09 21:10:54.189 DEBUG (MainThread) [zigpy.zcl] [0x8A1B:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=150, command_id=11, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-09 21:10:54.190 DEBUG (MainThread) [zigpy.zcl] [0x8A1B:1:0x0006] Decoded ZCL frame: TuyaZBOnOffAttributeCluster:Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
2022-12-09 21:10:54.194 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'000401060001010001000019ffdc1b8affff0718ab0a00001001'
2022-12-09 21:10:54.196 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=25), 255, -36, 0x8a1b, 255, 255, b'\x18\xab\n\x00\x00\x10\x01']
2022-12-09 21:10:54.196 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x8A1B), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=25, profile_id=260, cluster_id=6, data=Serialized[b'\x18\xab\n\x00\x00\x10\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-36)
2022-12-09 21:10:54.197 DEBUG (MainThread) [zigpy.zcl] [0x8A1B:1:0x0006] Received ZCL frame: b'\x18\xab\n\x00\x00\x10\x01'
2022-12-09 21:10:54.198 DEBUG (MainThread) [zigpy.zcl] [0x8A1B:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_respons [... cut ...] K_KEY_NOT_AUTHORIZED = 0, NWK_DECRYPTION_FAILURE = 0, APS_DECRYPTION_FAILURE = 0, ALLOCATE_PACKET_BUFFER_FAILURE = 0, RELAYED_UNICAST = 0, PHY_TO_MAC_QUEUE_LIMIT_REACHED = 0, PACKET_VALIDATE_LIBRARY_DROPPED_COUNT = 0, TYPE_NWK_RETRY_OVERFLOW = 0, PHY_CCA_FAIL_COUNT = 0, BROADCAST_TABLE_FULL = 0, PTA_LO_PRI_REQUESTED = 0, PTA_HI_PRI_REQUESTED = 0, PTA_LO_PRI_DENIED = 0, PTA_HI_PRI_DENIED = 0, PTA_LO_PRI_TX_ABORTED = 0, PTA_HI_PRI_TX_ABORTED = 0, ADDRESS_CONFLICT_SENT = 0, EZSP_FREE_BUFFERS = 249]

javicalle, 2022-12-09:
Ok, that logs are showing reports from the physical device. They come from 3 diferent endpoints/switches.
But all reports are `On` commands:
```
Attribute report received: on_off=<Bool.true: 1>
```

Is expected that the `Off` command also reports its status like:
```
Attribute report received: on_off=<Bool.false: 0>
```
Can you check and confirm?

Can you also attach the logs from the HA operation? I'm looking for some diferentce between both.

SpongyPMC, 2023-02-02:
Hi all, 

Did this ever get solved? I'm having the same issue with the TS0013 3 gang device where all lights switch on or off when I flick one on or off via HA. They all work fine when physically pushed on the switch but not via the app.

Would love some help to get it work as expected!

MattWestb, 2023-02-02:
Then you is having one TS0013 its not the same device family is this issue.
Can you posting `Device signature` and `Diagnostic information` from the device so can see how its looks ?

SpongyPMC, 2023-02-02:
Here's the signature...

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_qewo8dlz",
  "model": "TS0013",
  "class": "zigpy.device.Device"
}

And I've attached the diagnostics.

This is all very new to me so if I haven't given you it in the right format then let me know!

Appreciate your help!

[zha-9f1ac0f5c53d56ed8a2efc4f23c6efee-_TZ3000_qewo8dlz TS0013-0eb4fb98cc0455a845e82517cca05885.json (1).txt](https://github.com/zigpy/zha-device-handlers/files/10572775/zha-9f1ac0f5c53d56ed8a2efc4f23c6efee-_TZ3000_qewo8dlz.TS0013-0eb4fb98cc0455a845e82517cca05885.json.1.txt)

MattWestb, 2023-02-02:
Your device is very likely added in https://github.com/zigpy/zha-device-handlers/pull/2091 and you need updating your HA 2023.2  then you i not on that version or installing the quirk local (not recommended).

MattWestb, 2023-02-02:
I can guarantee 110% then i was not looking on every line but its looks its one of the added in zhaquirk ver 0.0.91 and you is running 0.0.90.

And you must doing the repairing or is the device not being feed and switching all gang then getting commands from ZHA.

ikreis, 2023-02-17:
> Have you tried to remove the device from HA and paired it again? Is the device loading the quirk? You can check it from the device's view.

I just did, and it fixed my problem. Looks like a recent HA update dealt with the bug. I only installed the switch 3 weeks ago.

javicalle, 2023-02-18:
@admons is your issue fixed? Can we close the ticket?

PR #1983 (2022-12-04): Add the TS0003 and TS0004 metering devices
And here comes the TS0003 and TS0004 devices with the metering cluster.
Fixes: #1981
Fixes: #1982
commit 2323715d (2022-12-04): Add the TS0003 and TS0004 metering devices (#1983)

And here comes the TS0003 and TS0004 devices with the metering cluster.
Fixes: #1981
Fixes: #1982
closed 2023-08-24 by github-actions[bot]