zigpy/zha-device-handlers #1982: [BUG] TS0004_TZ3000_ltt60asa lights effect each other

pervolianinen, 2022-12-03:
**Describe the bug**
I have a Tuya 4 gang with neutral light switch (TS0004_TZ3000_ltt60asa) connect to zha. When I turn on/off 1 light, it effect all other lights too.

**Expected behavior**
Every light should be independant

<details>

<summary>Device signature</summary>

```{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
[... 749 lines cut ...]
        },
        "out_clusters": {}
      },
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

**Additional context**
Add any other context about the problem here.

javicalle, 2022-12-04:
The quirk from https://github.com/zigpy/zha-device-handlers/pull/1983/files would fix the issue.
It will be helpful if you can validate it against your device.

Thanks in advance.

admons, 2022-12-04:
Thanks for the help @javicalle ,
I tried it and it didn't solved the problem with my module, although the description is equal to the one I have.
I tried 2 different (identical) modules, so the problem is not with the module itself.

PR #1983 (2022-12-04): Add the TS0003 and TS0004 metering devices
And here comes the TS0003 and TS0004 devices with the metering cluster.
Fixes: #1981
Fixes: #1982
commit 2323715d (2022-12-04): Add the TS0003 and TS0004 metering devices (#1983)

And here comes the TS0003 and TS0004 devices with the metering cluster.
Fixes: #1981
Fixes: #1982
closed 2022-12-04 by dmulcahey