zigpy/zha-device-handlers #1992: [Device Support Request] TS011F _TZ3000_rdfh8cfs

ChristophCaina, 2022-12-07:
**Is your feature request related to a problem? Please describe.**
After setting up a fresh Zigbee network I've seen the following warnings in the Logs of my HomeAssistant installation:

```
Logger: zigpy.zcl
Source: runner.py:119
First occurred: 00:51:30 (20 occurrences)
Last logged: 00:51:48

[0xA5B5:1:0xe001] Unknown cluster command 225 b'\x02\xef\x91"+9\x00\x00\x00\x07\x00\x00\x00\x00'
[0xE3C2:1:0xe001] Unknown cluster command 225 b'\x02\xed\x91"+\x15\x00\x00\x00\x00\x00\x00\x00\x00'
```

The devices are TS011F PPower Plugs
[image]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

<details>

<summary>Device signature</summary>

```yaml
"data": {
    "ieee": "**REDACTED**",
    "nwk": 58306,
    "manufacturer": "_TZ3000_rdfh8cfs",
    "model": "TS011F",
    "name": "_TZ3000_rdfh8cfs TS011F",
    "quirk_applied": true,
    "quirk_class": "zhaquirks.tuya.ts011f_plug.Plug",
    "manufacturer_code": 4417,
    "power_source": "Mains",
    "lqi": 48,
    "rssi": null,
    "last_seen": "2022-12-07T18:02:45",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x010a",
          "in_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x0702",
            "0x0b04",
            "0xe000",
            "0xe001"
          ],
          "out_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      }
    },
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

MattWestb, 2022-12-07:
The cluster we have implanted but only some attributes is in it:
https://github.com/zigpy/zha-device-handlers/blob/792d7e2ed0c40e16c7a58c9854fd0062649b6cc8/zhaquirks/tuya/__init__.py#L999.

Its remember of https://github.com/zigpy/zha-device-handlers/blob/792d7e2ed0c40e16c7a58c9854fd0062649b6cc8/zhaquirks/tuya/__init__.py#L895 commands that tuya is sending.

I can only speculating that tuya is sending some extra commands that  we is not knowing.
Can it being over voltage or current alarms ???

If you is "decoding" the revived data is it making some senses ??

Its also remember of tuya MCU commands but i dont knowing them so well.

ChristophCaina, 2022-12-08:
What additional information can / should I provide from my side?

javicalle, 2022-12-08:
> If you is "decoding" the revived data is it making some senses ??

I haven't had any luck trying to decode the data 🤷🏻‍♂️

> What additional information can / should I provide from my side?

Can you attach the full error from your logs?
Any clue if the exception is related to some device operation?

ChristophCaina, 2022-12-08:
I've tried to decode the data - but I think, this `\x91"+9` and `\x91"+` will be an issue with the decoding...
And maybe, this is why the warning was thrown?

Unfortunately, I don't have the full logs anymore... and so far the error did not come up again...

I'll try to see, if I can somehow reporduce it..
The device / Integration in HomeAssistant does provide the following information and operations:

- switch (On / Off)
- Active Power ( W )
- Power Factor
- RMS current ( A )
- RMS Voltage ( V )
- Summation delivered ( kWh )
- Backlight Mode:
  - off
  - LighthWhenOn
  - LightWhenOff
- child lock (On / Off)
- Power On state:
  - LastState
  - On
  - Off
 - Identify
 - LQI
 - RSSI

The only "issue" I can see in my setup is, that the Power factor is "unknown"
[image]

javicalle, 2022-12-08:
Related to https://github.com/zigpy/zigpy/issues/1121

MattWestb, 2022-12-12:
@ChristophCaina I think its the best option for getting one clue of the data is being used.
Also if you is having hardware for sniffing the Zigbee network is being one large help.

ChristophCaina, 2022-12-12:
ok, I will do this later today.
Can you recommend hardware / tools to sniff the Zigbee Network?

MattWestb, 2022-12-12:
If you have one spare EZSP (have sniffing mode in standard firmware) or TI CC-2531 (need sniff firmware) you can sniffing Zigbee easy. Good instruction in Z2M Wiki / Docks.

ChristophCaina, 2022-12-12:
hm... I just have my default sonoff Zigbee USB Dongle...
Maybe, I can use this to install sniffing functioniallity - as long as I don't loose any of the main functionalities :)
I think, it shoudl work with an TI CC-2531 (?)

MattWestb, 2022-12-12:
If the Sonoff is one EZSP version you can using it for sniffing without changing firmware.
But you cant sniffing and using it with ZHA at the same time then its different working mode its using.

IT CC-2531 is working but need sniffing firmware being flashed on it.

closed 2023-07-10 by github-actions[bot]