zigpy/zha-device-handlers #1995: [Device Support Request] Heiman CO_CTPG Carbon Monoxide sensor

pfefferle, 2022-12-08:
**Is your feature request related to a problem? Please describe.**
Similar to https://github.com/zigpy/zha-device-handlers/issues/1826

When Heiman CO_CTPG Monoxide Carbon sensor is added to HA it reports as a "Mains" powerd device and after some hours drops out of the network

**Describe the solution you'd like**
If possible a custom quirk to correct it.

<details>

<summary>Device signature</summary>

```JSON
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|MainsPowered: 132>, manufacturer_code=4627, maximum_buffer_size=64, maximum_incoming_transfer_size=0, server_mask=0, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.ExtendedSimpleDescriptorListAvailable|ExtendedActiveEndpointListAvailable: 3>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0009",
        "0x0500"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Heiman",
  "model": "CO_CTPG",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```JSON
"data": {
    "ieee": "**REDACTED**",
    "nwk": 53862,
    "manufacturer": "Heiman",
    "model": "CO_CTPG",
    "name": "Heiman CO_CTPG",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4627,
    "power_source": "Mains",
    "lqi": 255,
    "rssi": -74,
    "last_seen": "2022-12-08T00:05:24",
    "available": false,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|MainsPowered: 132>, manufacturer_code=4627, maximum_buffer_size=64, maximum_incoming_transfer_size=0, server_mask=0, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.ExtendedSimpleDescriptorListAvailable|ExtendedActiveEndpointListAvailable: 3>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0402",
          "in_clusters": [
            "0x0000",
            "0x0001",
            "0x0003",
            "0x0009",
            "0x0500"
          ],
          "out_clusters": [
            "0x0019"
          ]
        }
      }
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "binary_sensor.carbon_monoxide",
        "name": "Heiman CO_CTPG"
      },
      {
        "entity_id": "button.carbon_monoxide_identify",
        "name": "Heiman CO_CTPG"
      }
    ],
    "neighbors": [],
    "routes": [],
    "endpoint_names": [
      {
        "name": "IAS_ZONE"
      }
    ],
    "user_given_name": "Carbon Monoxide",
    "device_reg_id": "74a1c4e034572067293754c4e47218c7",
    "area_id": "746840d334f149c382e786dca8d15b11",
    "cluster_details": {
      "1": {
        "device_type": {
          "name": "IAS_ZONE",
          "id": 1026
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "Heiman"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "CO_CTPG"
              }
            },
            "unsupported_attributes": {}
          },
          "0x0003": {
            "endpoint_attribute": "identify",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0500": {
            "endpoint_attribute": "ias_zone",
            "attributes": {
              "0x0000": {
                "attribute_name": "zone_state",
                "value": 1
              },
              "0x0001": {
                "attribute_name": "zone_type",
                "value": 43
              },
              "0x0002": {
                "attribute_name": "zone_status",
                "value": 32
              },
              "0x0010": {
                "attribute_name": "cie_addr",
                "value": [
                  175,
                  231,
                  4,
                  255,
                  255,
                  46,
                  33,
                  0
                ]
              }
            },
            "unsupported_attributes": {}
          },
          "0x0001": {
            "endpoint_attribute": "power",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0009": {
            "endpoint_attribute": "alarms",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**

[image]

javicalle, 2022-12-08:
There is a proposed quirk:
<details open>
<summary>HeimanSmokCO_CTPG</summary>

```python
class HeimanSmokCO_CTPG(CustomDevice):
    """CO_CTPG quirk."""

    signature = {
        MODELS_INFO: [(HEIMAN, "CO_CTPG")],
        ENDPOINTS: {
            1: {
                # "profile_id": 260,
                # "device_type": "0x0402",
                # "in_clusters": ["0x0000","0x0001","0x0003","0x0009","0x0500"]
                # "out_clusters": ["0x0019"]
                PROFILE_ID: zigpy.profiles.zha.PROFILE_ID,
                DEVICE_TYPE: zigpy.profiles.zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Alarms.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }

    replacement = {
        NODE_DESCRIPTOR: zigpy.zdo.types.NodeDescriptor(
            logical_type=2,
            complex_descriptor_available=0,
            user_descriptor_available=0,
            reserved=0,
            aps_flags=0,
            frequency_band=8,
            mac_capability_flags=132 & 0b1111_1011,
            manufacturer_code=4627,
            maximum_buffer_size=64,
            maximum_incoming_transfer_size=0,
            server_mask=0,
            maximum_outgoing_transfer_size=0,
            descriptor_capability_field=3,
        ),
        ENDPOINTS: {
            1: {
                PROFILE_ID: zigpy.profiles.zha.PROFILE_ID,
                DEVICE_TYPE: zigpy.profiles.zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Alarms.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }
```

</details>

There is a guide about enabling custom quirks:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Copy inside the current `heidan/smoke.py` file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/heiman/smoke.py
Add the proposed code in the new file.
Restart HA and pair you device again. Check that the device is loading the quirk and that the device signature has changed.

pfefferle, 2022-12-09:
I tried it several times, but the local quirk wasn't loaded. Maybe I misconfigured something. Will check again later today.

javicalle, 2022-12-09:
The signature seems OK. Have you tried to remove and pair the device again?
Any error during the startup?
If not working, try to enable the debug logs and look for the device quirk resolution.

pfefferle, 2022-12-09:
I had the quirk in the wrong folder! it works like a charm! thanks a lot!

[image]

pfefferle, 2022-12-09:
Still available (after nearly an hour). I will check back later today, with an update.

And here is the screenshot with the expanded Zigbee Info (I forgot the in the last screenshot):

[image]

pfefferle, 2022-12-09:
Still active! Seems to work as it should!

pfefferle, 2022-12-10:
12h later: still available!

javicalle, 2022-12-10:
Thanks for confirming. 👍🏻
I have created the PR, so I hope that will come with the next version.
Once included, you can delete your local quirk.

pfefferle, 2023-07-31:
@javicalle since version 2023.06 or 2023.05 the device is offline again. After reconnecting I see it for about half an hour or an hour and then it disconnects again. The Energysource is "Battery or Unknown".

The signature is

```JSON
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4627, maximum_buffer_size=64, maximum_incoming_transfer_size=0, server_mask=0, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.ExtendedActiveEndpointListAvailable|ExtendedSimpleDescriptorListAvailable: 3>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0009",
        "0x0500"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Heiman",
  "model": "CO_CTPG",
  "class": "zhaquirks.heiman.smoke.HeimanSmokCO_CTPG"
}
```

javicalle, 2023-08-01:
There had not been changes in the applied quirk from the initial version.
Your device signature shows that quirk is applied and that the `mac_capability_flags` has the desired value.

It would be changes at ZHA level, but would affect several devices for sure (the `mac_capability_flags` fix is not uncommon). I haven't see any report in ZHA in this way.

Can it be a low battery issue or noise network?

pfefferle, 2023-08-01:
@javicalle I will check that, thanks for your reply!

PR #2001 (2022-12-12): Fix Heiman CO_CTPG Carbon Monoxide sensor
Heiman CO_CTPG Carbon Monoxide sensor reports as MainsPowered and drops from network. The quirk just fix the `NODE_DESCRIPTOR`

Fix: #1995