zigpy/zha-device-handlers #1996: [BUG] Moes 1 gang dimmer switch only working in lowest dimming mode TS0601 by _TZE200_la2c2uo9

rrooggiieerr, 2022-12-08:
**Describe the bug**
The Moes 1 gang dimmer switch I use only gives light as being in it's lowest dimmer position, changing the brightness in HA does not affect the real brightness of the attached lights.

It used to work properly, however stopped working since the latest upgrade of HA to 2022.12.0 Since this update also update to a ew version of zha-quirks I expect this to be the source of the problem.

Device info:
TS0601
by _TZE200_la2c2uo9

Zigbee info:
IEEE: 84:ba:20:ff:fe:67:35:b1
Nwk: 0x547a
Device Type: Router
LQI: 255
RSSI: -48
Last Seen: 2022-12-08T12:22:17
Power Source: Mains
Quirk: zhaquirks.tuya.ts0601_dimmer.TuyaSingleSwitchDimmer

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on the switch, it will light up on it's lowest brightness
2. Increase brightness, nothing will happen

**Expected behavior**
I expect the brightness of the light to go up if I do so trought the interface

**Screenshots**
If applicable, add screenshots to help explain your problem.

<details>

<summary>Device signature</summary>
I don't see any "Zigbee Device Signature" button in the device settings view, has this changed?

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.12.0",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "5.15.61-v8",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "zha_toolkit": {
      "version": "v0.8.27",
      "requirements": [
[... 453 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

javicalle, 2022-12-08:
Please, can you enable the debug logs and attach it?
* https://www.home-assistant.io/integrations/zha/#debug-logging

rrooggiieerr, 2022-12-09:
@javicalle do you still want me to make the log files since the source if the problem seems to be found?

PR #1999 (2022-12-12): Revert PR #1984
Users are reporting bad behavior with the dimming function after PR #1984:
* https://github.com/zigpy/zha-device-handlers/issues/1554#issuecomment-1343399518

Reverting to fix issues.

Fix: #1996
Fix: #1750
closed 2022-12-12 by dmulcahey