zigpy/zha-device-handlers #2000: Backlight mode on TS0001, TS0002, TS0003[BUG]

willmann, 2022-12-09:
Backlight mode don´t works on ZHA (TS0001, TS0002, TS0003). In configuration appear as select. 
But works on tuya or local tuya integrations and this option appear as a boolean switch.

Sorry, i´m using google translate

ZHA:
[image]

TUYA:
[image]

javicalle, 2022-12-09:
Can you attach the logs for the HA backlight operation.
I would expect that at least the `Off` and `LightWhenOn` should work. Any errors in the logs?

willmann, 2022-12-09:
@javicalle , 

Could you instruct me how to get the log?

Thanks

javicalle, 2022-12-09:
Is none of the values working? Neither the `Off` or the `LightWhenOn` options?

Here is some instructions to enable the debug logs:
* https://www.home-assistant.io/integrations/zha/#debug-logging

Be carefull because can be very verbose. Don't keep long time after get the logs you want.

willmann, 2022-12-09:
I tried all 3 options.
I'll follow the instructions to enable debug.
Are there any keywords to look for?

willmann, 2022-12-09:
[backlight.txt](https://github.com/zigpy/zha-device-handlers/files/10198613/backlight.txt)

@javicalle , 

I hope it helps.

javicalle, 2022-12-10:
I can't see the logs from the quirk.
Can you attach the signature or the diagnosis from any of the failing devices?
Can you check if devices are loading any quirk?

willmann, 2022-12-10:
[zha-dba4ffd11e8899871583d5f3cd82eea8-_TZ3000_bmqxalil TS0001-2a3c9be3eeb5b383eb0863a323f4c8cc.json (1).txt](https://github.com/zigpy/zha-device-handlers/files/10198966/zha-dba4ffd11e8899871583d5f3cd82eea8-_TZ3000_bmqxalil.TS0001-2a3c9be3eeb5b383eb0863a323f4c8cc.json.1.txt)

"Can you check if devices are loading any quirk?"
How do I do it?

willmann, 2022-12-10:
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/10198978/home-assistant.log)
Full log.

willmann, 2022-12-10:
@javicalle 
If there's anything else I can do, please let me know.

willmann, 2022-12-13:
Hi @javicalle !!

Any news?

javicalle, 2022-12-13:
Out of time right now.
I'll check in the weekend.

willmann, 2023-01-13:
Hi @javicalle !

Any good news?

azmat007, 2023-05-10:
Same issues here too. I need the backlight to go off as it really bothers me at night. Also if the backlight can go off on other devices like ZB-SW03
Or can anyone suggest which device can i buy that can switch off backlight in ZHA.

closed 2023-11-13 by github-actions[bot]

jhbcabral, 2024-07-11:
Any news here?

willmann, 2024-07-11:
It still doesn't work. I'm migrating to Z2M.

xg1990, 2025-01-16:
same problem here. No error in debug logs
```
2025-01-16 21:48:08.681 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0x1F51:2:0x0006]: cluster_handler[on_off] attribute_updated - cluster[On/Off] attr[backlight_mode] value[SwitchBackLight.Mode_0]
2025-01-16 21:48:08.681 DEBUG (MainThread) [zha] Emitting event cluster_handler_attribute_updated with data ClusterAttributeUpdatedEvent(attribute_id=32769, attribute_name='backlight_mode', attribute_value=<SwitchBackLight.Mode_0: 0>, cluster_handler_unique_id='84:27:12:ff:fe:26:c6:71:2:0x0006', cluster_id=6, event_type='cluster_handler_event', event='cluster_handler_attribute_updated') (3 listeners)
2025-01-16 21:48:08.682 DEBUG (MainThread) [zha] Emitting event state_changed with data EntityStateChangedEvent(event_type='entity', event='state_changed', platform=<Platform.SELECT: 'select'>, unique_id='84:27:12:ff:fe:26:c6:71-2-6-backlight_mode', device_ieee=84:27:12:ff:fe:26:c6:71, endpoint_id=2, group_id=None) (1 listeners)
2025-01-16 21:48:08.682 DEBUG (MainThread) [homeassistant.components.zha.entity] select.switch_dining_backlight_mode_2: Handling event from entity: EntityStateChangedEvent(event_type='entity', event='state_changed', platform=<Platform.SELECT: 'select'>, unique_id='84:27:12:ff:fe:26:c6:71-2-6-backlight_mode', device_ieee=84:27:12:ff:fe:26:c6:71, endpoint_id=2, group_id=None)
```