zigpy/zha-device-handlers #2003: Aqara Curtain Driver E1 (lumi.curtain.agl001)

circa1665, 2022-12-09:
**Is your feature request related to a problem? Please describe.**
Am able to open and close curtains using the controls in HA, however have to physically turn the device around to change opening direction. The device supports reversing direction, but an entity for this does not appear in ZHA, neither does the battery level. Power source is incorrectly listed as mains powered and device firmware version is not listed.
Lastly, the device has a light level sensor but again no entity for this in ZHA. 

**Describe the solution you'd like**

- Entity to reverse opening direction.
- Battery level sensor.
- Light level sensor.
- Correct Power source to battery.
- Display device firmware version.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|MainsPowered: 132>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000a",
        "0x0102",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0003",
        "0x000a",
        "0x0019",
        "0xfcc0"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.curtain.agl001",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.12.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "5.15.74",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.3",
    "docker_version": "20.10.18",
    "chassis": "embedded",
    "run_as_root": true
[... 314 lines cut ...]
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
https://zigbee.blakadder.com/Aqara_ZNCLBL01LM.html

There's a pr for this device [here](https://github.com/zigpy/zha-device-handlers/pull/1648), this currently adds support for the battery level sensor, but the reversing direction and light level sensor don’t work.

badrpc, 2022-12-11:
I have one of this for testing. The #1648 mentioned above works well to add battery level and direction control option. However there's something strange about this device. When I set `Curtain mode` to `Inverted` everything works fine, but when I set it to normal, the close/open directions seem to be swapped. For example, I if set position to 5 and send down_close command I'm expecting the device to move towards the side it is close to. However the device moves towards the farther edge and reports position 100. Once there the close button in UI is no longer available and open button does not work.

As I understand this device also has a luminosity sensor, would be nice to have one available in Home assistant or to set rules for automatic opening/closing.

circa1665, 2022-12-12:
For me, using the custom quirk only adds battery level, the direction control option doesn't show up for me. I tried updating the Aqara E1 Curtain driver to the latest firmware using an Aqara hub, but no joy.

TheJulianJES, 2023-01-04:
@circa1665 Did you check the clusters UI?

Go to the device page -> click the three dots -> Manage Zigbee Device. In the first dropdown menu, select ``XiaomiAqaraCurtainE1`` (if it shows up) and then you can modify the attributes.
Note: always leave the second text field empty. When you're reading an attribute, it'll appear in the first text field. If you want to write that attribute, just remove the text from the first text field and enter whatever you want to write in the first text field. (second text field needs to be empty), then press "Write Zigbee attribute."

TheJulianJES, 2023-01-04:
@circa1665 Can you restart HA now and see if you get the config entity?
Not sure why the battery entity isn't working though, maybe it comes back after some time

circa1665, 2023-01-04:
@TheJulianJES, yes! The curtain mode entity is showing up now - although it's showing as "Normal", rather than "Inverted".
[image]

TheJulianJES, 2023-01-04:
Changing the curtain mode to "inverted" (whilst the device is awakened) should properly update the status there.

https://github.com/home-assistant/core/pull/75132 did not add the attribute to ``ZCL_INIT_ARGS`` for the ``WindowCovering`` channel: https://github.com/home-assistant/core/blob/80c357c00f733a90534fc9999fe5b606849c3b16/homeassistant/components/zha/core/channels/closures.py#L120-L122

- it needs to be added to init args, so the attribute is read during pairing (and thus the config entity created)
- ``"window_covering"`` from that PR should also be changed to the constant ``CHANNEL_COVER``
- [the quirk](https://github.com/zigpy/zha-device-handlers/pull/1648) also needs to be merged into zha-quirks

circa1665, 2023-01-06:
Just to flag, the battery state is still "Unknown" so not a case of it coming back "after some time", thanks!

TheJulianJES, 2023-01-06:
What coordinator are you using?

circa1665, 2023-01-06:
I'm using SONOFF Zigbee 3.0 USB Dongle Plus

LouisAtGH, 2023-01-13:
I am trying to make to control a curtain set with two lumi.curtain.agl001 motors. Not so easy.  

Just as mentioned above I am missing
-    Entity to reverse opening direction.
-    Battery level sensor.
-    Light level sensor.
-    Correct Power source to battery.
-    Display device firmware version.

I can work around the reverse directions for the right curtain, but not around the other issues.

But I have another major problem ..... that is the position of the motors. There is a command ^cover.set_cover_position^ and according to info found position has a range from 0 .. 100 where  0 = closed and 100 is open,  (https://www.home-assistant.io/integrations/cover.template/)

However that is where the problems are starting:
- because "how to set the open position" and "how to set the closed position"
- and for the right side curtain it is of cause mirrored
- and you have to re-calibrate the "open position" and the "closed position" every time the roller slipped 

I have really no idea how to set those closed and open position reference points using the software
And the behavoir when changing the position slider .... is strange .... I do not understand

What ever I would be **very very very glad** with an updated ZHA-driver and perhaps also firmware for this device !!!

To work around the positioning issues, I consider to use sensors at the closed and open position at each site ..... 
That makes the whole thing of course more expensive and far more complex .... but I do not see an other option at the moment.  .... And it does not help against other issues like ..... unkown battery level etc ...

So, again updates will be highly appreciated

msolomos, 2023-01-29:
Am intersting for battery level too.. 
Yes the cover buttons are reversed but I don't really mind as I can completely control my curtain from Analog_output. 
However battery level is essential and I really can't figure out how it is still not fully supported from ZHA.

circa1665, 2023-01-29:
Have updated SONOFF Zigbee 3.0 USB Dongle Plus to latest firmware: https://github.com/Koenkk/Z-Stack-firmware/blob/Z-Stack_3.x.0_coordinator_20221226/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20221226.zip
Hasn't made any difference, battery state is still "Unknown".

circa1665, 2023-02-06:
So from the original list, looks like:
- ~~Entity to reverse opening direction.~~ - sorted but yet to be added to custom quirk (current workaround [here](https://github.com/zigpy/zha-device-handlers/issues/2003#issuecomment-1371503590)).
- Battery level sensor.
- Light level sensor.
- Correct Power source to battery.
- Display device firmware version.

Would be great to get the remaining points working, is there anything I can do to help?

LouisAtGH, 2023-02-07:
Actuality  I am using the latest Home Assistant version 2023.2.2 version and probably more relevant the recent Sonoff E USB (EZSP = Silicon Labs EmberZNet protocol: Elelabs, HUSBZB-1, Telegesis by ZHA). Before I used the older Sonoff P USB.

Since I made that move, the position slider and the related the  set_cover_position command is gone. 

Since I used to open and close the curtains with the cover_position command (which was more reliable then the open/close command) from the gui and even more relevant from node-red, two points:

- I think the now missing go to position -x should be added to the to do list
- I hope that someone knows how to get the slider back !!

msolomos, 2023-02-07:
I am using Aqara E1 curtain driver for a few years. So far I was able to control it only through the analog output.
Since the last update I now have cover working properly and the direction is not reverted anymore! Weldone to everyone!
This is great progress. Please note that I have to delete the node and add it again.
What is still missing and still a must for many of us is the battery level.

circa1665, 2023-02-10:
I added my Aqara E1 curtain driver to Home Assistant Bridge/HomeKit and now the battery sensor works.

I've noticed that the state of the device gets stuck on opening/closing so I've had to create an automation to listen for the events `cover.open_cover` and `cover.close_cover`, then `cover.stop_cover` after a 20 second delay.

I've also noticed that when I interact with I open/close the curtain that nothing shows up at all when I listen for `zha_event` in Developer Tools.

Lastly this doesn't seem to play nice with Home Assistant Bridge/HomeKit, can't operate this device from HomeKit.

Hamish1066, 2023-02-14:
> I added my Aqara E1 curtain driver to Home Assistant Bridge/HomeKit and now the battery sensor works.

How do you add it to HomeKit and why does the battery work with this method? I thought HomeKit didn't support Zigbee without a Zigbee bridge?

circa1665, 2023-02-15:
Hi @Hamish1066, I don’t know why, the battery sensor just started working when I did. I use the [HomeKit integration](https://www.home-assistant.io/integrations/homekit/) in Home Assistant to make some devices available to control via Homekit.

Hamish1066, 2023-02-15:
@circa1665 I'm also using the HomeKit integration but battery status isn't showing up in HA for me. Weird.

circa1665, 2023-03-07:
Recently migrated to Home Assistant Yellow, now the Aqara Curtain Driver E1 does not work at all with ZHA. This is the only zigbee device I had any trouble with. The device is unavailable, cannot re-pair the device with ZHA - I've tried many many times.

circa1665, 2023-03-13:
Much as I would love to get this working with ZHA, doesn’t appear that’s going to happening anytime soon.

Ended up purchasing an [Aqara E1 hub USB dongle](https://www.amazon.co.uk/Aqara-Interface-Repeater-Supports-Assistant/dp/B09BJCNCKM). Now have this working via HomeKit controller in Home Assistant.

The reversing direction isn't exposed as an entity in HA and has to be setup via the Aqara Home app. I now have a functioning cover entity and battery level in Home Assistant which will have to do for now.

LouisAtGH, 2023-03-14:
Tja I undestand, but for me that is the wrong direction. 

HA should be in control, not the suppliers cloud! 
I would like to stay away from all kind of cloud services, especially chinese ones!

I have the curtains working, using node-red and sensors at the begin and end of each curtain side. 
I is not ideal, but with limitations it works

Hamish1066, 2023-03-14:
I appreciate all the work that @circa1665 has done on this, but I also agree with @LouisAtGH that the goal should be local integration. 

For me, almost everything works using the device with ZHA. The only quirks are that (1) the battery sensor doesn't show up; and (2) I can't use the open/close function, but it works instead to set a position (i.e. 0-100%). So for me it is totally usable with ZHA, just not perfect.

bjvanst, 2023-04-03:
I have four of these.  Two on two windows.  I just applied the custom quirk and battery level is properly exposed on all four units.

I'm observing similar behaviours as others with regards to their state.  State, current_position and sending open or close commands through the GUI in HomeAssistant are very inconsistent.

Sending open or close commands through Home Assistant service calls works 100% of the time.

One interesting observation which may be a coincidence... before applying the custom quirk, I had no ability to change direction so I physically flipped two units.  After applying the quirk, the option to change direction appears for two units but not the other two.  The two it appears on are the flipped units.

Reading the attribute mentioned [here](https://github.com/zigpy/zha-device-handlers/issues/2003#issuecomment-1370419738) doesn't appear to make a difference.

Hamish1066, 2023-04-03:
> I just applied the custom quirk and battery level is properly exposed on all four units.

Could you please explain what the custom quirk is. Sorry if I’m missing something.

bjvanst, 2023-04-03:
> > I just applied the custom quirk and battery level is properly exposed on all four units.
> 
> Could you please explain what the custom quirk is. Sorry if I’m missing something.

It is the PR mentioned in the original post:
https://github.com/zigpy/zha-device-handlers/pull/1648

ds-sebastian, 2023-04-13:
Is there any testing that needs to be done? I've just received a new one of these and was hoping it would be easily integrated in to HA. The existing quick PR also has an unknown battery for me and did not reverse directions. or give me any new options to.

Nafania, 2023-06-30:
> Is there any testing that needs to be done? I've just received a new one of these and was hoping it would be easily integrated in to HA. The existing quick PR also has an unknown battery for me and did not reverse directions. or give me any new options to.

plus 1 - can help with testing, im using lumi.curtain.agl001, firmware 0x0000001b with sonoff Zigbee

TheJulianJES, 2023-10-08:
Please help with testing https://github.com/zigpy/zha-device-handlers/pull/2629 and let's continue the discussion there.

PR #2629 (2023-10-21): Add Xiaomi E1 driver curtain support
## Proposed change

~~https://github.com/zigpy/zha-device-handlers/pull/2620 should be merged first. This PR then needs to be rebased (for the test file).~~
This PR will be merged first, other than rebased. It'll use some cluster classes from this PR.

## Additional information

Supersedes https://github.com/zigpy/zha-device-handlers/pull/1648
Fixes #2003

Changes are all untested. I'm also not sure if this is the right way to use the motor mode. It's in the original quirk and a HA PR for the inverted motor mode has already been merged (even though the quir [... cut]
closed 2023-10-21 by TheJulianJES