zigpy/zha-device-handlers #2006: [BUG] Tuya TRV (MOES BRT-100) battery level is unknown

MaximumSU, 2022-12-10:
**Describe the bug**
Tuya TRV (MOES BRT-100) battery level shown as unknown. Attributes show 2 AA batteries (correct) but level always "unknown".

**To Reproduce**

**Expected behavior**
Battery level

**Screenshots**

<details>

<summary>{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_b6wax7g0",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_trv.MoesHY368_Type1new"
}</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>{
  "home_assistant": {
    "installation_type": "Home Assistant Core",
    "version": "2022.12.1",
    "dev": false,
    "hassio": false,
    "virtualenv": true,
    "python_version": "3.10.9",
    "docker": false,
    "arch": "armv7l",
    "os_name": "Linux",
    "os_version": "4.19.66-v7+",
    "run_as_root": false
  },
  "custom_components": {
    "linkplay": {
      "version": "3.1.8",
      "requirements": [
        "async-upnp-client>=0.27",
        "validators~=0.12",
        "chardet>=4.0.0"
      ]
[... 379 lines cut ...]
        }
      }
    }
  }
}</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

javicalle, 2022-12-12:
Well, the current code is a bit... optimistic?
https://github.com/zigpy/zha-device-handlers/blob/1d7b87e516cd4e3a1505cf979e5d4e42837cd01a/zhaquirks/tuya/ts0601_trv.py#L277-L280

So, when low battery signal comes, the 'battery_level' will be `5` (5%?)
Until then, no signal, so no `else 100` condition.

IMHO that code couldn't have been working ever. And no nice solution. Sorry.

MattWestb, 2022-12-12:
The TRV is only reporting battery low or full and it have not reporting the attribute (it dont do it very oft).
Take out the battery and wait one minute and putting it back and it shall reporting all attribute also the battery status.

MaximumSU, 2022-12-13:
> The TRV is only reporting battery low or full

Ok, i think, "full" is not reported, only "low" (there is only "battery_low" attribute).

PS: May be you know what is the switch "Switch" created after pairing? What he do? I don't see any influence of this switch to anything...

javicalle, 2022-12-13:
> Ok, i think, "full" is not reported, only "low" (there is only "battery_low" attribute).

Have you tried to remove the battery and put it again?

> May be you know what is the switch "Switch" created after pairing?

It seems to be a 'windows detection' (whatever it means):
https://github.com/zigpy/zha-device-handlers/blob/1d7b87e516cd4e3a1505cf979e5d4e42837cd01a/zhaquirks/tuya/ts0601_trv.py#L779-L793

I would say that sould activate some kind of 'target temp when window open' or similar, but I really don't know.

MaximumSU, 2022-12-13:
@javicalle 

> It seems to be a 'windows detection' (whatever it means)

Aha, thank you for the good idea. This TRV have some actions if window is opened. Need to check this direction.

MaximumSU, 2022-12-13:
OK, this is not an issue or bug, this is "by design" of TRV. I close the issue.

MattWestb, 2022-12-13:
We have testing some tuya TRVs Window detection and was not getting it triggered then putting it outside for longer time. If putting it in the freezer its being triggered and sending window open alarm attribute (= not very usebal function). My Siterwells is also open the valve if going under 10°C for frost protection.

MaximumSU, 2022-12-13:
Wait, i see here: 
https://www.zigbee2mqtt.io/devices/BRT-100-TRV.html
and here:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/f649fd86f17345757ab6722c73c0c004585e5b14/devices/moes.js#L273
what this TRV reporting battery percentage in the "battery" attribute, not only "battery_low" and also it is reporting the valve state - opened, or closed. But this quirk don;t have such attributes... So, reopen. Battery state uses not correct attribute.

MaximumSU, 2022-12-14:
So, device signature is the same, but it is defferent device... Great.
Thank you, i try the custom quirk.