zigpy/zha-device-handlers #2010: [BUG] Ikea Tradfri E1810 won't work anymore...

wes1993, 2022-12-15:
After i have updated HA the ikea tradfri 5 buttons won't work anymore... i can pair with ZHA but if i press or do something else won't work... Do no send the command to ZHA (On Off Color etc)

MattWestb, 2022-12-15:
From the device card do one reconfigure and pressing one button then sending commands to the remote so its waking up and can receiving the commands.

wes1993, 2022-12-15:
I have tried this but won't work....
Finally i have solved by changing the Quriks with the old version from May, probably there is something with this new quirks that broke something....

wes1993, 2022-12-15:
Probably the problem could stay in the missing import of IKEA_CLUSTER_ID = 0xFC7C  # decimal = 64636 from the fivebtnremote.py file, this variable it's imported in the fivebtnremotezha.py

[image]

MattWestb, 2022-12-15:
Do you getting
 ```
IKEA TRADFRI remote control Checkin event was fired
18:50:14 - 32 minutes ago
``` 

in the Logbook on the device card every 55 minutes ?

My remote is on the latest firmware and your 2 then its have loaded the version 3.

In the production quirk is OK defined:
https://github.com/zigpy/zha-device-handlers/blob/954745eaf47b205f99534e34e11679b8853d8878/zhaquirks/ikea/fivebtnremotezha.py#L45-L51

Do you have local quirks installed that is overriding the production ones ?

wes1993, 2022-12-15:
Yes, i have solved using the quirks from may i don't know why has worked with the old one, i have overrided this two quirks:

fivebtnremote.py
fivebtnremotezha.py

I have uploaded this files, i have rebooted the HA instance and paired again and now works well.

MattWestb, 2022-12-15:
The Oct 25 PR is moving the IKEA specific clusters to INIT and is being imported from it in all quirks and its working for all IKEA controllers (i have not tested knyplan then i cant getting it in AT)

One more time is you using custom quirks ?

wes1993, 2022-12-15:
Nope, i'm not using a custom quirks

MattWestb, 2022-12-15:
Then is very strange then it shall working with the October release and i have new installed one test system and all IKEA controllers is working well.

The most important thing is you have getting it  working OK and hope not more user is getting the same problem.

matrixes, 2023-05-21:
I initially had the same issue with no events fired after pairing. I got the controller to _partly_ work by using the pairing sequence suggested here https://github.com/zigpy/zha-device-handlers/issues/1752#issuecomment-1555840279

Now, the controller works in part. The buttons that fire events are
* on/off (toggle)
* brightness up
* brightness down

However, the left and right buttons fire no events.

I'm using SkyConnect and Home Assistant 2023.5.3 without any quirks connected to IKEA devices.

MattWestb, 2023-05-21:
You meed binding the scene cluster to the coordinator or the controller is not seeding the commands to it.

MattWestb, 2023-05-21:
Do one reconfigure device from the device card and waking up the device then sending commands to it

matrixes, 2023-05-22:
I did the Reconfigure-cycle and the report _looks_ promising.

[image]

 However, the result is the same; There are no events sent for the left and right buttons.

matrixes, 2023-05-22:
I've messed around with this and reading the value reported from `sw_build_id (id:0x4000)` HA states `24.4.5`

When I delete my remote and re add it, I reconfigure it, press the toggle button on the remote to wake it up and this is the result

[image]

Also, looking in HA debug logs, when pressing the left and right button, it echoes lines like these;

`2023-05-22 16:10:55.323 WARNING (MainThread) [zigpy.zcl] [0xD3D6:1:0x0005] Unknown cluster command 7 b'\x00\x01\r\x00'`

MattWestb, 2023-05-22:
Now the device is having the right cluster for the new device.
But its not being OK paired then its not communicating with the system.
Deleting it and resetting it and wait one minute (or restart HA) and adding it a gen.
Then its OK paired in the log window for adding devices pressing the toggle button some times so you see its sending commands OK.
Then going to the device card and do one reconfigure this device and pressing the toggle button some times so its not sleeping. If all is going well you shall getting all green.
If not restarting HA and trying reconfigure device one more time.

The V24.4.X firmware is little tricky getting paired OK but its working if getting the right timing also the reset procedure can being tricky getting working.

matrixes, 2023-05-23:
I've now done a number of re-pairs and reconfigurations (and HA reboots) but cannot get the remote to work with all 5 buttons.
The reconfiguration screen provide different results, most of them ending up like this

[image]

Unfortunately, the only buttons that work are toggle, brightness up and down. The left and right do not fire events other than the debug log "Unknown cluster command..."

Also, the remote consistently report a battery level of 128%.

matrixes, 2023-05-23:
This is the expanded Device info card

[image]

I had previously tried to use a quirk for a Tint remote control, but the config has quirks disabled as per

```
zha:
  enable_quirks: false
  custom_quirks_path: /config/zha_quirks/
  zigpy_config:
    ota:
      ikea_provider: true                        # Auto update Trådfri devices
      ledvance_provider: true                    # Auto update LEDVANCE/OSRAM devices
      salus_provider: true                       # Auto update SALUS/Computime devices
      inovelli_provider: true                    # Auto update INOVELLI devices
      thirdreality_provider: true                # Auto update 3REALITY devices
      #otau_directory: /path/to/your/ota/folder  # Utilize .ota files to update everything else
```

I have no other quirk related configurations as far as I am aware of. It's a _mostly_ clean/vanilla HomeAssistant installation run in Docker.

Is there something more specific in files/logs I can monitor or check to see if the right quirks are in place and being read? Or something else that could help in figuring out where things go wrong?

TheJulianJES, 2023-05-23:
Do not do this unless you want to break everything:
`enable_quirks: false`

That disables included quirks. Without those, these remotes will not work. Just remove that line and re-pair the remote.

If you have issues with custom quirks, then delete those custom quirks in the custom quirk folder.

matrixes, 2023-05-23:
> Do not do this unless you want to break everything: `enable_quirks: false`
> 
> That disables included quirks. Without those, these remotes will not work. Just remove that line and re-pair the remote.
> 
> If you have issues with custom quirks, then delete those custom quirks in the custom quirk folder.

Right on the money! I didn't connect the dots that setting that to false meant that _all_ quirks (including the integration provided ones) were going to be disabled.
I'm sorry for having us had to run down a deeper rabbit hole than necessary and I want to commend @MattWestb for putting up with my questions.

I have made changes to the configs accordingly, restarted HA and pesto! all the buttons now work!

Again, I appreciate all your efforts!

MattWestb, 2023-05-23:
Great work that you have getting it working !!
I have testing nearly all versions of IKEA controller and they is working OK so you was having one bad setting in for quirks and its working OK then you was fixing it :-)

closed 2023-11-26 by github-actions[bot]