zigpy/zha-device-handlers #2011: [Device Support Request] TS0601 sensor variant(?)

Migsi, 2022-12-15:
**Is your feature request related to a problem? Please describe.**
I've just bought three tuya temperature/humidity sensors (expected Aquara ones, got generic/tuya ones) and connected them to my HA. So far everything went good, they got interviewed, three entities were created (temp, hum, battery) and everything seemed fine. Then I noticed, all measured values were "0", so 0°C, 0% Humidity, 0% Battery. The only working property is LQI. I researched a bit and found these sensors get randomish manufactor descriptors, so they might not have been recognized properly. Mine are `_TZE200_a8sdabtg`, which do not seem to be covered by https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_sensor.py. But at least they are recognized as such sensor. I still don't get how reading "0" values only could happen.

**Describe the solution you'd like**
The sensor values should be read properly.

<details>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0402",
        "0x0405"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_a8sdabtg",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.12.5",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Vienna",
    "os_name": "Linux",
    "os_version": "5.10.142+truenas",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.28.4",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "zha_toolkit": {
      "version": "v0.8.28",
      "requirements": [
        "packaging>=20.8",
        "pytz"
      ]
    },
    "thermal_comfort": {
      "version": "1.5.4",
      "requirements": []
[... 282 lines cut ...]
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

**Additional context**
I've got them here: https://www.amazon.de/gp/product/B09NKCDXT9/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

javicalle, 2022-12-15:
It seems to be another variant for `TuyaTempHumiditySensor_Square` so I would suggest to enable the local quirk folder, copy the current `ts0601_sensor.py` file and add the manufacturer to the signature:
```python
class TuyaTempHumiditySensor_Square(CustomDevice):
    """Custom device representing tuya temp and humidity sensor with e-ink screen."""

    # RelativeHumidity multiplier
    # RH_MULTIPLIER = 100

    signature = {
        MODELS_INFO: [
            ("_TZE200_a8sdabtg", "TS0601"),
            ("_TZE200_qoy0ekbd", "TS0601"),
            ("_TZE200_znbl8dj5", "TS0601"),
        ],
        ENDPOINTS: {
```

Save changes, restart HA and re pair the device.

Migsi, 2022-12-17:
You were right, that worked flawlessly. Even though this model has no screen. Also I'm a little disappointed of it reporting humidity values as integers only, but that might be a limitation of the sensor itself.
I'll open a PR to add this variant.

closed 2022-12-18 by dmulcahey