zigpy/zha-device-handlers #2012: [BUG] Start-up behavior for Moes 1 gang dimmer switch TS0601 by _TZE200_la2c2uo9 not working

rrooggiieerr, 2022-12-16:
**Describe the bug**
The configurable start-up behavior for the Moes 1 gang dimmer switch TS0601 by _TZE200_la2c2uo9 does not change the start -up behaviour. The behaviour will always be previous state.

**To Reproduce**
Steps to reproduce the behavior:
1. in ZHA select the TS0601 by _TZE200_la2c2uo9 device
2. Under Configuration change Start-up behavior to anything else than PreviousValue
3. Switch off the device by breaking the power
4. After applying back power the state of the device will be the previous state, not the state as configured in Start-up behavior

The Start-up behavior can also be modified by Manage Zigbee Device > TuyaOnOf > start_up_on_off but behaviour is the same

**Expected behavior**
If the device supports a Start-up behavior then the configured Start-up behavior is expected to work. If the devicce does not support Start-up behavior then the option should be removed from the interface.

**Screenshots**
If applicable, add screenshots to help explain your problem.

<details>

<summary>Device signature</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
I just think the device doesn't support Start-up behavior, so it needs to be removed from the configuration

javicalle, 2022-12-16:
Hi, can you attach the device signature?
Which values do you get if you ask for the in the 'Manage Zigbee Device' for:
* `TuyaOnOff (Endpoint id: 1, Id: 0x0006, Type: in)` --> `start_up_on_off (id: 0x4003)`
* `MoesSwitchManufCluster (Endpoint id: 1, Id: 0xef00, Type: in)` (or the name you get for cluster `0xef00`) --> `power_on_state (id: 0x8002)`

Does this values change when you set the 'start-up' behavior from HA?

Any errors in your logs?

rrooggiieerr, 2022-12-17:
Yes, the value for `start_up_on_off (id: 0x4003)` does change according to the settings in HA, it's just like the device is ignoring the setting. Cluster `0xef00` doesn't have a `power_on_state (id: 0x8002)`

[image]

javicalle, 2022-12-17:
And the device signature is?

rrooggiieerr, 2022-12-17:
Is this what you mean?

```
IEEE: 84:ba:20:ff:fe:67:35:b1
Nwk: 0x547a
Device Type: Router
LQI: 255
RSSI: -46
Last Seen: 2022-12-17T10:54:44
Power Source: Mains
Quirk: zhaquirks.tuya.ts0601_dimmer.TuyaSingleSwitchDimmer
```

javicalle, 2022-12-17:
Go to the device view.
3 dots, admit zigbee device, signature tab.

Or just attach the device diagnostic.

rrooggiieerr, 2022-12-17:
Ah!
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_la2c2uo9",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_dimmer.TuyaSingleSwitchDimmer"
}
```

javicalle, 2022-12-17:
Ok, the behavior is not related to the quirk.
The device defines the attribute don't implement it.
From ZHA is not possible to distinguish if the implementation is fine or not.
IMHO we don't must to address this behavior with a quirk.
Any user can disable the entity in HA or claim to the manufacturer for a fix.

rrooggiieerr, 2022-12-17:
@javicalle I see what you mean. Alright, then I'll close this issue

closed 2022-12-17 by rrooggiieerr