zigpy/zha-device-handlers #2014: [Device Support Request] TS0601 by _TZE200_khx7nnka (light sensor)

neuquenfr, 2022-12-17:
**Is your feature request related to a problem? Please describe.**
The only entities returned are: LQI (ok), RSSI (unknown) and the information "Event are fired" each time I pair it.
No way to measure light, which is its purpose.

**Describe the solution you'd like**
Could this community please provide us with a quirk that makes this device work properly with HA (OS) 2022.12.6 and ZHA/Sonoff Zigbee 3.0 USB Dongle? Thank you!

<details>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_khx7nnka",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.12.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Madrid",
    "os_name": "Linux",
    "os_version": "5.15.76-v8",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.4",
    "docker_version": "20.10.19",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "google_geocode": {
[... 382 lines cut ...]
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
[image]

URL of the product on Aliexpress: https://www.aliexpress.com/item/1005004095233195.html?spm=a2g0o.productlist.main.1.5c9casUkasUkTV&algo_pvid=ae555c11-2f6e-4f03-b524-494017be0553&algo_exp_id=ae555c11-2f6e-4f03-b524-494017be0553-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000028017055693%22%7D&pdp_npi=2%40dis%21EUR%2115.76%214.35%21%21%21%21%21%402100b20d16713006786068965d0745%2112000028017055693%21sea&curPageLogUid=Kqct4PelnXdI

javicalle, 2022-12-17:
I would try to add it to the `TuyaIlluminance` quirk:
https://github.com/zigpy/zha-device-handlers/blob/954745eaf47b205f99534e34e11679b8853d8878/zhaquirks/tuya/ts0601_illuminance.py#L70

It will be something like:
```python
        MODELS_INFO: [
            ("_TZE200_khx7nnka", "TS0601"),
            ("_TZE200_yi4jtqq1", "TS0601"),
        ],
```

There is a good guide for enabling your local quirk:
https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

neuquenfr, 2022-12-17:
@javicalle , this worked at first attempt. Thank you so much for your help!

javicalle, 2022-12-17:
If the quirk is working for your device, do you mind create a PR and add it to the library.
You can do it editing the file from the web browser:
https://github.com/zigpy/zha-device-handlers/blob/954745eaf47b205f99534e34e11679b8853d8878/zhaquirks/tuya/ts0601_illuminance.py#L70

Just follow the indications.
This way others users would use the device thanks to you.

neuquenfr, 2022-12-18:
@javicalle , I am totally new to contributing in Github and have passed hours reading/testing how to place a PR. Since I could not modify the original file (which is logical) and I did not understand most of the github vocabulary, I found a way to edit (probably a copy of) the file, then submit the result to Github (to where? I don't know).
Maybe the procedure I followed was OK, but I doubt it. Sorry for that.
The only modification I placed, leading to the device working, was to add up a line with ***("_TZE200_khx7nnka", "TS0601"),*** as you recommended.
Here is a copy of what is displayed on my desktop screen.
I hope this helps the community.
[image]

javicalle, 2022-12-18:
Hi @neuquenfr you have done almost all the job.
Thanks for trying so hard.

Your submit must have gone to your project fork: 
* https://github.com/neuquenfr/zha-device-handlers/

And you have a banch `patch-1` branch here:
* https://github.com/neuquenfr/zha-device-handlers/tree/patch-1

Probably, when you go to your fork you will see a banner asking for create a PR to the master. Just push the button and follow the instructions.

neuquenfr, 2022-12-18:
Honestly, @javicalle , I didn't see "one" green button, but several of them, leading me to the next one.
I hope this worked, otherwise I'm sorry to be a nuisance for this so helpful community and hope I didn't mess up too much.
Feel free to tell me what I should do to make things get back to normal.
Thanks!

javicalle, 2022-12-18:
@neuquenfr it haven't work as expected, but don't worry it can be messy if you are not familiar.

I have created the PR with your changes.

Once again, thanks for your job.

neuquenfr, 2022-12-18:
Thank YOU, @javicalle !

closed 2022-12-18 by neuquenfr

alexanderwwagner, 2024-02-07:
I have the same problem like @neuquenfr  described in this issue with a actual Homeassistant version: Core 2024.1.5.
What could be the problem?

AkazaRenn, 2024-05-04:
Me too here, the version shows the same as OP but no functional entities are returned.

Sheridan, 2024-08-14:
```
                          "tuyaDataPoints":
                          {
                            "64": [{"name": "unknown", "type": "raw"}],
                            "2": [{"name": "illuminance", "type": "value"}],
                            "1": [{"name": "illuminanceType", "type": "enum"}]
                          },
                          "illuminanceType": {"enum": ["unknown", "unknown", "overload?"]}
```

PR #2020 (2022-12-19): Add `_TZE200_khx7nnka` device
Fix: #2014
commit eb51d015 (2022-12-19): Add `_TZE200_khx7nnka` device (#2020)

Fix: #2014