zigpy/zha-device-handlers #2019: [Device Support Request] Bticino L4027C

davix10, 2022-12-18:
I have a connection problem with the Bticino L4027C, it is found and can be used but the notification LED on the button remains flashing as if it were still in discovery status. In this way, when the power is cut off, the device is no longer found by the home assistant because it is offline, in fact the color of the LED on the button returns to being red as if it were not connected to anything.
After a lot of resetting the roller shutter button on only one i managed to get it to work and get the green led to go off. At home I also have the Bticino L4411C and these have never given me any problems, they work first time!
I've tried them all but there's no way to get it recognized. I have a Sonoff Zigbee v3 ZBDongle-E stick.

Thanks!

myrkoo, 2022-12-19:
Same problem with L4027C, after pairing the green led still flashing. Shutter work but if I remove the power supply the pair is lost and led is a fix red.
Thanks

TheJulianJES, 2023-01-04:
@davix10 Please fill out the issue template in the future. Now, just post the device signature and diagnostic information for it here.

@myrkoo What coordinator are you using?

davix10, 2023-01-06:
@TheJulianJES sorry but I am beginner, where can I find this information?

myrkoo, 2023-01-06:
I'm using sonoff ZBdongle-P ZNP = Texas Instruments Z-Stack ZNP protocol: CC253x, CC26x2, CC13x2 da ZHA
After a lot of attempts my devices was paired. During the pairing I pushed the connection switch many times and led goes purple and after few minutes off. Very strange!
Thanks

closed 2023-03-15 by github-actions[bot]

ImmineoII, 2023-11-21:
> I'm using sonoff ZBdongle-P ZNP = Texas Instruments Z-Stack ZNP protocol: CC253x, CC26x2, CC13x2 da ZHA After a lot of attempts my devices was paired. During the pairing I pushed the connection switch many times and led goes purple and after few minutes off. Very strange! Thanks

Hi, im facing the same issue, could you please give me some advice on the proedure you followed to pair the shutters? At the moment i paired two of those triggering opening wiht the physical button before put them in paring mode, but for the others seems to not work.