zigpy/zha-device-handlers #2023: [Device Support Request] Ysrsai ZB-DL01

Morhoo, 2022-12-19:
**Is your feature request related to a problem? Please describe.**
Recently acquired this tuya based device (dimmable LED switch) https://zigbee.blakadder.com/Ysrsai_ZB-DL01.html
Home Assistant using ZHA identifies it as "eWeLight ZB-DL01" and does not show "current_level" in controls section, only "Forceonlight" switch.
I wasn't able do create any custom quirk to make it work by myself.

**Describe the solution you'd like**
Full support for this device. Or at least ability to control current_level attribute (input cluster 0x0008 / 0x0000)

<details>

<summary>Device signature</summary>

```yaml
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4107, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0101",
          "in_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x0008",
            "0x1000"
          ],
          "out_clusters": []
        }
      }
    },
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.12.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Prague",
    "os_name": "Linux",
    "os_version": "5.15.80",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.4",
    "docker_version": "20.10.19",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.5",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
[... 406 lines cut ...]
                "attribute_name": "on_off_transition_time"
              },
              "0x0011": {
                "attribute_name": "on_level"
              },
              "0x0014": {
                "attribute_name": "default_move_rate"
              }
            }
          },
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {}
      }
    }
  }
}
```

**Additional context**
[image]

Morhoo, 2022-12-19:
Apparently I just didn't understand how dimmable light works. Setting brightness parameter when callilng light.turn_on service works as it should.

Sorry, closing.

closed 2022-12-19 by Morhoo

MattWestb, 2022-12-19:
If you is turning the device on from the device card and then clicking on the name of it do you getting one light level slider ?
Its only there then the light is i on state.

Morhoo, 2022-12-19:
> If you is turning the device on from the device card and then clicking on the name of it do you getting one light level slider ? Its only there then the light is i on state.

Thank you MattWestb, yes it works flawlessly, I was just dumb and confused by "Startup current level" slider in configuration section and expected the same in controls. Now I know better.