zigpy/zha-device-handlers #2024: TRADFRI on/off switch not firing DIM_DOWN events

xcyb0rg, 2022-12-19:
Hi,

I have a TRADFRI on/off switch which works mostly fine, except that it mistakenly fires DIM_DOWN events as DIM_UP. I looked over the zhaquirks.ikea.twobtnremote.IkeaTradfriRemote2Btn file, but nothing jumps at me as being incorrect. I tried tweaking commands and events, things do change, but I cannot make DIM_DOWN work.

The Logbook correctly lists the events when I keep the "Up" button pressed (first the long press and then the release):

> Ikea_Switch Move With On Off event was fired with parameters: {'move_mode': <**MoveMode.Up**: 0>, 'rate': 83}
> Ikea_Switch Remote Button Long Release - **Dim Up** event was fired

But when I keep the "Down" button pressed, even though the Move is clearly shown as "Down", the release event is fired as "Dim Up" !?

> Ikea_Switch Move event was fired with parameters: {'move_mode': <**MoveMode.Down**: 1>, 'rate': 83, 'options_mask': <bitmap8.0: 0>, 'options_override': <bitmap8.0: 0>}
> Ikea_Switch Remote Button Long Release - **Dim Up** event was fired

I'm using Home Assistant 2022.12.7.

Where is the problem?

Thank you,
Mihai

MattWestb, 2022-12-19:
The device is not firing one real down is sending one stop command after move.
We have not implanting the logic to "remember" if it was up or down from the proviso command fired so we dont knowing the direction.

If you like implanting it is PR more then welcome and you can looking on the IKEA open close remote it have it implanted.

TheJulianJES, 2023-01-04:
Indeed, the device sends a ``stop_with_on_off`` command after being released from a long-press.
We're currently parsing this as a release from a DIM_UP here:
https://github.com/zigpy/zha-device-handlers/blob/7788036230986cebe8e65281d7f58c7566172c8a/zhaquirks/ikea/twobtnremote.py#L121-L125

When does the dimmer send a simple ``stop`` event instead of a ``stop_with_on_off`` event though?
Asking because of this:
https://github.com/zigpy/zha-device-handlers/blob/7788036230986cebe8e65281d7f58c7566172c8a/zhaquirks/ikea/twobtnremote.py#L133-L137

MattWestb, 2023-01-07:
I think the stop after MoveMode.Down was working in one earlier firmware version (= i dont knowing).
The Open/Close is storing the last move command and its using it for "tagging" the stop with up or down.

elvismercado, 2023-03-19:
I also have this problem.

JR-Y, 2023-04-01:
Same issue, in homeassistant looks like this:

DOWN long press
[image]

UP long press
[image]

MattWestb, 2023-04-01:
With latest firmware 24.4.5
Long press up:
```
  endpoint_id: 1
  cluster_id: 8
  command: move_with_on_off
  args:
    - 0
    - 83
  params:
    move_mode: 0
    rate: 83
```
Long press down:
```
  endpoint_id: 1
  cluster_id: 8
  command: move
  args:
    - 1
    - 83
    - 0
    - 0
  params:
    move_mode: 1
    rate: 83
    options_mask: 0
    options_override: 0
 ```
Both is sending the same stop command:
```
  endpoint_id: 1
  cluster_id: 8
  command: stop_with_on_off
  args: []
  params: {}

  endpoint_id: 1
  cluster_id: 8
  command: stop_with_on_off
  args: []
  params: {}
```
And all is OK in the DA:
https://github.com/zigpy/zha-device-handlers/blob/be0fc65d9b074406bb16e43e5b0decc3665ed84f/zhaquirks/ikea/twobtnremote.py#L113-L138

closed 2024-04-02 by github-actions[bot]

lxs687, 2025-05-27:
I also have this problem with IKEA switch not firing DIM_DOWN events ! I'm running latest firmware and core version but looks like the triggers type: remote_button_long_press ; subtype: dim_down is definitively not working !

I do understand that for a Dim Down (long press on 0) the event are "RODRET Dimmer Remote Button Long Release - Dim Up event was fired" and "RODRET Dimmer Move event was fired with parameters: {'move_mode': <MoveMode.Down: 1>, 'rate': 83, 'options_mask': <OptionsMask: 0>, 'options_override': <Options: 0>}"

But now how do you implement that in the automation as following code is never executed : 

- id: '1748331844557
  alias: Button xxx Dim Down
  description: ''
  triggers:
  - device_id: 3dbdcdf4594908df4643dc3808077f62
    domain: zha
    type: remote_button_long_press
    subtype: dim_down
    trigger: device
  conditions: []
  actions:
  - action: light.turn_on
    metadata: {}
    data:
      brightness_step_pct: -10
    target:
      device_id: ff65832f1059e1c371c2f408ad4c82c0
  mode: single