zigpy/zha-device-handlers #2025: [Device Support Request] Support temperature offset for Tuya Saswell TRV _TZE200_c88teujp

jclsn, 2022-12-20:
**Is your feature request related to a problem? Please describe.**
I would like to set a temperature offset for the Saswell TRVs. It seems there needs to be a data point mapping implemented. The model is listed here https://github.com/Koenkk/zigbee-herdsman-converters/blob/927f352c9dc848f716c23abf9d2a0a56c13a9f67/devices/saswell.js#L11

and here https://github.com/Koenkk/zigbee-herdsman-converters/blob/927f352c9dc848f716c23abf9d2a0a56c13a9f67/lib/tuya.js#L516

, but I have no idea how to implement a mapping here 

https://github.com/zigpy/zha-device-handlers/blob/c2b2893d027fdcd3a225996fdf382d2ac72ff799/zhaquirks/tuya/ts0601_valve.py#L161

**Describe the solution you'd like**
Implement a data point mapping for a given model

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0201",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_c88teujp",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_trv_sas.Thermostat_TZE200_c88teujp"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.12.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.76-v8",
    "supervisor": "2022.11.2",
    "host_os": "Home Assistant OS 9.4",
    "docker_version": "20.10.19",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
[... 338 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-12-20:
I'm sorry to say I'm not familiar with TRVs, but we're going to help where we can.

The TRV device don't use the 'MCU approach' (not in the same way that other implementations) so you will not find any `DPToAttributeMapping` here 🤷🏻‍♂️ 

As you have reported, it seems that in Z2M it is possible to configure the `saswell_thermostat_calibration`. This part is done here:
* https://github.com/Koenkk/zigbee-herdsman-converters/blob/927f352c9dc848f716c23abf9d2a0a56c13a9f67/converters/toZigbee.js#L5743

As you have stated, the DP value for the attribute is `saswellTempCalibration: 27,` in hexadecimal will be `0x1B`.

Let's see how is it in ZHA...

Your device is already loading a quirk ([`zhaquirks.tuya.ts0601_trv_sas.Thermostat_TZE200_c88teujp`](https://github.com/zigpy/zha-device-handlers/blob/c2b2893d027fdcd3a225996fdf382d2ac72ff799/zhaquirks/tuya/ts0601_trv_sas.py#L195))

Most of the magic happens here:
* https://github.com/zigpy/zha-device-handlers/blob/c2b2893d027fdcd3a225996fdf382d2ac72ff799/zhaquirks/tuya/ts0601_trv_sas.py#L39

and here:
* https://github.com/zigpy/zha-device-handlers/blob/c2b2893d027fdcd3a225996fdf382d2ac72ff799/zhaquirks/tuya/ts0601_trv_sas.py#L96

...but no reference to any `calibration` attribute.

Interestingly, I did find a reference to the attribute elsewhere:
* https://github.com/zigpy/zha-device-handlers/blob/c2b2893d027fdcd3a225996fdf382d2ac72ff799/zhaquirks/tuya/ts0601_trv.py#L913

Right name and right DP, that can not be a coincidence.

So the `ZonnsmartTV01_ZG` seems to implement the temperature calibration:
* https://github.com/zigpy/zha-device-handlers/blob/c2b2893d027fdcd3a225996fdf382d2ac72ff799/zhaquirks/tuya/ts0601_trv.py#L1663

Since it doesn't seem like this quirk can be used for your device, I guess the functionality should be implemented in the `Thermostat_TZE200_c88teujp` quirk.

That would be more or less:
* all related with the `ZONNSMART_TEMPERATURE_CALIBRATION_ATTR`
  * a constant attribute (ie: `TEMPERATURE_CALIBRATION_ATTR = 0x021B`
  * to add the attribute at `ManufacturerThermostatCluster` class
  * handle in the `_update_attribute` the new attribute
  * a new `self.temperature_calibration_bus = Bus()` in `Thermostat_TZE200_c88teujp`
* for the UI part, an implementation like the `ZONNSMARTTemperatureOffset`:
  * https://github.com/zigpy/zha-device-handlers/blob/c2b2893d027fdcd3a225996fdf382d2ac72ff799/zhaquirks/tuya/ts0601_trv.py#L1328

I hope it's enough to guide you

jclsn, 2022-12-20:
Yeah thanks, that is plenty. I will see if I can implement this when I find the time. 

How can I test this after implementing?

javicalle, 2022-12-21:
> How can I test this after implementing?

Enable the local quirk folder in your instalation and put inside the new version of the quirk (the full file).
Restart HA and check if the device signature changes.

> Is it possible to expose this as a HA entity?

That is the function of the `ZONNSMARTTemperatureOffset` class.

jclsn, 2022-12-26:
Okay, got it to do "something". I have a slider now and get enter values. The heating setpoint changes sometimes and then it doesn't. The core also seems to crash sometimes and the frontend can't be accessed for a while. How can I debug this? Where are the logs?

closed 2023-01-03 by dmulcahey