zigpy/zha-device-handlers #2034: [Device Support Request] Loratap SS9600ZB-V2 TS0046

SkyaTura, 2022-12-24:
**Is your feature request related to a problem? Please describe.**
Just asking for a new device support.

**Describe the solution you'd like**
A specific new quirk.

<details>

<summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006",
        "0xe000"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_iszegwpd",
  "model": "TS0046",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```json
{
[... 1040 lines cut ...]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [TuyaSmartRemoteOnOffCluster, TuyaZBE000Cluster],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [TuyaSmartRemoteOnOffCluster, TuyaZBE000Cluster],
            },
            6: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [TuyaSmartRemoteOnOffCluster, TuyaZBE000Cluster],
            },
        },
    }
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

SkyaTura, 2022-12-24:
I don't know if is it worthy mentioning, but Home Assistant shows 4 switch entities that have no utility at all, and any events are recorded when I tap any button.

There is an error being thrown, tho:

```2022-12-24 02:04:39.229 WARNING (MainThread) [zigpy.zcl] [0xB78D:2:0x0006] Unknown cluster command 253 b'\x00'```

MattWestb, 2022-12-24:
You have doing nearly all things right in your test quirk !!
The nearly identical device is the `class TuyaSmartRemote0044TOPlusA` and it have the same endpoints and cluster settings.
In the signature you have adding endpoint 5 and 6 that is not in the orginal device signature so it not matching = not loaded.

Delete the endpoint 5 and 6 in the signature but leave it in the replacement for testing.

If getting problems with formatting use [Black Playground](https://black.vercel.app/) for fixing it.

Delete the` __pycache__ ` folder in your local quirk folder if its populated and restart HA and look if you is getting the quirk being loaded in the device signature.
It shall looks like this in the last line of device info on the  device card `Quirk: fourbtnremote.IkeaTradfriRemoteV2` (its for the new version of IKEA Styrbar).

SkyaTura, 2022-12-24:
Oh, now I got how it matches.

I should've think a little while over this hahaha

Although, the automations are not being triggered by tapping the buttons 😞

Also, there is nothing showing on the logs now.

I'll try resetting everything later today, for now, thanks for the support and merry Christmas 😊

MattWestb, 2022-12-25:
### Feliz Natalm !!
The good thing you is not getting errors and the quirk is being loaded OK.
Put debug logging for zha, zigpy and zhaquirks so you can see what is happening.
Its expected not getting events from button 5 and 6 but we is fixing it later :-)))

SkyaTura, 2022-12-25:
## Updates

I removed the device, cleared the cache, restarted the docker container, repaired the controller.

Finally, it now recognizes the events. Even the 5 and 6 buttons 🎉

However, nothing is all perfect...

Button 1 to 6: perfectly fires "single tap" events
Button 1 to 6: it does fires "double tap", but, sometimes it recognizes as "single tap"

Only button 1 fires "hold" events, also, not always...

Is that a problem with my quirk, my ZHA, my device, or should I just throw everything on the window? (Just kidding, I can't afford the last option)

MattWestb, 2022-12-25:
My experience of tuya TS004F is that double shall being very fast for working OK.
Can you trying doing 2 things with the quirk ?
First in the replacement par remove the `TuyaSmartRemoteOnOffCluster,` from `in` and putting it in the `out` cluster on all 6 endpoints and trying if its working OK.
The device is one remote and is only sending commands = out cluster and light shall receiving command and shall have it as in cluster (its tuya devices) then the 4 not working switches in  the GUI is not being made.
The second can you  try putting in this code i your device class:
https://github.com/zigpy/zha-device-handlers/blob/c2b2893d027fdcd3a225996fdf382d2ac72ff799/zhaquirks/tuya/ts004f.py#L189-L201
Perhaps some imports may need being added for getting it working.
Its the tuya magic that some devices is needing for working OK and Z2M is using it for this device.
With it you need deleting the device and resetting it and then restart HA and joining it new for it getting the magic spell being casted on it very early early in the joining process.
I hope getting the long press working on all buttons with it and if its being OK casted one time the device is keeping it until the battery is out **and** its being resettled.

SkyaTura, 2022-12-25:
Sure, I'll test it.

There is a way to see "raw packages" being emitted by the controller? Like IR blasters have bits?

I suppose that, as this is a radio protocol, there is much more information to work on.

MattWestb, 2022-12-25:
Did you getting long press and double working on all 6 ????
If yes that is working great !!

For logging i think with  `homeassistant.components.zha: debug` and `zigpy: debug` in HA config you can see the data in normal way from the device. You can also doing debug of the UART but  then its "zigbee formatted" and useless for you.

Can you trying changing the replacement like this for endpoint 2-6:
```
            6: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [
                    TuyaSmartRemoteOnOffCluster,
                ],
            },
```
Then the not working switch shall going away in the GUI and power is only needed on the first one. and for EP 1:
```
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    TuyaZBE000Cluster,
                ],
                OUTPUT_CLUSTERS: [
                    TuyaSmartRemoteOnOffCluster,
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
```
And some combinations or coordinators and devices dont like changing device type so put `DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,` on all EP in replacement so its the same as the device signature.

Timing of double and long pressing is fixed in the device firmware and we cant changing it :-(((

I think the TS0046 was one god Christmas gift for you and you have getting most working OK now.

Need little more testing and fixing then is best making one PR so its being updated with  ZHA in the future and also other user is getting there device working out of the box.

MattWestb, 2022-12-25:
Here is some more debug setting that is is possible and as you can see im using EZSP coordinator so EZSP and bellows is the debug handles for it and if you is using one other you shall changing the right  type.
```
logger:
  default: info
  logs:
    homeassistant.core: info
    homeassistant.components.zha: info
    zigpy: debug
#    zigpy.zdo: debug
#    zigpy.zcl: debug
#    zigpy.application: debug
#    bellows: debug
#    bellows.ezsp: debug
#    bellows.ezsp.protocol: debug
    bellows.zigbee.application: debug
#    bellows.uart: debug

#    zigpy_cc: debug
#    zigpy_deconz.zigbee.application: debug
#    zigpy_deconz.api: debug
#    zigpy_xbee.zigbee.application: debug
#    zigpy_xbee.api: debug
#    zigpy_zigate: debug
    zhaquirks: debug
    custom_zha_quirks: debug
    custom_components.zha_toolkit: debug
```
Also both quirk debugs can giving some information if the quirk have implanting debug logging.

SkyaTura, 2022-12-25:
It's working flawlessly, ~~I'll even buy a pack I saw on sales on Ali Express this night hahaha~~

Maybe I can manage the "long longpress timming" without messing with firmwares, just by designing a better automation interface with the controller, leaving long_press only for rarely used actions.

### Thank you, again, for the support you gave 🥰

javicalle, 2022-12-29:
Could you try with this modification of your code:
```python
class TuyaSmartRemote0046TOPlusA(EnchantedDevice, Tuya6ButtonTriggers):
    """Tuya 6-button remote device with time on out cluster."""

    signature = {
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=1, input_clusters=[0, 1, 6, 57344], output_clusters=[10, 25]))
        .../...
```

(Extend from `EnchantedDevice` and delete the `__init__` and `spell` methods)

You will need to import the new class:
```python
from zhaquirks.tuya.mcu import EnchantedDevice
```

SkyaTura, 2022-12-30:
I can test it tomorrow.

Could you explain what this modification do, tho?

javicalle, 2022-12-30:
Sure, the `EnchantedDevice` class wraps all the 'Tuya spell cast' function:
https://github.com/zigpy/zha-device-handlers/blob/a6556fea6ab6400984094dcc88d7bd772e322e97/zhaquirks/tuya/mcu/__init__.py#L648

Is just a cleaner way to implement it. And preferably if you creates the PR.

Thanks in advance.

SkyaTura, 2023-01-05:
I got two more controllers today, they are the same model, but I'll be able to test the proposed changes and check how they behave with multiple controllers as well.

DJBenson, 2023-01-12:
Hey - any update on this - I have one of these devices (LoraTap-branded) which I got to work with your custom quirk - what's the route to getting the device officially supported?

DJBenson, 2023-01-12:
@MattWestb thank you. The device is working fine. I wasn't sure what the process is for official support as I've just migrated from z2m to zha.

MattWestb, 2023-01-12:
Then some one have one quirk working OK (adding / fixing some or all possible function of the device) is recommended doing one PR for adding it to ZHA.
Then tuya is very exotic in there function its normally need more work getting them working OK as devices that is flowing the Zigbee standard = working out of the box.

This quirk need little more fixing for getting the not working switches in the GUI and also merger it to our tuya standard.

If some can doing more testing we can getting it merged for the next HA update.

SkyaTura, 2023-01-12:
Sorry folks, I had personal issues that impede me to test it again.

I don't want to promise an ETA to avoid not reaching it again, but I'll test again with the final version ASAP, probably until the end of the weekend.

There is other things I want to try with this device in HASS as well, I hope nothing else happens and I get time now hahaha

---

_Someone, by any chance, know any integration that allow one use button on these devices as "code inputs"? Like triggering an event if I press a sequence of buttons instead of individual actions?_

MattWestb, 2023-01-12:
Zigbee controller device (remotes) is sending events / commands like double click / Light Toggle and is sent from one OnOff **out** cluster.
tuya have making one OnOff **In** cluster = one light that is being switched on or off then receiving commands.
That is not Zigbee compatible but we cant change the firmware.
So we is getting 6 not working switches in ZHA GUI and the best we can doing is deleting them in the quirk by moving the OnOff clusters to out in the replacement.
Make some magic "hokus pockus" is never going thru the maintainers.

If like making switches in HA GUI use trample switches that you can doing all you like.

closed 2023-01-18 by dmulcahey

TheJulianJES, 2023-03-12:
Is `EnchantedDevice` (so reading the six attributes on pairing AND every HA restart) really necessary for this remote?

Did anyone ever test if there were any differences between just `CustomDevice` and `EnchantedDevice` for `TS0046`?

PR #2094 (2023-01-18): Add support for Tuya TS0046
Made in colab with @MattWestb and @javicalle _(mostly by them)_ as discussed in #2034

fixes: #2034