zigpy/zha-device-handlers #2036: [Device Support Request] _TZ3000_8nkb7mof plugs

ali1234, 2022-12-27:
After upgrading Home Assistant these plugs got three new entities:

The configuration "Power on state" works.

The configuration "Backlight mode" does nothing. The light on the unit never lights up regardless of the selected value. The selected value is saved on the device though.

The sensor "Power factor" always says "Unknown".

The diagnostic "Identify" still does nothing. It has always been present and has never done anything.

There is a cluster attribute "Child lock" which works as expected if you manually set it to true through the management interface.

The other sensors appear to continue to work correctly.

These plugs are round and have a blue LED under the only button. When they were brand new, before they had been connected to any network, the LED would turn on when the switch was turned on. This has not been the case ever since I connected them to home assistant. Now the LED only lights when in pairing mode.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_8nkb7mof",
  "model": "TS0121",
  "class": "zhaquirks.tuya.ts0121_plug.TS0121B"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.12.8",
    "dev": false,
[... 596 lines cut ...]
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

TheJulianJES, 2023-01-04:
> The diagnostic "Identify" still does nothing

It should blink an LED on the plug or turn on/off the attached device briefly. If it doesn't do anything, Tuya hasn't implemented it correctly for that device. You can just ignore/disable it.

> The sensor "Power factor" always says "Unknown".

Pretty sure it's not supported by the plug. You can manually disable it for now (or just ignore it), but could also prevent it from being added to HA at all.

> The configuration "Backlight mode" does nothing. The light on the unit never lights up regardless of the selected value. The selected value is saved on the device though.

Hmm, the config entity is only created when the cluster attribute exists, so it's likely this is not implemented correctly by the plug.

--- 

> There is a cluster attribute "Child lock" which works as expected if you manually set it to true through the management interface, but this attribute is not shown in dashboards etc.

So you're saying you do not see the following entity:

[image]

but have a ``child_lock`` attribute in the cluster UI which you can successfully change to ``0`` or ``1``?

[image]

I don't see that ``child_lock`` attribute in your diagnostic settings. It will show up in the clusters UI, but if it's not configurable there, an entity also won't be created.

ali1234, 2023-01-04:
Yes, that is what I am saying. Furthermore, changing the child lock value in the cluster disables the physical button as expected. It can be re-enabled by pressing double pressing it, or by changing the value in the cluster back to false.

ali1234, 2023-07-03:
There has been no change to anything I wrote previously.

closed 2024-01-06 by github-actions[bot]