zigpy/zha-device-handlers #2042: Another variance of TS0012

darek, 2022-12-30:
**Is your feature request related to a problem? Please describe.**
I do have couple of Tuya 2Ch no neutral switches (or another variance) and it seems those does not work properly because each time i enable one switch both of them are turning on / off.
I found out that this device support was added on November https://github.com/zigpy/zha-device-handlers/pull/1967 by adding new quirk named Tuya_Double_Var05 but all of my devices are using this quirk and it seems it does not work.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_jl7qyupf",
  "model": "TS0012",
  "class": "zhaquirks.tuya.ts001x.Tuya_Double_Var05"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.12.8",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Warsaw",
    "os_name": "Linux",
    "os_version": "5.15.80",
    "supervisor": "2022.12.1",
[... 423 lines cut ...]
              }
            }
          }
        },
        "out_clusters": {}
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

MattWestb, 2022-12-30:
The quirk is casting "tuya magic spell" on the device then its paring and it shall start behaving _normal_.
The problem its only working then the device is joining new and not later (on most devices but not all).
Delete the problem device and waiting one minute and restarting HA.
Then HA have starting OK reset the device and joining it normal and look if its working OK.
If not sett little more debug logging and redo the procidure and posting the log from the paring here.

darek, 2022-12-30:
Thanks for reply, will try your solution, but those devices where included this week so long after fix was merged.

MattWestb, 2022-12-30:
I was thinking that :-(((
It can being that the paring was taking to long time or timing out and the tuya magic spell was sent to late to the device.

Pleas post one debug log then pairing one device new.

darek, 2022-12-30:
Just tried to repair as you described but with no luck, here are some debug logs. Inclusion went quite fast and without error.
Logs are from home assistant with debug logs on zigpy.

```
2022-12-30 17:39:46.191 ERROR (SyncWorker_1) [miio.miioprotocol] Got error when receiving: timed out
2022-12-30 17:39:51.036 WARNING (MainThread) [homeassistant.components.zha.core.channels.base] [0xBB00:1:0x0300]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
2022-12-30 17:39:51.205 WARNING (MainThread) [homeassistant.components.zha.core.channels.base] [0xBB00:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
2022-12-30 17:39:51.642 WARNING (MainThread) [homeassistant.components.zha.core.channels.base] [0xBB00:1:0x0008]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
2022-12-30 17:39:56.250 ERROR (SyncWorker_0) [miio.miioprotocol] Got error when receiving: timed out
2022-12-30 17:40:16.136 ERROR (SyncWorker_4) [miio.miioprotocol] Got error when receiving: timed out
2022-12-30 17:40:46.118 ERROR (SyncWorker_0) [miio.miioprotocol] Got error when receiving: timed out
2022-12-30 17:41:36.095 ERROR (SyncWorker_7) [miio.miioprotocol] Got error when receiving: timed out
2022-12-30 17:42:43.264 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=114, profile_id=0, cluster_id=<ZDOCmd.Mgmt_Permit_Joining_req: 0x0036>, data=Serialized[b'r\xfe\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-12-30 17:42:43.265 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendBroadcast: (0xfffc, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=114), 0, 115, b'r\xfe\x00')
2022-12-30 17:42:43.267 DEBUG (MainThread) [bellows.ezsp.protocol] Send command addTransientLinkKey: (ff:ff:ff:ff:ff:ff:ff:ff, [90, 105, 103, 66, 101, 101, 65, 108, 108, 105, 97, 110, 99, 101, 48, 57])
2022-12-30 17:42:43.267 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'22fb21a9622ae94d59947c25aa5592489c4e55ab9ecd1575fd3bcd7e'
2022-12-30 17:42:43.285 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'23fba1a9622a151d0aa37e'
2022-12-30 17:42:43.286 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
2022-12-30 17:42:43.286 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'33fbb5a94d2a8980c77e'
2022-12-30 17:42:43.286 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2022-12-30 17:42:43.286 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'43fbb1a9112a10b259a24a25aa5593499ce1d8abedce9874feb49d891f257e'
2022-12-30 17:42:43.286 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2022-12-30 17:42:43.287 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'35f821a9fb2aea4da66bb5da55aac820fb0c42ceaca20be29ca800eccc47dbb97e'
2022-12-30 17:42:43.287 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 54 (sendBroadcast) received: b'00af'
2022-12-30 17:42:43.287 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 25 (stackStatusHandler) received: b'9c'
2022-12-30 17:42:43.287 DEBUG (MainThread) [bellows.zigbee.application] Received stackStatusHandler frame with [<EmberStatus.undefined_0x9c: 156>]
2022-12-30 17:42:43.287 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'0500003600000000010000afff000000ffff0372fe00'
2022-12-30 17:42:43.288 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST_LOOPBACK: 5>, EmberApsFrame(profileId=0, clusterId=54, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=175), 255, 0, 0x0000, 255, 255, b'r\xfe\x00']
2022-12-30 17:42:43.288 DEBUG (MainThread) [bellows.zigbee.application] Ignoring message type: <EmberIncomingMessageType.INCOMING_BROADCAST_LOOPBACK: 5>
2022-12-30 17:42:43.297 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'54f8a1a9fb2a15674f7e'
2022-12-30 17:42:43.297 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
2022-12-30 17:42:43.298 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 175 (addTransientLinkKey) received: b'00'
[... cut ...]
2022-12-30 17:42:55.444 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x049C](Remote Control N2): last_seen is 103500.25064277649 seconds ago and ping attempts have been exhausted, marking the device unavailable
2022-12-30 17:42:55.445 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x049C](Remote Control N2): Update device availability -  device available: False - new availability: False - changed: False
2022-12-30 17:43:06.099 ERROR (SyncWorker_8) [miio.miioprotocol] Got error when receiving: timed out

```

MattWestb, 2022-12-30:
I have looking in the log and the quirk is not casting any "tuya magic" and your device is needing it for working OK so you is having one new unknown version of it.

Can you configure local quirks in HA and putting this unziped quirk in the directory, deleting one device and restarting HA and joining it then HA have loading OK.
[ts001x.zip](https://github.com/zigpy/zha-device-handlers/files/10326043/ts001x.zip)

Instruction configure local quirk in ZHA https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274.

@javicalle Shall we migrating all device classes in this quirk for testing ?
I think all devices in it is unproblematic and shall working OK with it.

javicalle, 2022-12-30:
I also believe that adding the `EnchantedDevice` would not be a problem so it is fine to me if we add to all the quirks (in that file).

MattWestb, 2022-12-30:
Im awaiting the user feedback and is it positive i making one PR with all classes updated.
Thanks for feedback !!

darek, 2022-12-30:
I made test with one device, works great, tomorrow will test other devices. Thanks for help

MattWestb, 2022-12-31:
@darek You dont need restarting HA for getting the tuya magic being casted only deleting the device in ZHA and waiting one minute (so the database have written the deletion) and then adding the device new in the system (it was needed for getting the new setting for local quirks working).
If its not like it you can try "reloading" ZHA integration from device integration tab  between deleting + waiting and adding the device and its going faster then restarting HA.

I preparing making one PR for fixing the quirk so it can being added in next upcoming HA release.

darek, 2022-12-31:
Got 3 devices, all of them are working now properly.

MattWestb, 2022-12-31:
## Great !!
PR is made and waiting being merged if the maintainer is liking it (its closing the issue then being merged).

Dont forgetting deleting the local quirk then updating HA that have it integrated or you can getting problem in the future.

closed 2022-12-31 by dmulcahey

mkolesnik, 2023-01-01:
Thanks a lot @MattWestb , I can verify this fix works as well for my 2-gang no neutral tuya switches (_TZ3000_jl7qyupf)

Thanks for the quick fix! :D

PR #2049 (2022-12-31): Merger all TS001X device classes to "EnchantedDevice"
Use `EnchantedDevice` as some new devices is needing it and it shall not harming old devices wot we knowing.

Fixes: #2042