zigpy/zha-device-handlers #2044: [BUG] Xiaomi smart plug MAEU01 not working correctly since SkyConnect

Steephsel, 2022-12-30:
**Describe the bug**
Since i've received the SkyConnect, 2 out of my 4 Xiaomi MAEU01 show no readings and temperature 0,2/0,3°C. in Home Assistant the SkyConnect was automatically detected as ZHA integration, which I'm not used to myself but I know for sure it was working before the SkyConnect via z2m. The only thing actually working is the Switch function, which I don't even use it for. I've tried all I can think of. Deleting, repairing, reconfigure, etc.

**To Reproduce**
Steps to reproduce the behavior:
1. Not sure what is causing this, so can't tell yhow to reproduce.

**Expected behavior**
Readings from the power/voltage and power delivered.

**Screenshots**
[image]

TheJulianJES, 2023-01-03:
You did not fill out the issue template nor provide any diagnostic information for the plug.
Please download the diagnostic information for that plug (three dots on that device page -> diagnostic information).
Then drop the txt file into your comment field to upload.

Also, "Device temperature" being stuck at ~0°C or ~20°C is known for this plug. The entity should probably be removed by default.

Did you try completely removing the plug from HA and your network, then allow joins again ("Add Device" button) and **reset and re-join the plug**?

Steephsel, 2023-01-03:
> You did not fill out the issue template nor provide any diagnostic information for the plug or what version of Home Assistant you're running. Please download the diagnostic information for that plug (three dots on that device page -> diagnostic information). Then drop the txt file into your comment field to upload. (If you can't download the diagnostic information for that plug, just attach the device signature. (can be found by the three dots -> "Manage Zigbee device" -> "Device Signature" tab))
> 
> Also, "Device temperature" being stuck at ~0°C or ~20°C is known for this plug. The entity should probably be removed by default.
> 
> Did you try completely **removing the plug** from HA and your network, then allow joins again ("Add Device" button) and **reset and re-join the plug**? (not the same as reconfiguring)

I removed and readded them a couple of times already, but did not try the keeping alive trick yet during re-adding it. I've asked my wife if she's willing to help me out from home, she's getting pretty annoyed already with my smart home not working...

What I did also notice is that the plugs not working show a firmware version in ZHA, those that are working don't, see below image.
[image]

For now here's the text file from the plug not working. 
[zha-663a960a3e6ead69982b7d4f072b0a30-LUMI lumi.plug.maeu01-76d3d8148c41237eb81825a2242702d8.json.txt](https://github.com/zigpy/zha-device-handlers/files/10336845/zha-663a960a3e6ead69982b7d4f072b0a30-LUMI.lumi.plug.maeu01-76d3d8148c41237eb81825a2242702d8.json.txt)

TheJulianJES, 2023-01-03:
The "Firmware" field on the device page only shows up when an update checks for a device.
The Aqara plugs do not automatically check in and ask for firmware updates, so the "ota_notify" command has to be used.
Then, they'll show their firmware version there (in hexadecimal).

If you click the three dots -> Manage Zigbee device and read the ``app_version`` of the Basic cluster, the plug shows the same version ``41`` (in decimal).

Can you check the ``app_version`` for all of the plugs and let me know what the working ones and what the non-working ones are running? (We can downgrade the newer ones to an older versions that's way less buggier)

Also, you mention that the sensors worked with Z2M (and your previous stick). Are you 100% sure that the "active power" and "summation delivered" sensors actually work with Z2M on all of your plugs?

Steephsel, 2023-01-04:
> 

Thanks for the explanation, I still need to learn more about ZHA :') 

I checked all the plugs, they do indeed all show `app_version` 41. And I'm 100% sure they did all work on Z2M, aside from the fact the entities had different names for the readings. 

On a side note, I noticed my lights are not officially supported so that would explain why they were working buggy, I will post a device support request for those. Other than that I only have my Philips Hue outdoor motion sensor bugging out, which should be officially supported. I will try and repair that later on today (doesn't have priority)

TheJulianJES, 2023-01-04:
I'll do some more testing later with the plugs. I tried to re-pair my plugs many times and could not get them break.

The plugs are annoying because they try to detect if they're connected to a Xiaomi hub or something else. If they think it's a third-party hub, they'll disable part of their functionality.

ZHA (and Z2M) try to work around this by sending Xiaomi manufacturer codes during pairing when detecting a Xiaomi device. This works reliably for me.

You could also try updating your firmware: https://github.com/NabuCasa/silabs-firmware/blob/main/EmberZNet/beta/NabuCasa_SkyConnect_EZSP_v7.1.3.0_ncp-uart-hw_115200.gbl
(In the future, HA will allow updates without any command line tools, but for now you'll need universal-silabs-flasher. There's a tutorial and the full command for flashing here: https://github.com/NabuCasa/silabs-firmware/wiki/Flash-Silicon-Labs-radio-firmware-manually)
I don't think it will be solved with a SkyConnect firmware upgrade though, as I also had no issues with the older firmware.

Otherwise, I'd recommend downgrading the firmware on the affected plugs to a more stable version. A download link can be found in this comment: https://github.com/Koenkk/zigbee2mqtt/issues/13903#issuecomment-1244720072
A wiki entry for OTA is found here: https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates

Basically, add this to your ``configuration.yaml``:
```yaml
zha:
  zigpy_config:
    ota:
      otau_directory: /config/zigpy_ota
```
Put the file from the comment in a newly created ``zigpy_ota`` folder in your ``/config`` folder.

Then, restart Home Assistant. After you've done that, execute the following service call in Home Assistant to temporarily set your ZHA logs to debug:

```yaml
service: logger.set_level
data:
  homeassistant.components.zha: debug
  bellows.zigbee.application: debug
  bellows.ezsp: debug
  zigpy: debug
  zigpy_deconz.zigbee.application: debug
  zigpy_deconz.api: debug
  zigpy_xbee.zigbee.application: debug
  zigpy_xbee.api: debug
  zigpy_zigate: debug
  zigpy_znp: debug
  zhaquirks: debug
```

Then, you can either use the GUI or another service call to execute an "OTA image notify" command. This will make your plug ask if there any updates available:
```yaml
ieee: device_ieee # repplace with the IEEE of the plug
endpoint_id: 1 # sometimes also: 11, 2, or 3
cluster_type: out
command_type: client
cluster_id: 25
command: 0
params:
  payload_type: 0
  query_jitter: 100
```

Now, monitor the HA log file when the update is completed, remove the OTA file, and restart Home Assistant. Remove the plug from HA and then re-pair it. It *should* work now. (Also check if the app_version is lower now)

(Note: The OTA file needs to be removed, as the upgrade will restart otherwise (you can also remove the OTA file during an upgrade to cancel it (if it happens to restart)). This restart happens because ZHA normally just wants to upgrade stuff and it and the plug need to be fooled: the OTA header says it's version 42, but it's actually an older version.)

Regarding the lights and Hue sensors, please create separate issues and tag me in them (@TheJulianJES). (All Hue sensors work, but the OnOff entity should be disabled and the Occupancy one changed to Motion)

Steephsel, 2023-01-04:
I'm pretty sure I'm doing something wrong somewhere in this process. I've uploaded the file to the folder, restarted HA, set the loggers etc but the OTA image update thingy doesn't seem to work. I've checked both HA service and the ZHA commands, but neither seem correct? Forigive my newbie-ness with ZHA... 😢 

[image]

TheJulianJES, 2023-01-04:
On the left, you just need to type "issue cluster command" (zigbee.issue_cluster_command should come up).

(On the right, you'd have have needed to provide 0 and 100 for the args mentioned on the left)

Steephsel, 2023-01-04:
Yes! This seems to have done the trick. 😄 Thanks a lot!

closed 2023-01-04 by Steephsel

TheJulianJES, 2023-01-06:
It seems like a typo was found within a library for SkyConnect where Xiaomi manufacturing codes weren't properly sent for some devices only.

Can you post the first three blocks of an IEEE of a Xiaomi plug that did not work before the firmware downgrade?

Steephsel, 2023-01-06:
> It seems like a typo was found within a library for SkyConnect where Xiaomi manufacturing codes weren't properly sent for some devices only.
> 
> Can you post the first three blocks of an IEEE of a Xiaomi plug that did not work before the firmware downgrade?

Sure, 04:cf:8c is the one i used in my posts earlier